zigpy/zha-device-handlers #1124: [BUG] Aqara sensors not resetting motion

benblb, 2021-11-03:
**Describe the bug**
I hope this is the correct place to post this. 

Occasionally an Aqara sensor will remain in the "detected" state too long. Based on entity history, an Aqara's motion detection clears after 70 seconds normally. However, occasionally this value will be dramatically higher. Please see image. 

[image]

Please note: I have four Aqara motion sensors and was using them with deCONZ (Conbee II USB). Two weeks ago, three began missing detections occasionally and having long reset times so I tried changing the batteries last week. That didn't solve the problem, so I moved to ZHA last night. I am running ZHA on ch15 and my Hue hub is on ch25. WiFi 2.4GHz is on wifi-channel 10. Conbee II is on a USB extension cord. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add Aqara motion sensor to ZHA
2. Use normally
3. Note erratic behavior sometimes(?)

**Expected behavior**
Sensors should always clear after the default time, not get stuck as detected. 

**Screenshots**
Image in section above. 

**Additional context**
I have not ruled out this being a sensor issue and all 3 failing within a short time of each other. I have also not ruled out a single sensor failing and messing with the network, however 1/4 works perfectly every time (and is significantly newer than the other 3). All motion sensors are the "new" version which measure luminance as well. 

When I was running on deCONZ, I noticed that sometimes a luminance value would come through when I walk into the room even if motion did not trigger. Before this fiasco I had everything stable for ~18 months.

MattWestb, 2021-11-03:
You can using one app for your mobile and looking if you have neighbors have strong WiFi like WiFi analyzer that can interference with your Zigbee network but is not so easy then its looks like you dont have any routers that is keeping the network open with routing calls.

More likely if you is only having 3 motion sensors and one CorBee II is that the sensors is not getting contact with the coordinator then like sending commands and you need 3 or more good routers for getting the network working OK as one mesh network that is the key to getting it working good.

I suggesting 4 outlets / plugs in the near of your motion sensors so they have one router in the near and dont need using there radio with high power (SED is using only 1/10 of the chip possible power compared with one router)  and long time for sending commands.

benblb, 2021-11-03:
@MattWestb I heard that Aqara sensors dislike going through most routers so I avoided using them (also because my place is small). There is a WiFi network from a neighbor on wifi-channel 4 which I believe overlaps with Zigbee 15. My scanner app is showing -59dB near the sensor with the most problems. There's also non-setup WeMo Mini DE2 smart plug at -66dB on wifi-channel 3 which actually spans from channel 1-7! 

Is WiFi interference likely the issue?

My sensor is showing LQI of 255 and RSSI -56, both of which I believe to be more than satisfactory. However I notice you're a contributor and therefore know much more than me about Zigbee so please correct me if I am mistaken!

MattWestb, 2021-11-03:
I have one "standard setup" for my rooms with one IKEA outlet at router, one E1743 remote for the outlet or lights and one Aqara weather (Tmp, Humid and pressure) and dont have any problems with the Aqara sensors that all is having then they have one stable router and dont losing the contact to it (if they do they is leaving the network).

I have blocking my coordinator for having direct children then coordinators is not so good parents then they is doing many other things that can making them not replaying to there children and getting problems.

Also i was taking one of my test network with 2 routers and 20 end devices then i was on 3 weeks holiday and coming home and plugging the USB in the laptop and all was on line in one hour.
If having device is having the coordinator as parent and its offline the device must jumping to one new one but its not working bood with some device (Xiaomi and some more).

And dont using ORRAM plug then they is dropping packages to / from there children = problems.

If you is having  good routers the hole network is being more stable and immune from interference then can rerouting the traffic around the problem and its working if all end devices have good routers as parent then they cant (or not doing) jumping to on new parent so fast and can getting one goof rout out of the interference "cloud" that is not possible if not having routers nears them.

One more thing My RaspBee I is having one very good radio but its bad then many sleeping devices is using it as parent but its being one long way and is not good then getting interference that is blocking the radio channel in one part or the way to the coordinator then they is using very low power.

Also its not only one way you is needing good signal you need it both ways to and from the device or they cant communicating.

benblb, 2021-11-03:
@MattWestb I took your advice and moved the light on my stairs from Hue into ZHA. I connected my upstairs motion sensors to them, while everything else connects to the coordinator directly. I hope this helps solve the problem! Thanks for your replies.

benblb, 2021-11-03:
Unfortunately adding a nearby router doesn't appear to have solved the problem. The gap at the start of the screenshot is where I added the sensor via the Hue bulb. 

[image]

MattWestb, 2021-11-04:
Interesting !!
Can you putting up little debug logging so you can see what the sensor is sending if its sending all commands OK to the coordinator. 
I think  it shall sending one command for movement true and one false then its have timed out (some types is only sending movement and the system is adding the false then the rime have ending) ?
https://github.com/zigpy/zigpy/wiki/Zigpy-Advanced-Configuration

benblb, 2021-11-04:
Yep, here are just ~25 minutes of logs. Sorry there is so much, but debugging is enabled and there was a sensor problem + one that took too long to timeout within this timeframe. The log was too big to paste here or on PasteBin, sorry it's so long! 

**Password**: halogs
**Log file**: https://easyupload.io/jjh8t5

MattWestb, 2021-11-04:
I was only making on fast look and searching for error and somthing is not OK with your system.

Some user is getting problem with ZHA and deCONZ radios if some integrations is bad behaving and locking the system then deCONZ radios is critical if the continuation speed to them is stabile. If getting long IO time from the SD-card on one PI can making them reeking out also if have connection problem with some network service can do the same.

You was getting time out  from the light that is one router and is its having OK radio link to the coordinator it shall not happening. Also was the coordinator not successful with finding one rout (router discovery) and that is bad then its not knowing where the device is in the network.

I dont knowing witch version you have on your radio module but if its not on the latest i recommending updating it then its some fixes for the continuation stability.
Also try disabling or fixing the integrations that is not working / getting contact with its servers so HA is not getting locks.

Try getting HA running without integration problems and logging what is happening on the zigbee side can getting the clue what more can being wrong but i think its integrations that is locking and perhaps the firmware in your radio.

benblb, 2021-11-04:
I believe the integration you're referring to is "Amcrest" and I resolved it. I also resolved the "NWS" integration errors. 

I updated the firmware on my Conbee II to a slightly newer version (not newest) that a Reddit user is having success with. Can upgrade again later if the problem is not resolved. I'll keep an eye on the logs for "timeout". Are there any other keywords I should be looking for?

MattWestb, 2021-11-04:
Looking for time out is good then the zigbee network is getting is it bad and if getting error its being somthing being lost.

You shall not getting one timeout to one router then its normal bad in radio communication but with decCONZ its can being that the system is being slow / having IO problems.

Try and see if its being better or not and also different firmware on the deCONZ radio if you have time to do that.

benblb, 2021-11-04:
No help needed at this time but figured an update is in order. I flashed the newest firmware on my Conbee II but still noticed issues. I checked my WiFi analyzer app again and saw some signals on WiFi 3 and 4 at around -65dB so I decided to switch to channel 25 which is open. 

I deleted ZHA and used deCONZ to switch the Conbee II to channel 25. I then added the ZHA channel 25 config, restarted, and began to setup ZHA again. I deleted and re-added each device one-by-one. 

I don't have enough data yet to share if the change worked, but so far my sensors are behaving like they were before in their reset times. When they were working, my sensors would always reset between 1 and 2 minutes, but the timing varied. So far, they are all within that range. Fingers crossed!

MattWestb, 2021-11-04:
I do the same !!!

One thing with your Philips HUE lights, Most (or not all) have one bad thing in there firmware that making they is not deleting children that have moved to one other parent so end device can having old links in the network map. It normally no large problem but can good to knowing if some of your sensors is jumping around (therefor i have putting my HUE lights in the black box).

Hope that resetting the sensors and pairing them is not making the them working for some week and then start being bad ones !!

benblb, 2021-11-04:
Thanks! One thing I have noticed (and it's no big deal if everything works lol) is that ZHA seems to be slower than deCONZ. 

Also my longest sensor reset time is now 2m 12s :/ If it doesn't get longer than 3 minutes I can live with it.

MattWestb, 2021-11-04:
I have one Aqara magnet (Window /door sensor) and its reporting very fast then testing it in (Z)HA GUI but i have one "IKEA Billy EZSP" as coordinator on my PI3B+ so cant comparing it with CornBee II but i have one RaspBee I that is my X production system but i is using it for touch link resting problem lights now so cant comparing them.

benblb, 2021-11-05:
Some more logs, these have errors now. 

`1-11-04 19:38:35 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-11-04 19:38:35 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-11-04 19:38:35 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 176: f0
2021-11-04 19:38:35 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb0 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xa064 endpoint=1>, status: 0xf0
2021-11-04 19:38:35 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 176 req id frame: TXStatus.MAC_TRANSACTION_EXPIRED
2021-11-04 19:38:35 DEBUG (MainThread) [zigpy.device] [0xa064] Delivery error for seq # 0xaf, on endpoint id 1 cluster 0x0000: message send failure

### New block ###

2021-11-04 19:50:57 DEBUG (MainThread) [homeassistant.components.zha.entity] light.stairs: polling current state
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 210 under 211 request id, data: b'00d2000000'
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 211, 0, <DeconzAddressEndpoint address_mode=2 address=0xC65B endpoint=11>, 260, 6, 1, b'\x00\xd2\x00\x00\x00', 2, 0)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 211]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc65b>, 11, 260, 6, b'\x18\xd2\x01\x00\x00\x00\x10\x00', 0, 175, 255, 55, 99, 123, 0, -64]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy.zcl] [0xc65b:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=210 command_id=Command.Read_Attributes_rsp>
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc65b>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'18d2010000001000'
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 211: 00
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd3 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc65b endpoint=11>, status: 0x00
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 212 under 213 request id, data: b'00d4000000'
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 213, 0, <DeconzAddressEndpoint address_mode=2 address=0xC65B endpoint=11>, 260, 8, 1, b'\x00\xd4\x00\x00\x00', 2, 0)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 213]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 213: e9
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd5 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc65b endpoint=11>, status: 0xe9
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 213 req id frame: TXStatus.MAC_NO_ACK
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy.device] [0xc65b] Delivery error for seq # 0xd4, on endpoint id 11 cluster 0x0008: message send failure
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 214 under 215 request id, data: b'00d60008000700030004000240'
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (28, 215, 0, <DeconzAddressEndpoint address_mode=2 address=0xC65B endpoint=11>, 260, 768, 1, b'\x00\xd6\x00\x08\x00\x07\x00\x03\x00\x04\x00\x02@', 2, 0)
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 215]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
[... cut ...]
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 215: 00
2021-11-04 19:50:57 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd7 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc65b endpoint=11>, status: 0x00
2021-11-04 19:51:02 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC65B:11:0x0300]: failed to get attributes '['color_mode', 'color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster:`

MattWestb, 2021-11-05:
Pleas put ` ``` ` on one line before and after the code block so its missing the formatting and is easier to reading.

The `TXStatus.MAC_TRANSACTION_EXPIRED`  is somthing was sent but was not getting and response from the receiver but i dont knowing in witch layer but the request was ACKed of the device so it have received it but the CornBee II have not getting the expected replay from the device or the device have not getting the request and there for not replaying to it.

The `TXStatus.MAC_NO_ACK` is from the underlying IEEE 802.15.4 network that have sending somthing to one device and waiting for one very short ACK from the device but it have not getting it and i think its after the CornBee II have re sending the command 3 times or more and you is getting one `message send failure`.

Your radio channel is not blocked of WiFi or somthing then you is getting more error messages but somthing is not  good in the Zigbee network but i cant  saying what it is.

Then if the communication from the host system is not stable (high IO or bad timing) is the Cornbee not answer and receiving ACKs OK then it have much to do on the comport to the host (I have sniffing my RaspBee I and it can being very "lazy" ACKing packages from devices and they must re sending many times for getting it thru).

Then the coordinator have problems with communicating with routers then is having larger problem doing it good with end device (the routing problem in your previous post included).

I was thinking changing the light you is having as router if its lazy and putting one other from your HUE system there or the not supported device that you was removing then if its one route device its doing its routing work as router in the network only that its upper functions is not working in ZHA shall not being any problem.

If its not helping i thinking its looks more and more its one instability in the CornBee II then you is getting the same in deCONZ and ZHA but i cant saying for 100%.

benblb, 2021-11-05:
Previous post edited with ```, good formatting advice!

I just ordered a used TI CC2531 coordinator USB with Z Stack Firmware for $10 to test. I also believe the Conbee could be the issue. 

One thing to note is I am running Home Assistant Supervised on Debian 11. Maybe there's a setting/configuration I need to adjust?

MattWestb, 2021-11-05:
Thanks for the formatting but is now not needed mow then i was cutting all the dates for getting it easier to read :-)))

The TI CC-2531 is one very bad /  old hard and and software but is normally working very well in small networks and the good is that you can flashing it with sniffer firmware and you can see what is going wrong i your system so not wasted money then having problem !!

And you is getting 4 for the price of one CornBee II :-))))

I think model manager is need being disable or uninstalled in Linux or it can being that is trying using the CornBees comport and you can getting problems.

Corn / RaspBee is very nice device but they have very much no standard things in there firmware that is making the Zigbeee line hard and deCONZ is very closed code and is only living because some in the community is adding devices in there system but is like year 1363 with old development tools and crappy code.

benblb, 2021-11-05:
> The TI CC-2531 is one very bad / old hard and and software but is normally working very well in small networks

Figured I would buy something super cheap to see if Conbee is the problem. If the CC-2531 works, I'll but something nicer :) But my network is 9 devices in a small space so I figured it could work well, at least for testing. 

 > you can flashing it with sniffer firmware and you can see what is going wrong

That's very cool! I would need to buy the equipment to flash it, but good idea. 

> I think modem manager is need being disable or uninstalled

Thanks, I did disable that previously. 

> deCONZ is very closed code and is only living because some in the community is adding devices in there system but is like year 1363 with old development tools and crappy code

I left deCONZ because I didn't like the support/community, ZHA is better for collaboration.

MattWestb, 2021-11-05:
First testing if its working with the CC-2531 so you is knowing then re flashing if you like.
I have flashing my 2 test CC-2531 with my RPi 3B+ and it was working OK but taking some minutes and is not easy if not having one cable with for the debug connection then its 1/2 the normal 2.5 mm pitch and its good for testing if not having many expensive coordinator laying around (i dont).
Also if you is getting it working with the CC you can flashing the ConrnBee II with sniffing firmware but i have not testing that (then i dont have one) but shall working if reading deCONZ forum but that you do later then have all possible you need.

Im one of the few user that is being banned from there git and i have seen so bad things like one young France user with ConrBee II that was corrupting the IEEE addresses (its the long unique address of the device) and was not possible taking with devices and hi was not getting any help and in the end hi was putting his system in one box and trying forgetting all ;-( 
And all is user that have paying for there device **and** support.

I is normally using IKEA Zigbee modules flashed with standard coordinator firmware that is working great and can being used as sniffer or coordinator without re flashing and costing lesser then your CC-2531 and is have better performance at my old RaspBee I and around your CornBee II but is very good supported in the community.

benblb, 2021-11-05:
Oh! One thing I noticed and wanted to mention is that after a Home Assistant restart (just Home Assistant, not the host system), the sensors all behave perfectly for 30-60 minutes. It's only over time that they begin to send strange data.

MattWestb, 2021-11-05:
The system is making one soft reset of the coordinator then it starting so if its problem in the ConBee is starting from queerer one.

And is also looks like your Zigbee network is being slow and very likely is being the same  root for it.

Wait and testing the you have your CC :-))

benblb, 2021-11-05:
Sounds good! I'll let you know how it goes :)

benblb, 2021-11-06:
So far the sensors have been rock solid today! This morning I double checked that ModemManager was disabled, but I also made sure to disable WiFi (no network was set up) on my host system just in case of interference. I am fairly confident my change to channel 25 yesterday was helpful. 

I also got to the root cause of my initially filed "bug" (it's not a bug). Occasionally the sensor will remain in a detected state if motion is continuous. For example, if I keep moving around in a room for 5 minutes, motion will remain detected for 5+ minutes. Today I was cleaning in my living room for about 15 minutes and strangely enough the motion remained on. I figured this was more than just a coincidence (see pic). 

[image]

In deCONZ a new message is sent each time motion is triggered which is not how things work in ZHA. A couple of adjustments to my Node Red motion lights flow and everything is working great! 

My only (minor) complaint at this point is the speed of the network. I believe deCONZ responded a touch faster than ZHA so if anyone has tips on how to speed things up I would be grateful. Do routers slow sensors down? 

Going to close this issue out, but if someone else runs into this problem in the future I hope the above troubleshooting and resolution is helpful.

closed 2021-11-06 by benblb

MattWestb, 2021-11-06:
In ZHA its i think 4 different "working modes" of motion sensors that is working in different way "under the hood" (IKEA motions is sending on with off time 1 minute to lights and other only one "motion detect" without reset and the reset is being "made" of ZHA) and if one sensor is sending the same message ZHA is not updating the sate of the sensor but i think you is getting it as event or its logged in the system message bus.

Zigbee is using message forwarding but the routers is decrypting the frames and encrypting it before sending it the next router (hopes / jumps) but its being made very fast like in this sniff of one device is using one router for attribute changes to the coordinator:
```
No.	Time	Protocol	IEEE Src	IEEE Dst	Zigbee Src	Zigbee Dst	ZBN Dst	Group Nr	ZBN Seq	ZBA Seq	ZDP Seq	Nwk Seq	Src EP	Dst EP	Info
788	20:18:50,659422	ZigBee HA	0x5c1f	0x6701	0x5c1f	0x0000	0x0000		210	155		233	1	1	ZCL: Report Attributes, Seq: 7
789	20:18:50,666573	ZigBee HA	0x6701	0x0000	0x5c1f	0x0000	0x0000		210	155		157	1	1	ZCL: Report Attributes, Seq: 7
790	20:18:50,667673	ZigBee	0x0000	0x6701	0x0000	0x5c1f	0x5c1f		45	155		83	1	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
791	20:18:50,678230	ZigBee	0x6701	0x5c1f	0x0000	0x5c1f	0x5c1f		45	155		158	1	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
```
from the time stamp: 666573 - 659422 = 0.007151 seconds for one jump = no visibly time. 
If the network is working OK it shall being like that but you can sniffing and looking then you have getting your CC stick. (the sniff above is not showing the real low level acks but is not interesting in this case)

15 min for cleaning the living room i very good i need mush longer time for dot that ;-((

benblb, 2021-11-08:
Thanks. That small amount of time is definitely not noticeable. Everything worked perfectly for me over the weekend so my issues are officially resolved. I appreciate all your help. Thanks again!

MattWestb, 2021-11-08:
Sounds great and glad to helping you getting your system working OK !!

May i getting remote access to your system so i can checking that you is doing the cleaning well ?   ;-))  (its one joke).

If you have one micro oven its 150% killing your network near it then its flooding heavy RF in all spectrum and if one Zigbee device is near it they knocked out but bust normally user dont knowing that.

benblb, 2021-11-08:
> May i getting remote access to your system so i can checking that you is doing the cleaning well ? ;-)) (its one joke).
Lol 

> If you have one micro oven its 150% killing your network near it then its flooding heavy RF in all spectrum and if one Zigbee device is near it they knocked out but bust normally user dont knowing that.

I noticed that my WiFi security camera crashes when I run the microwave! Horrible for WiFi and Zigbee.