zigpy/zha-device-handlers #1134: [Device Support Request] Aqara Wall Switch EU version (no neutral, double rocker)

vsergeenko777, 2021-11-08:
**Is your feature request related to a problem? Please describe.**
I have several Aqara Smart Wall Switches (no neutral, single/double rocker) EU versions and they work almost perfectly.

The problem exists only with double rocker version (model WS-EUK02), state of the right rocker is not changed if physically switch it.

**Describe the solution you'd like**
This issue has fixed in [zigbee2mqtt#6524](https://github.com/Koenkk/zigbee2mqtt/issues/6524) and [this](https://github.com/Koenkk/zigbee2mqtt/issues/6524#issuecomment-872421599) looks like a solution.

**Device signature**
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l2aeu1",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2021-11-15:
@dmulcahey sorry David i was putting in the wrong issue number so it was closing the wrong issue. Can you pleas reopen this issue ?

Also closing the https://github.com/zigpy/zha-device-handlers/issues/1149

Sorry for making troubles for you ;-(((

Ditskys, 2022-04-25:
Just bought one of these and looking to setup this week. Does the right rocker still not report the state correctly? How could I help with fixing it?

khrazik, 2022-04-26:
It seems that ZHA does not expose zigbee clusters to enable decoupled mode on these switches, even though they should be supported. [https://www.zigbee2mqtt.io/devices/WS-EUK02.html](Zigbee2MQTT seems to have these clusters supported).

Is it possible to add these features to the ZHA quirk?

vsergeenko777, 2022-04-27:
Best solution is zigbee2mqtt, it supports much more devices than zha. Since the creation of this issue, I have moved to z2m and have no problems with these wall switches or any other zigbee devices.

Lampyridae1984, 2022-05-18:
I just ordered the double rocker version, thinking it would work just like the single one. Bummer. 😞 Anything I can do to help, once I get it? Not really an expert on debugging, but I'm willing to give it a go if someone can shove me in the right direction.

Lampyridae1984, 2022-05-18:
@vsergeenko777, is it still working properly for you with zigbee2mqtt? It seems it has, yet again, broken down for someone: [https://github.com/Koenkk/zigbee2mqtt/issues/6524](url)

vsergeenko777, 2022-05-18:
> @vsergeenko777, is it still working properly for you with zigbee2mqtt? It seems it has, yet again, broken down for someone: [https://github.com/Koenkk/zigbee2mqtt/issues/6524](url)

Yes, I have two WS-EUK02 working fine on zigbee2mqtt 1.25.0.
[image]

Lampyridae1984, 2022-05-18:
> Yes, I have WS-EUK02 working fine on zigbee2mqtt 1.25.0.

Fair enough.

vsergeenko777, 2022-05-18:
I also have WS-EUK01, WRS-R02 and many other Aqara devices that work great with zigbee2mqtt.

Lampyridae1984, 2022-05-18:
> I also have WS-EUK01, WRS-R02 and many other Aqara devices that work great with zigbee2mqtt.

Is this your way to lure me over to zigbee2mqtt? 😉😂

vsergeenko777, 2022-05-18:
> > У меня также есть WS-EUK01, WRS-R02 и многие другие устройства Aqara, которые отлично работают с zigbee2mqtt.
> 
> Это твой способ заманить меня на zigbee2mqtt?😉😂

Yeah 🙂

HemiBob, 2022-05-25:
None of my Aqara Wall Switch EU versions (no neutral) is supported in zigbee2mqtt, will support appear soon or shall I return them to Amazon?

closed 2022-11-28 by github-actions[bot]

plantoschka, 2023-07-21:
Switched to ZHA a couple of days ago and everything works great but this is definitely the thing I miss most. I have a couple of those switches I used in decoupled mode and it's no longer necessary to set this in ZHA.
Would be great if anyone could look into this.

PR #1152 (2021-11-15): Moving tuya 0xE000X cluster to INIT and adding TS000X quirk
Moving the tuya PITA clusters to INIT and adding functions for switch types to one of them.
Adding TS000X device quirks that  is using it and also the TS011F have it but i dont knowing if its being implanted in the devices.

New TS000X devises is testes but not  the last have the user not reported back.

The device type is ON_OFF_LIGHT for all cluster but the user like have it as ON_OFF_PLUG_IN_UNIT then its getting on flash as icon  in ZHA is that  OK ?

The calss class TuyaZBExternalSwitchTypeCluster(CustomCluster): is not having an [... cut]