zigpy/zha-device-handlers #1135: [Device Support Request] Update for lumi.plug quirk for total energy consumption

Brazen00, 2021-11-08:
The Xiaomi smart plug (lumi.plug) is not appearing in the Home Assistant Energy integration under Individual Devices.

See https://github.com/home-assistant/core/issues/59032

According to @Adminiuga "The quirk for lumi plug in zha-device-handlers needs to get updated to provide summation attribute for energy consumed."

Device signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug",
  "class": "zhaquirks.xiaomi.aqara.plug.Plug"
}
```

Pairing log (hopefully I managed to capture all the relevant info):

```
2021-11-09 08:16:58 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0xa5cd, 00:15:8d:00:01:b7:97:df
2021-11-09 08:16:58 INFO (MainThread) [zigpy.application] New device 0xa5cd (00:15:8d:00:01:b7:97:df) joined the network
2021-11-09 08:16:58 DEBUG (MainThread) [zigpy.device] [0xa5cd] Scheduling initialization
2021-11-09 08:16:58 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5CD ieee=00:15:8d:00:01:b7:97:df is_initialized=False> from ep 0 to ep 0, cluster 19: b'\xa1\xcd\xa5\xdf\x97\xb7\x01\x00\x8d\x15\x00\x8e'
2021-11-09 08:16:58 DEBUG (MainThread) [zigpy.zdo] [0xa5cd:zdo] ZDO request ZDOCmd.Device_annce: [0xA5CD, 00:15:8d:00:01:b7:97:df, 142]
2021-11-09 08:16:58 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-11-09 08:16:58 INFO (MainThread) [zigpy.device] [0xa5cd] Requesting 'Node Descriptor'
[... 61 lines cut ...]
2021-11-09 08:17:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA5CD:3:0x000c]: 'async_initialize' stage succeeded
2021-11-09 08:17:26 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA5CD](lumi.plug): power source: Mains
2021-11-09 08:17:26 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xA5CD](lumi.plug): completed initialization
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_plug_df97b701_electrical_measurement
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_plug_df97b701_electrical_measurement_rms_current
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_plug_df97b701_electrical_measurement_rms_voltage
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_plug_df97b701_analog_input
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_plug_df97b701_analog_input_2
2021-11-09 08:17:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA5CD:1:0x0006]: attempting to update onoff state - from cache: False
2021-11-09 08:17:26 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (00:15:8d:00:01:b7:97:df, 1, 0, 5, 'lumi.plug') params: FOREIGN KEY constraint failed
2021-11-09 08:17:26 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=62 command_id=Command.Read_Attributes_rsp>
2021-11-09 08:17:26 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new switch.zha entity: switch.lumi_lumi_plug_df97b701_on_off
2021-11-09 08:17:31 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=124 command_id=Command.Read_Attributes>
2021-11-09 08:17:31 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:1:0x000a] ZCL request 0x0000: [[0]]
2021-11-09 08:17:33 INFO (MainThread) [homeassistant.components.nmap_tracker] No MAC address found for 192.168.1.133
2021-11-09 08:17:44 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:2:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=138 command_id=Command.Report_Attributes>
2021-11-09 08:17:44 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:2:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=48.619998931884766>)]]
2021-11-09 08:17:44 DEBUG (MainThread) [zigpy.zcl] [0xa5cd:2:0x000c] Attribute report received: present_value=48.619998931884766
2021-11-09 08:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA5CD:1:0x0b04]: async_update
2021-11-09 08:17:49 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (00:15:8d:00:01:b7:97:df, 1, 2820, 1293) params: FOREIGN KEY constraint failed
2021-11-09 08:17:49 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (00:15:8d:00:01:b7:97:df, 1, 2820, 1288) params: FOREIGN KEY constraint failed
2021-11-09 08:17:49 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (00:15:8d:00:01:b7:97:df, 1, 2820, 1290) params: FOREIGN KEY constraint failed
2021-11-09 08:17:49 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (00:15:8d:00:01:b7:97:df, 1, 2820, 1285) params: FOREIGN KEY constraint failed
2021-11-09 08:17:49 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (00:15:8d:00:01:b7:97:df, 1, 2820, 1287) params: FOREIGN KEY constraint failed
```

erik-de-neve, 2021-11-29:
Any update on this?

hacshacdgacs, 2022-05-30:
Have updated and am confirming that this plug still doesn't work.

tias2001, 2022-08-03:
yes still not working

SpeedFire0, 2022-08-16:
Please fix this. Energy consumption is a very important part of this device and we don't want to loose it in ZHA.
Please add!

cobch, 2023-01-03:
Same issue here. Did you find a fix?

hacshacdgacs, 2023-01-03:
I did not, I've been slowly replacing this device because of this issue

cobch, 2023-01-03:
Alternativ is to switch to Z2m? I see... I have just received the SkyConnect but did not see any support yet on z2m. Maybe I should wait in that case since I want to switch from deconz to something more futur proof...

hacshacdgacs, 2023-01-03:
My hub doesn't allow me to use ZigBee2Mqtt so unfortunately that's not an
option for me

TheJulianJES, 2023-01-04:
Please post the diagnostic information from a ``lumi.plug`` that has been plugged in for some time with a load attached.

> I have just received the SkyConnect but did not see any support yet on z2m. 

It's experimental support but you just need to set the adapter type to "ezsp".

hacshacdgacs, 2023-01-04:
[image]
Attached is an image that shows the sensors I can see.

TheJulianJES, 2023-01-04:
The "0.4" seems to be a power consumption of "0.4 W", right?
And is "130.7" the total energy consumption? (So does that value only ever increase?)

If so, we'd just need to put the values on a ``LocalDataCluster`` (energy and metering).

Also, what's the binary input sensor for the plug doing?

hacshacdgacs, 2023-01-04:
That value only ever increases, yes. I assume the binary sensor is the
power for the plug (on-off)

TheJulianJES, 2023-01-04:
Does the second AnalogInput automatically increase (when the attached device is powered on) or does it only update on HA restarts?

hacshacdgacs, 2023-01-04:
Both analog inputs only update when HomeAssistant restarts, and the binary
input is not tied into whether or not the switch is on (currently running a
dishwasher cycle and the binary input still says off)

TheJulianJES, 2023-01-04:
Can you use the "reconfigure this device" option for the plug and see if one or both AnalogInputs then update without restarting?

(At least the first one might need a power change of like 10W)

hacshacdgacs, 2023-01-04:
No luck, device reconfiguration fails and the inputs don't update.
[image]

TheJulianJES, 2023-01-14:
### Download
Please try this custom quirk: [TJJ_lumi_plug_test_v1.zip](https://github.com/zigpy/zha-device-handlers/files/10416313/TJJ_lumi_plug_test_v1.zip)

### Install Instructions
In your `configuration.yaml`, you might need to define the following if you don't have it already:
```yaml
zha:
  custom_quirks_path: /config/zha_custom_quirks
```
This assumes you're running Home Assistant OS (points to `/config`).

Then, put the `tjj_lumi_plug_v1` folder in that `zha_custom_quirks` folder, reboot HA, and then re-pair the plugs! (This is important)

In the end, the folder structure should look like this: `/config/zha_custom_quirks/tjj_lumi_plug_v1`
The last folder should contain two files: `__init__.py` and `plug.py`

## More Info
(Make sure you re-paired the plug!) (It's possible that you even need to remove the plug from ZHA and then re-add it)

Let me know what works with this quirk and what doesn't (if values update and so on).

If the total power consumption (i. e. the second ``AnalogInputCluster``) never updates, try uncommenting the commented lines in the ``plug.py`` file (marked with`UNCOMMENT BELOW TO TEST 2nd PART OF POSSIBLE FIX`):
Then, restart HA. A re-pairing *shouldn't* be necessary here.

hacshacdgacs, 2023-01-15:
Okay so enabling this
zha:
  custom_quirks_path: /config/zha_custom_quirks
  in my config causes my Zigbee network to crash.
 
[image]

TheJulianJES, 2023-01-15:
Ah yeah, (remove it for now). I'll fix it soon-ish

TheJulianJES, 2023-01-15:
Please try this: 
[TJJ_lumi_plug_test_v2.zip](https://github.com/zigpy/zha-device-handlers/files/10420799/TJJ_lumi_plug_test_v2.zip)

hacshacdgacs, 2023-01-15:
Okay now the first analog input is updating but the second even with the
possible fix is still refusing to update.
(image showing the values of the first)
[image: image.png]

TheJulianJES, 2023-01-15:
Images apparently aren't sent correctly via email. (Can you send it again using GitHub directly?)
And just to confirm, you tried removing the plug from HA and re-pairing it while the "v2 quirk" is installed?

Also, what coordinator do you have (and what firmware)?

jonwaland, 2023-01-16:
updated quirk, restarted HA, removed device, re-paired.

without 2nd part enabled, minimal load (xiaomi bedside lamp powered, but lamp off):
[image]

without 2nd part enabled, higher load (xiaomi bedside lamp on):
[image]

without 2nd part enabled, higher load after 5 minutes (xiaomi bedside lamp on) - no change on second analogue input
[image]

WITH 2nd part enabled, higher load (xiaomi bedside lamp on) enabled:
[image]

WITH 2nd part enabled, higher load (xiaomi bedside lamp on) enabled, after one hour:
[image]

TheJulianJES, 2023-01-16:
Please try this: [TJJ_lumi_plug_test_v3.zip](https://github.com/zigpy/zha-device-handlers/files/10424079/TJJ_lumi_plug_test_v3.zip)

You'll need to re-pair the plug again though. A new sensor for the total consumption *should* show up. (The ``AnalogInput`` sensors can be ignored then). Let me know if the total consumption sensor updates properly.

I've also increased the logging to ``warning`` of Xiaomi attribute reports for that plug, so you'll (hopefully) see some logs containing: ``DEBUG TJJ QUIRK``. These are expected (please send them here).

jonwaland, 2023-01-16:
Will be another 12 hours before **I** can test. But will do.

TheJulianJES, 2023-01-16:
Test it without uncommenting that part first.

jonwaland, 2023-01-17:
updated quirk, restarted HA, removed device, re-paired.

without 2nd part enabled, minimal load (xiaomi bedside lamp powered, but lamp off):
[image]

without 2nd part enabled, maxload (xiaomi bedside lamp powered, and lamp on full brightness):
[image]

without 2nd part enabled, maxload (xiaomi bedside lamp powered, and lamp on full brightness) - 13 minutes later - summation delivered has increased.
[image]

debug:
```
2023-01-17 14:56:04.678 WARNING (MainThread) [lumi2] 00:15:8d:00:02:8b:71:cf - (DEBUG TJJ QUIRK) Attribute report. attribute_id: [65281] value: [{'0xff01-100': <Bool.false: 0>, 'temperature': 25, 'power': 0.0, 'consumption': 11.694486618041992, 'voltage': 2380, 'X-attrib-5': 51, '0xff01-154': 16, '0xff01-8': 4886, '0xff01-7': 0, '0xff01-9': 512}]
2023-01-17 15:01:47.622 WARNING (MainThread) [lumi2] 00:15:8d:00:02:8b:71:cf - (DEBUG TJJ QUIRK) Attribute report. attribute_id: [65281] value: [{'0xff01-100': <Bool.true: 1>, 'temperature': 39, 'power': 5.296999931335449, 'consumption': 11.694499969482422, 'voltage': 2380, 'X-attrib-5': 51, '0xff01-154': 0, '0xff01-8': 4886, '0xff01-7': 0, '0xff01-9': 512}]
2023-01-17 15:06:52.621 WARNING (MainThread) [lumi2] 00:15:8d:00:02:8b:71:cf - (DEBUG TJJ QUIRK) Attribute report. attribute_id: [65281] value: [{'0xff01-100': <Bool.true: 1>, 'temperature': 39, 'power': 13.088000297546387, 'consumption': 11.69559097290039, 'voltage': 2380, 'X-attrib-5': 51, '0xff01-154': 0, '0xff01-8': 4886, '0xff01-7': 0, '0xff01-9': 512}]
2023-01-17 15:13:54.615 WARNING (MainThread) [lumi2] 00:15:8d:00:02:8b:71:cf - (DEBUG TJJ QUIRK) Attribute report. attribute_id: [65281] value: [{'0xff01-100': <Bool.true: 1>, 'temperature': 39, 'power': 13.088000297546387, 'consumption': 11.697200775146484, 'voltage': 2380, 'X-attrib-5': 51, '0xff01-154': 0, '0xff01-8': 4886, '0xff01-7': 0, '0xff01-9': 512}]
```

TheJulianJES, 2023-01-17:
So "active power", "summation delivered", and "voltage" are updating correctly now (without restarting HA)?

(Ignore power factor and AnalogInput sensors)

jonwaland, 2023-01-17:
yes - active power, summation delivered, and voltage appear to be correctly updating (Note that I restarted HA as part of redoing the quirk)

[image]
[image]
[image]

(summation is increasing, but the graph scale hides it)
[image]

closed 2023-01-23 by dmulcahey