zigpy/zha-device-handlers #1136: Ok I achieved to test my quirk, and now my device is working, see the result :

juhanjuku, 2021-11-09:
Ok I achieved to test my quirk, and now my device is working, see the result :
[image]
 I will try to make a pull request soon, it is just a tiny change in the `siren.py` file.

_Originally posted by @CharlieBailly in https://github.com/zigpy/zha-device-handlers/issues/648#issuecomment-754956900_

juhanjuku, 2021-11-10:
I'm running  core-2021.11.2 on Rasperry Pi4
For me TS0601 by _TZE200_d0yu2xgi is still not working
[image]

CharlieBailly, 2021-11-10:
Hi, can you post your device signature and some logs as I did in the issue ?

juhanjuku, 2021-11-10:
Sinature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_d0yu2xgi",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_siren.TuyaSiren2"
}
and logfile
[log.txt](https://github.com/zigpy/zha-device-handlers/files/7514007/log.txt)

juhanjuku, 2021-11-10:
Now even worse. After removing and adding siren again, I don't have temperature and humidity sensors there anymore.
Tried several times.
[image]

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_d0yu2xgi",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_siren.TuyaSiren2"
}
```
[log.txt](https://github.com/zigpy/zha-device-handlers/files/7514719/log.txt)

MattWestb, 2021-11-10:
Delete the device and wait one minute and restarting HA and pairing it then ZHA is stable after the restart.

I was having somthing like that yesterday the ZHA was reusing old cashed data and i was not getting all sensors on my TRV.

juhanjuku, 2021-11-10:
You are right. After HA restart temp and humid are back, but still 'Unknown'
[image]

Hedda, 2021-11-12:
Are you also looking at adding support for the new siren entity platform for Zigbee sirens, doorbells and chimes to the ZHA integration component in Home Assistant core? As per the feature requested here:

https://community.home-assistant.io/t/zha-integration-to-use-siren-entity-platform-for-zigbee-sirens-doorbells-and-chimes/343075

_Request ZHA integration 1 use recently new siren entity platform introduced in Home Assistant 2021.8_

_" A siren entity is a device whose main purpose is to control siren devices like a doorbell or chime."_

[https://www.home-assistant.io/blog/2021/08/04/release-20218/#siren ](https://www.home-assistant.io/blog/2021/08/04/release-20218/#siren)

[https://www.home-assistant.io/integrations/siren/ ](https://www.home-assistant.io/integrations/siren/)

[https://developers.home-assistant.io/docs/core/entity/siren/ ](https://developers.home-assistant.io/docs/core/entity/siren/)

_So far Z-Wave JS, DeCONZ, and Tuya integrations (Devolco is WIP) has added siren entity platform:_

https://github.com/home-assistant/architecture/issues/375

https://github.com/home-assistant/core/pull/52948

https://github.com/home-assistant/core/pull/56397

https://github.com/home-assistant/core/pull/57780

https://github.com/home-assistant/core/pull/53400

alexsands, 2021-11-13:
I'm also experiencing this issue as are [others in the HA forums](https://community.home-assistant.io/t/no-entities-in-zigbee-tuya-siren-with-zha/256927/42). It looks like Tuya updated the siren firmware while keeping the model and manufacturer IDs the same, so @CharlieBailly's PR to add Tuya2 is picking these up as well.

Charlie - do you have any guidance on this, having gotten the Tuya2 quirk to work in the past? My signature is the same as what @juhanjuku posted above. I'm happy to help or open a PR with a possible fix, but I'm not exactly sure where to start / how to test it (HA running in VirtualBox).

@Hedda siren support for ZHA would be great, but in the meantime I think this can also be solved with something similar to Charlie's original PR.

CharlieBailly, 2021-11-14:
Hi, it's been a while but basically when i was testing, I set up a virtual box VM(Virtual Machine) of home assistant on my laptop (you can follow the official instructions of home assistant), shut down my actual home assistant setup (on my raspberry pi) so it does not interfer with the testing. Then setup ZHA on the VM, and using the portainer addon and following the instructions on the readme of this repo you can setup custom quirks and they should be applied on restart of your home assistant VM.
Once the testing was done I simply shut down the VM power back on the raspberry and my original zigbee network was still working (I was using a Sonoff Zigbee bridge flashed with tasmota).
Don't hesitate to ask if it's not clear.

juhanjuku, 2021-11-14:
Sorry, but I'm no programmer. I have made only one very basic program in Python.
I understand what is quirk, but how to add my changes to my HA is mystery for me
So I rather relay on more experience guys work and try to help with testing and feedback.

CharlieBailly, 2021-11-14:
Ok, so can you try to remove the python file where the siren quirk is and give us back the signature and logs without the quirk TuyaSiren2 applied ? Because when the quirk is applied, the signature is the one replaced by the quirk.

juhanjuku, 2021-11-14:
I'm glad to do this, but where is this python file with quirks? Also, what is signature?
[image]

[image]

CharlieBailly, 2021-11-14:
Everything is explain in the Readme of this repo :

Signature : https://github.com/zigpy/zha-device-handlers#signature

Testing using portainer :
https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

Then you would need to remove this file :
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_siren.py
Then reboot.

You need portainer to access the file inside the home assistant container (unfortunately, you can't access theses file from your `192.168.1.10/config` folder)

jacekk015, 2021-11-24:
@CharlieBailly It's not needed.
You need just to add that to configuration.yaml
```
zha:
  custom_quirks_path: /config/zha_custom_quirks/
```
and put your quirk there. It will be processed before installed quirk.

CharlieBailly, 2021-11-25:
Ok good to know, but can you remove an existing quirk using this method ?

juhanjuku, 2021-11-25:
@CharlieBailly I'm confused with portainer. Is it supported or not?
[image]

MattWestb, 2021-11-25:
@CharlieBailly If you is installing one local quirk its overriding the quirk that is in the (Z)HA container then its have higher priority so you dont need deleting the original quirk only making your local quirk have the right signature so ZHA is loading it for your device.

CharlieBailly, 2021-11-25:
> @CharlieBailly I'm confused with portainer. Is it supported or not? [image]

Mmm apparently it is not supported anymore. But there is also a way to get a shell in the container using ssh and port 2222, check the home assistant doc.

CharlieBailly, 2021-11-25:
> @CharlieBailly If you is installing one local quirk its overriding the quirk that is in the (Z)HA container then its have higher priority so you dont need deleting the original quirk only making your local quirk have the right signature so ZHA is loading it for your device.

Yes but i wanted his ZHA to not use any quirk so we can easily see the original device signature in the UI. If you have another solution it's ok.

MattWestb, 2021-11-25:
Then deleting / inactivating the quirk in the container or repairing it and looking  in the log for then ZHA is requesting the cluster information of the device and endpoints.

juhanjuku, 2021-11-27:
I have finally succeeded to run custom quirk ts0601_siren. For start I didn't changed anything in original quirk except adding class TuyaSiren3(TuyaSiren)
After deleting and rejoining my siren I see in log:
[0xD1B9:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xD1B9:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0xD1B9:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
but then comes error:
Error handling '_save_attribute' event with (80:4b:50:ff:fe:04:74:f8, 1, 0, 4, '_TZE200_d0yu2xgi') params: FOREIGN KEY constraint failed

''' [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=183 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=38>)]]
[0x7714:1:0x0b04] Attribute report received: rms_current=38
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=194 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=0>), Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=0>)]]
[0x7714:1:0x0b04] Attribute report received: active_power=0, rms_current=0
[0x82f9:9:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x82f9:9:0x0702] ZCL request 0x000a: [[Attribute(attrid=1024, value=<TypeValue type=int24s, value=0>)]]
[0x82f9:9:0x0702] Attribute report received: instantaneous_demand=0
[0x82f9:9:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
[0x82f9:9:0x0702] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint48_t, value=236368>)]]
[0x82f9:9:0x0702] Attribute report received: current_summ_delivered=236368
[0x7714:1:0x0b04]: async_update
[0x800F:1:0x0b04]: async_update
[0x9868:1:0x0b04]: async_update
[0xE9C9:1:0x0b04]: async_update
[0x800f:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Read_Attributes_rsp>
[0x9868:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=63 command_id=Command.Read_Attributes_rsp>
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Read_Attributes_rsp>
[0xe9c9:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=65 command_id=Command.Read_Attributes_rsp>
[0x82f9:9:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
[0x82f9:9:0x0702] ZCL request 0x000a: [[Attribute(attrid=1024, value=<TypeValue type=int24s, value=1>)]]
[0x82f9:9:0x0702] Attribute report received: instantaneous_demand=1
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=231 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=70>)]]
[0x7714:1:0x0b04] Attribute report received: rms_current=70
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=237 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=42>)]]
[0x7714:1:0x0b04] Attribute report received: rms_current=42
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=243 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=2>)]]
[0x7714:1:0x0b04] Attribute report received: active_power=2
[0x7714:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=250 command_id=Command.Report_Attributes>
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=77>)]]
[0x7714:1:0x0b04] Attribute report received: rms_current=77
[... cut ...]
[0x7714:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=0>)]]
[0x7714:1:0x0b04] Attribute report received: rms_current=0
[0x82f9:9:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=9 command_id=Command.Report_Attributes>
[0x82f9:9:0x0702] ZCL request 0x000a: [[Attribute(attrid=1024, value=<TypeValue type=int24s, value=2>)]]
[0x82f9:9:0x0702] Attribute report received: instantaneous_demand=2 '''

Hedda, 2021-12-06:
FYI, support for siren entities in ZHA integration component has now been added to the Home Assistant core:

https://github.com/home-assistant/core/pull/60920

(that will maybe not make it into HA 2021.12 release? https://rc.home-assistant.io/blog/2021/12/03/release-202112/ )

attenzione, 2021-12-09:
> (that will maybe not make it into HA 2021.12 release? https://rc.home-assistant.io/blog/2021/12/03/release-202112/ )

I see it's included in the list of release changes ;)

juhanjuku, 2021-12-12:
Updated to HA 2021.12, removed siren, added again - still not working.
[image]
There are some errors in log
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/7698897/home-assistant.log)
Reconfiguration gives no errors
[image]

dmulcahey, 2021-12-14:
What isn’t working? Do the 2 services in developer tools work?

juhanjuku, 2021-12-14:
Humidity and temperature sensors show 'Unknown'
On_off switch does nothing.
With 'Manage clusters' I can't read any attribute.
Not sure if I make service call from developers tools correctly, but no response to 'siren.turn_on'

[image]

I'm ready to make any tests if you can advise me what to do

tom-winkler, 2021-12-29:
Added logsof my device - any hint on where to start bugfixing/debugging?
`New device 0x4378 (b4:e3:f9:ff:fe:02:e1:80) joined the network
[0x4378] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x78\x43\x80\xE1\x02\xFE\xFF\xF9\xE3\xB4\x80'
[0x4378:zdo] ZDO request ZDOCmd.Device_annce: [0x4378, b4:e3:f9:ff:fe:02:e1:80, 128]
Tries remaining: 3
[0x4378] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4378] Extending timeout for 0x38 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x38\x00\x78\x43\x02\x40\x80\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x4378] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x4378] Discovering endpoints
Tries remaining: 3
[0x4378] Extending timeout for 0x39 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x39\x00\x78\x43\x01\x01'
[0x4378] Discovered endpoints: [1]
[0x4378] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x4378:1] Discovering endpoint information
Tries remaining: 3
[0x4378] Extending timeout for 0x3a request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x3A\x00\x78\x43\x14\x01\x04\x01\x51\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xEF\x02\x19\x00\x0A\x00'
[0x4378:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x4378] Extending timeout for 0x3b request
[0x4378:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Read_Attributes_rsp>
[0x4378] Read model 'TS0601' and manufacturer '_TZE200_d0yu2xgi' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x4378] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0x4378 ieee=b4:e3:f9:ff:fe:02:e1:80 is_initialized=True>
Checking quirks for _TZE200_d0yu2xgi TS0601 (b4:e3:f9:ff:fe:02:e1:80)
Considering <class 'zhaquirks.tuya.ts0601_siren.TuyaSiren2'>
Found custom device replacement for b4:e3:f9:ff:fe:02:e1:80: <class 'zhaquirks.tuya.ts0601_siren.TuyaSiren2'>
device - 0x4378:b4:e3:f9:ff:fe:02:e1:80 entering async_device_initialized - is_new_join: True
device - 0x4378:b4:e3:f9:ff:fe:02:e1:80 has joined the ZHA zigbee network
[0x4378](TS0601): started configuration
[0x4378:ZDO](TS0601): 'async_configure' stage succeeded
[0x4378:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x4378:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x4378:1:0x0006]: finished channel configuration
[0x4378:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0x4378:1:0x0405]: Successfully configured reporting for '{'measured_value': (30, 900, 100)}' on 'humidity' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x4378:1:0x0405]: finished channel configuration
[0x4378:1:0x0000]: finished channel configuration
[0x4378:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x4378:1:0x0402]: Successfully configured reporting for '{'measured_value': (30, 900, 50)}' on 'temperature' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x4378:1:0x0402]: finished channel configuration
[0x4378:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (b4:e3:f9:ff:fe:02:e1:80, 1, 0, 4, '_TZE200_d0yu2xgi') params: FOREIGN KEY constraint failed
[0x4378:1:0x0006]: 'async_configure' stage succeeded
[0x4378:1:0x0405]: 'async_configure' stage succeeded
[0x4378:1:0x0000]: 'async_configure' stage succeeded
[0x4378:1:0x0402]: 'async_configure' stage succeeded
[0x4378:1:0x0019]: 'async_configure' stage succeeded
Error handling '_save_attribute' event with (b4:e3:f9:ff:fe:02:e1:80, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0x4378](TS0601): completed configuration
[... cut ...]
Ignoring message (b'000c000700') on cluster 10: unknown endpoint or cluster id: 'No cluster ID 0x000a on (b4:e3:f9:ff:fe:02:e1:80, 1)'
Ignoring message (b'000c000700') on cluster 10: unknown endpoint or cluster id: 'No cluster ID 0x000a on (b4:e3:f9:ff:fe:02:e1:80, 1)'
[0x4378:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=13 command_id=36>
[0x4378:1:0xef00] ZCL request 0x0024: [[0, 15]]
[0x4378:1:0xef00] No handler for cluster command 36`

alexsands, 2022-06-29:
I just upgraded to Home Assistant Core 2022.6.7 (+ latest ZHA), removed and re-added this device, and am still having issues. Like @juhanjuku mentioned above, the on/off switch on the siren entity does not do anything and both the temperature and humidity sensors are `Unknown`. Has anyone been able to get this device to work or any suggestions on where to start?

The device signature is below:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_d0yu2xgi",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_siren.TuyaSiren2"
}
```

amokrunner, 2022-07-06:
Experiencing same issue.
Home Assistant Core 2022.6.7
Home Assistant Supervisor 2022.06.2
Texas Instruments CC1352/CC2652, Z-Stack 3.30+ (build 20211217)

closed 2023-01-11 by github-actions[bot]