zigpy/zha-device-handlers #1137: [Device Support Request] TS0601 Cover. Zemismart (Tuya) bcm500ds-tyz curtain motor

dirkpitt35, 2021-11-09:
**Is your feature request related to a problem? Please describe.**
I would like to use this cover with HASS via ZHA. When added via ZHA the device is added but no entities are created.

**Describe the solution you'd like**
All control features for a curtain i.e. Open, Close, %

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_xaabybja",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Additional context**
None

MattWestb, 2021-11-10:
I have founding 2 candidate for adding your device ID`     ("_TZE200_xaabybja", "TS0601"),` :
https://github.com/zigpy/zha-device-handlers/blob/10dbcc4d64fdf748989e6d72ea169bc7163eb6ad/zhaquirks/tuya/ts0601_cover.py#L80-L82
or:
https://github.com/zigpy/zha-device-handlers/blob/10dbcc4d64fdf748989e6d72ea169bc7163eb6ad/zhaquirks/tuya/ts0601_cover.py#L176-L184
Both have the time cluster as out other have it as in cluster.
I dont knowing but i think the function is the same on both and the last have the time cluster in the replacement and the first not but i think its not relevant but thest the last first if it working OK.

Your device is inverted by default but its already being added of the your device in the INIT.
You also need putting the definition for the remote keys in the INIT here and finding the right combinations for your device:
https://github.com/zigpy/zha-device-handlers/blob/10dbcc4d64fdf748989e6d72ea169bc7163eb6ad/zhaquirks/tuya/__init__.py#L91-L104
The change of the `ts0601_cover.py` you can doing as local quirk but the INIT must being patched in the HA container for working with  the local quirk.

If you cant doing the patching i can do test files for you that you must copy to the right place but try if you can getting it working then its easier testing then having the device in the hand.

Instruction of editing quirks in the HA container and local quirks you finding here: https://github.com/zigpy/zha-device-handlers/discussions/693

dirkpitt35, 2021-11-10:
Hi, thanks for the in depth response. I've had a read of the instructions for editing quirks but I'm not able to make any sense of it. I admit that I'm not really a programmer but can take instruction well. If there's a step by step instruction on what I need to do then I can manage that I think.

I have supervised HA running in a VM if that helps at all?

Really appreciate the help with this.

Matt

MattWestb, 2021-11-10:
Supervised is OK !
I was forgetting my linking is made after renaming the quirks and your must using the old names like `suplemon /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts0601.py` and i think the lines is being changes then i have deleting some devices then moving them to on own quirk so the start  line for the best candidate shall being at 230.

The INIT is `suplemon /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py`  and i think the lines is the same but not in my test system then i have patching it so its not standard but you is finding then around the linked one.

Sorry for doing you more confused but i was forgetting the renaming and splitting the quirks.

You need having the device ID in both files or ZHA is not loading.
If you is getting the `ts0601.py` OK you can see that the device is using the quirk on the device card and Zigbee device signature like this:
 `Quirk: ts0601_trv_siterwell.SiterwellGS361_Type1` (from my tuya TRV).

Then the quirk is loading you can testing different key mappings in the INIT.

dirkpitt35, 2021-11-10:
Aw man, I'm having trouble following this. lol.

So I've figured out with your help how to edit the files on the docker container. You're right, I have the old .py names so no ts0601_cover.py. 

Can you explain this part please?

"If you is getting the ts0601.py OK you can see that the device is using the quirk on the device card and Zigbee device signature like this:
Quirk: ts0601_trv_siterwell.SiterwellGS361_Type1 (from my tuya TRV)."

Matt

MattWestb, 2021-11-10:
For my TRV its looks like this:
[image]
The last line is having the name of the quirk is its loaded OK and if not you is **not** getting the last line with `quirk:`

dirkpitt35, 2021-11-10:
Ok that makes sense then. I'm not seeing the final line.

I've added the following line to /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts0601.py around line 230

`MODELS_INFO: [          
            ("_TZE200_zah67ekd", "TS0601"),
            ("_TZE200_xuzcvlku", "TS0601"),
            ("_TZE200_rddyvrci", "TS0601"),
            ("_TZE200_nueqqe6k", "TS0601"),
            ("_TZE200_gubdgai2", "TS0601"),
            ("_TZE200_yenbr4om", "TS0601"),
            ("_TZE200_5sbebbzs", "TS0601"),
            ("_TZE200_xaabybja", "TS0601"),`

I've added the following line to /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py around line 91

`TUYA_COVER_COMMAND = {                                
    "_TZE200_zah67ekd": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},     
    "_TZE200_fzo2pocs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xuzcvlku": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_rddyvrci": {0x0000: 0x0002, 0x0001: 0x0001, 0x0002: 0x0000},
    "_TZE200_3i3exuay": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nueqqe6k": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},        
    "_TZE200_gubdgai2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zpzndjez": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_cowvfni3": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},
    "_TYST11_wmcdj3aq": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_yenbr4om": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_5sbebbzs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xaabybja": {0x0000: 0x0002, 0x0001: 0x0002, 0x0002: 0x0001},`

Then restarted hass. Did I miss a step?

Thanks
Matt

MattWestb, 2021-11-10:
Can you putting ` ``` ` on one line before and after the code you have putting it so its not losing the formatting ?

MattWestb, 2021-11-10:
It shall looking like this:
```
        MODELS_INFO: [
            ("_TZE200_zah67ekd", "TS0601"),
            ("_TZE200_xuzcvlku", "TS0601"),
            ("_TZE200_rddyvrci", "TS0601"),
            ("_TZE200_nueqqe6k", "TS0601"),
            ("_TZE200_gubdgai2", "TS0601"),
            ("_TZE200_yenbr4om", "TS0601"),
            ("_TZE200_5sbebbzs", "TS0601"),
            ("_TZE200_xaabybja", "TS0601"),
        ],  # Not tested
```
If its being wrong formatted ZHA is not starting and if its matching you is getting the quirk line in the device signature.

Sorry the ` ``` ` on one new line or its not working :-(

MattWestb, 2021-11-10:
No HA restart is needed for ZHA reading the new quirk.
Normally repairing is not needed.
Can you posting one new device signature as from the first posting and with the code brackets ?
I have problem that one working quirk was not loaded yesterday but i was doing very bad things in the system that was making problems with all things.

dirkpitt35, 2021-11-10:
```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_xaabybja",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}```

MattWestb, 2021-11-10:
I have doing one more check and all looks OK.
Can you putting little more debug logging in your HA config for:
```
    homeassistant.components.zha: debug
    zigpy: debug
```
And restarting HA.
Then its possible see then ZHA is trying loading quirks for the device but its being very messy log.

MattWestb, 2021-11-10:
Sorry i was writing wrong debug is only needed for:
`zigpy.quirks` and ZHA is shoeing like this in the log then stating:
```
2021-11-10 10:09:10 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fourbtnremote.IkeaTradfriRemote'>
2021-11-10 10:09:10 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 84:2e:14:ff:fe:5f:89:0e: <class 'zhaquirks.ikea.fourbtnremote.IkeaTradfriRemote'>
2021-11-10 10:09:10 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden TRADFRI remote control (00:0d:6f:ff:fe:62:53:2a)
2021-11-10 10:09:10 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'fivebtnremotezha.IkeaTradfriRemote3'>
2021-11-10 10:09:10 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:0d:6f:ff:fe:62:53:2a: <class 'fivebtnremotezha.IkeaTradfriRemote3'>
```
and you can see if its mashing for your device.

dirkpitt35, 2021-11-10:
I'm seeing an unknown cluster error in the log (before I enabled the other debug logs)

Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:95
First occurred: 13:53:49 (3 occurrences)
Last logged: 13:53:49

Unknown cluster 61184

After I enabled debug log:

2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_xaabybja TS0601 (84:71:27:ff:fe:73:0c:7e)
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2021-11-10 13:53:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint

dirkpitt35, 2021-11-10:
Ok so here's the log entry.

```
2021-11-10 15:22:03 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_xaabybja TS0601 (84:71:27:ff:fe:73:0c:7e)
2021-11-10 15:22:03 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_cover.TuyaMoesCover0601'>
2021-11-10 15:22:03 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 84:71:27:ff:fe:73:0c:7e: <class 'ts0601_cover.TuyaMoesCover0601'>
```
However I get an error when trying to use the device:
```
2021-11-10 15:22:43 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140130118722896] '_TZE200_xaabybja'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
await handler.job.target(service_call)
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
await self.hass.helpers.service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 667, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 863, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 704, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 126, in async_open_cover
res = await self._cover_channel.up_open()
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 59, in wrapper
result = await command(*args, **kwds)
File "/config/deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 917, in command
TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_xaabybja'
```

MattWestb, 2021-11-10:
## GREAT !!
Take on look in the container in `/config/deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py` at line 917.
If its one extra space or extra after the string.

I normal copy one of the "old" and only editing the shartes so not braking the formatting.

dirkpitt35, 2021-11-10:
I haven't even touched that file though.. the file I was working on was /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py

MattWestb, 2021-11-10:
The quirk is loading OK but its depending on the extra data from the INIT.
Make sure that the line you was adding its 110%.
`"_TZE200_xaabybja": {0x0000: 0x0002, 0x0001: 0x0002, 0x0002: 0x0001},`

The file in the error is the file that is is using the INIT and cant finding your device ID.

Very near !!!

dirkpitt35, 2021-11-10:
[image]

The controls are now greyed out. Should I be tweaking these settings?
```
{0x0000: 0x0002, 0x0001: 0x0002, 0x0002: 0x0001}
```

MattWestb, 2021-11-10:
Yes that is what the INIT code is doing but is only working if you is getting the error fixed that is coming from the same file (the INIT).

Looks nice !!!

dirkpitt35, 2021-11-10:
I'm not seeing the error anymore but then I'm unable to initiate a command (the error appeared when the open command was sent). I do see this error though, do you know what it means?
```
2021-11-10 16:43:43 WARNING (MainThread) [zigpy.zcl] Unknown cluster 61184
2021-11-10 16:43:43 WARNING (MainThread) [zigpy.zcl] Unknown cluster 61184
2021-11-10 16:43:43 WARNING (MainThread) [zigpy.zcl] Unknown cluster 61184
```

MattWestb, 2021-11-10:
Then the quirk has not being loaded then cluster 61184 = 0xef00is the tuya cluster.
Lock if the device have one "quirk" on the device card.

dirkpitt35, 2021-11-10:
[image]

MattWestb, 2021-11-10:
OK then trying on other method.
Copy the files from the zipfile to your HA configuration directory (there the configurtion.yaml
 is and its outside from the HA container).
Then copy the 2 feels with this commands in the container CLI:
```
cp /config/__init__.py  /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py 
cp /config/ts0601_trv.py  /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts0601.py 
```
Delete (or move) the files from you local quirk directory and restarting HA.

MattWestb, 2021-11-10:
If its offline trying repairing it if its have going off line then you have restarting your system so oft.

dirkpitt35, 2021-11-10:
Which zip file? Was there something supposed to be attached to the comment?

MattWestb, 2021-11-10:
This ;-)
[TS0601.zip](https://github.com/zigpy/zha-device-handlers/files/7514925/TS0601.zip)

I was trying all commands in my test system and copy the files from it for getting it right and was forgetting it ;-((

dirkpitt35, 2021-11-10:
I do it ALL the time! 👍 

Thanks, I'll try this.

dirkpitt35, 2021-11-10:
Now it's not picking up the quirk.
[image]
I removed the __pycache__ folder again and restarted but the folder isn't being recreated.

MattWestb, 2021-11-10:
I think you have getting some bad in your HA container.
Can you looking if ZHA is loading the quirk then its starting (shall being around 215 quirks)
Its looks like this and the last is the local quirk (only the end is shown):
```
2021-11-10 10:07:19 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zen.thermostat
2021-11-10 10:07:19 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing
2021-11-10 10:07:19 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing.motion
2021-11-10 10:07:19 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2021-11-10 10:07:19 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts0601_trv_siterwell
```

MattWestb, 2021-11-10:
The last seen is is not  updated (for some hour) so i think your device is not online.
Deleting it, waiting 1 minute and then adding it a gen.

dirkpitt35, 2021-11-10:
So it is loading the quirks
```
2021-11-10 18:28:35 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.tuya.ts0601
```
I'll try removing and readding it.

MattWestb, 2021-11-10:
Good was the system loading the other 215 quirks or not ?

dirkpitt35, 2021-11-10:
Yes it was. Didn't want to spam the comments.

I removed and re added the device but the quirk wasn't found and entity not created. Some errors in the log:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:114
First occurred: 18:32:38 (36 occurrences)
Last logged: 18:33:01

[0x81c2:1:0xef00] Unknown cluster-specific command 1
[0x81c2:1:0xef00] Unknown cluster-specific command 36
[0x81c2:1:0xef00] Unknown cluster-specific command 17
```
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:95
First occurred: 18:28:35 (4 occurrences)
Last logged: 18:32:31

Unknown cluster 61184
```
```
Logger: frontend.js.latest.202111080
Source: components/system_log/__init__.py:190
First occurred: 18:28:39 (1 occurrences)
Last logged: 18:28:39

http://homeassistant.chequers.cloud:8123/config/server_control:0:0 Uncaught
```

MattWestb, 2021-11-10:
Your device have sending somthing on the tuya cluster so its working.
Restart HA and look is the quirk is loading.
I was having that problem yesterday but i was doing little more bad things in my test system for getting it.

dirkpitt35, 2021-11-10:
If I re-enable the local quirks folder as below and include the ts0601_cover.py.
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```
It picks up the quirk and adds the entity but when clicking the buttons on the entity I get this error:
```
Logger: homeassistant.components.websocket_api.http.connection
Source: deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:917
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 18:41:55 (1 occurrences)
Last logged: 18:41:55

[140622941236768] '_TZE200_xaabybja'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 667, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 863, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 704, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 126, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 59, in wrapper
    result = await command(*args, **kwds)
  File "/config/deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 917, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_xaabybja'
```

dirkpitt35, 2021-11-10:
> Your device have sending somthing on the tuya cluster so its working. Restart HA and look is the quirk is loading. I was having that problem yesterday but i was doing little more bad things in my test system for getting it.

It doesn't pick up the quirk this way, plus the __pycache__  isn't recreated.

MattWestb, 2021-11-10:
So back where you was before.

Do you have one large system that  take long time to getting it up and running of you deleting little brutal things like the HA container and most local configuration ?

dirkpitt35, 2021-11-10:
I'm just starting out on this so it's not a huge pain to nuke it all and start again but as it's a VM I have a snapshot of the system before I started working on this. BTW I've already tried reverting back :)

I can try powering it off and rebuilding on a new VM just to see. Keep it simple and just try and get the curtains working. You've been a great help today, thank you so much for your time. I'll try from scratch again and revert back tomorrow.

MattWestb, 2021-11-10:
I was thinking deleting the HA container and trying the supervisor installing it new if its somthing that is changed by mistake in the HA container.

Also deleting the `Zigbbb.DB` in the config directory and the `.storage` folder that is holding all HA devices and entries but then more or less all you setting is away.

Do you running the latest HA with the fix version ? if not updating it and its installing one new HA container and deleting all changes you have doing i it.

dirkpitt35, 2021-11-10:
Interesting.. I deployed another 'latest version' OVA of Home Assistant, installed HACs, Portainer, Terminal, Samba Share and ZHA. Copied over the files you sent in the ZIP and rebooted. The device added fine and I tested the open button.. heard a scream from upstairs (think the wife was changing).. seems to work. Must be something broken in my version of HASS, perhaps all the fiddling about with Zigbee2Mqtt and ZHA has done something.

I'm going to set everything up again from scratch. Thanks again for your help!
Matt

MattWestb, 2021-11-10:
I hope you is not  getting **real bad** problems later then its being dark after the testing the open button !!!!

I was having the feeling that somthing was not OK but the quirk was working then loading (local) but ZHA was not doing all right with the INIT in the container (or all quirks in the container).

Great that is working !!

I like to knowing (not urgent) if the setting for the keys in INIT is the right then i can making one PR and its likely coming in next HA release and you and other users dont need hacking the HA container all the time.

Great work done !!

dirkpitt35, 2021-11-11:
Just an update on the testing. All functionality appears to be fine and any manual changes are reported back to HA just fine.

I think you can roll this into the next HA release.

Just to confirm the device details:

Zemismart BCM500DS-TYZ Curtain Motor
[image]
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_xaabybja",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601.TuyaMoesCover0601"
}
```
Thanks for all your help. I think we can close the issue if you don't have any further questions?
Matt

MattWestb, 2021-11-11:
Great i hope you was not getting more problem from upstairs then done the last tests.

Can you confirming the key mapping in the INIT is it the same as i was putting in the ziped one or have you changing it ?

I can putting one fixing issue in the PR and its closing it then its being merged or you can closing it then you like :-))

dirkpitt35, 2021-11-11:
Confirmed. All I did was copy your files into the respective folders.

MattWestb, 2021-11-11:
PR made and waiting for being merged :-))

closed 2021-11-11 by dmulcahey

MattWestb, 2021-11-11:
And merged by David !!
Only waiting for HA 2021.12 and you is having it without copy files.

PR #1142 (2021-11-11): Adding Zemismart Bcm500ds tyz curtain motor
Tested by user with ZHA 2012.11 before renaming the quirk but all changes is the same and shall working.

Fixes #1137