zigpy/zha-device-handlers #1145: [Device Support Request] - TS0003 [_TZ3000_odzoiovu] - Tuya 3 gang switch module

Flopescu, 2021-11-11:
**Is your feature request related to a problem? Please describe.**
Feature request - quirk to allow the change of the "Switch_type" to momentary button.
Currently the automated discovery and configuration leaves the switch to "rocker button".

**Describe the solution you'd like**
I would like to have the option to change this behaviour and use a momentary button to toggle the switch.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
    "node_descriptor": "NodeDescriptor(
       logical_type=<LogicalType.Router: 1>,
       complex_descriptor_available=0,
       user_descriptor_available=0,
       reserved=0,
       aps_flags=0,
       frequency_band=<FrequencyBand.Freq2400MHz: 8>,
       mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>,
       manufacturer_code=4417,
       maximum_buffer_size=66,
       maximum_incoming_transfer_size=66,
       server_mask=10752,
       maximum_outgoing_transfer_size=66,
       descriptor_capability_field=<DescriptorCapability.NONE: 0>,
       *allocate_address=True,
       *is_alternate_pan_coordinator=False,
       *is_coordinator=False,
       *is_end_device=False,
       *is_full_function_device=True,
       *is_mains_powered=True,
       *is_receiver_on_when_idle=True,
       *is_router=True,
       *is_security_capable=False)",
    "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_odzoiovu",
  "model": "TS0003",
  "class": "zigpy.device.Device"
}

```

Device Discovery log:

```
Device 0x16b3 (a4:c1:38:69:2b:af:96:50) joined the network
[0x16b3] Requesting 'Node Descriptor'
[0x16b3] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x16b3] Discovering endpoints
[0x16b3] Discovered endpoints: [1, 2, 3, 242]
[0x16b3] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=3 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x16b3:1] Discovering endpoint information
[0x16b3:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=256, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 57344
Unknown cluster 57345
[0x16b3:2] Discovering endpoint information
[0x16b3:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=256, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 57345
[0x16b3:3] Discovering endpoint information
[0x16b3:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=256, device_version=1, input_clusters=[4, 5, 6, 57345], output_clusters=[])
Unknown cluster 57345
[0x16b3:242] Discovering endpoint information
[0x16b3:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x16b3] Read model 'TS0003' and manufacturer '_TZ3000_odzoiovu' from <Endpoint id=1 in=[identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, None:0xE000, None:0xE001, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x16b3] Discovered basic device information for <Device model='TS0003' manuf='_TZ3000_odzoiovu' nwk=0x16B3 ieee=a4:c1:38:69:2b:af:96:50 is_initialized=True>
```

**Additional context**
This device is added in zigbee2mqtt, info here: https://www.zigbee2mqtt.io/devices/TS0003_switch_module.html

If there is anything that I can test or any other log that I can retrieve, I would be happy to help.

Thank you!

MattWestb, 2021-11-11:
Some user and devs is testing implanting the mode switch but its need little more work for getting it done for production. #1118

Your device looks being easy fixing then we have the ground function working OK.

MattWestb, 2021-11-12:
I have making one quirk but its little tricky to testing is its working on your device then its have being some changes in ZHA and it must being coped into HA docker container.

Unzip the files in your HA configure folder.
Open one CLI in the HA docker container and use these commands for copy the new files.
```
cp /config/__init__.py /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py
cp /config/ts011f_plug.py /us/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts011f_plug.py
cp /config/ts0121_plug.py /us/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts0121_plug.py
cp /config/ts001x.py /us/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ts001x.py
```
If not all files is updated is ZHA not stating then some cluster names is changed and is braking the quirk loading.

Restart HA and if its working the system is loading the quirk if i have doing all OK.

[custom_zha_quirks.zip](https://github.com/zigpy/zha-device-handlers/files/7530573/custom_zha_quirks.zip)

Pleas reporting back if its working or not.

Flopescu, 2021-11-12:
thanks!
I can try this ... or at least I think I can!
let me test it and I;ll update this ticket

Cheers!

MattWestb, 2021-11-12:
No problem if i cant getting any user testing i doing one PR with adding the function to the one gang module but its no hurry then the other user is running it and its more weeks to next HA update so take the time you need :-)))

MattWestb, 2021-11-15:
The PR is merged but your device is not verified working OK or not so if you can trying it is great or you is waiting until next release of HA and if its not working OK then we is fixing it later :-))

wingertge, 2021-11-25:
It's working perfectly with the quirk I submitted in the PR. Tried color temp, brightness and color and it all works great.

MattWestb, 2021-11-25:
You is meaning https://github.com/zigpy/zha-device-handlers/issues/1173 ?
That looks great !!

Did you also trying the quirk for your switch (TS0003) in this issue ??

Hedda, 2022-05-10:
> Feature request - quirk to allow the change of the "Switch_type" to momentary button.

FYI, there is a similar request for "switch_type" configuration in https://github.com/zigpy/zha-device-handlers/issues/1231 for Aqara T1 with neutral (lumi.switch.n0agl1).

See related developers discussion here for reference -> https://github.com/zigpy/zigpy/discussions/934

_Originally posted by @dumpfheimer in https://github.com/zigpy/zigpy/discussions/934#discussioncomment-2654475_

> dmulcahey has been working on exposing the configuration entities here: https://github.com/dmulcahey/home-assistant/tree/dm/zha-zcl-select
> 
> Also, I have a similar PR pending here: [zigpy/zha-device-handlers#1391](https://github.com/zigpy/zha-device-handlers/pull/1391) And I have added the necessary entites here: https://github.com/dumpfheimer/core/tree/dumpfheimer/zha-expose-configuration-entities
> 
> All three together make this setting available **for Aqara H1 Switches** but porting it for T1 _should_ not be very difficult.

closed 2022-11-13 by github-actions[bot]