zigpy/zha-device-handlers #1146: [Device Support Request] ZLinky_TIC (French Linky electricity meter)

pdecat, 2021-11-11:
**Is your feature request related to a problem? Please describe.**

Add support for ZLinky_TIC, an add-on to transmit all data from the French Linky electricity meter.

Firmware for this device is open source, and clusters are documented at https://github.com/fairecasoimeme/Zlinky_TIC

**Describe the solution you'd like**

I'd like to be able to query the manufacturer specific cluster attributes:
[image]

**Device signature**

```
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0xE837, SrcIEEE=00:15:8d:00:04:ff:ff:ff, ParentNwk=0x0BA5)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xE837, SrcIEEE=00:15:8d:00:04:ff:ff:ff, ParentNwk=0x0BA5)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0xE837, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0xE837, NWK=0xE837, IEEE=00:15:8d:00:04:ff:ff:ff, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0xE837, NWK=0xE837, IEEE=00:15:8d:00:04:ff:ff:ff, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 INFO (MainThread) [zigpy.application] New device 0xe837 (00:15:8d:00:04:ff:ff:ff) joined the network
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.device] [0xe837] Scheduling initialization
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.zigbee.application] Pretending we received a ZDO message: b'\xFF\x37\xE8\xFF\xFF\xFF\x04\x00\x8D\x15\x00\x8E'
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xE837 ieee=00:15:8d:00:04:ff:ff:ff is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x37\xE8\xFF\xFF\xFF\x04\x00\x8D\x15\x00\x8E'
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.zdo] [0xe837:zdo] ZDO request ZDOCmd.Device_annce: [0xE837, 00:15:8d:00:04:ff:ff:ff, 142]
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 INFO (MainThread) [zigpy.device] [0xe837] Requesting 'Node Descriptor'
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy.device] [0xe837] Extending timeout for 0xfa request
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0xE837, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
Nov 06 16:08:58 myhost hass[1643]: 2021-11-06 16:08:58 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.FAIL: 1>)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xE837), dst_ep=0, src_ep=0, cluster=ZDOCmd.Node_Desc_req, sequence=250, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'\xfa7\xe8'
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Node_Desc_req({'NWKAddrOfInterest': 0xE837}) and replaced with ZDO.NodeDescReq.Req(DstAddr=0xE837, NWKAddrOfInterest=0xE837)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.NodeDescReq.Req(DstAddr=0xE837, NWKAddrOfInterest=0xE837)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.NodeDescReq.Rsp(Status=<Status.SUCCESS: 0>)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.NodeDescRsp.Callback(Src=0xE837, Status=<Status.SUCCESS: 0>, NWK=0xE837, NodeDescriptor=NullableNodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False))
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.zigbee.application] Pretending we received a ZDO message: b'\xFA\x00\x37\xE8\x01\x40\x8E\x37\x10\x7F\x64\x00\x00\x2C\x64\x00\x00'
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xE837 ieee=00:15:8d:00:04:ff:ff:ff is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\xFA\x00\x37\xE8\x01\x40\x8E\x37\x10\x7F\x64\x00\x00\x2C\x64\x00\x00'
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 INFO (MainThread) [zigpy.device] [0xe837] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 INFO (MainThread) [zigpy.device] [0xe837] Discovering endpoints
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0xE837, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.SUCCESS: 0>)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xE837), dst_ep=0, src_ep=0, cluster=ZDOCmd.Active_EP_req, sequence=251, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'\xfb7\xe8'
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Active_EP_req({'NWKAddrOfInterest': 0xE837}) and replaced with ZDO.ActiveEpReq.Req(DstAddr=0xE837, NWKAddrOfInterest=0xE837)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ActiveEpReq.Req(DstAddr=0xE837, NWKAddrOfInterest=0xE837)
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ActiveEpReq.Rsp(Status=<Status.SUCCESS: 0>)
[... cut ...]
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xE837](ZLinky_TIC): completed configuration
Nov 06 16:09:00 myhost hass[1643]: 2021-11-06 16:09:00 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xE837](ZLinky_TIC): stored in registry: ZhaDeviceEntry(name='LiXee ZLinky_TIC', ieee='00:15:8d:00:04:ff:ff:ff', last_seen=1636211340.9443586)
```

**Additional context**

Device is distributed directly by its independent French manufacturer at https://lixee.fr/produits/37-zigate-usb-ttl-3770014375148.html

pdecat, 2022-05-10:
Still waiting on firmware update to unblock PR #1165

mathuieu, 2022-09-02:
Hello,
any news on this integration ?
thanks for your help

Hedda, 2022-10-03:
> any news on this integration ?

FYI, there is some news about new quirk code for ZHA Device Handers now as well as a new firmware needed for it, see ->  https://github.com/zigpy/zha-device-handlers/pull/1165

closed 2022-10-16 by dmulcahey

Hedda, 2022-10-17:
FYI, you still need to manually update ZLinky TIC firmware but at least quirk code has now been merged into ZHA Device Handlers:

https://github.com/zigpy/zha-device-handlers/pull/1165

Another problem with Lixee ZLinky TIC needing firmware update now instead really becomes a user experience problem as now the first impression without a such Zigbee OTA Provider code firmware for ZLinky TIC is that it can not be updated automatically with the ZHA integration so users who can pair/join it will still not be able to use it until they manually updated the firmware on ZLinky TIC via other means.

Would now be great if someone with the necessary skills and interest would consider submitting a pull request to zigpy with ZLinky TIC Zigbee OTA Provider code now that ZHA Device Handlers support for your Lixee ZLinky TIC has now been merged into the zha-quirks package that will likely be added around the time that the next upcoming major Home Assistant core version is released.

See details in related feature request for zigpy which is Zigbee library Home Assistant ZHA depends on -> https://github.com/zigpy/zigpy/issues/1060

fairecasoimeme, 2022-10-17:
Sorry, but I think it's not a ZLinky problem but a zigpy-zigate problem. I think you can update ZLinky with OTA with other compatible coordinator than ZiGate.
By the way, on my side, I try to find and motivate a developer to add OTA functionnality to Zigpy-Zigate

pdecat, 2022-10-17:
I guess @Hedda means adding Lixee to the list of OTA supported manufacturers here: https://www.home-assistant.io/integrations/zha/#ota-firmware-updates

In the mean time, people can use @mdeweerd's excellent https://github.com/mdeweerd/zha-toolkit to download OTA updates from [Koenkk/zigbee-OTA](https://github.com/Koenkk/zigbee-OTA) and install them automatically without having to manipulate file by hand.

One issue that needs to be addressed in 100% automated mode is how to select the router / non-route OTA firmware.

fairecasoimeme, 2022-10-17:
OK,
I think I'll use a different way to manage route / non-route OTA firmware. Don't hesitate to give suggestions to be easier for OTA plugins ;)

Hedda, 2022-10-17:
> I guess @Hedda means adding Lixee to the list of OTA supported manufacturers here: https://www.home-assistant.io/integrations/zha/#ota-firmware-updates

Yes in practice I mean zigpy needs to have a downloader code for LiXee/ZLinky firmware Zigbee OTA provider added to it here:

https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py

ZHA only supports automatic OTA firmware update for those Zigbee OTA providers that is available in the zigpy code there.

https://github.com/zigpy/zigpy/issues/1060

zigpy library is the Zigbee framework library that the ZHA integration component that Home Assistant core depends on, and Home Assistant is not alone in depending on it as zigpy is partially also used by Jeedom Zigbee Plugin and Domoticz Zigbee Plugin (which I believe both are also very popular in France).

https://www.home-assistant.io/integrations/zha

https://www.domoticz.com/wiki/ZigbeeForDomoticz

https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/

So there is a very large userbase of Zigbee users that is indirectly using the zigpy library and by default only get its OTA providers.

Hedda, 2022-10-17:
> OK,
> I think I'll use a different way to manage route / non-route OTA firmware. Don't hesitate to give suggestions to be easier for OTA plugins ;)

@fairecasoimeme Regarding managing router or non-router firmware; could you not make unified firmware that is configurable?

I do not know much about Zigbee application firmware development but I know that for example, the "premium version" of PTVO Zigbee Configurable Firmware is configurable so that you can control the RF transmit power output dBm as "signal level " remotely in run-time, see an explanation of that feature here with instruction on how to use change that signal level "remotely via a custom converter (a.k.a. quirk) for Zigbee2MQTT -> https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/

fairecasoimeme, 2022-10-17:
not possible because non router and router mode have differents flash structures. :(

mdeweerd, 2022-10-17:
Rather than updating ZHA specific settings, I still strongly believe that it is better to maintain a central open list such as [Koenkk/zigbee-OTA](https://github.com/Koenkk/zigbee-OTA).  Maybe that list does need some extra info such as:
- A quality indicator (reverse (from packet capture), pre-release, from manufacturer);
- More than one FW version for a device (to be able to update to another version than the latest one.

Regarding ZLinky: I think it should create two "models" so that the versions and updates apply to only one "branch" of the software versions.  Which implies that in the future the end user should be able to choose an apparently not compatible FW update for his device.

Hedda, 2022-10-18:
> Rather than updating ZHA specific settings, I still strongly believe that it is better to maintain a central open list such as [Koenkk/zigbee-OTA](https://github.com/Koenkk/zigbee-OTA). Maybe that [...]

Interesting ideas but such new zigpy OTA feature ideas probably instead belong in a more general zigpy discussion about OTA features and functions so therefor suggest starting a new discussion about that here -> https://github.com/zigpy/zigpy/discussions

Or alternatively, if the ideas are specifically related to Zigbee OTA for fairecasoimeme ZLinky TIC device then in that existing zigpy feature request here -> https://github.com/zigpy/zigpy/issues/1060

There is also an existing feature request for zigpy that is specifically about getting OTA files from [Koenkk/zigbee-OTA](https://github.com/Koenkk/zigbee-OTA) repository here -> https://github.com/zigpy/zigpy/issues/535

Hedda, 2022-10-18:
> By the way, on my side, I try to find and motivate a developer to add OTA functionality to **zigpy**

@fairecasoimeme May I suggest that you re-post that comment as reply to that zigpy feature request here -> https://github.com/zigpy/zigpy/issues/1060

mdeweerd, 2022-10-18:
> Interesting ideas but such new zigpy OTA feature ideas probably instead belong in a more general zigpy discussion

See https://github.com/zigpy/zigpy/discussions/654#discussioncomment-3775270

I was "reminding" my view because of the comment:

> I guess @Hedda means adding Lixee to the list of OTA supported manufacturers here: https://www.home-assistant.io/integrations/zha/#ota-firmware-updates

Hedda, 2022-10-20:
> I think I'll use a different way to manage route / non-route OTA firmware. Don't hesitate to give suggestions to be easier for OTA plugins ;)

Please also see the related discussion in zigbee-herdsman-converters repo which have the same OTA firmware image selection  dilemma with there being two firmware with the same "Zigbee Device Signature" -> https://github.com/Koenkk/zigbee-herdsman-converters/issues/4785

PR #1165 (2022-10-16): Add Lixee ZLinky_TIC manufacturer specific cluster
This PR adds support for the manufacturer specific cluster `0xFF66` of the Lixee ZLinky_TIC device.

Reference available at:
- https://github.com/fairecasoimeme/Zlinky_TIC
- https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/ZLinky/Source/LixeeCluster.h#L74-L125

Submitted as draft as there's only one of many attribute for now.

Resolves #1146