zigpy/zha-device-handlers #1156: [Device Support Request]

Draghmar, 2021-11-15:
**Is your feature request related to a problem? Please describe.**
There are series of devices `TS0041` that had issues with getting to work properly through ZHA so the quirk was made. I just got such a device with one button and after pairing it all I see is power sensor. There's no button to check the status. I've seen through out few issues here and on the HA forum that was quite normal but as I understand the quirk was changed to be more universal instead of maintaining manufactures list. I even tried to make local quirk from the newest file to make sure it wasn't HA's old file fault: `Quirk: ts0041.TuyaSmartRemote0041TO`.

**Describe the solution you'd like**
I'd like to have a button in the HA. ;)

This is the device signature and below you may find pairing log.
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_axpdxqgu",
  "model": "TS0041",
  "class": "ts0041.TuyaSmartRemote0041TO"
}
```
```
Sending 'zdo_leave_req' failed: 
New device 0x9856 (60:a4:23:ff:fe:e2:9e:7b) joined the network
[0x9856] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x56\x98\x7B\x9E\xE2\xFE\xFF\x23\xA4\x60\x80'
[0x9856:zdo] ZDO request ZDOCmd.Device_annce: [0x9856, 60:a4:23:ff:fe:e2:9e:7b, 128]
Tries remaining: 3
[0x9856] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9856] Extending timeout for 0xf8 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\xF8\x00\x56\x98\x02\x40\x80\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x9856] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x9856] Discovering endpoints
Tries remaining: 3
[0x9856] Extending timeout for 0xf9 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\xF9\x00\x56\x98\x01\x01'
[0x9856] Discovered endpoints: [1]
[0x9856] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x9856:1] Discovering endpoint information
Tries remaining: 3
[0x9856] Extending timeout for 0xfa request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\xFA\x00\x56\x98\x12\x01\x04\x01\x00\x00\x01\x03\x00\x00\x01\x00\x06\x00\x02\x19\x00\x0A\x00'
[0x9856:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0x9856] Extending timeout for 0xfb request
[0x9856:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=251 command_id=Command.Read_Attributes_rsp>
[0x9856] Read model 'TS0041' and manufacturer '_TZ3000_axpdxqgu' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, on_off:0x0006] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x9856] Discovered basic device information for <Device model='TS0041' manuf='_TZ3000_axpdxqgu' nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=True>
Device is initialized <Device model='TS0041' manuf='_TZ3000_axpdxqgu' nwk=0x9856 ieee=60:a4:23:ff:fe:e2:9e:7b is_initialized=True>
Checking quirks for _TZ3000_axpdxqgu TS0041 (60:a4:23:ff:fe:e2:9e:7b)
Considering <class 'ts0041.TuyaSmartRemote0041TI'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts0041.TuyaSmartRemote0041TO'>
Found custom device replacement for 60:a4:23:ff:fe:e2:9e:7b: <class 'ts0041.TuyaSmartRemote0041TO'>
device - 0x9856:60:a4:23:ff:fe:e2:9e:7b entering async_device_initialized - is_new_join: True
device - 0x9856:60:a4:23:ff:fe:e2:9e:7b has joined the ZHA zigbee network
[0x9856](TS0041): started configuration
[0x9856:ZDO](TS0041): 'async_configure' stage succeeded
[0x9856] Extending timeout for 0xfc request
[0x9856:1:0x0000]: finished channel configuration
[0x9856:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (60:a4:23:ff:fe:e2:9e:7b, 1, 0, 4, '_TZ3000_axpdxqgu') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (60:a4:23:ff:fe:e2:9e:7b, 1, 0, 5, 'TS0041') params: FOREIGN KEY constraint failed
[0x9856:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x9856] Extending timeout for 0xfd request
[0x9856:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=253 command_id=Command.Configure_Reporting_rsp>
[0x9856:1:0x0001]: Successfully configured reporting for '{'battery_voltage': (3600, 10800, 1), 'battery_percentage_remaining': (3600, 10800, 1)}' on 'power' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x9856:1:0x0001]: finished channel configuration
[0x9856:1:0x0001]: 'async_configure' stage succeeded
[0x9856:1:0x0000]: 'async_configure' stage succeeded
[0x9856:1:0x0019]: 'async_configure' stage succeeded
[0x9856](TS0041): completed configuration
[0x9856](TS0041): stored in registry: ZhaDeviceEntry(name='_TZ3000_axpdxqgu TS0041', ieee='60:a4:23:ff:fe:e2:9e:7b', last_seen=1636998916.4670825)
[0x9856](TS0041): started initialization
[0x9856:ZDO](TS0041): 'async_initialize' stage succeeded
[0x9856:1:0x0001]: initializing channel: from_cache: False
[0x9856] Extending timeout for 0xfe request
[0x9856:1:0x0000]: initializing channel: from_cache: False
[0x9856:1:0x0000]: finished channel initialization
[0x9856:1:0x0019]: initializing channel: from_cache: False
[0x9856:1:0x0019]: finished channel initialization
[0x9856:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Read_Attributes>
[0x9856:1:0x000a] ZCL request 0x0000: [[7]]
[0x9856:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=254 command_id=Command.Read_Attributes_rsp>
[0x9856] Extending timeout for 0xff request
[0x9856:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=255 command_id=Command.Read_Attributes_rsp>
[0x9856:1:0x0001]: finished channel initialization
[0x9856:1:0x0001]: 'async_initialize' stage succeeded
[0x9856:1:0x0000]: 'async_initialize' stage succeeded
[0x9856:1:0x0019]: 'async_initialize' stage succeeded
[0x9856](TS0041): power source: Battery or Unknown
[0x9856](TS0041): completed initialization
```

MattWestb, 2021-11-15:
Your device is getting the quirk loaded so its looks OK.
You cant switching one remote on  / off like light and they is sending events like on off toggle click and hold.

If you is going to development tools, events, and putting `zha_event` in `Event to subscribe to` and clicking start listening and you shall getting all events the switch is sending in the log window.

This events you can using in automatons and also one the device card you can clicking the ` + `  by Automations and you is getting suggestions using events from the remote.

Draghmar, 2021-11-15:
Oh...that is kinda unintuitive as everything else I've had in HA had some endpoints available directly. So it's new to me. Thanks for the info then. It works this way! :)
Is there any reason why those events aren't available as a entities like it's normal in HA?

MattWestb, 2021-11-15:
Its the architecture of HA then all is about events is coming in and the system is reacting on them.

You can using [Template](https://www.home-assistant.io/docs/configuration/templating/) and making switches that is following your remotes commands sent.

Draghmar, 2021-11-15:
Hm, that does makes sense.
Nah, don't need to do that. But thanks for the tip!

closed 2021-11-15 by Draghmar