zigpy/zha-device-handlers #1158: [Device Support Request] Pull Bead Curtain Motor M515EGZT -V2 (Tuya?)

PremiumUsername, 2021-11-18:
The manual included with the device says the model is: M515EGZT -V2version-12v. When I bought it it said Tuya in the listing but The product is not branded. The device says TM616EGZT on the back.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x3a33:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=114 command_id=Command.Report_Attributes>
[0x3a33:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0x3a33:1:0x0006] Attribute report received: on_off=1
[0x3a33:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=115 command_id=Command.Report_Attributes>
[0x3a33:1:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=131>)]]
[0x3a33:1:0x0008] Attribute report received: current_level=131
[0x3A33:1:0x0008]: received attribute: 0 update with value: 131
[0x2929:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Report_Attributes>
[0x2929:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0x2929:1:0x0006] Attribute report received: on_off=1
[0x8417:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=196 command_id=Command.Report_Attributes>
[0x8417:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=LVBytes, value=b'\x01!\xbd\x0b\x03(\x1c\x04!\xa8\x13\x05!%\x04\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x00'>)]]
[0x8417:1:0x0000] Attribute report received: 65281=b'\x01!\xbd\x0b\x03(\x1c\x04!\xa8\x13\x05!%\x04\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x00'
[0x8417:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3005 < [Max]:3100, Battery Percent: 66.0
New device 0xcbea (b4:e3:f9:ff:fe:f4:f8:2e) joined the network
[0xcbea] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x85\xea\xcb.\xf8\xf4\xfe\xff\xf9\xe3\xb4\x8e'
[0xcbea:zdo] ZDO request ZDOCmd.Device_annce: [0xCBEA, b4:e3:f9:ff:fe:f4:f8:2e, 142]
Tries remaining: 3
[0xcbea] Requesting 'Node Descriptor'
Tries remaining: 2
[0xcbea] Extending timeout for 0x27 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xcbea] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xCBEA ieee=b4:e3:f9:ff:fe:f4:f8:2e is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00,\x00\x07\x00'
[0xcbea] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=44 comman [... cut ...]
It pairs OK. No buttons at first. Tried using a custom quirk using the same key settings as for TZE200_nueqqe6k:
`TUYA_COVER_COMMAND = {
    "_TZE200_hsgrhjpf": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
}`
... then I see buttons but they can't be used. 

When I chose a point on the slider the device blinks once but nothing happens. I haven't set limits for the device (don't know how), so I can't see how the slider could ever work. Pressing the stop, up or down buttons just gives an error message "Failed to call service..."

PremiumUsername, 2021-11-18:
I see now that my custom quirk doesn't seem to be loading ok.
```
File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 917, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_hsgrhjpf'
```
Seems to me like it's not using the init file I put in the local folder. Why is that?

MattWestb, 2021-11-18:
The INIT must being put in the HA docker container or your local quirk can getting data from it (local quirks is getting all from the container and cant getting from local quirks).

Instruction editing files in the HA container: https://github.com/zigpy/zha-device-handlers/discussions/693

PremiumUsername, 2021-11-19:
I've now edited the init file directly in the container and now the buttons work.

in the file ts0601.py I used:
```
        MODELS_INFO: [
            ("_TZE200_zah67ekd", "TS0601"),
            ("_TZE200_xuzcvlku", "TS0601"),
            ("_TZE200_rddyvrci", "TS0601"),
            ("_TZE200_nueqqe6k", "TS0601"),
            ("_TZE200_gubdgai2", "TS0601"),
            ("_TZE200_yenbr4om", "TS0601"),
            ("_TZE200_5sbebbzs", "TS0601"),
            ("_TZE200_hsgrhjpf", "TS0601"),
        ], 
```
and in __init__.py:
```
TUYA_COVER_COMMAND = {
    "_TZE200_zah67ekd": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_fzo2pocs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xuzcvlku": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_rddyvrci": {0x0000: 0x0002, 0x0001: 0x0001, 0x0002: 0x0000},
    "_TZE200_3i3exuay": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nueqqe6k": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_gubdgai2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zpzndjez": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_cowvfni3": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},
    "_TYST11_wmcdj3aq": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_yenbr4om": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_5sbebbzs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_hsgrhjpf": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
}
```

I still don't know how or if it is possible to calibrate the end positions somehow. So the slider doesn't work. How is it I'm supposed to set the end positions, what clusters/commands should I use?

MattWestb, 2021-11-19:
I dont knowing but i think it can being made with the buttons but you need reading in the forum how to do that.

If the Up, Down and Stop working OK (then 100% open the up is grayed out and the down is grayed out then 100% closed) ?

If it OK i can doing one PR for adding it do it being supported in next release of HA.

PremiumUsername, 2021-11-19:
> I dont knowing but i think it can being made with the buttons but you need reading in the forum how to do that.
>
> If the Up, Down and Stop working OK (then 100% open the up is grayed out and the down [...]

If I manually set the slider to somewhere in the middle I can use both the UP and Down button, otherwise one button is greyed out. I think this part is working ok. The missing functionality stems form me not having limits set so I think adding it would be fine.

I may add that the manual only shows how to set the limits via the Queqi gateway. There are no instructions for using the set-buttons and the method for first version of M515EGZT doesn't seem to work, so I think I need to do this via zigbee. Maybe someone could sniff what the gateway is doing when it sets the limits.

closed 2021-11-20 by dmulcahey

davehendo, 2021-11-25:
@PremiumUsername did you have any luck in finding out how to set limits manually (with buttons on the device), instead of through Tuya app?
Or have you found an inx manual for TM616EGZT?
Thanks

PremiumUsername, 2021-11-28:
> 

Sadly, I haven't found a way to do so. I also tried to use a remote from a zemismart zigbee curtain motor, but I wasn't able to pair it, the manuals for these products seems incomplete. 

I did find this on the zemismart [website:](https://www.zemismart.com/products/m515egbztn)

```
/**
 *  Tuya Window Shade (v.0.1.0)
 *	Copyright 2020 iquix
 *
 *	Licensed under the Apache License, Version 2.0 (the "License"); you may not use this file except
 *	in compliance with the License. You may obtain a copy of the License at:
 *
 *		http://www.apache.org/licenses/LICENSE-2.0
 *
 *	Unless required by applicable law or agreed to in writing, software distributed under the License is distributed
 *	on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the License
 *	for the specific language governing permissions and limitations under the License.
 This DTH is coded based on iquix's tuya-window-shade DTH.
 https://github.com/iquix/Smartthings/blob/master/devicetypes/iquix/tuya-window-shade.src/tuya-window-shade.groovy
 */

import groovy.json.JsonOutput
import physicalgraph.zigbee.zcl.DataType

metadata {
	definition(name: "ZemiSmart Vertical Blind", namespace: "ShinJjang", author: "ShinJjang-iquix", ocfDeviceType: "oic.d.blind", vid: "generic-shade") {
		capability "Actuator"
		capability "Configuration"
		capability "Window Shade"
		capability "Window Shade Preset"
		capability "Switch Level"
		capability "Sensor"

		command "pause"
		command "levelOpenClose"
        
        attribute "Direction", "enum", ["Reverse","Forward"]
        attribute "OCcommand", "enum", ["Replace","Original"]
        attribute "stapp", "enum", ["Reverse","Forward"]
        attribute "remote", "enum", ["Reverse","Forward"]

		fingerprint endpointId: "0x01", profileId: "0104", inClusters: "0000 0004 0005 EF00", outClusters: "0019, 000A", manufacturer: "_TZE200_xuzcvlku", model: "TS0601", deviceJoinName: "Zemismart Vertical Blind"        
	}

	preferences {
		input "preset", "number", title: "Preset position", description: "Set the window shade preset position", defaultValue: 50, range: "0..100", required: false, displayDuringSetup: false
        input name: "Direction", type: "enum", title: "Direction Set", options:["01": "Reverse", "00": "Forward"], required: true, displayDuringSetup: true
        input name: "OCcommand", type: "enum", title: "Replace Open and Close commands", options:["2": "Replace", "0": "Original"], required: true, displayDuringSetup: true
        input name: "stapp", type: "enum", title: "app opening,closing Change", options:["2": "Reverse", "0": "Forward"], required: true, displayDuringSetup: true
        input name: "remote", type: "enum", title: "RC opening,closing Change", options:["1": "Reverse", "0": "Forward"], required: true, displayDuringSetup: true
	}

	tiles(scale: 2) {
		multiAttributeTile(name:"windowShade", type: "generic", width: 6, height: 4) {
			tileAttribute("device.windowShade", key: "PRIMARY_CONTROL") {
				attributeState "open", label: 'Open', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#00A0DC", nextState: "closing"
				attributeState "closed", label: 'Closed', action: "open", icon: "http://www.ezex.co.kr/img/st/window_close.png", backgroundColor: "#ffffff", nextState: "opening"
				attributeState "partially open", label: 'Partially open', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#d45614", nextState: "closing"
				attributeState "opening", label: 'Opening', action: "close", icon: "http://www.ezex.co.kr/img/st/window_open.png", backgroundColor: "#00A0DC", nextState: "closing"
				attributeState "closing", label: 'Closing', action: "open", icon: "http://www.ezex.co.kr/img/st/window_close.png", backgroundColor: "#ffffff", nextState: "opening"
			}
		}
		standardTile("contPause", "device.switch", inactiveLabel: false, decoration: "flat", width: 2, height: 2) {
			state "pause", label:"", icon:'st.sonos.pause-btn', action:'pause', backgroundColor:"#cccccc"
		}
		standardTile("presetPosition", "device.presetPosition", width: 2, height: 2, decoration: "flat") {
			state "default", label: "Preset", action:"presetPosition", icon:"st.Home.home2"
		}
[... 126 lines cut ...]
def updated() {
	log.debug "val(${Direction}),valC(${OCcommand}),valR(${remote})"
	DirectionSet(Direction ?:"00")
}	

def DirectionSet(Dval) {
	log.info "Dset(${Dval})"
   sendHubCommand(zigbee.command(CLUSTER_TUYA, SETDATA, "00" + zigbee.convertToHexString(rand(256), 2) + "05040001" + Dval))
}

def configure() {
	log.info "configure()"
}

private sendTuyaCommand(dp, fn, data) {
	log.info "${zigbee.convertToHexString(rand(256), 2)}=${dp},${fn},${data}"
	zigbee.command(CLUSTER_TUYA, SETDATA, "00" + zigbee.convertToHexString(rand(256), 2) + dp + fn + data)
}

private rand(n) {
	return (new Random().nextInt(n))
}
```

Maybe that could be of some help.

davehendo, 2021-11-28:
Thanks for getting back to me.
Yes - I am using iquix DH in Smartthings, and it works well.
Unfortunately, still not found a way to set the limits using physical buttons, or through Smartthings. Only option is to set limits in Smart Life.
Thanks

mrJudahBella, 2022-06-11:
Ran into the same issue today not being able to set upper and lower limits, the device says "V2-RA" at the back:
[image]
 
The devices labeled "ZM" at the back are able to set upper and lower limits at the device itself:
[image]

@dhc25 you mentioned that you used "Smart Life" to set the upper and lower limits. Are you using a Tuya ZigBee gateway with it?

mrJudahBella, 2022-06-11:
I'm using Zigbee2MQTT, so https://github.com/Koenkk/zigbee2mqtt/issues/12300 helped.

PR #1163 (2021-11-20): Pull Bead Curtain Motor M515EGZT -V2
Adding Pull Bead Curtain Motor M515EGZT -V2 `("_TZE200_hsgrhjpf", "TS0601")`.
Fixes #1158
Tested of user and working but have problem calibrating the device.