zigpy/zha-device-handlers #1159: [Device Support Request] TS0601 _TZE200_a4bpgplm tuya trv

joaoasilva, 2021-11-18:
**Is your feature request related to a problem? Please describe.**
Bought this thermostat from AliExpress looking at this https://zigbee.blakadder.com/Haozee_TRV601.html that says it works with ZHA but it doesn't. It's possible to get the temperature and set but most of the stuff doesn't work.

**Describe the solution you'd like**
For it to work ex: window status, schedule, turn on/off.

Someone already converted it successfully to work on HA but I don't understand how can I port that to ZHA here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/245fc8273e3afac8d71ec196d0e3ae71e413cba9/lib/tuya.js#L426

Please help.
Thanks

MattWestb, 2021-11-18:
Is it one Rti-Tek or one SHENZHEN RSH white labeled device ?
Looks in the tuya TRV matrix with design it heaving, https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395
I think both is having TRVs that is very similar in commands and can being adapter or adjusted for getting them working.

joaoasilva, 2021-11-18:
It's the Rti-Tek 1. TRV 601 LED display on the side and rotating knob on the top.

MattWestb, 2021-11-18:
You can trying adding the device ID after line 1061
https://github.com/zigpy/zha-device-handlers/blob/c5f1b2dd8f2a8e69fa2cd5e09bd9683ec4989fad/zhaquirks/tuya/ts0601_trv.py#L1058-L1062
in one new line as the other ID is made.

Configure local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and and putting your edited file in the new folder you have made for local quirk and restart HA.

If all is going well you is getting one TRV on the device card with the device class of your quirk.

You can testing some function but not all is working until patching ZHA.

joaoasilva, 2021-11-18:
So sorry, I forgot to mention that it "works" getting the temperature and set because I've added it exactly there. 😅
But can't turn it off/on, get the window detection or schedule and sometimes it goes crazy and I can't set it anymore, have to restart again for it to work.
It's not really usable.
Is it easy to patch based on https://github.com/Koenkk/zigbee-herdsman-converters/blob/245fc8273e3afac8d71ec196d0e3ae71e413cba9/lib/tuya.js#L426 ?

MattWestb, 2021-11-18:
I think its better **jacekk015** taking on look on this then hi is cooking some new TRVs and have it in the fingers.

@jacekk015 can you taking on look on one new tuya TRV ???

joaoasilva, 2021-11-18:
Thanks, @jacekk015 if you need any debug from the tuya developer portal I can get you.

jacekk015, 2021-11-18:
@joaoasilva Yep. For start please put here a pairing log. And after pairing put here also Zigbee Device Signature

joaoasilva, 2021-11-18:
Sure:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_a4bpgplm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

joaoasilva, 2021-11-18:
```
2021-11-18 20:27:12 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0x9817, 50:32:5f:ff:fe:91:dd:82
2021-11-18 20:27:12 INFO (MainThread) [zigpy.application] New device 0x9817 (50:32:5f:ff:fe:91:dd:82) joined the network
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.device] [0x9817] Scheduling initialization
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x9817 ieee=50:32:5f:ff:fe:91:dd:82 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x8d\x17\x98\x82\xdd\x91\xfe\xff_2P\x80'
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.zdo] [0x9817:zdo] ZDO request ZDOCmd.Device_annce: [0x9817, 50:32:5f:ff:fe:91:dd:82, 128]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-11-18 20:27:12 INFO (MainThread) [zigpy.device] [0x9817] Requesting 'Node Descriptor'
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.device] [0x9817] Extending timeout for 0x3a request
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 58 under 59 request id, data: b'3a1798'
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e7c000700aa00
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x9817>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'8d179882dd91feff5f325080'
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 59, 0, <DeconzAddressEndpoint address_mode=2 address=0x9817 endpoint=0>, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, b':\x17\x98', 2, 0)
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x127c00190012003b0002179800000002000003003a17980200
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x127c00090002002a3b
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, 59]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e7d000700aa00
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c7e000c000500021798ffe4
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x177d000800010001
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x177d002b0024002a0200000102310d01040100ef0d000960010000030200040000006400afff4d0a0c00c0
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0d31>, 1, 260, 61184, b'\t`\x01\x00\x00\x03\x02\x00\x04\x00\x00\x00d', 0, 175, 255, 77, 10, 12, 0, -64]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=96 command_id=1>
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=515, function=0, data=[4, 0, 0, 0, 100])]
2021-11-18 20:27:12 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:7a:31 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 100]]
2021-11-18 20:27:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0D31:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 0
2021-11-18 20:27:12 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:7a:31 Tuya Attribute Cache : [{32769: 0, 32770: 0, 8: 0}]
2021-11-18 20:27:12 DEBUG (MainThread) [homeassistant.components.zha.cover] setting position: 0
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e7e000700aa00
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0d31>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09600100000302000400000064'
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e7f000700ae00
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2021-11-18 20:27:12 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c80000c000500021798ffe4
[... cut ...] b'\t`\x01\x00\x00\x07\x04\x00\x01\x01', 0, 175, 255, 107, 10, 12, 0, -53]
2021-11-18 20:27:13 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=96 command_id=1>
2021-11-18 20:27:13 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 1])]
```

joaoasilva, 2021-11-18:
Thanks, @jacekk015 💯

jacekk015, 2021-11-19:
Sorry but I don't see any communication between TRV and HA.
Only pairing logs.
Try to get more time of logs. If it wont help we try to give you some file to force the communication.

We are searching for something like this:
"Received value [0, 0, 0, 42] for attribute 0x0265"
```
2021-09-24 15:01:03 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=16, command_id=528, function=0, data=[4, 0, 0, 0, 34])]
2021-09-24 15:01:03 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 34] for attribute 0x0210 (command 0x0002)
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=2>
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=24, command_id=536, function=0, data=[4, 0, 0, 0, 233])]
2021-09-24 15:01:13 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 233] for attribute 0x0218 (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=9 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=30, command_id=286, function=0, data=[1, 0])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0] for attribute 0x011e (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=10 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=34, command_id=546, function=0, data=[4, 0, 0, 0, 131])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 131] for attribute 0x0222 (command 0x0002)
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=2>
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=101, command_id=613, function=0, data=[4, 0, 0, 0, 42])]
2021-09-24 15:01:25 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 42] for attribute 0x0265 (command 0x0002)
```

joaoasilva, 2021-11-19:
I'm not getting those unless I add the line ("_TZE200_a4bpgplm", "TS0601")
If I add them it seems to consume some of them and others doesn't
Can you send the file to force the communication? is there a way to filter by device too?
THanks

jacekk015, 2021-11-19:
OK - so if you already have some quirk inside just add that line and:
remove and re-pair the device
Still wait for some time 2-3 minutes min, because after pairing TRV start to send all data, but it takes some time.
Put here the logs as a file, if they're too big, or to some pastebin.com service and give here a link.
Don't worry if there are 1000 lines or more - I filter that out easily.

joaoasilva, 2021-11-19:
I'm not sure what's happening but I've added the device and see the logs when adding but then I can control some of the things in ex: Lovelace and see the actual set time but no logs from this specific device appear in zhaquirks logs
I can see other logs from the cover, wall switches, etc

joaoasilva, 2021-11-19:
`021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 513, 17) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1cfc000c00050002eb63ffe7
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17fc002b0024002a0200000102310d01040100ef0d000959010000030200040000000000afff2b470c00d3
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0d31>, 1, 260, 61184, b'\tY\x01\x00\x00\x03\x02\x00\x04\x00\x00\x00\x00', 0, 175, 255, 43, 71, 12, 0, -45]
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=89 command_id=1>
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=515, function=0, data=[4, 0, 0, 0, 0])]
2021-11-19 14:59:09 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0D31:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 100
2021-11-19 14:59:09 DEBUG (MainThread) [homeassistant.components.zha.cover] setting position: 100
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1cfd000c0005000247fbe4d2
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0d31>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09590100000302000400000000'
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17fd000800010001
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17fd0028002100220200000102f97b01040100ef0a000972010000010400010000afff2b470c00d3
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7bf9>, 1, 260, 61184, b'\tr\x01\x00\x00\x01\x04\x00\x01\x00', 0, 175, 255, 43, 71, 12, 0, -45]
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.zcl] [0x7bf9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=114 command_id=1>
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.zcl] [0x7bf9:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0])]
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7bf9>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09720100000104000100'
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17fe000800010001
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17fe050800010022
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 513, 19) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17ff000800010001
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17ff050800010022
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 513, 20) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1701000800010001
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 513, 30) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1701050800010022
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 513, 41) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1702000800010001
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e02000700aa00
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1702002b0024002202000001023e5a01040100ef0d00092b010000030200040000000000aff82b470c00b8
[... cut ...]
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 1, 33) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 1, 49) params: FOREIGN KEY constraint failed
2021-11-19 14:59:09 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (50:32:5f:ff:fe:91:dd:82, 1, 1, 51) params: FOREIGN KEY constraint failed
`

Getting a lot of this _unsupported_attribute_added for the device

jacekk015, 2021-11-19:
Are you sometimes not messing Deconz with ZHA??
Did you enable all debug elements properly??
https://github.com/zigpy/zha-device-handlers/issues/1123#issuecomment-962602119

joaoasilva, 2021-11-19:
Finally managed to get something:
Setting temperature:
```
2021-11-19 16:31:04 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 125] for attribute 0x0202 (command 0x0001)
2021-11-19 16:31:07 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 130] for attribute 0x0202 (command 0x0001)
2021-11-19 16:31:09 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 135] for attribute 0x0202 (command 0x0001)
2021-11-19 16:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 140] for attribute 0x0202 (command 0x0001)
```
enabling auto mode:
```
2021-11-19 16:32:09 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x0401 (command 0x0001)
2021-11-19 16:32:09 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 150] for attribute 0x0202 (command 0x0001)
```
disabling auto mode:
```
2021-11-19 16:33:25 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0001)
2021-11-19 16:33:25 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 60] for attribute 0x0202 (command 0x0001)
```
from 5 degrees to "off" mode:
```
2021-11-19 16:34:15 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [2] for attribute 0x0401 (command 0x0001)
2021-11-19 16:34:15 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
```
from "off" to 5.5 degrees:
```
2021-11-19 16:36:23 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0001)
2021-11-19 16:36:23 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 55] for attribute 0x0202 (command 0x0001)
```
child lock on:
`2021-11-19 16:37:34 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [1] for attribute 0x010c (command 0x0001)`

child lock off:
`2021-11-19 16:37:50 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x010c (command 0x0001)`

from 30 degrees to "on":
```
2021-11-19 16:49:28 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [3] for attribute 0x0401 (command 0x0001)
2021-11-19 16:49:30 DEBUG (MainThread) [zhaquirks.tuya] [0xfb47:1:0xef00] Received value [0, 0, 0, 214] for attribute 0x0203 (command 0x0001)
```

room temperature?
```2021-11-19 16:53:37 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 235] for attribute 0x0203 (command 0x0001)```

random responses 0x0202 ?
```
2021-11-19 16:53:30 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0001)
2021-11-19 16:53:30 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 1, 39] for attribute 0x0202 (command 0x0001)
2021-11-19 16:53:32 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [3] for attribute 0x0401 (command 0x0001)
2021-11-19 16:53:32 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 1, 44] for attribute 0x0202 (command 0x0001)
```

random not sure what they are but it seemed during a calibration? I think I've taken the batteries out by mistake
```
2021-11-19 16:43:38 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 1, 44] for attribute 0x0202 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0203 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x0406 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x0407 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x0108 (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0] for attribute 0x010c (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 91] for attribute 0x020d (command 0x0001)
2021-11-19 16:43:39 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x020f (command 0x0001)
2021-11-19 16:43:40 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 1, 44] for attribute 0x0210 (command 0x0001)
2021-11-19 16:43:40 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [255, 255, 255, 246] for attribute 0x0265 (command 0x0001)
2021-11-19 16:43:40 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0266 (command 0x0001)
2021-11-19 16:43:40 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 0, 117] for attribute 0x0296 (command 0x0001)
2021-11-19 16:43:44 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [1, 6, 0, 0, 200, 12, 0, 0, 150, 18, 0, 0, 220, 22, 0, 0, 150] for attribute 0x0011 (command 0x0001)
2021-11-19 16:43:44 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [2, 6, 0, 0, 200, 12, 0, 0, 150, 18, 0, 0, 220, 22, 0, 0, 150] for attribute 0x0012 (command 0x0001)
[... cut ...]
2021-11-19 16:43:44 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [5, 6, 0, 0, 200, 12, 0, 0, 150, 18, 0, 0, 220, 22, 0, 0, 150] for attribute 0x0015 (command 0x0001)
2021-11-19 16:43:44 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [6, 6, 0, 0, 200, 12, 0, 0, 150, 18, 0, 0, 220, 22, 0, 0, 150] for attribute 0x0016 (command 0x0001)
2021-11-19 16:43:44 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [7, 6, 0, 0, 200, 12, 0, 0, 150, 18, 0, 0, 220, 22, 0, 0, 150] for attribute 0x0017 (command 0x0001)
```

Also, it takes 3xAA batteries... battery status isn't working.

joaoasilva, 2021-11-19:
If it helps:
   // Haozee TS0601 TRV
    haozeeSystemMode: 1,
    haozeeHeatingSetpoint: 2,
    haozeeLocalTemp: 3,
    haozeeBoostHeating: 4,
    haozeeBoostHeatingCountdown: 5,
    haozeeRunningState: 6,
    haozeeWindowState: 7,
    haozeeWindowDetection: 8,
    haozeeChildLock: 12,
    haozeeBattery: 13,
    haozeeFaultAlarm: 14,
    haozeeMinTemp: 15,
    haozeeMaxTemp: 16,
    haozeeScheduleMonday: 17,
    haozeeScheduleTuesday: 18,
    haozeeScheduleWednesday: 19,
    haozeeScheduleThursday: 20,
    haozeeScheduleFriday: 21,
    haozeeScheduleSaturday: 22,
    haozeeScheduleSunday: 23,
    haozeeTempCalibration: 101,
    haozeeValvePosition: 102,
    haozeeSoftVersion: 150,

taken from another library

joaoasilva, 2021-11-19:
I have put it outside so the temp would go down quickly to see if I could get the window state but not sure if this is right:
```
2021-11-19 18:23:54 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 2, 228] for attribute 0x0266 (command 0x0001)
2021-11-19 18:25:58 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 3, 222] for attribute 0x0266 (command 0x0001)
2021-11-19 18:28:02 DEBUG (MainThread) [zhaquirks.tuya] [0xeb37:1:0xef00] Received value [0, 0, 3, 232] for attribute 0x0266 (command 0x0001)
```

not sure what it means, seems the same as **heating on**

jacekk015, 2021-11-19:
Window detect never works on window open ;-). Try to put it to the freezer. Checked personally

joaoasilva, 2021-11-19:
Don't I need to enable the detection first though? Not sure how to do it

jacekk015, 2021-11-19:
@MattWestb where do we put that TRV?? New file?? Haozee?

MattWestb, 2021-11-19:
The manufacture is Rti- Tek and Haizee is one white labeling of it.
I think Rti sounds good or what do think ?

They have making 2 models one with LED and one with OLED color display.
The good they is using the same Manufacturer Name and Model ID on them.

OEM / Brand | Model | Manufacturer Name | Model ID | ZHA Quirk | Request
-- | -- | -- | -- | -- | --
Rti-Tek | TRV 601 | _TZE200_a4bpgplm | TS0601 | ⭕ (Moes I ?) |  
Rti-Tek | TRV 602 | _TZE200_a4bpgplm | TS0601 | ⭕ (Moes I ?) |

jamjam9, 2021-11-20:
I have one of these and am happy to test also. However when I follow the instructions above and add in the line in the custom quirks copy of the ts0601_trv file i get...

ImportError: cannot import name 'TuyaPowerConfigurationCluster2AA' from 'zhaquirks.tuya' 

Showng in the logs , any ideas?

jacekk015, 2021-11-20:
@jamjam9 We haven't made a proper quirk for that device yet. Give use few days. You will get it before winter for sure ;-) You will be notified when the file will be ready for test.

jamjam9, 2021-11-20:
Cheers thanks. Obviously my rank amaturism showing, poking about files I have no clue about :)

jacekk015, 2021-11-29:
@joaoasilva @jamjam9 
Remove old quirk from the folder and replace it with the file below.
Since you've tested other code it's better to remove device, restart HA, re-pair device.
After pairing it takes a minute or two to fill-in the fields, or battery level.
[ts0601_trv_rtitek.py.zip](https://github.com/zigpy/zha-device-handlers/files/7621584/ts0601_trv_rtitek.py.zip)

In case problems be sure to enable debug logging:
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
And paste here the logs, don't remove any line from the logs.

joaoasilva, 2021-11-30:
Hi @jacekk015 ,
Thank you so much!
I'm getting:
```
File "/config/zha_custom_quirks/ts0601_trv_rtitek.py", line 71, in __init__
self.endpoint.device.thermostat_onoff_bus.add_listener(self)
AttributeError: 'Rti' object has no attribute 'thermostat_onoff_bus'
```

jacekk015, 2021-11-30:
@joaoasilva Corrected file
[ts0601_trv_rtitek.py.zip](https://github.com/zigpy/zha-device-handlers/files/7625420/ts0601_trv_rtitek.py.zip)

MattWestb, 2021-11-30:
### Futures found / used in the TRV in the tuya TRV Matrix:
Missed or wrong pleas correcting my.
- - -
Modes:

0 Schedule ✅
1 Manual ✅
2 Off ?
3 On ?

Setting / reporting current setpoint temperature ✅
Reporting local temperature ✅
Calibrating local temperature ✅ (=> device sending new updated local temperature)
Setting min heat setpoing limit ✅
Setting max heat setpoing limit ✅
Setting Boost (On/Off) ✅
Setting Boost time (sec) ✅
Setting and reporting Schedules ✅ (local on device and on cluster attributes)
Setting / reporting child lock ✅
Setting / reporting window detection (on/off) ✅
Reporting valve position (%) ✅
Reporting valve state (Is Opened/Closed) ✅
Reporting battery % ✅

ZHA support:
✅ = Yes in ZHA Extended GUI function in local quirk.

- - -

I cant understand the On , Off and boost if they is implanted as mode / presets or not or as switches in GUI.

If boost is implanted as one presets and start doing countdown and going back to manual looks logic for my and can being user in HA automatons.
On/Off is problematic then its normally braking the heating algorithm in the device but can being used of some users.

joaoasilva, 2021-11-30:
Error when changing mode:
```
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
await handler.job.target(service_call)
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
await self.hass.helpers.service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 667, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 863, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 704, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/zha/climate.py", line 438, in async_set_hvac_mode
if await self._thrm.async_set_operation_mode(HVAC_MODE_2_SYSTEM[hvac_mode]):
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 243, in async_set_operation_mode
if not await self.write_attributes({"system_mode": mode}):
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 292, in write_attributes
res = await self.cluster.write_attributes(data, **kwargs)
File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 544, in write_attributes
manufacturer_attrs.update(new_attrs)
TypeError: 'NoneType' object is not iterable
```

jacekk015, 2021-11-30:
Only user can confirm, since in Z2M I've seen some comments that Boost/Boost Countdown doesn't work. So we need to wait for tests.
I always map a Boost to Presets, same like Eco mode and Comfort mode if it exists.
On/Off - there's no such Presets. not sure what to do with it.

Maxsmart has On/Off in Manual mode, when you go to 0 or over max temp.

jacekk015, 2021-11-30:
@joaoasilva 

In case of problems be sure to enable debug logging.

logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
**And paste here the logs, don't remove any line from the logs.**

Did you remove old quirk file?
Did you re-paired device and restarted HA??
I need logs from full logs.
They should look like:
```
manufacturer=None tsn=219 command_id=Command.Report_Attributes>
2021-11-30 13:07:04 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=30983>)]]
2021-11-30 13:07:04 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0400] Attribute report received: measured_value=30983
2021-11-30 13:07:20 DEBUG (MainThread) [zigpy.zcl] [0x669a:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=234 command_id=Command.Report_Attributes>
2021-11-30 13:07:20 DEBUG (MainThread) [zigpy.zcl] [0x669a:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=26274>)]]
2021-11-30 13:07:20 DEBUG (MainThread) [zigpy.zcl] [0x669a:2:0x0400] Attribute report received: measured_value=26274
2021-11-30 13:07:37 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=220 command_id=1>
2021-11-30 13:07:37 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0019] ZCL request 0x0001: [0, 4107, 269, 1107323831, None]
2021-11-30 13:07:37 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0019] OTA query_next_image handler for 'Philips SML002': field_control=0, manufacture_id=4107, image_type=269, current_file_version=1107323831, hardware_version=None
2021-11-30 13:07:37 DEBUG (MainThread) [zigpy.zcl] [0x53a4:2:0x0019] No OTA image is available
2021-11-30 13:08:38 DEBUG (MainThread) [zigpy.zcl] [0x98fd:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=31 command_id=Command.Report_Attributes>
2021-11-30 13:08:38 DEBUG (MainThread) [zigpy.zcl] [0x98fd:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int8s, value=0>)]]
2021-11-30 13:08:38 DEBUG (MainThread) [zigpy.zcl] [0x98fd:1:0x0006] Attribute report received: on_off=0
2021-11-30 13:08:58 DEBUG (MainThread) [zigpy.zcl] [0x980f:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=141 command_id=Command.Report_Attributes>
2021-11-30 13:08:58 DEBUG (MainThread) [zigpy.zcl] [0x980f:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=4423>)]]
2021-11-30 13:08:58 DEBUG (MainThread) [zigpy.zcl] [0x980f:1:0x0405] Attribute report received: measured_value=4423
```

MattWestb, 2021-11-30:
@joaoasilva need one updated preset patch for ZHA :-((

joaoasilva, 2021-11-30:
```
2021-11-30 12:36:18 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:18 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:7a:31 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Data: [1, 0]]
2021-11-30 12:36:18 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=45 command_id=1>
2021-11-30 12:36:18 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=515, function=0, data=[4, 0, 0, 0, 0])]
2021-11-30 12:36:18 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:00 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 0]]
2021-11-30 12:36:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE5CB:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 100
2021-11-30 12:36:18 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:00 Tuya Attribute Cache : [{32769: 0, 32770: 0, 8: 100}]
2021-11-30 12:36:18 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=112 command_id=1>
2021-11-30 12:36:18 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:18 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:5a Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Data: [1, 0]]
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=45 command_id=1>
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0xe5cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=515, function=0, data=[4, 0, 0, 0, 0])]
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:00 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 0]]
2021-11-30 12:36:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE5CB:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 100
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:00 Tuya Attribute Cache : [{32769: 0, 32770: 0, 8: 100}]
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x9ec8:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=1>
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x9ec8:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=522, function=0, data=[4, 0, 0, 116, 48])]
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:8d:e1:c3 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x020a, Function: 0x00, Data: [4, 0, 0, 116, 48]]
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=1>
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:7a:31 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Data: [1, 0]]
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=113 command_id=1>
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:5a Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Data: [1, 0]]
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=1>
2021-11-30 12:36:19 DEBUG (MainThread) [zigpy.zcl] [0x0d31:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:19 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:7a:31 Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Da [... cut ...]
2021-11-30 12:36:23 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=122 command_id=1>
2021-11-30 12:36:23 DEBUG (MainThread) [zigpy.zcl] [0xc3cb:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1031, function=0, data=[1, 0])]
2021-11-30 12:36:23 DEBUG (MainThread) [zhaquirks.tuya] 04:cd:15:ff:fe:7f:79:5a Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0407, Function: 0x00, Data: [1, 0]]
```

joaoasilva, 2021-11-30:
Set max temperature and min temperature works but with a bug
5 minimum in HA will put 0.5 in the TRV
35 maximum in HA will put 3.5 in the TRV

jacekk015, 2021-11-30:
Are you using some very, very, very, very old HA version???
That error is nor related to TRV code itself, but for missing some internal files.
Which version of HA u use?

joaoasilva, 2021-11-30:
Operating System
Home Assistant OS 6.6

Version
supervisor-2021.10.8
Newest Version
supervisor-2021.10.8

Version
core-2021.11.5
Newest Version
core-2021.11.5

jacekk015, 2021-11-30:
OK what is Changing mode?? Since your TRV is not supported you do not have presets.
I assume that you click a Fire Icon to get that error - that's correct??

joaoasilva, 2021-11-30:
> OK what is Changing mode?? Since your TRV is not supported you do not have presets.
> I assume that you click a Fire Icon to get that error - that's correct??

Yes, clicked the fire icon

jacekk015, 2021-11-30:
OK, so most of the TRV's doesn't have OFF or ON mode. So clicking Fire Icon does nothing. Presets are used to change modes like Auto/Manual and etc. To have Presets working @MattWestb can prepare a file to upload or you need to wait for the official update of your TRV - then HA will show Presets list under 3 dots of Lovelace view of TRV.
You're suppose to have ON/OFF mode.
I can map Off Icon to OFF mode, and I'll try to map Fire Icon to iterate between Presets.

jacekk015, 2021-11-30:
Clicking Fire/Off icon should iterate between TRV modes.
[ts0601_trv_rtitek.py.zip](https://github.com/zigpy/zha-device-handlers/files/7626287/ts0601_trv_rtitek.py.zip)

MattWestb, 2021-11-30:
@jacekk015 is the normal "moes" working then you is only using presets 1, 2 and 5  or must i doing one patched of your latest ?

jacekk015, 2021-11-30:
Standard MOES was the reason the Presets exists:
```
MOES_MODE_ATTR = 0x0404  # [0] away [1] scheduled [2] manual [3] comfort [4] eco [5] boost [6] complex
```
.
```
    class Preset(t.enum8):
        """Working modes of the thermostat."""

        Away = 0x00
        Schedule = 0x01
        Manual = 0x02
        Comfort = 0x03
        Eco = 0x04
        Boost = 0x05
        Complex = 0x06
```
.
```
        MODELS_INFO: [
            ("_TZE200_ckud7u2l", "TS0601"),
            ("_TZE200_ywdxldoj", "TS0601"),
            ("_TZE200_cwnjrr72", "TS0601"),
        ],
        MODELS_INFO: [
            ("_TYST11_ckud7u2l", "kud7u2l"),
            ("_TYST11_ywdxldoj", "wdxldoj"),
            ("_TYST11_cwnjrr72", "wnjrr72"),
        ],
```

MattWestb, 2021-11-30:
@joaoasilva
Copy the unziped `climate.py` to your HA config directory.
Open one CLI to the HA docker container and copy the file in to it with:
```
cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py
```
Instruction for getting one docker container CLI https://github.com/zigpy/zha-device-handlers/discussions/693
[climate_rtitek.zip](https://github.com/zigpy/zha-device-handlers/files/7626418/climate_rtitek.zip)

Restart HA and you shall getting preset in your TRV card like this (but with more presets):
[image]

jacekk015, 2021-11-30:
@MattWestb He has mapped also Boost there.

joaoasilva, 2021-11-30:
This is what happens when I change mode:
```
2021-11-30 16:53:56 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0x0201] Mapping standard operation_preset (0x4002) with value <Preset.Boost: 5> to custom {260: 1}
2021-11-30 16:53:57 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: wrote {'operation_preset': 5} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2021-11-30 16:53:57 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=2>
2021-11-30 16:53:57 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=67, command_id=260, function=0, data=[1, 1])]
2021-11-30 16:53:57 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0xef00] Received value [1] for attribute 0x0104 (command 0x0002)
2021-11-30 16:53:57 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[programing_oper_mode] = ProgrammingOperationMode.Simple
2021-11-30 16:53:57 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[occupancy] = Occupancy.Occupied
2021-11-30 16:53:57 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[operation_preset] = Preset.Boost
2021-11-30 16:53:57 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=80 command_id=Command.Default_Response>
2021-11-30 16:53:57 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x000b: [0, <Status.SUCCESS: 0>]
2021-11-30 16:54:00 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0x0201] Mapping standard operation_preset (0x4002) with value <Preset.Manual: 2> to custom {1025: 1, 260: 0}
2021-11-30 16:54:00 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=2>
2021-11-30 16:54:00 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=68, command_id=1025, function=0, data=[1, 1])]
2021-11-30 16:54:00 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0002)
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[programing_oper_mode] = ProgrammingOperationMode.Simple
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[occupancy] = Occupancy.Occupied
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[operation_preset] = Preset.Manual
2021-11-30 16:54:00 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=88 command_id=Command.Default_Response>
2021-11-30 16:54:00 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x000b: [0, <Status.SUCCESS: 0>]
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: wrote {'operation_preset': 2} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=92 command_id=Command.Default_Response>
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x000b: [0, <Status.SUCCESS: 0>]
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=2>
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=69, command_id=260, function=0, data=[1, 0])]
2021-11-30 16:54:01 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0002)
```
Doesn't seem to make any difference.
Also, I can see all the modes, shouldn't just be the supported ones?

The temperature calibration goes between -20 and 20, is that right?
When I change it it not sure what it does though
Here are the logs:
```
2021-11-30 16:53:56 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0x0201] Mapping standard operation_preset (0x4002) with value <Preset.Boost: 5> to custom {260: 1}
2021-11-30 16:53:57 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=2>
[... cut ...]
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=2>
2021-11-30 16:54:01 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=69, command_id=260, function=0, data=[1, 0])]
2021-11-30 16:54:01 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0002)
```

jacekk015, 2021-11-30:
> Doesn't seem to make any difference.

I don't understand your responses. If you want it to work somehow you need to explain it a bit more.
It works? It don't work?

> This is what happens when I change mode:
Yes I know that. Modes are changing - what we expected more?
You've switched to Manual mode.
```
2021-11-30 16:54:00 DEBUG (MainThread) [zigpy.zcl] [0xb2a7:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=68, command_id=1025, function=0, data=[1, 1])]
2021-11-30 16:54:00 DEBUG (MainThread) [zhaquirks.tuya] [0xb2a7:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0002)
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[programing_oper_mode] = ProgrammingOperationMode.Simple
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[occupancy] = Occupancy.Occupied
2021-11-30 16:54:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB2A7:1:0x0201]: Attribute report 'RtiThermostat'[operation_preset] = Preset.Manual
```

> The temperature calibration goes between -20 and 20, is that right?
When I change it it not sure what it does though

It changes the same parameter that when you change it on your TRV directly. There's manual for that TRV for sure.
Temperature calibration often shifts local temperature or target temperature readings.
Again - see manual.

> Also, I can see all the modes, shouldn't just be the supported ones?
Yes, but Matt gave you workaround to have Presets right away, instead of waiting for next HA release.

We are more interested if those function works or not.
Do you can set temperature from HA? Is HA see your TRV room temp? Is battery level reported? etc...
Are Min/Max temps now OK?
How's usability.

joaoasilva, 2021-11-30:
Thanks for all the work!
Overall all seems to work, it's more than usable!
Can set the temperature, get the room temperature and everything. Max/Min is working too!

jacekk015, 2021-11-30:
Please test it few days. In case of problems or questions write here. If all is OK we try to publish it so everyone will have support.
@jamjam9 You're also welcome to tests.

jamjam9, 2021-11-30:
Interview gets completed but device still on 'configuring' . No option to set device name.

logs ....

Log Details (WARNING)
Logger: zigpy_znp.utils
Source: custom_zha_quirks/ts0601_trv_rtitek.py:71
First occurred: 21:57:03 (1 occurrences)
Last logged: 21:57:03

Caught an exception while executing callback
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/utils.py", line 138, in _resolve
    result = self.callback(response)
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 924, in on_zdo_tc_device_join
    self.handle_join(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 293, in handle_join
    dev.schedule_initialize()
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 120, in schedule_initialize
    self._application.device_initialized(self)
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 137, in device_initialized
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 44, in get_device
    return _DEVICE_REGISTRY.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/registry.py", line 148, in get_device
    device = candidate(device._application, device.ieee, device.nwk, device)
  File "/config/custom_zha_quirks/ts0601_trv_rtitek.py", line 728, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 683, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 98, in __init__
    self.add_endpoint(endpoint_id, replace_device=replaces)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 113, in add_endpoint
    ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 137, in __init__
    cluster = c(self, is_server=True)
  File "/config/custom_zha_quirks/ts0601_trv_rtitek.py", line 71, in __init__
    self.endpoint.device.thermostat_onoff_bus.add_listener(self)
AttributeError: 'Rti' object has no attribute 'thermostat_onoff_bus'

jamjam9, 2021-11-30:
Some more logs from the second attempt at adding device.....

2021-11-30 22:00:01 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xA2A4, SrcIEEE=50:32:5f:ff:fe:40:47:2f, ParentNwk=0xB79A)
2021-11-30 22:00:01 INFO (MainThread) [zigpy.application] Device 0xa2a4 (50:32:5f:ff:fe:40:47:2f) joined the network
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.application] Device 50:32:5f:ff:fe:40:47:2f changed id (0xa2a4 => 0xa2a4)
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.device] [0xa2a4] Skipping initialization, device is fully initialized
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_a4bpgplm' nwk=0xA2A4 ieee=50:32:5f:ff:fe:40:47:2f is_initialized=True>
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_a4bpgplm TS0601 (50:32:5f:ff:fe:40:47:2f)
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_trv_rtitek.Rti'>
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 50:32:5f:ff:fe:40:47:2f: <class 'ts0601_trv_rtitek.Rti'>
2021-11-30 22:00:01 WARNING (MainThread) [zigpy_znp.utils] Caught an exception while executing callback
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/utils.py", line 138, in _resolve
    result = self.callback(response)
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 924, in on_zdo_tc_device_join
    self.handle_join(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 293, in handle_join
    dev.schedule_initialize()
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 120, in schedule_initialize
    self._application.device_initialized(self)
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 137, in device_initialized
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 44, in get_device
    return _DEVICE_REGISTRY.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/registry.py", line 148, in get_device
    device = candidate(device._application, device.ieee, device.nwk, device)
  File "/config/custom_zha_quirks/ts0601_trv_rtitek.py", line 728, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 683, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 98, in __init__
    self.add_endpoint(endpoint_id, replace_device=replaces)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 113, in add_endpoint
    ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 137, in __init__
    cluster = c(self, is_server=True)
  File "/config/custom_zha_quirks/ts0601_trv_rtitek.py", line 71, in __init__
    self.endpoint.device.thermostat_onoff_bus.add_listener(self)
AttributeError: 'Rti' object has no attribute 'thermostat_onoff_bus'
2021-11-30 22:00:01 DEBUG (MainThread) [zigpy.appdb] Device 50:32:5f:ff:fe:40:47:2f already exists. Updating it.
2021-11-30 22:00:02 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0xA2A4, NWK=0xA2A4, IEEE=50:32:5f:ff:fe:40:47:2f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded: 128>)
2021-11-30 22:00:02 INFO (MainThread) [zigpy.application] Device 0xa2a4 (50:32:5f:ff:fe:40:47:2f) joined the network
2021-11-30 22:00:02 DEBUG (MainThread) [zigpy.zdo] [0xa2a4:zdo] ZDO request ZDOCmd.Device_annce: [0xA2A4, 50:32:5f:ff:fe:40:47:2f, 128]
2021-11-30 22:00:02 DEBUG (MainThread) [zigpy.device] [0xa2a4] Extending timeout for 0x67 request
2021-11-30 22:00:03 DEBUG (MainThread) [zigpy.zcl] [0xa2a4:1:0x0004] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=103 command_id=2>
2021-11-30 22:00:03 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0xA2A4, NWK=0xA2A4, IEEE=50:32:5f:ff:fe:40:47:2f, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded: 128>)
2021-11-30 22:00:03 INFO (MainThread) [zigpy.application] Device 0xa2a4 (50:32:5f:ff:fe:40:47:2f) joined the network
2021-11-30 22:00:03 DEBUG (MainThread) [zigpy.zdo] [0xa2a4:zdo] ZDO request ZDOCmd.Device_annce: [0xA2A4, 50:32:5f:ff:fe:40:47:2f, 128]
2021-11-30 22:00:03 DEBUG (MainThread) [zigpy.device] [0xa2a4] Extending timeout for 0x68 request
2021-11-30 22:00:03 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xA2A4, SrcIEEE=50:32:5f:ff:fe:40:47:2f, ParentNwk=0xDB37)
2021-11-30 22:00:03 INFO (MainThread) [zigpy.application] Device 0xa2a4 (50:32:5f:ff:fe:40:47:2f) joined the network
2021-11-30 22:00:03 DEBUG (MainThread) [zigpy.device] [0xa2a4] Cancelling old group rescan
2021-11-30 22:00:03 DEBUG (MainThread) [zigpy.device] [0xa2a4] Extending timeout for 0x69 request
2021-11-30 22:00:03 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xA2A4, SrcIEEE=50:32:5f:ff:fe:40:47:2f, ParentNwk=0xDB37)
[... cut ...]
2021-11-30 22:02:25 DEBUG (MainThread) [zigpy.zcl] [0xa2a4:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=42 command_id=Command.Report_Attributes>
2021-11-30 22:02:25 DEBUG (MainThread) [zigpy.zcl] [0xa2a4:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65503, value=<TypeValue type=CharacterString, value=�Z9)�Z9)>)]]
2021-11-30 22:02:25 DEBUG (MainThread) [zigpy.zcl] [0xa2a4:1:0x0000] Attribute report received: 65503=�Z9)�Z9)

jacekk015, 2021-11-30:
@jamjam9 use never version

https://github.com/zigpy/zha-device-handlers/issues/1159#issuecomment-982671444

laurius, 2021-11-30:
Also have the version with the display (Salcar), for me the newer version did work.
Setpoint and current temperature value look fine:
[image]
[image]
Min max values work but are displayed multiplied by 10:
[image]
Did not use the modified climate.py yet.

MattWestb, 2021-12-01:
### Looks very nice @laurius !!
I think the wrong scaling is fast fixed by **jacekk015** then hi is getting time looking on the code.
Is the TRV reporting the local temperature every minute ?
My Siter is doing only one time in the hour and cant being tricked ;-((

If it's possible copy the presets file to your HA docker container so you can trying the mode setting is working OK to and from the TRV and you can using the boos mode with 
 the countdown..

Did you finding more information of the TRV (then its having 3 AA batteries the mass) ?
Is it possible doing some setting in the TRV like the min and max setpoint temperature ?
We dont knowing them for the moment also the local temperature correction min and max is not known.

If you can getting them it's easy fixing it in the code so the device is not getting problem then sending out of range data to it.

Is the valve position always 100% or do it dynamic updating then heating more or less ?
I think the on/off switch is children clock, is it working both ways (local and in ZHA reporting OK for both in ZHA) ?

Brave user !!!

jacekk015, 2021-12-01:
@laurius Could you test this version. Please also enable debug logging and in case of problems send them also.
Min/Max temps are under attributes 0x020F and 0x0210:
```
RTI_MIN_TEMPERATURE_ATTR = 0x020F  # minimum limit of temperature setting (decidegree)
RTI_MAX_TEMPERATURE_ATTR = 0x0210  # maximum limit of temperature setting (decidegree)
```
[ts0601_trv_rtitek.py.zip](https://github.com/zigpy/zha-device-handlers/files/7632809/ts0601_trv_rtitek.py.zip)

laurius, 2021-12-01:
Local temperature is reported about every minute. The TRV has 3 batteries, the batterie level is reported in percent, looks correct to me. The valve is not always at 100% on the TRV display, the value displayed in output valve state is not the same. Changing the valve state does nothing on the TRV. The on_off button works for child lock, displays also correctly when changing on TRV. Will test the other things later after work. Thanks for making this work, thought i wouldn't be able to use those functions.

MattWestb, 2021-12-01:
The valve % is made as one read / write sensor one but its cant being written (We was not getting the read only working OK in ZHA) so its looks being OK.

laurius, 2021-12-01:
One quick question, i don't run a docker installation but Hassos on a raspberry pi. when i try:
`cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py`
the path does not exist. Where do i need to copy it or should i create that path?

MattWestb, 2021-12-01:
It was some years i was running native pi for HA but i cant remember where is hidden.
Try searching for `climate.py` in the fie system and you is finding it.

[206 more replies]