zigpy/zha-device-handlers #1162: [BUG] hvac_action value of Bitron Termostat is always off

kovbalazs89, 2021-11-19:
The HVAC_ACTION attribute of my Bitron Home 902010/32 thermostat is always off , never changes.  The associated sensor in HomeAssistant shows always off too. Because of this, I cant see the hvac status of my thermostat in HomeAssistant.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Integrations
2. Click on ZHA
3. Pair the Bitron Thermostat to HomeAssistant
4. See the attributes of the climate entity

**Expected behavior**
Climate state changes in HomeAssistant after the thermostat turned physically the relay on and off.

**Screenshots**
I was able the query the real cluster attribute of this termostat, when the heating is on or off.

[image]
[image]

**Additional context**
I have created an sql query sensor in Home Assistant, to track the state of the running_state attribute. It can show the corract state from zigbee.db file.

```
  - platform: sql
    db_url: sqlite:////config/zigbee.db
    scan_interval: 10
    queries:
      - name: hvac_mode
        query: "SELECT value FROM attributes_cache_v7 where ieee = '00:0d:6f:00:17:0c:df:fc' and cluster = 513 and attrid = 41"
        column: "value" 
```

[image]

I think, we need a replacement in this device_handler, to get the real state of the termostat's hvac_action.

jacekk015, 2021-11-22:
Device signature needed

kovbalazs89, 2021-11-23:
I have attached the technical documentation of the thermostat:
[91ZbuTU-duS.pdf](https://github.com/zigpy/zha-device-handlers/files/7587168/91ZbuTU-duS.pdf)

The relevant section is:
[image]
[image]

jacekk015, 2021-11-23:
Again: Device signature needed

marcospunkt, 2022-03-13:
Hello,

i have the same Problem. Here is de Zigbee signature:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4209, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0020",
        "0x0201",
        "0x0204",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Bitron Home",
  "model": "902010/32",
  "class": "zhaquirks.bitron.thermostat.Av201032"
}`

closed 2022-09-16 by github-actions[bot]