zigpy/zha-device-handlers #1164: [Device Support Request] Tuya SNTZ007 Zigbee contact sensor

slawslom, 2021-11-20:
**Is your feature request related to a problem? Please describe.**
I bougth this contact sensor:  https://www.houseiq.pl/pl/p/Czujnik-otwarcia-zamkniecia-ZIGBEE-3.0-TUYA-SMART/844 
I think it is model SNTZ00. I use conbee II with home assistant (in docker) when I discover this device it is recognized as _TZE200_fwoorn8y TS0601 and I cant do anything with it.

**Describe the solution you'd like**
For it to work events status etc.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fwoorn8y",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Pairing log:
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x5437:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>
[0x39b7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=91 command_id=Command.Read_Attributes_rsp>
[0x39B7:1:0x0008]: received attribute: 0 update with value: 91
[0x39b7:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=95 command_id=Command.Read_Attributes_rsp>
[0x5437:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=89 command_id=Command.Read_Attributes_rsp>
[0x5437] Extending timeout for 0x61 request
[0x5437:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=97 command_id=Command.Default_Response>
[0x5437:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: [0, <Status.SUCCESS: 0>]
light.bathroom_secondary_on_off: turned off: [0, <Status.SUCCESS: 0>]
[0x5437:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x5437:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x5437:1:0x0006] Attribute report received: on_off=0
[0xa0a4:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=83 command_id=Command.Read_Attributes_rsp>
[0x4b36:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=81 command_id=Command.Read_Attributes_rsp>
[0xa0a4:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=99 command_id=Command.Read_Attributes_rsp>
[0xA0A4:1:0x0008]: received attribute: 0 update with value: 76
[0xa0a4:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=103 command_id=Command.Read_Attributes_rsp>
[0xe994:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>
[0xe994:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=105 command_id=Command.Read_Attributes_rsp>
[0xE994:1:0x0008]: received attribute: 0 update with value: 76
[0xe994:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=107 command_id=Command.Read_Attributes_rsp>
[0x4b36:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Read_Attributes_rsp>
[0x4B36:1:0x0008]: received attribute: 0 update with value: 76
[0x4b36:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Read_Attributes_rsp>
[0x015f] Delivery error for seq # 0x43, on endpoint id 1 cluster 0x0006: message send failure
[0x015F:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: [0x015f:1:0x0006]: Message send failure
[0x015F:1:0x0006]: async_initialize: retryable request #2 failed: [0x015f:1:0x0006]: Message send failure. Retrying in 1.1s
[0xa407:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=130 command_id=2>
[0xa407:1:0xef00] Unknown cluster-specific command 2
[0xa407:1:0xef00] ZCL request 0x0002: b'\x00\x02\x02\x02\x00\x04\x00\x00\x00d'
[0xa407:1:0xef00] No handler for cluster command 2
[0x015F:1:0x0006]: initializing channel: from_cache: False
[0xce83] Delivery error for seq # 0x4d, on endpoint id 1 cluster 0x0300: message send failure
[0xCE83:1:0x0300]: failed to get attributes '['color_loop_active', 'current_x', 'current_y', 'color_temperature']' on 'light_color' cluster: [0xce83:1:0x0300]: Message send failure
[0xCE83:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('[0xce83:1:0x0300]: Message send failure'), DeliveryError('[0xce83:1:0x0300]: Message send failure'), DeliveryError('[0xce83:1:0x0300]: Message send failure'), DeliveryError('[0xce83:1:0x0300]: Message send failure')]
[0xCE83:1:0x0006]: 'async_initialize' stage succeeded
[0xCE83:1:0x0300]: 'async_initialize' stage succeeded
[... 34 lines cut ...]
[55255:1:0x0000]: finished channel configuration
[55255:1:0x0019]: finished channel configuration
[55255:1:0x0000]: 'async_configure' stage succeeded
[55255:1:0x0019]: 'async_configure' stage succeeded
[55255](TS0601): completed configuration
[55255](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_fwoorn8y TS0601', ieee='a4:c1:38:e4:7c:3f:7c:aa', last_seen=1637404860.2452004)
[55255](TS0601): started initialization
[55255:ZDO](TS0601): 'async_initialize' stage succeeded
[55255:1:0x0000]: initializing channel: from_cache: False
[55255:1:0x0000]: finished channel initialization
[55255:1:0x0019]: initializing channel: from_cache: False
[55255:1:0x0019]: finished channel initialization
[55255:1:0x0000]: 'async_initialize' stage succeeded
[55255:1:0x0019]: 'async_initialize' stage succeeded
Device a4:c1:38:e4:7c:3f:7c:aa already exists. Updating it.
[55255](TS0601): power source: Battery or Unknown
[55255](TS0601): completed initialization
[0x015F:1:0x0006]: initializing channel: from_cache: False
[0xd7d7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=133 command_id=2>
[0xd7d7:1:0xef00] Unknown cluster-specific command 2
[0xd7d7:1:0xef00] ZCL request 0x0002: b'\x00\x04\x01\x01\x00\x01\x01'
[0xd7d7:1:0xef00] No handler for cluster command 2

Please help,
Thank you

jacekk015, 2021-11-22:
Device needs a quirk to be written.
BTW
Better return this device and buy cheaper Sonoff Zigbee SNZB-04
I use two of them since a year on same battery.
They use standard Zigbee protocols and works out of the box. Plug&Play
Pozdrawiam

slawslom, 2021-11-23:
Thank you - I will use the cheaper Sonoff then :)

jacekk015, 2021-12-05:
@slawslom Be so kind and Close this Issue.
Thanks!

slawslom, 2021-12-06:
Ok, I have closed the issue. Are all issues here resolved by "just by another device that works"? :)

closed 2021-12-06 by slawslom

jacekk015, 2021-12-06:
Definitely not. But this time I gave you a better solution, probably.

MattWestb, 2021-12-06:
You can look for PR adding devices and most is adding one or many devices or fixing issues that is coming then devices is getting upgraded firmware and changing there functionality.
https://github.com/zigpy/zha-device-handlers/pulls?q=is%3Apr+is%3Aclosed

Tuya is making many nice devices but all TS0601 using one extra MCU that sending no standard commands that need being converted to "normal" Zigbee commands and its not easy getting all in place if no one have the device, tuya TBGW and can sniffing the device communicating with tuya Cloud.

I have one list with devices i like but i dont  baying them then its not being Zigbee certified and is not working out of the box with normal Zigbee system (most TS0601 devices) like ZHA.

w0lfp4ck, 2023-01-08:
I have a few of these devices and they are MUCH cheaper than the Sonoff sensors.  That aside, I'd like to be able to use this sensor as a learning lesson.  Could you point me in the right direction?  I found a quirk that does pull in the contact state, but not the battery nor the tamper resistant function.  Here's the listed outputs of the device from Tuya (https://developer.tuya.com/en/docs/iot/s?id=K9gf48hm02l8m).  
doorcontact_state | Status of door window sensor | Boolean | {}
temper_alarm | Tamper alarm | Boolean | {}
battery_percentage | Battery capacity percentage | Integer | {“unit”:“%”,“min”:0,“max”:100,“scale”:0,“step”:1}
battery_value | Battery capacity value | Integer | {“unit”:“”,“min”:0,“max”:30000,“scale”:0,“step”:1}
battery_state | Battery capacity status | Enum | {“range”:[“low”,“middle”,“high”]}
signal_strength | Signal strength | Integer | {“min”:-255,“max”:255,“step”:1,“unit”:“dBm”,“scale”:0}

Here's the code from the quirk that DOES pull back the contact state but nothing else works.  Apologies for the long post here, any direction you can point me is more than appreciated.

"""Tuya contact sensor device."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import LocalDataCluster, DoublingPowerConfigurationCluster

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
)

from zhaquirks.tuya import TuyaLocalCluster

from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaAttributesCluster, TuyaMCUCluster, TuyaDPType

IASZONE_ZONE_STATUS_ATTR = 0x0002 

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""
 

class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: 10000.0 * math.log10(x + 1.0),
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
    }

class TuyaContactSensor(CustomDevice):
    """Tuya contact sensor device."""

    signature = {
        # NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
        # device_version=1
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1,
        # input_clusters=[0, 1, 1280],
        # output_clusters=[10, 25])
        MODELS_INFO: [("_TZE200_fwoorn8y", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                    TuyaIlluminanceMeasurement, 
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

TheJulianJES, 2023-01-08:
@w0lfp4ck I think it's best if you [open a new issue](https://github.com/zigpy/zha-device-handlers/issues/new/choose), fill-out the provided issue template, and provide your added information there.