zigpy/zha-device-handlers #1168: [Device Support Request] TS0211 by _TZ1800_ladpngdx Lidl Silvercrest Doorbell

ernstmosch, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

After Pairing to ZHA (Raspi 4/ Conbee2) the device shows a Sensor 'Ias_zone' which is in state 'Closed' - after pushing the Doorbell- button this state changes to "Open" and stays in that state.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I want to use this doorbell as a push-button for an automation so i expect it to change back to the old state.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_ladpngdx",
  "model": "TS0211",
  "class": "zigpy.device.Device"
}

Sent 'mgmt_permit_joining_req' to 84:18:26:00:00:0f:84:80: [<Status.SUCCESS: 0>]
Device 0x42d7 (84:2e:14:ff:fe:f8:09:83) joined the network
Device 84:2e:14:ff:fe:f8:09:83 changed id (0x42d7 => 0x42d7)
[0x42d7] Scheduling initialization
[0x42d7] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\xf8\xd7B\x83\t\xf8\xfe\xff\x14.\x84\x80'
[0x42d7:zdo] ZDO request ZDOCmd.Device_annce: [0x42D7, 84:2e:14:ff:fe:f8:09:83, 128]
Tries remaining: 3
[0x42d7] Discovering endpoints
Tries remaining: 3
[0x42d7] Extending timeout for 0x71 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=False> from ep 0 to ep 0, cluster 6: b'C\xfd\xff\x04\x01\x00\x01\x00\x05'
[0x42d7:zdo] ZDO request ZDOCmd.Match_Desc_req: [0xFFFD, 260, [], [1280]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'q\x00\xd7B\x01\x01'
[0x42d7] Discovered endpoints: [1]
[0x42d7] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x42d7:1] Discovering endpoint information
Tries remaining: 3
[0x42d7] Extending timeout for 0x74 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=False> from ep 0 to ep 0, cluster 32772: b't\x00\xd7B\x14\x01\x04\x01\x02\x04\x01\x05\x00\x00\x01\x00\x03\x00\x00\x05\x05\x0b\x01\x19\x00'
[0x42d7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1, input_clusters=[0, 1, 3, 1280, 2821], output_clusters=[25])
[0x42d7] Extending timeout for 0x76 request
[0x42d7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=118 command_id=Command.Read_Attributes_rsp>
[0x42d7] Read model 'TS0211' and manufacturer '_TZ1800_ladpngdx' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, ias_zone:0x0500, diagnostic:0x0B05] out=[ota:0x0019] status=<Status.ZDO_INIT: 1>>
[0x42d7] Discovered basic device information for <Device model='TS0211' manuf='_TZ1800_ladpngdx' nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=True>
Device is initialized <Device model='TS0211' manuf='_TZ1800_ladpngdx' nwk=0x42D7 ieee=84:2e:14:ff:fe:f8:09:83 is_initialized=True>
Checking quirks for _TZ1800_ladpngdx TS0211 (84:2e:14:ff:fe:f8:09:83)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
[... 56 lines cut ...]
[0x42d7:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=136 command_id=Command.Read_Attributes_rsp>
[0x42d7] Extending timeout for 0x8e request
[0x42d7:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=99 command_id=0>
[0x42d7:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Supervision_reports|Tamper: 20>, <bitmap8.0: 0>, 0, 0]
[17111:1:0x0500]: Updated alarm state: ZoneStatus.0
[0x42d7:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=142 command_id=Command.Read_Attributes_rsp>
[17111:1:0x0001]: finished channel initialization
[17111:1:0x0500]: 'async_initialize' stage succeeded
[17111:1:0x0001]: 'async_initialize' stage succeeded
[17111:1:0x0000]: 'async_initialize' stage succeeded
[17111:1:0x0019]: 'async_initialize' stage succeeded
[17111](TS0211): power source: Battery or Unknown
[17111](TS0211): completed initialization

**Additional context**
Add any other context or screenshots about the feature request here.

Accidently i just found out that the Button on the (back under the lid) changes the State back to "Closed".
It seems to me as if the two buttons "doorbell" on front "and "Manipulation Detection/Open lid" are messed up in "one sensor".

[image]

Unfortunatly i am not able to reproduce that.

Thanks in advance - Michael

magicbenny, 2021-12-10:
would appreciate that too. Button works fine in Deconz and Z2M but not in ZHA.

Joerg76, 2021-12-18:
i am also interested in a solution.

pbvdven, 2022-02-28:
Same here the state wont go back to closed after pushing the button.

pfak, 2022-03-04:
PR #1400

closed 2022-03-11 by dmulcahey

moritzbeck01, 2023-01-04:
> Did you try listening to `zha_event` and see what events come through? (Can be done in HA -> Developer -> Events -> listen for `zha_event`, then press the doorbell)

After pressing on the doorbell button, I get these three events in about the same interval.

Event 0:
```
event_type: zha_event
data:
  device_ieee: 84:2e:14:ff:fe:f6:d3:9b
  unique_id: 84:2e:14:ff:fe:f6:d3:9b:1:0x0500
  device_id: f461037a8c12c5b0b35b665927357440
  endpoint_id: 1
  cluster_id: 1280
  command: single
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-01-04T20:07:14.527905+00:00"
context:
  id: 01GNZ6A7TZR278TNNSFC2AFT2Z
  parent_id: null
  user_id: null
```

Event 1:
```
event_type: zha_event
data:
  device_ieee: 84:2e:14:ff:fe:f6:d3:9b
  unique_id: 84:2e:14:ff:fe:f6:d3:9b:1:0x0500
  device_id: f461037a8c12c5b0b35b665927357440
  endpoint_id: 1
  cluster_id: 1280
  command: single
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-01-04T20:07:16.646759+00:00"
context:
  id: 01GNZ6A9X6QZHBA42GH0MQP32A
  parent_id: null
  user_id: null
```

Event 2:
```
event_type: zha_event
data:
  device_ieee: 84:2e:14:ff:fe:f6:d3:9b
  unique_id: 84:2e:14:ff:fe:f6:d3:9b:1:0x0500
  device_id: f461037a8c12c5b0b35b665927357440
  endpoint_id: 1
  cluster_id: 1280
  command: single
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-01-04T20:07:19.666115+00:00"
context:
  id: 01GNZ6ACVJQJQXJSJCQ1QDDZHV
  parent_id: null
  user_id: null
```

moritzbeck01, 2023-01-13:
> Did you try listening to `zha_event` and see what events come through? (Can be done in HA -> Developer -> Events -> listen for `zha_event`, then press the doorbell)

Should I open a new issue for this? Or can this one be reopened.

PR #1400 (2022-03-11): get button press events from Tuya/LIDL/SilverCrest doorbell TS0211
Fixes #1168.

This PR generates events (3, because the doorbell sends state 3 times in 5 seconds after a press) when the doorbell is pushed.

The event looks like this:

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:f8:2a:08",
        "unique_id": "84:2e:14:ff:fe:f8:2a:08:1:0x0500",
        "device_id": "e3f05ce1cdd24bfe305af5c1ff166fec",
        "endpoint_id": 1,
        "cluster_id": 1280,
        "command": "single",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired" [... cut]