zigpy/zha-device-handlers #1171: [Device Support Request] EFEKTA Plant-Watering-Sensor-Zigbee

reven, 2021-11-23:
Name: efektalab.ru EFEKTA_PWS.
Custom DIY device. Pairs and appears in ZHA but fails to report soil moisture. Reported temperature and battery level are incorrect.
Info and source: https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
Sent 'mgmt_permit_joining_req' to 00:0d:6f:ff:fe:8e:7b:59: [<Status.SUCCESS: 0>]
[0xe213] Delivery error for seq # 0x72, on endpoint id 0 cluster 0x0034: message send failure
Sending 'zdo_leave_req' failed: [0xe213:0:0x0034]: Message send failure
Device 0xe213 (00:12:4b:00:24:42:d3:d2) left the network
Device 0xe213 (00:12:4b:00:24:42:d3:d2) left the network
New device 0xbf06 (00:12:4b:00:24:42:d3:d2) joined the network
[0xbf06] Scheduling initialization
Tries remaining: 3
[0xbf06] Requesting 'Node Descriptor'
Tries remaining: 2
[0xbf06] Extending timeout for 0x76 request
Device 0xbf06 (00:12:4b:00:24:42:d3:d2) joined the network
[0xbf06] Scheduling initialization
[0xbf06] Canceling old initialize call
Tries remaining: 3
[0xbf06] Requesting 'Node Descriptor'
Tries remaining: 2
[0xbf06] Extending timeout for 0x78 request
Device 0xbf06 (00:12:4b:00:24:42:d3:d2) joined the network
[0xbf06] Scheduling initialization
[0xbf06] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x06\xbf\xd2\xd3B$\x00K\x12\x00\x80'
[0xbf06:zdo] ZDO request ZDOCmd.Device_annce: [0xBF06, 00:12:4b:00:24:42:d3:d2, 128]
Tries remaining: 3
[0xbf06] Requesting 'Node Descriptor'
Tries remaining: 2
[0xbf06] Extending timeout for 0x7a request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 54: b'\x02\xb4\x01'
[0xbf06:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [180, <Bool.true: 1>]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 1 to ep 255, cluster 3: b'\x11\x01\x01'
[0xbf06] Received ZCL while uninitialized on endpoint id 1, cluster 0x0003 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=1 command_id=1>, payload: b''
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'x\x00\x06\xbf\x02@\x80\x00\x00P\xa0\x00\x00\x00\xa0\x00\x00'
[0xbf06:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xBF06, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0xbf06:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'z\x00\x06\xbf\x02@\x80\x00\x00P\xa0\x00\x00\x00\xa0\x00\x00'
[0xbf06] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xbf06] Discovering endpoints
Tries remaining: 3
[0xbf06] Extending timeout for 0x7c request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'|\x00\x06\xbf\x01\x01'
[0xbf06] Discovered endpoints: [1]
[0xbf06] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xbf06:1] Discovering endpoint information
Tries remaining: 3
[0xbf06] Extending timeout for 0x7e request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'~\x00\x06\xbf\x10\x01\x04\x01\x0c\x00\x02\x04\x00\x00\x01\x00\x02\x04\x08\x04\x00'
[0xe9c7:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:0d:6f:00:15:44:ef:91, 0, 0]
[0xbf06:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=12, device_version=2, input_clusters=[0, 1, 1026, 1032], output_clusters=[])
Unknown cluster 1032
[0xbf06] Extending timeout for 0x81 request
[0xbf06:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=129 command_id=Command.Read_Attributes_rsp>
[0xbf06] Read model 'EFEKTA_PWS' and manufacturer 'efektalab.ru' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, temperature:0x0402, None:0x0408] out=[] status=<Status.ZDO_INIT: 1>>
[0xbf06] Discovered basic device information for <Device model='EFEKTA_PWS' manuf='efektalab.ru' nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=True>
Device is initialized <Device model='EFEKTA_PWS' manuf='efektalab.ru' nwk=0xBF06 ieee=00:12:4b:00:24:42:d3:d2 is_initialized=True>
Checking quirks for efektalab.ru EFEKTA_PWS (00:12:4b:00:24:42:d3:d2)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
[... 46 lines cut ...]
[0xBF06:1:0x0000]: 'async_initialize' stage succeeded
[0xBF06](EFEKTA_PWS): power source: Battery or Unknown
[0xBF06](EFEKTA_PWS): completed initialization
Ignoring message (b'110201') on cluster 3: unknown endpoint or cluster id: 'No cluster ID 0x0003 on (00:12:4b:00:24:42:d3:d2, 1)'
[0x8012:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=Command.Report_Attributes>
[0x8012:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=30138>), Attribute(attrid=4, value=<TypeValue type=uint16_t, value=26909>), Attribute(attrid=7, value=<TypeValue type=uint16_t, value=394>)]]
[0x8012:1:0x0300] Attribute report received: current_x=30138, current_y=26909, color_temperature=394
[0xa0da:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=82 command_id=Command.Report_Attributes>
[0xa0da:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0xa0da:1:0x0006] Attribute report received: on_off=0
[0x8483:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=111 command_id=Command.Report_Attributes>
[... cut ...]
Ignoring message (b'110401') on cluster 3: unknown endpoint or cluster id: 'No cluster ID 0x0003 on (00:12:4b:00:24:42:d3:d2, 1)'
Ignoring message (b'110501') on cluster 3: unknown endpoint or cluster id: 'No cluster ID 0x0003 on (00:12:4b:00:24:42:d3:d2, 1)'
Ignoring message (b'110601') on cluster 3: unknown endpoint or cluster id: 'No cluster ID 0x0003 on (00:12:4b:00:24:42:d3:d2, 1)'
Ignoring message (b'110701') on cluster 3: unknown endpoint or cluster id: 'No cluster ID 0x0003 on (00:12:4b:00:24:42:d3:d2, 1)'
```
**Additional context**
[image]

Hedda, 2021-11-23:
FYI, Water Content Measurement clusters for Zigbee Soil Moisture and Leaf Wetness sensors was just added to the ZHA integration component in Home Assistant core dev branch just over two weeks ago so it has not made it into a new HA release as of yet, see:

https://github.com/home-assistant/core/pull/59300

Also, for reference; I believe EFEKTA Plant Watering Sensor by @smartboxchannel uses the same design and code "DIYRuZ Flower" soil moisture sensor which is an open-source hardware project that is also used by a few other DIY project in the community (or?).

Know that these are already supported by [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) (used by Zigbee2MQTT) is; DIYRuZ Flower, EFEKTA Plant Watering Sensor, ZeeFlora DIY Soil Moisture (keywords for searching their pull requests are; "flower", "plant", "soil", and "moisture").

### DIYRuZ Flower

https://zigbee.blakadder.com/DIYRuZ_Flower.html

https://www.zigbee2mqtt.io/devices/DIYRuZ_Flower.html

https://github.com/diyruz/flower

https://modkam.ru/?p=1700

### EFEKTA Plant Watering Sensor

https://zigbee.blakadder.com/EFEKTALAB_EFEKTA_PWS.html

https://www.zigbee2mqtt.io/devices/EFEKTA_PWS.html

https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee/

https://www.tindie.com/products/diyberk/plant-watering-sensor-efekta-zigbee2mqtt/

### ZeeFlora by Tech4You / TrimVa

https://zigbee.blakadder.com/Zeeflora.html

https://www.zigbee2mqtt.io/devices/ZeeFlora.html

https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/ZeeFlora.md

https://oshwlab.com/DIY_and_Co/capteur-plantes-co_copy

PS: There is by the way discussions about these and other Zigbee Soil Moisture Sensors in Home Assistant community forum here:

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803/

Hedda, 2021-11-23:
@majkrzak Slightly off-topic, but FYI, there is also a Zigbee sensor called "Spruce" by Plaid Systems (that is marketed for "smart irrigation") which is another product that will likely need to use Water Content Measurement cluster for soil moisture sensors:

### Spruce Soil Moisture Sensor Gen 3 (Model: PS-SPRZMS-SLP3) has Zigbee ID = "PS-SPRZMS-SLP3"

https://zigbee.blakadder.com/Plaid_PS-SPRZMS-SLP3.html

https://spruceirrigation.com/sensors/

https://spruceirrigation.com/order/

https://support.spruceirrigation.com/knowledge-base/gen3-spruce-sensor-guide/

https://support.spruceirrigation.com/article-categories/spruce-sensor/

If you don't count EFEKTA Plant Watering Sensor as a commercial product then Plaid Systems Spruce is the only commercially available Zigbee soil moisture sensor product on the market today (which is sad since it get very expensive if want to buy a lot).

majkrzak, 2021-11-23:
In my case temperature is correct. It takes some time for this device to report state correctly. Temperature resolution is 1 degree.
[image]

I'm currently waiting when my changes to Home Assistant will be released. They seems to have awkward release strategy of cherypicking random commits from `dev` branch to `rc` and then merging to `master`.

smartboxchannel, 2021-11-23:
the other day I tested a firmware specially created for zha, in which data from the humidity sensor is transmitted to the humidity cluster. I think this is also a working option. About the transmitted initial incorrect values. It is enough to press the button several times with a short click, this will lead to the fact that the data will not be sent according to the schedule. I think next week I will fix it and post it on github.

reven, 2021-11-24:
@Hedda, thanks for all the links and info. I was aware there was already zigbee2mqtt support for it, but didn't know if that would be useful to make the quirk for zha.
@majkrzak , you are 100% correct, the readings have stabilized after a (long) while. The battery no reads 100% instead of 126% and the temperature sensor reads 14C, though that is at least 6 to 7 degrees below ambient.
@smartboxchannel , thanks for your great work! I love the sensor, though it's also helped me realize that I'm getting to old to solder 0603 components. Looking forward to try the new firmware and the new addition to moisture content clusters to zha.

Hedda, 2021-12-06:
FYI, @smartboxchannel wrote in the Home Assistant community forum about testing new firmware with ZHA compatibility:

https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee/tree/main/PROJECT%20SOURCE

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803/52

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803/53

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803/54

aog2000a, 2022-03-27:
Hi @reven @smartboxchannel,

i'm trying to get an Efektalab Plant Watering Sensor to work with HomeAssistant and ZHA, it connects and pairs but it does not show any sensor value, even if the sensors seem to be listed in the "out_clusters". Do i need to switch to the custom FW package even if i am using a HA version which has the latest changes in?

I am using
- Efektalab Plant Watering Sensor with e-ink display with default FW
- Electrolama zig-a-zig-ah! coordinator [TI CC2652, Z-Stack 3.30+ (build 20201026)]
- Home Assistant 2022.3.7

Thanks!

aog2000a, 2022-03-28:
Thanks @Hedda ! Will try that first, and then see how it goes :-)

aog2000a, 2022-04-02:
[Not sure if it would be better to start a new Device Support Request, or keep hijacking this since same device...]

I updated the Zigbee Coordinator firmware to Z-Stack 3.x.0 version 20211217 first, and to version 20220219 now, and scratched the NAND to start from a clean config/re-pairing, but still can't see the sensor values (only the LQI, RSSI "sensors" but no measurements).

This is the device signature according to ZHA:

> {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0001",
        "0x0402",
        "0x0408"
      ]
    }
  },
  "manufacturer": "efektalab.ru",
  "model": "EFEKTA_ePWS",
  "class": "zigpy.device.Device"
}

However, i only see "Diagnostics" entities, but no Sensors (nor Controls, but i guess that is expected):

[image]

HA is updated to the latest version: Home Assistant 2022.3.8
These are the logs from the pairing:

> 2022-04-02 11:15:53 INFO (MainThread) [zigpy_znp.zigbee.application] Permitting joins for 60 seconds
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.MgmtPermitJoinReq.Req(AddrMode=<AddrMode.Broadcast: 15>, Dst=0xFFFC, Duration=60, TCSignificance=0)
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinReq.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinRsp.Callback(Src=0x0000, Status=<Status.SUCCESS: 0>)
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32822, SecurityUse=0, TSN=3, MacDst=0x0000, Data=b'\x00')
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.Broadcast: 15>, address=0xFFFC), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=54, TSN=8, Options=<TransmitOptions.NONE: 0>, Radius=0, Data=b'\x08\x3C\x00')
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
2022-04-02 11:15:53 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-02 11:16:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-04-02 11:16:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-04-02 11:16:34 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x3CF5, SrcIEEE=00:12:4b:00:19:98:5c:7e, ParentNwk=0x0000)
2022-04-02 11:16:34 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x3CF5, SrcIEEE=00:12:4b:00:19:98:5c:7e, ParentNwk=0x0000)
2022-04-02 11:16:34 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x3CF5, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2022-04-02 11:16:34 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x3CF5, NWK=0x3CF5, IEEE=00:12:4b:00:19:98:5c:7e, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded: 128>)
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x3CF5, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=0, MacDst=0x0000, Data=b'\xF5\x3C\x7E\x5C\x98\x19\x00\x4B\x12\x00\x80')
2022-04-02 11:16:35 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: nwk=0x3CF5, ieee=00:12:4b:00:19:98:5c:7e, capabilities=128
2022-04-02 11:16:35 INFO (MainThread) [zigpy.application] New device 0x3cf5 (00:12:4b:00:19:98:5c:7e) joined the network
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy.device] [0x3cf5] Scheduling initialization
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x3CF5 ieee=00:12:4b:00:19:98:5c:7e is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xf5<~\\\x98\x19\x00K\x12\x00\x80'
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy.zdo] [0x3cf5:zdo] ZDO request ZDOCmd.Device_annce: [0x3CF5, 00:12:4b:00:19:98:5c:7e, 128]
2022-04-02 11:16:35 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2022-04-02 11:16:35 INFO (MainThread) [zigpy.device] [0x3cf5] Requesting 'Node Descriptor'
[... cut ...]
2022-04-02 11:16:37 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x3CF5](EFEKTA_ePWS): completed initialization
2022-04-02 11:16:45 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-04-02 11:16:45 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-04-02 11:17:15 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-04-02 11:17:15 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)

Thanks for any hint!!

aog2000a, 2022-04-03:
@majkrzak Are you using the default FW in the Efekta sensor or the _ZHA variant (needed before your PR #59300)?
I see that my sensor gets discovered with "input_clusters=[0], output_clusters=[1, 1026, 1032]" which looks OK, i'd say, right?
But then I do not see any messages like

> bound 'power' cluster: Status.SUCCESS
bound 'temperature' cluster: Status.SUCCESS

I only see 

> [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
[homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']

majkrzak, 2022-04-03:
Yes, I was working on the firmware which was shipped to me and reporting the moisture cluster data. I'm not sure some recent changes did not break support, but I can't check it right now as my device is not in the network at the moment.

aog2000a, 2022-04-03:
I'm very confused now, since in @majkrzak 's pull request the soil_humidity cluster 0x408 seems to be an **input_cluster**, while my sensor is being detected with cluster 0x408 as **output_cluster**!
@Hedda is it possible to "fix" that with a quirk?

majkrzak, 2022-04-03:
I'm confused to 😅
I've added the cluster to the place where TODO marker was. This was enough to make my pwa work. Maybe there is really something wrong with firmware of yours?

aog2000a, 2022-09-17:
Just for the record: my device had a messed up FW version which has the sensor clusters wrongly set as Output Clusters (there is even an #ifdef for that in the source code in the Efektalab repo). I could get the sensor to work without issues, by adding a ZHA quirk which sets the right type for the sensing clusters.

closed 2023-03-23 by github-actions[bot]