zigpy/zha-device-handlers #1173: [Device Support Request] Lidl RGB Smart Bulb (Tuya TS0505A whitelabel RGBCCT controller)

wingertge, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
I bought some Lidl RGB smart bulbs which appear to be built on the Tuya TS0505A RGBCCT controller. The only control that shows up when adding them is a single toggle labeled `Level, light_color, on_off` which toggles the lamp but doesn't allow any color, color temperature or brightness settings.

**Describe the solution you'd like**
I'd like for the lamp to be fully controllable with brightness, color, color temperature, etc.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_dbou1ap4",
  "model": "TS0505A",
  "class": "zigpy.device.Device"
}
```

Pairing log:

```
New device 0xc3f2 (84:2e:14:ff:fe:22:d4:fe) joined the network
[0xc3f2] Scheduling initialization
Tries remaining: 3
[0xc3f2] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc3f2] Extending timeout for 0x0f request
Device 0xc3f2 (84:2e:14:ff:fe:22:d4:fe) joined the network
[0xc3f2] Scheduling initialization
[0xc3f2] Canceling old initialize call
Tries remaining: 3
[0xc3f2] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc3f2] Extending timeout for 0x11 request
Device 0xc3f2 (84:2e:14:ff:fe:22:d4:fe) joined the network
[0xc3f2] Scheduling initialization
[0xc3f2] Canceling old initialize call
Tries remaining: 3
[0xc3f2] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc3f2] Extending timeout for 0x13 request
Device 0xc3f2 (84:2e:14:ff:fe:22:d4:fe) joined the network
[0xc3f2] Scheduling initialization
[... 242 lines cut ...]
[0xc3f2:1:0x0008] ZCL request 0x0001: [[ReadAttributeRecord(attrid=0, status=<Status.SUCCESS: 0>, value=<TypeValue type=uint8_t, value=255>)]]
[0xC3F2:1:0x0300]: initializing channel: from_cache: False
[0xc3f2:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=113 command_id=Command.Read_Attributes_rsp>
[0xc3f2:1:0x0300] ZCL request 0x0001: [[ReadAttributeRecord(attrid=16386, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>), ReadAttributeRecord(attrid=3, status=<Status.SUCCESS: 0>, value=<TypeValue type=uint16_t, value=45874>), ReadAttributeRecord(attrid=4, status=<Status.SUCCESS: 0>, value=<TypeValue type=uint16_t, value=19660>), ReadAttributeRecord(attrid=7, status=<Status.SUCCESS: 0>, value=<TypeValue type=uint16_t, value=153>)]]
[0xC3F2:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
[0xC3F2:1:0x0006]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
[0xC3F2:1:0x0000]: initializing channel: from_cache: False
[0xC3F2:1:0x0000]: finished channel initialization
[0xC3F2:1:0x1000]: initializing channel: from_cache: False
[0xC3F2:1:0x1000]: finished channel initialization
[0xC3F2:1:0x0019]: initializing channel: from_cache: False
[0xC3F2:1:0x0019]: finished channel initialization
[0xC3F2:1:0x0008]: failed to get attributes '['current_level']' on 'level' cluster: 
[0xC3F2:1:0x0008]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
[0xC3F2:1:0x0300]: failed to get attributes '['color_loop_active', 'current_x', 'current_y', 'color_temperature']' on 'light_color' cluster: 
[0xC3F2:1:0x0300]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
[0xC3F2:1:0x0006]: 'async_initialize' stage succeeded
[0xC3F2:1:0x0300]: 'async_initialize' stage succeeded
[0xC3F2:1:0x0008]: 'async_initialize' stage succeeded
[0xC3F2:1:0x0000]: 'async_initialize' stage succeeded
[0xC3F2:1:0x1000]: 'async_initialize' stage succeeded
[0xC3F2:1:0x0019]: 'async_initialize' stage succeeded
[0xC3F2](TS0505A): power source: Mains
[0xC3F2](TS0505A): completed initialization
```

MattWestb, 2021-11-23:
Wot is happening then you is clicking on the `Level, light_color, on_off` or the light bulb icon ?
Normally you shall getting one card with the control of the light that is only on/off if the light is off. 
If it being on you shall getting light level if one dim light and TC if its one CT light and RGB if the light is RGB.

If not working open manage  cluster and "Color (Endpoint id: 1, Id: 0x0300, Type: in)" and attribute "color_capabilities (id: 0x400a)" and clicking "get zigbee attribute" and you shall getting somthing like: `bitmap16.16|8|4|2|1`.

wingertge, 2021-11-23:
There's not 3 controls, there's just one which toggles the light. color_capabilities returns None which is a known quirk of the sister device TS0502A CCT (non-RGB). The quirk for that device just hardcodes it to 0x16.

MattWestb, 2021-11-23:
I have one LIDL LED RGBTC Stripe and its having RGB and CT but is also returning "none" then reading the color capacity so is shall also not having any controls for RGB and CT but it have ;-(

First re starting HA and see if its loading the device OK and you is getting the controller in ZHA GUI.

If not working configure local quirks in HA config and copy the TS0502A quirk in the local quirk folder but you need adding your device ID and changing the https://github.com/zigpy/zha-device-handlers/blob/fbee5ba7a378b42c0196b21ef42ebfc361af526a/zhaquirks/lidl/cct.py#L55
to what your device is having (EXTENDED_COLOR_LIGHT = 0x010D) or its not loading the quirk. Also double check that the cluster is the same.
Then restarting HA and all is going OK you shall getting TC controllers and also the quirk name as `device class` in the signature.

After that you can trying changing the attribute https://github.com/zigpy/zha-device-handlers/blob/fbee5ba7a378b42c0196b21ef42ebfc361af526a/zhaquirks/lidl/cct.py#L33
so you is getting all function for your light.

MattWestb, 2021-11-24:
Very nice done !!!!!
Under 24 hours :-))

closed 2021-11-25 by wingertge