zigpy/zha-device-handlers #1175: Tuya TRV TZE200: hvac_action reports still heating

radekposvar, 2021-11-24:
I have a TS0602 , _TZE200_b6wax7g0
The applied quirks iszhaquirks.tuya.ts0601_trv.MoesHY368_Type1new

I´m running ZHA integration using Sonoff zigbee bridge flashed with tasmota. HASS OS on RPi 4.

I would like to ask if it is possible to repair a bug which is not updating the hvac_action option of this exact TRV. In the developer it is still reporting heating, even when the set temperature is below the actuall temperature. In my stats it is heating 24/7.  It never reports the idle state.

Thank you for your h
[image]
[image]
elp,
Radek

MattWestb, 2021-11-24:
The `MoesHY368_Type1new` is one  "hot fix" for setpoint that was not working OK but its still futures that is not working OK so one user is trying making one new quirk that shall using all function of the device https://github.com/zigpy/zha-device-handlers/issues/1123.
I dont have the device but its looks that most functions is being pin pointed and is active trying fixing them.

I think in the end the TRV shall being moved to its own quirk for getting all function working OK then its one other hard and software then the "normal Moes" ones.
Pleas help testing the work that is being done with the new quirk so we can getting it working well.

radekposvar, 2021-11-24:
Thank you for your fast answers. @jacekk015 could you send me some tutorial or describe how to upload and enable local quirks. I´m using HA for some time, but I am not skilled with these OS things. Usually I can follow steps, but my coding and Linux experiences are quite limited.
Thanks

jacekk015, 2021-11-24:
In configuration.yaml add:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Create "custom_zha_quirks" in the config folder and restart HA.
If you place/change a quirk file in that folder you always need to restart HA

jacekk015, 2021-11-27:
@radekposvar In case you missed it here's the latest, fully functional, dev version:
https://github.com/zigpy/zha-device-handlers/issues/1123#issuecomment-980314632

dmulcahey, 2021-12-14:
So, I was helping someone with a hive thermostat and discovered that the core ZHA handling of hvac_action needed a bit of reworking. The changes are in dev and will be in the release of HA in February. It may be worth validating that the quirks for the Tuya stuff still work as intended

dmulcahey, 2021-12-14:
If the normal functions don’t work via the UI something is probably not standard and needs changes in quirks

MattWestb, 2021-12-14:
Its the same type as my and the state is having occupied_heating_setpoint.

<html>
<body>

climate.tyst11_trv_thermostatTYST11 TRV thermostat | heat | hvac_modes: heat min_temp: 5 max_temp: 30 preset_modes: none, away, schedule current_temperature: 23 temperature: 22.4 hvac_action: idle preset_mode: none system_mode: [4]/heat occupancy: 1 occupied_heating_setpoint: 2240 unoccupied_cooling_setpoint: 2200 icon: mdi:thermostat friendly_name: TYST11 TRV thermostat supported_features: 17
-- | -- | --

</body>
</html>

Only ZHA is making this wrong:
`unoccupied_cooling_setpoint: 2200`
Shall being: `unoccupied_heating_setpoint`

Perhaps better open in issue in HA ?

dmulcahey, 2021-12-14:
I’m not following. Where does this appear in HA

MattWestb, 2021-12-14:
Developer Tools > States and the TRV.

jacekk015, 2021-12-14:
Probably here's some mixed:
https://github.com/home-assistant/core/blob/master/homeassistant/components/zha/climate.py#L213

MattWestb, 2021-12-14:
Great @jacekk015 one bug find :-))

But do you have automations on the device card for setting one new `occupied_heating_setpoint` ?
That is missing and its little sad for users making easy automations.

MattWestb, 2021-12-14:
OPS !!
Its one more place mixed up:
https://github.com/home-assistant/core/blob/5df747276f3a1b787c24b3225c457ca55bb5c84d/homeassistant/components/zha/climate.py#L209

jacekk015, 2021-12-14:
All quirks maps they target temp to ```occupied_heating_setpoint```
For away mode ```unoccupied_heating_mode``` should be used as a target temp.
But the bug above makes it unusable because unoccupied_**cooling**_setpoint appears so - this is how it is.

MattWestb, 2021-12-14:
PR made for fixing heating and cooling setpoint in ZHA https://github.com/home-assistant/core/pull/61791.

MattWestb, 2021-12-14:
Test system patched :-))
```
hvac_modes:
  - heat
min_temp: 5
max_temp: 30
preset_modes:
  - none
  - away
  - schedule
current_temperature: 23
temperature: 22.4
hvac_action: idle
preset_mode: none
system_mode: '[4]/heat'
occupancy: 1
occupied_heating_setpoint: 2240
unoccupied_heating_setpoint: 2200
icon: mdi:thermostat
friendly_name: TYST11 TRV thermostat
supported_features: 17
```

closed 2022-06-19 by github-actions[bot]