zigpy/zha-device-handlers #1176: Aqara vibration sensor sensitivity. lumi.vibration.aq1

PlusPlus-ua, 2021-11-24:
**Is your feature request related to a problem? Please describe.**
I'm trying to use Aqara vibration sensor (lumi.vibration.aq1) to detect state of one device which produces low frequency vibration when works. Unfortunately, default level of sensitivity of the sensor doesn't detect the vibration.

Since I haven't Xiaomi ZigBee gateway, I'm trying to change sensitivity of the sensor from HA GUI.

There are in and out VibrationBasicCluster with custom attribute "sensitivity" (id: 0xff0d). I've tried to get a value of the attribute in both clusters, but nothing is returned. It ever doesn't return "None" value.

**Describe the solution you'd like**
Obtain an ability to set sensitivity of the sensor.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0019",
        "0x0101",
        "0x0500"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x5f02",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0012"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.vibration.aq1",
  "class": "zhaquirks.xiaomi.aqara.vibration_aq1.VibrationAQ1"
}
```

Adminiuga, 2021-11-24:
What radio are you using?

PlusPlus-ua, 2021-11-24:
TI CC2531 with source_routing firmware for now. In a couple of days will upgrade it to CC2652P based SONOFF dongle.

Adminiuga, 2021-11-24:
I don't know if they made new firmware for cc2531, but for the newer chips, the dev firmware contains a fix which adjust manufacturer id for aqara E1 devices and it may fix access to that attribute Id as well.

PlusPlus-ua, 2021-11-24:
Ok, thank you. What does it mean "dev firmware"? Is firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin fits?

MattWestb, 2021-11-24:
Not in the master branch in the dev branch.
https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

dmulcahey, 2021-11-24:
put 4447 in the mfg code field when issuing the set or get attribute commands and see if that works. Also ensure that you press the button on the sensor right after you issue the command

PlusPlus-ua, 2021-11-24:
@dmulcahey thanks, it seems like it works. Is button pressing required with new dev firmware?

MattWestb, 2021-11-24:
From Z2M:
```
    DJT11LM_vibration_sensitivity: {
        key: ['sensitivity'],
        convertSet: async (entity, key, value, meta) => {
            value = value.toLowerCase();
            const lookup = {'low': 0x15, 'medium': 0x0B, 'high': 0x01};
            utils.validateValue(value, Object.keys(lookup));

            const options = {...manufacturerOptions.xiaomi, timeout: 35000};
            await entity.write('genBasic', {0xFF0D: {value: lookup[value], type: 0x20}}, options);
            return {state: {sensitivity: value}};
        },
    },
``` 
Basic cluster and attribute 0xFF0D with 'low': 0x15, 'medium': 0x0B, 'high': 0x01 ? 
and also the 4447 as manufacture.

dmulcahey, 2021-11-24:
It’s a battery powered device and it’s Xiaomi… it’s recommended to press a button for all sleepy devices (battery powered) and pretty much required for all Xiaomi devices

Adminiuga, 2021-11-24:
Ok, so no need for the new coordinator firmware, but do need provide manufacturer id when reading or setting the attribute. Also do need to wake up the device by pressing button on it.

closed 2021-12-05 by dmulcahey