zigpy/zha-device-handlers #1178: [Device Support Request] HY367 TRV with new manufacturer ID

D-ominik, 2021-11-25:
My newly purchased HY-367 (https://zigbee.blakadder.com/Tuya_HY367.html) got a new manufacturer ID, please add "_TZE200_2atgpdho" to the device handler "MoesHY368_Type1". Thanks a lot!

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_2atgpdho",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2021-11-25:
Have you trying adding the ID to the quirk and getting it  loaded ?
Its need the "MoesHY368_Type1" for the basic function and also being added in ZHA for getting the presets working OK.

If you need help i can making the patched files for you.

D-ominik, 2021-11-25:
I made a custom device handler based on zhaquirks/tuya/ts0601_trv.py and put it in my folder defined by "custom_quirks_path". Basically I only added the new signature to MoesHY368_Type1 and after pairing the entities are available and controlling the temperature works fine.

Presets via the GUI are not available, but I didn't miss them either. I let the units run autonomously and only switch between away and schedule mode with Home Assistant when windows are opened or closed.

I assume the unit is identical to _TZE200_cwnjrr72, just another random ID...

MattWestb, 2021-11-25:
Great !!
I can do one updated `climate.py` for getting the presets working in ZHA that  is needed for getting all working OK with this TRVs.

If then all is working OK i can making one (more correct 2) PR for getting it supported in the next HA.

Looks like CN-Hysen have making one new version of the same TRV with updated ID. 
They was using the same for all but it can they have doing some changes that we is not knowing.

MattWestb, 2021-11-25:
Copy the `climate.py`  to your HA local configure folder.
Copy it into the HA container with this command from docker CLI in the HA container:
```
cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py 
```
Restar HA and is all is OK you is getting working presets.
[climateNTH.zip](https://github.com/zigpy/zha-device-handlers/files/7602425/climateNTH.zip)

D-ominik, 2021-11-25:
Thanks for your efforts. I'm using Home Assistant OS (6.6) on RPi4, not the docker environment. I think the process is a bit different in this case, isn't it?

MattWestb, 2021-11-25:
I think installing docker is blocked in OS :-((
Can being done with SSH or terminal but i have not trying it.

I have not example how but i thing HA community have some.

MattWestb, 2021-11-25:
I was testing on my production system (supervised HA) with:
```
ls -la /usr/src/homeassistant/homeassistant/components/zha/
total 276
drwxr-xr-x    1 root     root          4096 Nov  6 15:20 .
drwxr-xr-x    1 root     root          4096 Nov  3 18:00 ..
-rw-r--r--    1 root     root          6265 Nov  3 17:25 __init__.py
drwxr-xr-x    1 root     root          4096 Nov  6 15:22 __pycache__
-rw-r--r--    1 root     root          5251 Nov  3 17:25 alarm_control_panel.py
-rw-r--r--    1 root     root         44985 Nov  3 17:25 api.py
-rw-r--r--    1 root     root          5234 Nov  3 17:25 binary_sensor.py
-rw-r--r--    1 root     root         26027 Nov  6 15:20 climate.py
-rw-r--r--    1 root     root          9157 Nov  3 17:25 config_flow.py
drwxr-xr-x    1 root     root          4096 Nov  3 18:01 core
-rw-r--r--    1 root     root         10222 Nov  3 17:25 cover.py
-rw-r--r--    1 root     root          2909 Nov  3 17:25 device_action.py
-rw-r--r--    1 root     root          3317 Nov  3 17:25 device_tracker.py
-rw-r--r--    1 root     root          2947 Nov  3 17:25 device_trigger.py
-rw-r--r--    1 root     root         11405 Nov  3 17:25 entity.py
-rw-r--r--    1 root     root          7896 Nov  3 17:25 fan.py
-rw-r--r--    1 root     root         23269 Nov  3 17:25 light.py
-rw-r--r--    1 root     root          6504 Nov  3 17:25 lock.py
-rw-r--r--    1 root     root          1211 Nov  3 17:25 manifest.json
-rw-r--r--    1 root     root          9858 Nov  3 17:25 number.py
-rw-r--r--    1 root     root         18907 Nov  3 17:25 sensor.py
-rw-r--r--    1 root     root         11845 Nov  3 17:25 services.yaml
-rw-r--r--    1 root     root          4527 Nov  3 17:25 strings.json
-rw-r--r--    1 root     root          4645 Nov  3 17:25 switch.py
drwxr-xr-x    2 root     root          4096 Nov  3 17:25 translations 
```
in the HA docker container.
So i think you is in the host OS and not in the container.
If trying from host OS only` /usr/src/` exist then the rest in the the HA container.

MattWestb, 2021-11-25:
Some have using SSH and connecting into the docker container but i dont knowing how to install / enabling all things for getting it working OK.

D-ominik, 2021-11-25:
It was hard to find among a lot of wrong information, so here is the link to the correct instructions on how to get access to this data within Home Assistant OS:
https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host
Maybe others searching for the same information will find it faster this way...

I added the line with the new ID to climate.py and restarted it. The presets are now available. The preset "manual" is missing, but that is not important.

D-ominik, 2021-11-25:
Forgot: switching the presets works fine.

MattWestb, 2021-11-25:
## Great !!
I copy your finding to the discussion so can linking to it then user cant using portainer !!

My Site TRVs have lesser presets but i think manual mode on the TRV = preset `none` for Moes TRVs in ZHA.

Shall I making PR for adding the needed changes ?
The ZHA fix is better waiting some days then its one PR in the pipe and need redoing it if not have the changes in the PR.

I updating the tuya TRV matrix that  is working in the device request :-))

D-ominik, 2021-11-25:
You are right, mode "none" corresponds to "manual".

I cannot tell whether this version of the unit has new features, as I do not have any other units of this device series.

After inserting the batteries, before setting the clock, the display shows "IP" or "1P" and one can select a number between 1 and 8. This step is not mentioned in the instructions, so this could be something new. However, the instructions are poor anyway, so that doesn't have to mean anything...

In the settings, this option can also be found as item 7 (after "SET").
I could not determine the purpose of this setting.

Yes, please create a PR. Thanks! :-)

MattWestb, 2021-11-25:
I waiting some days and and doing one PR for ZHA and one for the quirk !!
(Pleas reminding my if i forgetting it !!!)

You can taking the battery out and putting them back then "tuya DP devices" is sending all DP commands they is using on power on. 
You need having debut for`     zhaquirks: debug` in HA for getting the response from the quirk.

Perhaps its some new nice commands implanted :-)))

D-ominik, 2021-11-26:
Hey Matt,

activated the debug log and did a power cycle on one of the TRVs. Here's what I got:

```
2021-11-26 12:10:40 INFO (MainThread) [zigpy.application] Device 0x6e4f (84:fd:27:ff:fe:a2:56:9c) joined the network
2021-11-26 12:10:40 DEBUG (MainThread) [zigpy.zdo] [0x6e4f:zdo] ZDO request ZDOCmd.Device_annce: [0x6E4F, 84:fd:27:ff:fe:a2:56:9c, 128]
2021-11-26 12:10:40 DEBUG (MainThread) [zigpy.device] [0x6e4f] Extending timeout for 0xe7 request
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0x000a] ZCL request 0x0000: [[7]]
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x309d:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=125 command_id=Command.Report_Attributes>
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x309d:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=31120>)]]
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x309d:2:0x0400] Attribute report received: measured_value=31120
2021-11-26 12:10:43 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0x0004] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=231 command_id=2>
2021-11-26 12:10:44 DEBUG (MainThread) [zigpy.zcl] [0x3b98:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=203 command_id=Command.Report_Attributes>
2021-11-26 12:10:44 DEBUG (MainThread) [zigpy.zcl] [0x3b98:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=LVBytes, value=b'\x01!\xb3\x0b\x03(\x17\x04!\xa8\x13\x05!\x17\x00\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x00'>)]]
2021-11-26 12:10:44 DEBUG (MainThread) [zigpy.zcl] [0x3b98:1:0x0000] Attribute report received: 65281=b'\x01!\xb3\x0b\x03(\x17\x04!\xa8\x13\x05!\x17\x00\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x00'
2021-11-26 12:10:44 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:02:f2:45:c3 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 2995, 'temperature': 23, 'X-attrib-4': 5032, 'X-attrib-5': 23, 'X-attrib-6': 1, 'path': 0, '0xff01-100': <Bool.false: 0>}]
2021-11-26 12:10:44 DEBUG (MainThread) [zigpy.zcl] [0x3b98:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:2995 < [Max]:3100, Battery Percent: 62.5
2021-11-26 12:10:46 DEBUG (MainThread) [zigpy.zcl] [0x6d87:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=28 command_id=Command.Report_Attributes>
2021-11-26 12:10:46 DEBUG (MainThread) [zigpy.zcl] [0x6d87:3:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2021-11-26 12:10:46 DEBUG (MainThread) [zigpy.zcl] [0x6d87:3:0x0006] Attribute report received: on_off=0
2021-11-26 12:10:47 DEBUG (MainThread) [zigpy.zcl] [0x43f2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=51 command_id=Command.Report_Attributes>
2021-11-26 12:10:47 DEBUG (MainThread) [zigpy.zcl] [0x43f2:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int8s, value=0>)]]
2021-11-26 12:10:47 DEBUG (MainThread) [zigpy.zcl] [0x43f2:1:0x0006] Attribute report received: on_off=0
2021-11-26 12:10:47 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=36>
2021-11-26 12:10:47 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL request 0x0024: [[0, 0]]
2021-11-26 12:10:47 DEBUG (MainThread) [zhaquirks.tuya] [0x6e4f:1:0xef00] Got set time request (command 0x0024)
2021-11-26 12:10:48 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=236 command_id=Command.Default_Response>
2021-11-26 12:10:48 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL request 0x000b: [36, <Status.SUCCESS: 0>]
2021-11-26 12:10:50 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=36>
2021-11-26 12:10:50 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL request 0x0024: [[0, 5]]
2021-11-26 12:10:50 DEBUG (MainThread) [zhaquirks.tuya] [0x6e4f:1:0xef00] Got set time request (command 0x0024)
2021-11-26 12:10:50 DEBUG (MainThread) [zigpy.zcl] [0x6e4f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufact [... cut ...]
2021-11-26 12:12:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4DE2:1:0x0500]: Updated alarm state: 0
2021-11-26 12:12:22 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:05:cf:4c:da - Resetting motion sensor
2021-11-26 12:12:22 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA767:1:0x0500]: Updated alarm state: 0
2021-11-26 12:12:27 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:05:cf:40:f7 - Resetting motion sensor
2021-11-26 12:12:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x41A6:1:0x0500]: Updated alarm state: 0
```

Is it this what you meant?

MattWestb, 2021-11-26:
PR made waiting merging :-)
And shall doing one for the presets in ZHA 2.

jacekk015, 2021-11-26:
Moes quirk extended to almost all functionality.
I've omitted MOES_FORCE_VALVE_ATTR because there's no such entity in ZHA GUI - only attribute exists.
Didn't touch schedule since original code writer probably did it properly.
Nice if someone were willing to test the quirk locally.
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/7610338/ts0601_trv_moes.py.zip)

closed 2021-11-28 by dmulcahey

MattWestb, 2021-12-01:
@Adminiuga Sorry for pinging  you directly !!

If i remember right is you having some tuya TRV type CN-Hysen / Moes.
If you have time can you trying the new quirk and doing some feedback if you is liking the approach for getting more function in ZHA ?

I knowing the best wold being doing all in ZHA or other module over Zigpy but this is the best we can do for the moment.

And we have over 20 devices that have many not supported and / or broken functions that is making user going away from ZHA that i think it bad for all in the end (Siterwell is the worse implanted device but it was in the first commitment).

Thanks in advance !!

Mvh  Mattias W

MattWestb, 2021-12-04:
@D-ominik Do you have one brand and model of your device ?
Its looks like the  manufacturer have start using new software IDs for all new devices they is selling to other companies so i need putting one name on your new device.

And the PR is merged in ZHA but not  the presets that is little late so it can being you must updating / copy the `climate.py` then you is updating to HA 2021.12 but the local quirk shall not being needed for it.

ferrdo4, 2021-12-06:
> Moes quirk extended to almost all functionality. I've omitted MOES_FORCE_VALVE_ATTR because there's no such entity in ZHA GUI - only attribute exists. Didn't touch schedule since original code [...]

Hello, i am not sure where to post this ... but i was pointed here to test out moes quirk. It is working fine so far, many setting params to set. i just have question ... im not sure what 3 on_off switches does.

jacekk015, 2021-12-06:
@ferrdo4 Child Lock, Window Open detect, Valve state [open/close]
Last two should be changed to binary sensors, probably

ferrdo4, 2021-12-06:
> @ferrdo4 Child Lock, Window Open detect, Valve state [open/close] Last two should be changed to binary sensors, probably

Ok that makes sense. Then i have one observation. In manual open window feature have two setting "--" is off, 5°C is enabled in range 5-25°C and second setting is running time 5-60 minutes.

I dont need this functionality, but maybe you would be interested, since ive started messing around with this quirk :)
Thanks for quick respond.

D-ominik, 2021-12-06:
> @D-ominik Do you have one brand and model of your device ? Its looks like the manufacturer have start using new software IDs for all new devices they is selling to other companies so i need putting [...]

From this valve type (HY367) I have only three devices, all from the same delivery in mid-November from Sandy Beach (https://www.amazon.co.uk/gp/product/B093B4SCBT/). More HY367 are on their way directly from China, but this will take some time. I'm curious to see which ID is used there.

jacekk015, 2021-12-06:
> Ok that makes sense. Then i have one observation. In manual open window feature have two setting "--" is off, 5°C is enabled in range 5-25°C and second setting is running time 5-60 minutes.
>
> I dont [...]

There's no much you could add to it.

@D-ominik 
Here, in Poland, are dirty cheap TRVs Maxsmart/AWOW
10pcs for 781zł - about 171 EUR for all set of 10pcs.
They have also smaller packages.
https://allegro.pl/oferta/glowica-termostatyczna-awow-10-szt-11407902831

I bought 2pcs of Maxsmart year ago, and now 10pcs of those. I'm starting to install them.

MattWestb, 2021-12-06:
CN-Hysen was only making 2 versions with the old and new Zigbee module for there devices for all customers until for one month ago its have coming 3 new IDs for there 4 models they is making.

Give one sign if they is having one new ID and i adding it in the quirk and the presets in ZHA.

I trying updating the tuya TRV matrix then i getting new ID from users so they dont need freezing all the time :-)) https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395

MattWestb, 2021-12-06:
The extra function in the new firmware  i hope we can digging little deeper later then we have getting the "normal" functions working OK for not messing all up and making all bad.

ferrdo4, 2021-12-06:
> The extra function in the new firmware i hope we can digging little deeper later then we have getting the "normal" functions working OK for not messing all up and making all bad.

Sure, but if you would like to try out anything Id gladly test it out. You guys do great work!

jacekk015, 2021-12-06:
@ferrdo4 If you miss some option, or you see that something could work better, just give us a hint.

ferrdo4, 2021-12-06:
> @ferrdo4 Child Lock, Window Open detect, Valve state [open/close] Last two should be changed to binary sensors, probably

So i got curious about the switches ... first one "switch.on_off" is a window open detection, "switch.on_off_2" is a lock - these can be set by app. and last one "switch.on_off_3" is a valve switch - if changed, it will set itself back

> @ferrdo4 If you miss some option, or you see that something could work better, just give us a hint.

If ill find something ill will, for now im just happy it works :)

jacekk015, 2021-12-06:
> "switch.on_off_3" is a valve switch - if changed, it will set itself back

That quirk wasn't actually made by me. I don't have this TRV so can't test it physically.
Valve switch in code is described as:
```
MOES_VALVE_DETECT_ATTR = 0x0114  # [0] do not report [1] report
```
Only you[user] or PDF manual can tell us what it is actually

ferrdo4, 2021-12-06:
> [edit] oh and Window detection best works... in freezer. Don't count on it. Just tested my 4 pcs with Windows opened. Guess what do they do. Yes they heat a lot ;-) We should rename it to Freezer detection.

maybe it is caused by missing temperature setting, being 5C default. on TRV itself it can be set up to 25C

D-ominik, 2021-12-06:
Maybe a misunderstanding because other devices work differently?

With a lot of other devices, the window-open detection is a little smarter and will be activated when the temperature drops by a certain value within a predefined period of time. Then the thermostat is set to the selected temperature (e.g. 5°C) for the selected duration (e.g. 20 min).

With these devices, however, the function is rather primitive: you set a temperature below which the window-open mode (= valve closed) will be active for the selected duration.

ferrdo4, 2021-12-06:
> Maybe a misunderstanding because other devices work differently?
>
> With a lot of other devices, the window-open detection is a little smarter and will be activated when the temperature drops by a [...]

Maybe, im not sure, but from my observation on one i have, when i set value like 17C on TRV itself, window open detection will activate when TRV shows 17C (and stay for defined time). However when i toggle it in HA off and on again, this value is set back to 5C

mhombach, 2021-12-09:
> 

Hey, I have some issues with my Tuya valves and was pointed to your comment here. Maybe I can help you testing?
See my original issue: https://github.com/zigpy/zha-device-handlers/issues/1212
I'm pretty new to HA even though I'm a developer my self. If I can help you in any way, just tell me how :)

ferrdo4, 2021-12-15:
i found a strange behavior with "analog_output_temperature_calibration" on moes quirk the set bar gives user possibility to set temperature calib in range <-9, 9>... but it gets somehow rounded <-9, 5> will be set to value -50 (-0.5C) <-4, 4> = 0 and <5, 9> is 50 (+0.5C) when i try to set it in developer tool to -100 (expecting -1C) it will be set back to value in device options ... i would like to set this calib based on difference of trv current temp and ext. sensor current temp :)

edgardavim, 2022-01-01:
Hi all,

I am living in Poland and also decided to enter the world of Zigbee TRV to use with my HomeAssistant. Bought also the Tuya HY367 
After sync with HA via zigbee2mqtt, this is what shows:
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_2atgpdho
Manufacturer TuYa
Model TS0601_thermostat

It connects, but disconnects after, not possible to do anything with the valve. The screen shows a huge 1P and cant do antyhing else.

Already removed and added again and keeps the same. Anyone had this issue? 
https://live.staticflickr.com/65535/51722394114_d14a1f429a_c.jpg

jacekk015, 2023-01-09:
Moes quirk corrected to work after recent HA update
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/10376551/ts0601_trv_moes.py.zip)

dmulcahey, 2023-01-09:
Can you please PR these?

PR #1184 (2021-11-28): Adding "_TYST11_czk78ptr", "zk78ptr"
Adding "_TZE200_czk78ptr", "TS0601", "_TYST11_czk78ptr", "zk78ptr", "_TZE200_2atgpdho", "TS0601" and ("_TYST11_2atgpdho", "atgpdho")
I also deleting the ("_TYST11_b6wax7g0", "6wax7g0") then its look Beca Smart is not making one with the old Zigbee module.
The "2atgpdho" versions also need getting presets in ZHA in one other PR for working OK.

Fixes #1178
Fixes #1180