zigpy/zha-device-handlers #1180: [Device Support Request]

accessiblepixel, 2021-11-25:
**Is your feature request related to a problem? Please describe.**
I cannot pair and access features of a TP-Smart ZigBee TRV Valve. Originally it came with a Tuya compatible WiFi gateway, but I've since picked up a CC2531 USB Zigbee Gateway, but when I try and find the device in home assistant it says This Device Has No Entities.

**Describe the solution you'd like**
It looks like it's very similar to a model listed on https://zigbee.blakadder.com/Alecto_SMART-HEAT10.html it physically looks the same.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add 
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_czk78ptr",
  "model": "zk78ptr",
  "class": "zigpy.device.Device"
}

**Additional context**
If I have put this in the wrong place, please move it to the correct one. I'm very new to zigbee integration but if you need any other logfiles, please let me know. I think one of the quirks that exists for similar devices might work for this one, but it's all a bit beyond me. I hope someone can help, I'd like to move my thermostats away from the Tuya Zigbee gateway to local control with Home Assistant.

Kind regards,
Jessica

accessiblepixel, 2021-11-25:
Updated my original post, because apparently I can't github properly.
Added some links to the device on Amazon, and corrected the device manufacturer.

Have looked at the device, and without disassembly, I can't find any additional model numbers on it, which is a pain.

The device (as bought) is a TRV smart radiator valve, which claimed at one point on it's listing page Zigbee 3 support. It's supposed to connect to the TCP Smart Wireless Hub, which is a Zigbee to Tuya/Smart Life (or TCP Smart's own branded version of Tuya/Smart Life) via WiFi, so any control requires the cloud.

My thought was to get a USB Zigbee gateway to remove the problems I'm having getting local control of the device with Home Assistant, but looks like it's going to be more "plug 'n' pray" to someone smarter than I am with these things. If I can give any more info, let me know as we might stop someone else wanting to pull their hair out :D

Kind regards,
Jessica

MattWestb, 2021-11-25:
Hi Jessica your device looks 100% like my Siterwell white labeled so i think its it shall being no problem getting it working with ZHA.

Do you have one brand and name of the TRV ?

I have patched 2 feels that need being added in your system and one is easy and the second its little tricky getting in right place..

First configuring "local quirks" https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.
Copy the `ts0601_trv.py` the the new local quirk folder.
Next you need getting access to your HA docker container with CLI https://github.com/zigpy/zha-device-handlers/discussions/693#discussion-1237451.
Copy the `climate.py` to your HA config folder and then from the HA docker container CLI copy it in to the folder with this command:
```
cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py
```
Restart HA and if all is going OK you shall getting one TRV device in ZHA.

If you is getting problem  pleas ask !!

This is the current "old" implementation with not all in place and not working but we have one with more function that  you can trying later if you like.

MattWestb, 2021-11-25:
Sorry i was writing one wrong thing !!
The `climate.py` you shall not copy for using the "normal" quirk only if you later like testing the new updated quirk !!

Only copy the `ts0601_trv.py` to the local quirk folder and it shall working.

Its being little late for my and need little sleep i think ;.)

accessiblepixel, 2021-11-25:
@MattWestb 
Thanks for the reply.

Where do I find ts0601_trv.py ?

The manufacturer is TCP-Smart and model number on Amazon is TBUWTRVZIG and has the Zigbee Device Fingerprint as above.

Other than that it doesn't have a model number, it's just called the TCP Smart Thermostatic Radiator Valve & Wireless Hub (linked both the kit and the extra valves in first comment)

Kind regards,
Jessica

MattWestb, 2021-11-25:
Some where under the . . .
[ts0601_trv_new.zip](https://github.com/zigpy/zha-device-handlers/files/7605385/ts0601_trv_new.zip)
 
You see im needing little sleep ;-(

Config local quirk and copy the `ts0601_trv.py` to the new folder and restart HA.

I can helping more tomorrow if needed.

MattWestb, 2021-11-25:
I have putting it in the tuya TRV matrix with brand and model
https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395 so other user can see its supported (then we have doing one working PR for it).

accessiblepixel, 2021-11-25:
Ahh, thank you. Just trying it out now.

I've copied over the ts0601_trv.py to the custom folder in config and added the entry to my configuration yaml, tried doing a restart from "server control" but haven't had any success yet, so I'm trying a host reboot.

There's no real rush in fixing this, given I should probably sleep too :)
Thanks for your help so far, you're a star.

Kind regards,
Jessica.

MattWestb, 2021-11-25:
If ZHA is matching the device OK for the quirk and loading it you shall getting in the last line on the device card like (not exact)  `Quirk: ts0601_trv_siterwell.SiterwellGS361_Type1` and also in the device signature.

accessiblepixel, 2021-11-25:
@MattWestb Hmm, after putting that folder and copying the ts061_trv.py file into /config/custom_zha_quirks now when I try and use the ZHA integrations thing it says "Failed to set up" and gives a link to view the logs, which are blank. I must be doing something wrong!

Is the tr061_trv file the only one I need to put in to get it working, or do I need to put the climate.py in there too?

Don't need an answer right now, though, I'll just update my trials as I go on for a bit before I crash :)

Kind regards,
Jessica

MattWestb, 2021-11-25:
Can you posting the error from HA ?
Local quirk can caching ZHA and its not starting but normally its mismatch of versions or bad copy paste from my side.

accessiblepixel, 2021-11-25:
What I posted was the error, it's not showing anything in my HA logfile.

The only entries I get are unrealated (about having HACS loaded and localtuya)

It looks like it's crashing out the ZHA integration, just says "Failed to setup click for logs" so I click logs, and there aren't any :(

Is there a way to turn a "debug" mode on, so I can help a bit better?

accessiblepixel, 2021-11-25:
These are screenshots of what I can see... (which are as useless as a chocolate fire guard!)

[https://i.lowrex.com/i/CzTa.png](Logs)
[https://i.lowrex.com/i/CFoh.png](Integration)

MattWestb, 2021-11-25:
No `climate.py` for the "old" quirk that we is testing first.

OK its one updated line that is not updated in HA 2021.11.
Can you looking for `TuyaPowerConfigurationCluster2AA`  and relapsing it with `TuyaPowerConfigurationCluster` in all lines you can finding (its only the `2AA` that is changed) ?

accessiblepixel, 2021-11-25:
That seems to have worked, it seems to be responding now 👯 

Thank you very much. After you and I have had some sleep, I'm more than willing to help you test out the "new" integration to help support another device.

Thank you so much <3

Until tomorrow.

Jessica.

MattWestb, 2021-11-25:
The first is 2 warning that you is using external components = OK then its no error.
The second can being that the quirk is not OK and crashing ZHA.
For eliminating the problem with the local quirk you can renaming it to `ts0601_trv.Xpy` then ZHA is not trying loading it.

You can also clicking "load full home assistant log" in the log window and you can see more.

accessiblepixel, 2021-11-25:
That was with the full home assistant log shown. I only have two errors, and they're both for custom compoenents (hacs and localtuya)... however, it indeed looks like the custom zha "quirk" was casuing an issue in the ZHA integration, but those edits you got me to make have made it respond (and seemingly respond pretty well!)

Go have some well deserved rest. Can pick up the next testing tomorrow. <3

accessiblepixel, 2021-11-26:
So just to log some things I've noticed for us tomorrow.

It doesn't seem to communicate with ZHA very often, and seems a bit... "laggy", and doesn't accept updates as fast as I would expect, but I'm all out of energy so will check in tomorrow :) Definitely closer to a solution than we were 👍

accessiblepixel, 2021-11-26:
So I put lots of updates in the previous comment to avoid loads of comment notifications...

It seems to be working alright, just a bit slow on the temperature updates, but I guess that's just because it's battery powered.

Whenever you're ready let me know how to test the new integration :)

Also, does the ZHA-Device_Handlers project have a 'coffee donation' type thing, or a charity that they'd prefer donations to be sent to (in their name)? Feel like sending a couple of bucks in support.

Kind regards,
Jessica

MattWestb, 2021-11-26:
The last question you can find in https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md#how-to-contribute-as-an-end-user.

And all tuya TRV is sleeping end devices so they is not have the radio on all the time and can taking some seconds before they is catching waiting commands from its parent.

"Normal" Zigbee device is its possible configuration of attribute reporting all normal clusters. But tuya is reusing  MCUs from other platforms like WiFi and BT and is only using the ZIgbee module a one tunnel for there "MQTT to cloud" service (sending DP commands = Data Points).
So on the Zigbee level we cant do any thing at all. My TRV  is reporting local temperature around 60 min and if the temperature is changing more then 1°C its updating it faster.

Some TRVs is possible sending one updating local temperature calibration and the MCU is sending one old and new local temperature but our MCU have that not implanted.

Now to the funny thing !!
The TRV was added in the first commit for 10 mount ago but with only basic function and most other have getting updates from nice coding user but  our not.
I have starting trying copy and paste for getting things working but it was not 100% but now one very good code warier have fixing nearly all lost broken things plus implanting some new good things.

Dow load the new quirk from this post https://github.com/zigpy/zha-device-handlers/discussions/1084#discussioncomment-1701973 and deleting the "old one" and restart HA.
If you can copy the `climate.py` you is also getting presets working in ZHA but is not  one must.

You is getting some new switches and sensors that needs being renamed but is not easy to finding them.
I can explaining later how to finding the function but i must looking little on the last version before i writing that.

If all is going well you shall getting somthing like this on the device card (after adding new custom classed and icon to the entry's:
[image]

And from loverns card with TRV and sensor graf:
[image]
I posting more info for doing the custom things !!

MattWestb, 2021-11-26:
The switches and sensors is having this configuration in the quirk:
EP 1: Window Detection switch On / Off
EP 2: Valve Detection switch On / Off
EP 3: Window Open Alarm sensor
EP 4: Valve Jammed Alarm sensor
EP 5: Child Lock switch On / Off
EP 6: Install Mode (FI) Switch On / Off
EP 7: Local Temp Calibration sensor (slider + - 10 ° / 0.1°)

And the first on/off is the window detection and you can trying switching it on and off and you shall see it changing on the TRV display.
The second is val detect is have no icon in the display so you cant see it.
The third Children lock you can pressing the button on the TRV for 5 seconds and the switch is toggled in ZHA.
The forth is FI (= taking the battery out and putting them back and doing calibration of the valve) and if you is switching it on the TRV is showing FI1 and you must calibrating the valve.

The sensors is:
First Window Open / anti freeze Alarm sensor
Second Valve Jammed Alarm sensor
and the last is one slider for local temperature calibration.

The name of all is not 110% but its possible finding them out.

For valve jammed put the setpoint to high and the TRV is sending valve jammed and one ` ! ` in the window (if activated).
Putting the TRV in the freezer and you is getting window open / antifreeze alarm (if activated).

I have making this customization in HA for getting good icons and functions:
```
From my customize.yaml in HA:

binary_sensor.tyst11_trv_binary_input_2:
  device_class: problem
binary_sensor.tyst11_trv_binary_input:
  device_class: opening
switch.tyst11_trv_on_off:
  icon: mdi:window-open-variant
switch.tyst11_trv_on_off_2:
  icon: mdi:pipe-valve
switch.tyst11_trv_on_off_3:
  icon: mdi:lock
switch.tyst11_kfvq6avy_fvq6avy_c1db03fe_on_off:
  icon: mdi:wrench
```
The name of the sensors / switches is always little trickey.

Its also possible giving all names and and looking / editing in `\.storage\core.entity_registry` and finding the right end point:
```
            {
                "entity_id": "switch.tyst11_kfvq6avy_fvq6avy_c1db03fe_on_off",
                "config_entry_id": "1509216dc5c6aac3464ac35895885f7b",
                "device_id": "55b2995a1e4187422eef81c61b572b3f",
                "area_id": null,
                "unique_id": "bc:33:ac:ff:fe:03:db:c1-6-6",
                "platform": "zha",
                "name": "TYST11 TRV FI",
                "icon": "mdi:wrench",
                "disabled_by": null,
                "capabilities": {},
                "supported_features": 0,
                "device_class": null,
                "unit_of_measurement": null,
                "original_name": "_TYST11_kfvq6avy fvq6avy c1db03fe on_off",
                "original_icon": null,
                "entity_category": null
            }
        ]
    }
```
Then the `unique_id` `bc:33:ac:ff:fe:03:db:c1` is the long address for the device and in the end the end point and cluster number `6-6`. Cross reference with the En point list.
`name` is the name the entry is having in the HA GUI.

The local temperature calibration is only for ZHA showing heating / idle and also getting the graf looking nice. I have -0.9° for the moment and it can being adjusted for making it looks right then the TRV is not sending  any status of the valve is open or closed so the quirk is comparing and saying heating or idle.

MattWestb, 2021-11-26:
Sorry i was missing your communicating problems (sorry TRV . . ).
Sending commands can being little tricky but resiving is normal working very fast.
But its one sleeping end device and we can do so much about it.
If having larger problem delete the device in ZHA waiting one minute and restarting HA.
Then ZHA is stable adding the device and look if all looks OK.
If the quirk is not loaded OK restart HA and it shall doing it OK.

accessiblepixel, 2021-11-26:
Sorry if I was a little confusing!

I'll try using the new climate.py as suggested and report back :)

MattWestb, 2021-11-26:
Yes !  :-)

accessiblepixel, 2021-11-26:
This folder
`/usr/src/homeassistant/homeassistant/components/zha/`
doesn't exist. Should I create it?

MattWestb, 2021-11-26:
Its not existing in the OS.
You must connecting to the docker container and open one CLI in it for getting  it then its only in the container.

Do you running on bare metal , VM or OS ?

MattWestb, 2021-11-26:
You looks being much better in the Linux CLI then my ;-((

accessiblepixel, 2021-11-26:
It's running in a virtual machine. I'm running Home Assistant OS 6.5.

I've installed the addon to let me SSH to it. I'll try having a look from the CLI.

Ah... Yes. I can see that folder if I login from the CLI, but I don't have SSH access to that with the addon SSH thing... so I needed to setup a private key and insert a usb stick to get access to the "real" SSH...

Anyway, I have copied that file over and am now rebooting... Certainly nothing simple about HA is there? 🤣

accessiblepixel, 2021-11-26:
Okay, now I've replaced climate.py and used that file from the linked thread, but now the integration isn't working again, it just says 
Failed to set up
Check the logs

And when I check the logs, nothing is listed there :(

MattWestb, 2021-11-26:
I hope you was only copying the  `climate.py` to the container and not the quirk for the device.

You can delete the quirk in the local quirk folder so its not making problems.

If you is not having one very large set up little more debug logging like:
```
logger:
  default: info
  logs:
    homeassistant.core: info
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```

accessiblepixel, 2021-11-26:
Yes, only copied the climate.py to /usr/src/homeassistant/homeassistant/components/zha/climate.py.

D'oh! I only COPIED climate.py to that folder, I didn't delete it from the custom_zha_components folder where I uploaded it to :( 🤦 

Okay so the integration had loaded up again, but now there's no controls shown, so I don't think it's recognising to use the new quirk file with the device.

MattWestb, 2021-11-26:
Perhaps `some one` was forgetting adding your device id in the quirk ?

Adding  it in or after line 648 (you can using one of the old one than you dont having then if you like)

I can get enough sleep . . .

accessiblepixel, 2021-11-26:
> Perhaps `some one` was forgetting adding your device id in the quirk ?
> 
> Adding it in or after line 648 (you can using one of the old one than you dont having then if you like)
> 
> I can get enough sleep . . .

Whoops! Well, adding it has got some functionality working again :)

I think it's just it doesn't know what the names of the things are like you said in one of your other posts.

Here's a screenshot of the device in ZHA. I've gotta disappear for a bit, but I'll have a crack at figuring out what is what when I get back.

I'm super appreciative of your help, thanks just doesn't seem to be enough.
Hopefully since we've gone through this, the next person who tries to do what I'm doing won't have to loose so much hair xD

MattWestb, 2021-11-26:
The grayed out you can delete its some old thing from the old quirk.
The first with the slider is local temperature correction set it to -0.9° and adjusting it later if you need.
I hope your device is communicating OK but you dont have  any battery reading but its only reported one or 2 times  day.

Try switching the children lock on the TRV and see if you is getting on of the switches changing state.

MattWestb, 2021-11-26:
PR made waiting merging :)

accessiblepixel, 2021-11-26:
> PR made waiting merging :)

Amazing :) I'll have a play with the device to see if I can get some names figured out for each of the sensors and configurations after I've defrosted from being outside. It's about 3C out, so quite tepid :rofl:

My partner and I are very grateful for your help!

I do have a question when the change is merged and finally added upstream, will I just have to remove the custom quirk or will it be an "automagic" process at some point when it gets into the main HA branch?

MattWestb, 2021-11-26:
I was going to LIDL on noon and it was raining snow and not its 10 cm of the col well JUNCK outside !!!

The merger is for the old implementation then we have not getting OK for the new then its one braking thing with many users is using it in there systems and its not popular braking working systems.

If leaving the local quirk its overriding the one in the HA docker container.
So if you like using the old you deleting the new one in the local quirk folder and and ZHA is using the internal one.

The only thing is the the presets (the modified climate.py) is gone and need being copy in to the container with SSH but its shall not being used if using the old quirk then its not having presets.

accessiblepixel, 2021-11-26:
I'll leave it running the new one for now then, and see if I can get the whole network swapped over, do some testing to find out which switches correspond to which function.

I agree, it's probably best not to break people's systems (although Home Assistant did that massive breaking change when they changed the Tuya API, although Tuya only gave them 2 months, but now they extended it)...

However, any support is much better than no support 👯‍♀️

closed 2021-11-28 by dmulcahey

JyothishGeorge, 2021-12-12:
Hi 

I have similar trv with me trying to pair and the pairing is successful. The HA is updated to core-2021.12.0 and I copied zha-quirks 0.0.65 to the  zha-quirks folder. Removed the __pycache__ and HA restarted.  After pairing the TRV not all the entities are loaded for me. Please the device signature below.

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_czk78ptr",
  "model": "zk78ptr",
  "class": "zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type1"
} 

All I get is thermostat and hvac_action.  Could you please help me figuring out to load the rest of the attributes? Any help on this is greatly appreciated. Not sure what am I missing.

[image]

Thanks George

MattWestb, 2021-12-13:
You dont need updating the the quirk in HA 2021.12 then all is in for the basic function for your TRV.
But for the moment its looks like your is not communicating then you dont have any battery status reported.

For getting te extended quirk you need configuring local quirks in HA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and puting the unziped file from this post https://github.com/zigpy/zha-device-handlers/discussions/1084#discussioncomment-1756246 in the new local quirk forlder.
After restart of HA you shall getting all new entries but not presets in the thermostat card.
Unzip the climate.py from this file [climate.zip](https://github.com/zigpy/zha-device-handlers/files/7701803/climate.zip) and copy it to your HA configuring folder.
Then you need open one CLI in the HA docker container and copy it in the the ZHA folder with  this command:
```
 cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py
```
restart HA anad you shall having presets in the thermostat card.

MattWestb, 2021-12-13:
Sorry its my fault !!!

The new device DI was not added in the quirk then its one new device.
I have making on updated version of the last one i have in my system with your device added.

[ts0601_trv_siterwell.zip](https://github.com/zigpy/zha-device-handlers/files/7704501/ts0601_trv_siterwell.zip)

Replacing the old one in your local quirk folder and deleting the cash folder and restarting HA and look how its working.

JyothishGeorge, 2021-12-13:
Hi MattWest

Its working for lot more attributes , you are super helpful, appreciate your time to check this, I will look around and understand the code in detail and looking forward to contributing if I can,  I have few more TRVs (Tuya) and going to put in testing soon,

Thanks a lot, 
George 
[image]

MattWestb, 2021-12-13:
Great !!

I wondering why you not having any battery reported ?

You can trying deleting the device and waiting one minute and paring it and look if you is getting data from the device OK.
Also you have one old switch that i think is from the old quirk have saving in the device database that is not have any function.

For getting the functions of the switches and sensors look in the quirk for the endpoints and ZHA is pupotling them with first found is without number in the entity ID  and the second one is getting 2 and so on.

All thanks is going to @jacekk015 that have doing the coding !!!

accessiblepixel, 2021-12-29:
Just a slight update, the "old" (current integration) in home assistant works fine with my valve.

I don't get battery info, but other than that it gets the temperature, I can set the temperature and can turn heat on/off.

I deleted the custom quirk when upgrading to home assistant 2021.12.6 and it all works magically fine (and reports in every 10 minutes if there's no change, where as the "new" beta method didn't update itself for some reason at all).

Thanks again for all the help :)

james-fry, 2022-09-06:
Is battery reporting working in ZHA? I can see above a few commments that its not... I am trying to add support for TCP valve (modelID: 'zk78ptr', manufacturerName: '_TYST11_czk78ptr') to zigbee2mqtt but I dont get battery reported.

MattWestb, 2022-09-07:
Try its with ZHA only disable Z2M and installing ZHA and you can testing it.

accessiblepixel, 2022-09-16:
I've never had battery indication work in my home assistant... I use the Lithium AA batteries which will do a whole year so long as the valves aren't too active, but it would be nice to have it reporting battery status.

MattWestb, 2022-09-16:
Nop i have 5 in production and test network and all is reporting battery % and one is eating double battery then the other.

Perhaps is your Lithium but i dont think it shall doing any different as ling you is getting enough voltages for the device from them.

You can try taking the battery out and waiting one minute and putting them back then the MCU shall reporting all DP to the host system.

Its also possible that this device MCU is having one different firmware that is using other functions but then you shall see the DP in the log that is changing then it reporting it (also after repower).

accessiblepixel, 2022-09-28:
Hmm. I'm not sure that it's because of the lithium AA batteries... I have a couple that are running standard Alkalines and they don't report either...

It might be because mine are the TCP Smart ones, maybe it reports it's battery info in a different variable (or a different way), maybe just slightly different to the device the quirk is based on? @MattWestb Is there anything I can grab from them to show all the data that they have available, to see if we can find that battery info and "map" it to the right field if this is what's happening? :)

MattWestb, 2022-09-28:
@accessiblepixel  If you have one TRV that is pared OK  with ZHA and have logging enabled and then taking the battery out and waiting one minute and outing it back the tuya MCU shall sending all DPs (commands) its using and you is getting them in the HA log.

I can being that its one updated version of the MCU firmware that have changing commands its using (some other manufacture have doing it) and if we is getting the DPs and then its possible doing one new mapping with them and its working OK but we is needing the information.
If i remember is the TRV not reporting the battery status so oft but i think its around 23 hours.

Have you looking if Z2M and deCONZ have getting the device supported and if the is using the came code for other devices ?

One current snapshot from my test TRV:
[image]

jacekk015, 2022-10-14:
@accessiblepixel From screenshot you've posted you are using some old quirk - I assume.
Remove old quirk from quirks folder and replace it with the one below.
Restart of HA is mandatory.
[ts0601_trv_siterwell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9785870/ts0601_trv_siterwell.py.zip)

jacekk015, 2022-10-14:
Put above quirk to the quirk folder and restart HA

accessiblepixel, 2022-10-14:
Ok. Done. I have more binary switches that have appeared, and there's now a "temperature correction" slider, power still says unknown but I'll watch it over the next few hours and see what it does. x

jacekk015, 2022-10-14:
Need to see some logs from HA to see what's going on there. Best would be to get Debug level logs from pairing, and 5 minutes after

accessiblepixel, 2022-10-15:
> Need to see some logs

Here are the logs you requested @jacekk015 

```
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=6, profile_id=0, cluster_id=32822, data=Serialized[b'\x06\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=7, profile_id=0, cluster_id=32822, data=Serialized[b'\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6821), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=129, profile_id=0, cluster_id=19, data=Serialized[b'\x81!h\xec%\x87\xfe\xff\xbd\x1b\xec\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
New device 0x6821 (ec:1b:bd:ff:fe:87:25:ec) joined the network
[0x6821] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6821 ieee=ec:1b:bd:ff:fe:87:25:ec is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81!h\xec%\x87\xfe\xff\xbd\x1b\xec\x80'
[0x6821:zdo] ZDO request ZDOCmd.Device_annce: [0x6821, ec:1b:bd:ff:fe:87:25:ec, 128]
Tries remaining: 3
[0x6821] Requesting 'Node Descriptor'
Tries remaining: 2
[0x6821] Extending timeout for 0x65 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6821), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=101, profile_id=0, cluster_id=32770, data=Serialized[b'e\x00!h\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6821 ieee=ec:1b:bd:ff:fe:87:25:ec is_initialized=False> from ep 0 to ep 0, cluster 32770: b'e\x00!h\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'
[0x6821] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x6821] Discovering endpoints
Tries remaining: 3
[0x6821] Extending timeout for 0x66 request
[0x778F](TS0502A): last_seen is 15412.453288078308 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0x778F](TS0502A): Update device availability -  device available: False - new availability: False - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6821), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=102, profile_id=0, cluster_id=32773, data=Serialized[b'f\x00!h\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6821 ieee=ec:1b:bd:ff:fe:87:25:ec is_initialized=False> from ep 0 to ep 0, cluster 32773: b'f\x00!h\x01\x01'
[0x6821] Discovered endpoints: [1]
[0x6821] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x6821:1] Discovering endpoint information
Tries remaining: 3
[0x6821] Extending timeout for 0x67 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6821), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=103, profile_id=0, cluster_id=32772, data=Serialized[b'g\x00!h\x10\x01\x04\x01\x00\x00\x00\x02\x00\x00\x03\x00\x02\x03\x00\x19\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6821 ieee=ec:1b:bd:ff:fe:87:25:ec is_initialized=False> from ep 0 to ep 0, cluster 32772: b'g\x00!h\x10\x01\x04\x01\x00\x00\x00\x02\x00\x00\x03\x00\x02\x03\x00\x19\x00'
[... cut ...]
I've also noticed since running the quirk file you provided me, none of the temperature recordings are appearing now for any of them, but I have found out some of the "switch options", one of them is "Enable Window Open Detection", "Child Lock", and "Re-run valve positioning test". The "Identify" button doesn't cause the device to identify itself (In other zigbee devices usually that causes an LED on it to flash to let you know which device it is).

Hopefully this information is useful.

Kind regards,
Jessica

jacekk015, 2022-10-15:
Need HA full logs from pairing process and min 5 minutes after. Thoose seems to be pairing window logs.
Also debug level needs to be enabled first.

accessiblepixel, 2022-10-15:
> Need HA full logs from pairing process and min 5 minutes after. Thoose seems to be pairing window logs. Also debug level needs to be enabled first.
> 
> Logs should look like those from link:
> 
> [#1159 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1159#issuecomment-1279352203)

Ah. You were not clear about that... but yes, here they are.

Couldn't post them here, as they were too long. I've linked them here instead with a week's expiry: https://pastecry.pt/X5Am5x#Up-Pyt_Num%5EEt8Zur3Yg6Uc8Em

I did not know what information I could trim to make them smaller, and there's no doubt information about other components in there...

jacekk015, 2022-10-16:
1. As for battery  - someone earlier made assumption that battery level is sent using 0x0215 attribute - I don't see any of that in the communication process, so that's why level is unknown. Maybe he was wrong or you have a bit different device.
`SITERWELL_BATTERY_ATTR = 0x0215  # [0,0,0,98] battery charge`
3. For temps - I see some strange errors, but code is based on my own devices so it should be working OK.

To be sure - you are using latest HA version with all updates?

Need some time to debug this.
Do NOT ever trim any data - it's possible that you cut out lines needed to see communication messages.

accessiblepixel, 2022-10-16:
I'm running 2022.10.1 at the minute, I haven't upgraded to 2022.10.4 yet, and there's an update to HA OS 9.2.

I'll do that now, and I'll do another pair/sync process and post the entire logfile from that process, since when I added the device HA was already up and running (It's used by us for a lot of stuff so... Kinda have to plan times for it to be offline.)

Standby.

jacekk015, 2022-10-16:
@accessiblepixel Try version below - temps should be OK.
[ts0601_trv_siterwell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9795329/ts0601_trv_siterwell.py.zip)

@MattWestb I'm pretty sure you have those TRVs
Could you put your logs here - we are searching for 0x0215 attribute which I don't see in the logs above.
That's battery level value

MattWestb, 2022-10-16:
I was moving my test TRV to tuya / LIDL ZBGW for cecking for updates but was not any :-((
But then adding it back i was getting:
```
2022-10-16 12:33:53.309 DEBUG (MainThread) [zhaquirks.tuya] [0x3489:1:0xef00] Received value [0, 0, 0, 57] for attribute 0x0215 (command 0x0001)
2022-10-16 12:34:33.739 DEBUG (MainThread) [zhaquirks.tuya] [0x3489:1:0xef00] Received value [0, 0, 0, 56] for attribute 0x0215 (command 0x0001)
```
= 65% battery in ZHA.

Idont know how oft its reporting but it not oft so best is removing the battery and putting it back so the MCU is sending all DPs its knowing.

By the way great welcome back Jacekk !!!

[7 more replies]

PR #1184 (2021-11-28): Adding "_TYST11_czk78ptr", "zk78ptr"
Adding "_TZE200_czk78ptr", "TS0601", "_TYST11_czk78ptr", "zk78ptr", "_TZE200_2atgpdho", "TS0601" and ("_TYST11_2atgpdho", "atgpdho")
I also deleting the ("_TYST11_b6wax7g0", "6wax7g0") then its look Beca Smart is not making one with the old Zigbee module.
The "2atgpdho" versions also need getting presets in ZHA in one other PR for working OK.

Fixes #1178
Fixes #1180