zigpy/zha-device-handlers #1185: [Device Support Request] LivingWise LVS-SC7

jfrancais, 2021-11-26:
**Is your feature request related to a problem? Please describe.**

Events aren't captured when pressing buttons on this paired device.  Here are the log entries I see for each button press.
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x01\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x02\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x03\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x04\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x05\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x06\x00\x00'
[0xb1d0:1:0x0017] Unknown cluster-specific command 8
[0xb1d0:1:0x0017] ZCL request 0x0008: b'\x07\x00\x00'

**Describe the solution you'd like**
Would like the particular evens captures.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0005"
      ],
      "out_clusters": [
        "0x0000",
        "0x0005",
        "0x0017"
      ]
    }
  },
  "manufacturer": "欧瑞博",
  "model": "75d430d66c164c26ac8601c05932dc94",
  "class": "zigpy.device.Device"
}

**Additional context**
In using home assistant zha, this pairs but events don't get captured.  As such device can't be used.

This device does seem to be supported in zigbee2mqtt but not zha
https://www.zigbee2mqtt.io/devices/LVS-SC7.html

jfrancais, 2021-11-26:
Might be some more useful information in here as well, same device:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2639

jfrancais, 2021-11-26:
Also may have useful information, same device:
https://github.com/bradlee-s/det-SmartThings/blob/master/Orvibo%207%20Button%20Scene%20Controller.groovy

jfrancais, 2022-01-06:
Still on the hunt for this

closed 2022-08-07 by github-actions[bot]

jfrancais, 2022-08-08:
Still an issue

jfrancais, 2022-12-13:
Still looking for this to be resolved.  Anyone?

dominic195, 2023-07-18:
would also like to know if a solution was found