zigpy/zha-device-handlers #1186: [Device Support Request] IKEA FYRTUR block-out roller blind zhaquirks.ikea.blinds.IkeaTradfriRollerBlinds2

mclever, 2021-11-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Battery level always shows 0%. Doing reconfigure device in Home Assistant shows an error on DoublingPowerConfigurationCluster.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Make it so the battery level is reported correctly

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "FYRTUR block-out roller blind",
  "class": "zhaquirks.ikea.blinds.IkeaTradfriRollerBlinds2"
}`

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you!

MattWestb, 2021-11-26:
Pleas post the error from the DoublingPowerConfigurationCluster.
Put ` ``` ` on one line before and after the code block so its not loosing the formation.

mclever, 2021-11-26:
@MattWestb I ended up updating zigbee firmware on nortek HUSBZB-1 and then the battery level started working. closing request

closed 2021-11-26 by mclever

MattWestb, 2021-11-26:
Great that it was fixing it.

I have not hearing one so clean fix for device problems then updating the old firmware on EM35X / EFR32 coordinators.

IKEA is also having new firmware in the pipe (its Zigbee certificated) that also can helping with some problems then its being released in the production feed.

mgrcic, 2022-07-05:
hello, 
I have the same issue. I am integrating this device into the 3rd party hub and the battery measurement reports are always 0.
This is confirmed by the ZigBee sniffer logs.
The configure report command is successfully processed by the roller blind, but it always reports 0 for battery voltage and battery percentage remaining attributes.
The device firmware version is 1.99.34-2.2.009
Do you have any suggestions?

MattWestb, 2022-07-05:
@mgrcic Your blinds is running on very old firmware then the current production is `2.3.087`.
Update the firmware and deleting the blind and waiting one minute and paring it new so it can being OK configurated and if not working try `reconfigure` the device from device card.

abjugard, 2023-11-02:
> @mgrcic Your blinds is running on very old firmware then the current production is `2.3.087`. Update the firmware and deleting the blind and waiting one minute and paring it new so it can being OK configurated and if not working try `reconfigure` the device from device card.

Is there any other way to get (Z)HA to change which quirk is used? I really would like to not have to change all my HomeKit-configured automations which I suspect will break if I remove and re-add the devices.