zigpy/zha-device-handlers #1187: Osram Smart+ Plug won’t pair

flosen1304, 2021-11-27:
Hi,
I want to move from deconz integration to ZHA (conbee II stick). Everything works fine expect my Osram Smart+ Plugs. I’m not able to pair them after resetting them (hold power button for 10 sec). I’ve tried different devices and locations. I think they should be supported.
Do you have any hint for me, what to do?

The log shows:

```[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xa46b:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=72 command_id=Command.Report_Attributes>
[0xa46b:3:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0xa46b:3:0x0006] Attribute report received: on_off=0
[0xa46b:3:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=73 command_id=Command.Report_Attributes>
[0xa46b:3:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=254>)]]
[0xa46b:3:0x0008] Attribute report received: current_level=254
[0xA46B:3:0x0008]: received attribute: 0 update with value: 254
```

I’m running 2021.11.2 on a RPI 4 in a python virtual enviroment.

Thanks in advance
Florian

puddly, 2021-11-29:
Is the plug far away from your coordinator?  If so, try upgrading to the latest release of zigpy-deconz:

```console
$ pip install --upgrade zigpy-deconz
$ hass --skip-pip  # otherwise HA will downgrade it
```

flosen1304, 2021-11-30:
@puddly Thanks! With zigpy-deconz 0.14 it works like a charme (before it was 0.13).

If I start the service with '--skip-pip' is this a permanent stable solution? Or do I have do leave away '--skip-pip' when version 0.14 will be integrated in hass? How can I notice?

puddly, 2021-11-30:
> If I start the service with '--skip-pip' is this a permanent stable solution?

Not really. New dependencies won't be installed when you restart HA so it'll eventually start breaking things when you upgrade.

The only difference between 0.13 and 0.14 is that joins are now permitted through every router on the network.  If you don't plan on joining any more devices far away from the coordinator, you can let HA downgrade.

The dependency PR was merged into HA Core a few days ago (https://github.com/home-assistant/core/pull/60474) but I'm not sure when it'll be released. Possibly in the next beta?

closed 2021-11-30 by puddly

Adminiuga, 2021-11-30:
Beta should be out this Wednesday i think. Release a week after