zigpy/zha-device-handlers #1188: FeiBit Light switch showing as dimmable when it's not

Smiley1244, 2021-11-28:
Hello,

Please forgive me I'm very new to ZHA but i'm in the process of moving all my Hue devices to ZHA.
I have a couple of FeiBit switches which worked fine in Hue and worked 100% in ZHA until a recent update. With the update (sorry I don't remember which one, RL has got in the way and it's been some time since the change). They went from being std on/off to reporting as dimmable which they are not.

In some circumstances the brightness gets set to 1 which really only affects the bightness of the icon in HA, everything works fine so it's just an annoyance but i thought I'd report it as it doesn't seem correct.

The devices I have are the 
FeiBit FNB56-ZSW02LX2.0 2 way switch
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4478, maximum_buffer_size=127, maximum_incoming_transfer_size=90, server_mask=10752, maximum_outgoing_transfer_size=90, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "12": {
      "profile_id": 49246,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "FeiBit",
  "model": "FNB56-ZSW02LX2.0",
  "class": "zigpy.device.Device"
}`

FeiBit FNB56-ZSW01LX2.0 1 way switch
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4478, maximum_buffer_size=127, maximum_incoming_transfer_size=90, server_mask=10752, maximum_outgoing_transfer_size=90, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "FeiBit",
  "model": "FNB56-ZSW01LX2.0",
  "class": "zigpy.device.Device"
}`

Please let me know if I can contribute anything else that might help or if anyone can advise if I can cahnge somethign in ZHA or HA to make them NOT dimmable?

MattWestb, 2021-11-28:
I was looking on your device profile and cluster:
```
"endpoints": { 
"11": { "profile_id": 49246, "device_type": "0x0000", 
"in_clusters": [ "0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0008", "0x1000" ], 
"out_clusters": [ "0x0019" ] }, 

"12": { "profile_id": 49246, "device_type": "0x0000", 
"in_clusters": [ "0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0008" ], 
"out_clusters": [ "0x0019" ] } },
```
The profile is saying:
```
PROFILE_ID = 49246 = ZLL
ON_OFF_LIGHT = 0x0000
DeviceType.ON_OFF_LIGHT: ([0x0004, 0x0005, 0x0006, 0x0008, 0x1000], []), (cluster that the device shall having)
```
And the cluster functions is:
```
"0x0000", Basic
"0x0003", Identity
"0x0004", Group
"0x0005", Sene
"0x0006", OnOff
"0x0008", Level
"0x1000", TL
"0x0019", OTA
```
And all this is not helping you more then all locks being OK for one ON_OFF_LIGHT Zigbee Light Link device :-((

I only have IKEA outlets but they is updated to Zigbee 3 and have different profile but  also having Level cluster and not getting and dim function in ZHA.

I can thinking that somthing was not going well in ZHA then updating your database around 5 times the last mounts and have getting some bad saved data that is making ZHA thinking that the Level cluster is "active" then it shall not being (= I dont knowing).

I knowing its not easy then its in wall mounted switches but you can trying deleting one switch from the device card and waiting one minute and then adding it and see if its being  added with Level or only On Off device in ZHA.

I hope some more with good experiences can bringing little more light in this problem.

Smiley1244, 2021-11-28:
Hi, thanks for the comments. I tried removing it and waiting 1 minute, came back the same. I then did it again and waiting 5 minutes, same result.
I'm just deleting the device from ZHA, this is enough? When deleted the switch goes back in to pairing mode.
I only ask if its enough as it repopulates in ZHA as before and reattaches all my automations etc so I wonder how much it's hanging on to even when removed?

If I try a reconfigure i get the following
[image]

MattWestb, 2021-11-28:
The good thing your switch is doing the leave OK and nice going in paring mode !!

ZHA is retrying deleting device more times if not getting  one "Left" from  the device therefor waiting one minute.

You can also trying waiting one minute and then restarting HA so its cant reusing old information (that shall being deleted in the database) but i dont knowing if its helping.

Dot play with all switches so you is loosing control of all your light and messing all your automation you have made, its enough "destroying" one for testing.

I hope some of the devs from US is waking up then in US you have many ZLL switches with and without dimmer function and they have more experience if them.

Smiley1244, 2021-11-28:
Removing, waiting a few minutes and then rebooting HQ before rejoin resulted in the the same. I'll keep monitoring this thread, i'm happy to try anything suggested :)

Smiley1244, 2021-12-26:
Anyone able to suggest anything to resolve this?

Smiley1244, 2022-06-24:
> There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved with the latest updates. Please make sure to update to the latest version and check if that solves the issue. Let us know if that works for you by adding a comment 👍 This issue has now been marked as stale and will be closed if no further activity occurs. Thank you for your contributions.

This is still an issue, even on the latest version.

Smiley1244, 2022-12-21:
This is still an issue on the latest build. Switch still have brightness exposed which doesn't exist. When using the physical switch the brightness drops to 1.

Smiley1244, 2022-12-21:
Closed by mistake

TheJulianJES, 2023-01-04:
Can't we just remove the ``LevelControl`` cluster with a custom quirk to remove the brightness slider from HA?

TheJulianJES, 2023-01-04:
@Smiley1244 Download and extract the ZIP please: [feibit_binarylight.py.zip](https://github.com/zigpy/zha-device-handlers/files/10346650/feibit_binarylight.py.zip)
Then, install it as a custom quirk and restart HA. ([Tutorial for installing a custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274))

Let me know if the device picks up the quirk (can be seen if ``"class"`` of the signature changes) and if the level control is removed for both switches.

Smiley1244, 2023-01-04:
@TheJulianJES that appears to work correctly on both switches, thanks! :)
Will this eventually be built-in to ZHA or will it stay a custom quirk?

TheJulianJES, 2023-01-04:
@Smiley1244 See:
- https://github.com/zigpy/zha-device-handlers/pull/2066

Smiley1244, 2023-01-04:
Superstar!! :) Thanks for sorting

closed 2023-01-04 by Smiley1244

TheJulianJES, 2023-01-04:
The quirk should be available in the final release of the next major HA version: **2023.2.0**