zigpy/zha-device-handlers #1189: [Device Support Request] Saswell Tuya Thremostat Family Improved Quirk

PLTorrent, 2021-11-28:
This quirk supports following devices:
Saswell | SEA801 | _TYST11_KGbxAXL2 | GbxAXL2 | ✅ Sas II | 798
Saswell | SEA802 | _TZE200_c88teujp | TS0601 | ✅ Sas I | 576
Saswell | SEA802 | _TYST11_c88teujp | 88teujp | ✅ Sas II | 576
HiHome | WZB-TRVL | _TZE200_zuhszj9s | TS0601 | ✅ Sas I | 1064
HiHome | WZB-TRVL | _TYST11_zuhszj9s | uhszj9s | ✅ Sas II | 798
Hama | 00176592 | _TZE200_yw7cahqs | TS0601 | ✅ Sas I | 923
Hama | 00176592 | _TYST11_yw7cahqs | w7cahqs | ✅ Sas II | 798
RTX | ZB-RT1 | _TZE200_azqp6ssj | TS0601 | ✅ Sas I | 923
RTX | ZB-RT1 | _TYST11_azqp6ssj | zqp6ssj | ✅ Sas II | 1064
Sanico | - | _TZE200_9gvruqf5 | TS0601 | ✅ Sas I | 1064
Sanico | - | _TYST11_9gvruqf5 | gvruqf5 | ✅ Sas II | 1064
Saswell | SEA802 | TZE200_zr9c0day | TS0601 |  ✅ Sas I

This quirk is experimental and supports following functions of the TRV:

- On/Off
- Temperature Setpoint
- Child Lock On/Off (actual child lock has to be activated manually on the device)
- Window Detection On/Off
- Anti-Freeze On/Off
- Limescale Protection On/Off
- Away Mode On/Off
- Schedule Mode On/Off 
- Temperature Correction setting
- Battery reporting (Full/Low)

Not implemented:
- Scheduling changing/setting

The quirk:
[Saswell.zip](https://github.com/zigpy/zha-device-handlers/files/7613730/Saswell.zip)

How to activate the quirk: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Please re-pair the device after new quirk is used.

[image]

[image]

Since currently we are not able to change names of on_off entities below is the description:
1. Window Detection
2. Child Lock
3. Anti-Freeze
4. Limescale Protection
5. Schedule Mode On/Off
6. Away Mode

Please test and provide feedback. Especially the _TYST11 lineage since my device is _TZE200 based.

Special thanks to @MattWestb @jacekk015

supersebbo, 2021-11-28:
This is great!  Thank you.

I'm curious though, I still see no mention of the "Valve Position" attribute, or the ability to force the valve position.  Zigbee2MQTT has been able to both read and set the valve position on the TS0601 series for quite a long time which makes it a lot more powerful for creating heating control systems in HA.  Can you take a look into this?

https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

https://github.com/Koenkk/zigbee-herdsman-converters/blob/b239676de9f191ba3ee76c1d96a493b4db9d3f3a/converters/fromZigbee.js#L4134

https://github.com/Koenkk/zigbee-herdsman-converters/blob/b239676de9f191ba3ee76c1d96a493b4db9d3f3a/converters/toZigbee.js#L4468

https://github.com/Koenkk/zigbee-herdsman-converters/blob/b239676de9f191ba3ee76c1d96a493b4db9d3f3a/converters/toZigbee.js#L4580

PLTorrent, 2021-11-29:
@supersebbo TS0601 is a very wide family of products. AFAIK on Moes likes you have valve position reporting. Saswell and clones do not report valve position in any way. Even in the original tuya app. Also there is no endpoint that reports it as it changes so no way to implement it.

supersebbo, 2021-11-29:
> @supersebbo TS0601 is a very wide family of products. AFAIK on Moes likes you have valve position reporting. Saswell and clones do not report valve position in any way. Even in the original tuya app. Also there is no endpoint that reports it as it changes so no way to implement it.

Thanks.  So the only way to 'control' these clone valves is by manipulating the set-point?  Do you by any chance know the internal logic?  It seems like the only open the valve if the set-point is >1 degree over the TRV measured temperature, but I don't know if this is true for all cases. 

I am trying to work out if it's more efficient in terms of Zigbee traffic to keep changing the set-point in order to meet the heat demand of my room temperature sensors, or whether to apply the "temperature correction".

MattWestb, 2021-11-29:
@supersebbo what is the device signature for your device ?
TS0601 is all tuya "DP devices" that is using tuya MCU and some with older zigbee modules.
Some tuya TRV is only supporting setpoint manipulation and other have more function but must knowing your device for knowing what is possible.

By manipulating one TRV by forcing the valve you is braking the regulation algorithm and you cant getting it working well (its takes days for one good TRV getting it self calibrated and working well).

By manipulating the local temperature calibrating in small steps you can still getting the TRV working well but is not easy getting it right and if you is doing it it too large steps its losing the calibrations for its algorithm.

I think the best is getting one good offset for the local temperature and then letting the TRV doing there work and calculating the heat demand from how much the TRV heating from val % reported or if not possible software calculated as being done for the Site TRVs.

supersebbo, 2021-11-29:
@MattWestb 

  "manufacturer": "_TZE200_c88teujp",
  "model": "TS0601",

It's not performing badly in the current weather using the set-point manipulation quoted above.  But I suspect when the weather changes this will cause some big over-shoots. 

[image]

MattWestb, 2021-11-29:
I have one Siterwell in my kitchen for nearly one year but its little complicated then its in 2 levels and 3 hearings and 2 have normal mechanical ones and the Siterwell is doing the "fin work" and have working very well only one time getting the valve jammed.

The Saswell can also being off / on / auto mode but using on / off its cant using the algorithms for getting good regulation so better using  the setpoint changing.

Testing and see how its working but i think if its not have to less heating it shall work OK or its getting problem then need heating and cant getting the heating from the heating system.

supersebbo, 2021-11-29:
The ciritcal thing is that the offset from the actual room temperature is not constant.  Look the light-blue line versus the purple line.  When the radiator is off the TRV temperature is almost the same as the Room temperature, but when the radiator is on, the TRV temperature is as much as 10 degrees hotter than the room temperature.

PLTorrent, 2021-11-29:
@supersebbo to be honest I am not very satisfied with how this TRV performs. I can compare it to netatmo and netatmo performs much better so far. I have just started testing it on the radiator, before I was just developing the quirk so it was easier to have it on the table. My general remark is that you should not be forced to manipulate the setpoint, you should just set the temperature and it should work as set and more or less keep the temperature. I mean the non-connected TRVs do it pretty well so why the intelligent one should not be capable of doing it equally well? If you have to set it all the time then for me it means... find another TRV... but that's just me.

jacekk015, 2021-11-29:
Only PID can probably make it to work properly.

MattWestb, 2021-11-29:
CN-Hysen "Moes" HY366, HY 367, HY368 an HY369 is having PID if looking on some data from http://www.xmhysen.com/ProductInfoCategory?categoryId=126920

jacekk015, 2021-11-29:
@supersebbo 

> My impression is that the TRVs are deliberately quite ‘dumb’ and I expect a lot of processing, learning and feedback happens within the Tuya App that you do not get when using them as raw Zigbee devices.

That's actually not true, most of them has PID and need to still work without Tuya gateway.
Tuya app is for temp set and etc. - what's the difference if you set a temp form App or on TRV locally.
Some of TRV has static open valve positions, like 25%, 50%, 75% or 100%
My Maxsmarts now has 29% opened valve and it never go back up/down to same value.

It's nothing new that the TRV is mounted on a heat source. Standard TRVs are also working same.
The thing is that, because of construction, they are slowly reacting devices, so they smooth the reaction to changes.
The same reason is why most of Window Detection functions don't even react.
If the temp doesn't drop rapidly TRV will open and heat more, instead of closing the valve for Window open reason.

supersebbo, 2021-11-29:
> @supersebbo
>
>
> That's actually not true, most of them has PID and need to still work without Tuya gateway. Tuya app is for temp set and etc. - what's the difference if you set a temp form App or on [...]

We are saying the same thing.  As I showed above, the PID control on the TRV is quite good, it is able to control the temperature at the TRV accurately and consistently, but it can only do PID with the inputs it knows about, which for a TRV on it's own is just the valve position, the set-point and the temperature measured at the TRV.   There is no way the TRV on it's own can know about the temperature in other parts of the room it is heating.  

The Tuya app allows you to add additional temperature sensors and assign them to rooms to help the TRV, in this case it must be the Tuya app doing the additional processing, or at the very least forwarding set-points/corrections values to the TRV based on the in-room temperature sensors.

PLTorrent, 2021-11-29:
>The Tuya app allows you to add additional temperature sensors and assign them to rooms to help the TRV, in this case it must be the Tuya app doing the additional processing, or at the very least forwarding set-points/corrections values to the TRV based on the in-room temperature sensors.

@supersebbo TBH I have sniffed this TRV with ZBOSS on the original tuya gateway with tuya app and seeing how quiet the device is zigbee-wise I am not so sure it does anything more with tuya app. I'm not sure it is able to have any use from external thermometer, even if you assign one to the same "area" in tuya app as the temp correction settings are in the range -6 to +6 degrees with a full degree step. It is also not possible to send data to device on the room temp endpoint and this is what one would expect in case of such "external thermometer" adjustment.  It even does not report the valve position.

pticon91, 2022-01-13:
Hello guys,

I really appreciate your work here. My _TZE200_yw7cahqs starts working like I would expect it to work, specially with that calibration entity. Anyway is there a way, how to force the quirk to send to TRV decimal numbers? I just tested to set resolution to 0.5deg and set manually the value to like 1.5deg, TRV enregister it. But When trying to do the same using Home Assistant, it sends to TRV just rounded values.

jacekk015, 2022-01-13:
@pticon91 You also have to change resolution of the GUI field
from[line 361]
```
self._update_attribute(self.attridx["resolution"], 1)
```
to
```
self._update_attribute(self.attridx["resolution"], 0.5)
```

pticon91, 2022-01-13:
@jacekk015 , yeah, that's done, but thank you for additional info. It will be definitely handy to have this info here, when someone else would like to achieve the same result.

dieneuser, 2022-04-09:
Since version HA2022.4 the quirk does not work properly for my Hama _TZE200_yw7cahqs & _TYST11_yw7cahqs devices. I had to adjust line 159 in the Saswell.py file to get ZHA to work at all. (change var name from "manufacturer_attributes" to "attributes") 
ZHA detects the device correctly, but the temperature cannot set or get. A repair of the devices did not bring any change.
Without this quirk I have only basic functions - no additional functions like the offset temperature that I need.

zillion42, 2022-10-12:
I have 9 Hama _TZE200_yw7cahqs devices. I have not installed this quirk but judging from what @dieneuser wrote, it's not worth trying at this point. Any chance someone can look into this ? I would really like to get all the modes working. Thanks in advance.

jacekk015, 2022-10-14:
@zillion42 That quirk code wasn't modified to apply HA code changes made in April.
Will try too look at it. It needs to be rewritten in some areas.

zillion42, 2022-10-14:
> @zillion42 That quirk code wasn't modified to apply HA code changes made in April. Will try too look at it. It needs to be rewritten in some areas.

Sounds promising... thx a lot !

jacekk015, 2022-10-15:
Someone need to test the quirk.
I didn't touch quirk programming since over a year, so it's possible that a bugs are present.
[trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9792683/trv_saswell.py.zip)

dieneuser, 2022-10-15:
Hello @jacekk015,

Thank you for the effort.
Your quirk works perfectly. All additional functions are available.

[image]

wtom, 2022-10-16:
For me it also works perfectly.
I own the Hama _TZE200_yw7cahqs thermostat. I just have one question, is it possible to get the "heating" state as a binary sensor, i mean this little flame it shows when the valve is open.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/saswell.js#L46
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L505

If i compare the attributes:
SASWELL_ANTI_FREEZE_ATTR = 0x010A  # [0/1] on/off 266
SASWELL_WINDOW_DETECT_ATTR = 0x0108  # [0/1] on/off 264
saswellWindowDetection: 8,
saswellFrostDetection: 10,
it's matching, so the heating attr should be 0x0103? Does anyone know if this value works in zigbee2mqtt?

jacekk015, 2022-10-16:
@wtom check your HA logs, but first enable debug level
Add to configuration.yaml

```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
and search for:
0x0103 or 0x0203 or 0x0303 etc... 03 needs to be on the end part

wtom, 2022-10-16:
@jacekk015 Thanks for you quick answer. It seems like there is no value reported. :( I can't find anything in the logs.

Then i need to build a binary sensor myself. It looks like, if the target_temp - current_temp > 1.5 degrees, it opens the valve.

zillion42, 2022-10-18:
Hi, 

I also wanted to report back, it works for me too. Some things I have noticed though. 

- The Child Lock can only be disabled from HA but enabled. To enable it I have to press the button on the Thermostat for 5 seconds. 
- I haven't really testet the Limestone Protection, Window detection, Away Mode or Anti Freeze yet. Some of them are not really easy to test.
- Schedule Mode on/off can be enabled and disabled from HA  

@jacekk015 Thanks for fixing the quirk so quickly.

jacekk015, 2022-10-18:
@zillion42 To be honest those functions are useless. After a year I didn't use them. Open window detection works best if you put the TRV to the freezer.

@wtom If you have enabled debug logs. Take out batteries and put them back. Wait 10 minutes and post the logs here. TRV should report all it could. Just post those logs from that 10 minutes part.

zillion42, 2022-10-18:
@offtopic, I couldn't get the code to format properly
I have made some cards, basically the whole day, to make those modes accessible and looking similar to custom:simple-thermostat, just in case anyone is interested.
[image]

  - cards:
      - cards:
          - cards:
              - entity: climate.wohnzimmer_heizung_thermostat_2
                header:
                  name: Wohnzimmer
                type: custom:simple-thermostat
              - type: custom:layout-card
                layout_type: grid
                layout:
                  grid-template-columns: 33% 33% 33%
                  grid-template-rows: auto
                  grid-template-areas: |
                    "main main2 main3"
                    "main main2 main3"
                    "header header header"
                cards:
                  - type: vertical-stack
                    cards:
                      - type: entities
                        entities:
                          - entity: >-
                              number.wohnzimmer_heizung_number_temperature_calibration_2
                            name: Number Temperature Calibration
                    view_layout:
                      grid-area: header
                  - type: custom:button-card
                    entity: switch.wohnzimmer_heizung_switch_9
                    show_name: true
                    name: Anti Freeze
                    show_state: false
                    color_type: card
                    color: rgb(254, 129, 0)
                    icon: mdi:snowflake
                    size: 16%
                    styles:
                      name:
                        - font-size: 15px
                    action: service
                    service:
                      domain: input_boolean
                      action: toggle
                      data:
                        entity_id: switch.wohnzimmer_heizung_switch_9
                  - type: custom:button-card
                    entity: switch.wohnzimmer_heizung_switch_12
                    show_name: true
                    name: Away Mode
                    show_state: false
                    color_type: card
                    color: rgb(254, 129, 0)
                    icon: mdi:logout
                    size: 16%
                    styles:
                      name:
                        - font-size: 15px
                    action: service
                    service:
                      domain: input_boolean
                      action: toggle
                      data:
                        entity_id: switch.wohnzimmer_heizung_switch_12
                  - type: custom:button-card
                    entity: switch.wohnzimmer_heizung_switch_8
                    show_name: true
                    name: Child Lock
[... 311 lines cut ...]
                      domain: input_boolean
                      action: toggle
                      data:
                        entity_id: switch.kuche_heizung_switch_17
                  - type: custom:button-card
                    entity: switch.kuche_heizung_switch_13
                    show_name: true
                    name: Window Detection
                    show_state: false
                    color_type: card
                    color: rgb(254, 129, 0)
                    icon: mdi:window-open-variant
                    size: 16%
                    styles:
                      name:
                        - font-size: 15px
                    action: service
                    service:
                      domain: input_boolean
                      action: toggle
                      data:
                        entity_id: switch.kuche_heizung_switch_13
            type: vertical-stack
        type: custom:swipe-card
    type: horizontal-stack

wtom, 2022-10-19:
@jacekk015 here is the log, it's long because of all the devices.
0x0B77 (climate.heizung_nico_thermostat): Here i took off the batteries. 
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9820040/home-assistant.log)

jacekk015, 2022-10-20:
@wtom There are no valve state attributes in the log.

wtom, 2022-10-20:
@jacekk015 I think it would be a nice addition. But it's very specific for each thermostat? As said, the shit Hama Thermostat goes to idle 2 degrees before "temp set" is reached. :)
An on top of that, the thermostat temperature is of course a lot higher than the room if the heater is on, because it's so close to the heater.

state = 0 if (int(temp_current) >= int(temp_set + 2)) else 1
this would be right in my case.

jacekk015, 2022-10-21:
Added HASS Heating state based on temp difference
[trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9842931/trv_saswell.py.zip)

zillion42, 2022-10-24:
@wtom My hama TRV'S don't stop heating 2° before they reach target temp. They normally overshoot by around 2°
This is a typical scenario for my  _TZE200_yw7cahqs, I set 22° it heats until 24°
[image]
@jacekk015 I will try the valve state tonight

wtom, 2022-10-24:
@zillion42 How hot are your heaters? Because mine run with a very low temperature and i think, in your case it's just a little slow in reacting of the temperature measurement. I even have the confirmation from Hama, that they work like this, with a tolerance of +-1.5 degrees, to save battery by not opening and closing the valves all the time.

So my case, low temperature heaters: Thermostat Temp is 19 degrees. I set target to 22, it takes 1h for the thermostat temp to reach 20.5 degrees (room still colder) and the valve closes, because it's in that range of +-1.5 to the target. (they think, heater is hot, will heat up the room in the next hour. (which it doesn't in my case)

Your case i think, hot temperature heaters: Thermostat Temp is 19.5 degrees. You set target to 22. valve opens. 15min later it checks, temp between 20 and 20.5, keeps valve open. 15min later checks again, temp between 21 and 21.5 and closes the valve. But temp still rising since the heater is very hot.

But for example, mine don't even open like at your 6am case. Temp is 19 and target 20. My valves would keep closed. :(

pticon91, 2022-10-24:
Agree with @zillion42  , from my personal measurements it looks like the threshold is 1.8° . I tried to contact Hama to get more technical details about that TRV, but unfortunately as they are not manufacturer, they do not have such kind of information (which is sad as they are selling this Saswell devices under their name). However they've been very kind and at least tried to help. That's not a case of Saswell. Conversation with Saswell was very cold and only answer I got from them was, that technical details/behaviour of TRV is "Company's secret" - mind blowing.

Also what you are saying @wtom  is correct, TRV opens just a little bit when needed - I've measured that the TRV has probably like 5 stages of opening (don't have the percentage numbers anymore, but it wasn't even very accurate). So when heating is needed, it opens a bit, and if temperature does not raise much in certain time frame it opens the valve bit more and this repeats till fully opened if necessary. I was experiencing same issue in my flat as you do with slow water run in heater, it's most probably calcium blockage.

However because of that initial almost 2° hysteresis, the TRV is useless unless you control the logic of target through some external automation.

wtom, 2022-10-24:
@pticon91 
> However because of that initial almost 2° hysteresis, the TRV is useless unless you control the logic of target through some external automation.

Yes, i control it with the better_thermostat integration with an external temperature sensor and some extra lines of code to keep the target temp always 2 degrees higher than the temp the thermostat shows. Like this it works.

zillion42, 2022-10-24:
@wtom I have a Buderus GB172/24K. Its a dhw (domestic hot water) flow type. Which means I have no storage tank to save hot water. I also have an EMS-ESP32 connected to the EMS bus, so I have all kinds of data on it. In general yes, my Heating Circuit (hc) is pretty hot, at the moment the hc1 current flow temperature measures 81.5°C in the boiler but it varies depending on hot water use. More informative is the target flow temperature. You can clearly see the night time reduction.
[image]
I also noticed that some of my heaters are much more prone to overshooting than others. Logically smaller rooms overshoot more often than larger rooms, so its not really easy to tell. I guess +/- 1.5° or 1.8° it really depends on the room size. But the general behavior of overshooting in my case and not reaching target temp in your case can probably be explained by the general temperature of the heating circuit. more on the hot side vs more cold side, as you wrote above. By the way, I'm also considering using ["better thermostat"](https://github.com/KartoffelToby/better_thermostat) to measure the temperature on the other side of the room, like you do

wtom, 2022-10-24:
@zillion42 My water flow temperature at the exit of the boiler (i also measure it externally with a d1 mini) is, at the moment 27°C. On colder days, with like 5°C outside, its around 35°C.

zillion42, 2022-10-24:
@offtopic, I'm really trying to make sense of my gas consumption, my hot water usage and my thermostats. The more I look at the graphs, the more I feel like, everything lives. Nothing really makes sense to me. If anyone feels like it, please analyze the following.
[image]

wtom, 2022-10-25:
@zillion42 I think your boiler is configured wrong. I don't know who did it and why, but it's running way too hot. Those are my settings not even when it's -10°C outside and our house is also not very efficient. And your boiler seems to be a bit more modern than ours. You said you don't have a water tank, then i would start there and check on which temperature is the warm water setting. I think around 40°C should be enough without losing any comfort. We have a water tank and i boil it to 55°C, but just because of the Legionellen (i don't know the english word).

Next check the "Raum Soll Temperature" that is set on the boiler and put it to 21 or 22. And then check the Heizkurve (Neigung and Niveau). Since you have a boiler with Brennwerttechnik, you should set it, so it's not much hotter than 55°C when it's like -10°C outside i think. But that also depends on your house, you need to test a lot. But normally you don't need a Vorlauftemp as you have it right now, also because you have like 15-18°C outside during the day.

wtom, 2022-10-25:
> Added HASS Heating state based on temp difference [trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9842931/trv_saswell.py.zip)

@jacekk015 Thank you, works perfect for me.
To note for everyone else, the +2 degrees are added in the code, so it might changes the state to idle too early for you.

jacekk015, 2022-10-25:
> We have a water tank and i boil it to 55°C, but just because of the Legionellen (i don't know the english word).

To kill Legionella you not need to keep 55°C, to be honest 50-55 degrees are ideal params to born Legionella!
To really kill Legionella you should occasionally boil water to 75°C
It has to be temp between 70-80°C

My Vaillant boiler has ability to make it occasionally, once on two weeks. Normally I keep 42°C.
Boiler overshots that to 44°C
[image]
To be hones I've talked with service man when boiler was installed and that function Isn't needed if you normally use tap water.
You can manually heat it when back from holidays.

wtom, 2022-10-25:
[https://en.wikipedia.org/wiki/Legionella#Heat](https://en.wikipedia.org/wiki/Legionella#Heat)
We also just heat it once a day, we don't keep the water. But it's actually a good point. Now that i have built the integration, that i send commands to the boiler, i could warm up the water to a lower temperature, and just maybe once a week boil it to 60°C.

jacekk015, 2022-10-25:
> At 50 °C (122 °F) – 90% die in 80–124 minutes, depending on strain ([D](https://en.wikipedia.org/wiki/D-value_(microbiology)) = 80–124 minutes)

Loss of resources. Even once/day is too much, since after boil taps give boil water - home users can burn themself.
https://www.vaillant.co.uk/downloads/product-manuals/vrc-700/vrc-700-installation-instructions-1968307.pdf
[image]

wtom, 2022-10-26:
> > Added HASS Heating state based on temp difference [trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/9842931/trv_saswell.py.zip)
> 
> @jacekk015 Thank you, works perfect for me.

Just one thing. If you turn off the thermostats, it's swiching the state to idle/heating.

jacekk015, 2022-10-26:
Need logs from that operation to service it. Some attribute needs to send something unique so I can treat it as OFF.

zillion42, 2022-10-26:
Hi again,

yeah Legionella came to mind for me too. I didn't change any settings on my boiler, it's actually on default settings. If I look at my Gas consumption, the previous two years, it is very much in the normal to slightly high range.
[image]
The night time reduction works again, I don't know whether the EMS-ESP32 messes with my settings. The only setting I play with is the Set-Point or target temperature, playing with it after 23:00 seems to disable my night time reduction completely. 

The goal, of course, is to save energy, not to waste more.

niebergaeller, 2022-12-01:
After upgrade to 2022.11.5, I got this error from the quirk:
  ...
    File "/config/custom_zha_quirks/trv_saswell.py", line 21, in <module>
      from zhaquirks.tuya.ts0601_trv_sas import BATTERY_REPORTED
   ImportError: cannot import name 'BATTERY_REPORTED' from 'zhaquirks.tuya.ts0601_trv_sas' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/ts0601_trv_sas.py)

I had to comment out line 21 to make ZHA start correctly.

zillion42, 2022-12-02:
Release 2022.11.5 - November 30
Bump ZHA dependencies ([@puddly](https://github.com/puddly) - [#82509] https://github.com/home-assistant/core/pull/82509

zillion42, 2022-12-02:
Thx for all your work @jacekk015, would you please fix this ?
[image]
[image]
[image]
[image]

jzielke84, 2022-12-04:
> After upgrade to 2022.11.5, I got this error from the quirk: ... File "/config/custom_zha_quirks/trv_saswell.py", line 21, in from zhaquirks.tuya.ts0601_trv_sas import BATTERY_REPORTED ImportError: cannot import name 'BATTERY_REPORTED' from 'zhaquirks.tuya.ts0601_trv_sas' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/ts0601_trv_sas.py)
> 
> I had to comment out line 21 to make ZHA start correctly.

Based on https://github.com/zigpy/zha-device-handlers/commit/c2f2b52f054cccb99bcd77d657e7ac9cc103a7ad this seems to fix the problem:
```diff
--- trv_saswell.py.bak	2022-12-04 16:44:57.012227268 +0100
+++ trv_saswell.py	2022-12-04 16:45:59.016507666 +0100
@@ -18,7 +18,7 @@
     TuyaThermostatCluster,
     TuyaUserInterfaceCluster,
 )
-from zhaquirks.tuya.ts0601_trv_sas import BATTERY_REPORTED
+from zhaquirks.tuya.ts0601_trv_sas import SASSWEL_BATTERY_LOW_ATTR
 from zigpy.profiles import zha
 from zigpy.zcl import foundation
 from zigpy.zcl.clusters.general import (
```

BigBadPhysicist, 2022-12-07:
Hello,

I am running a Hama TRV with the identification _TZE200_h4cgnbzg

I added the TRV as a TS0601 in line 467 and can read the Temp and Target value from the TRV.

But if I toggle any switches or try to set the target temperatur nothing changes.
ZHA is reporting back that the value is set but there seems to be some callback and then the temp is set back to the value bevor.

2022-12-07 09:07:29.022 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=climate, service=set_temperature, service_data=temperature=16, device_id=['7d61eb4c558255c1eb2b325e8b9cc24a']>

2022-12-07 09:07:29.029 DEBUG (MainThread) [zhaquirks.tuya] [0x0ef3:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 1600 to custom {615: 160}

2022-12-07 09:07:29.030 DEBUG (MainThread) [zigpy.zcl] [0x0EF3:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=111, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)

2022-12-07 09:07:29.031 DEBUG (MainThread) [zigpy.zcl] [0x0EF3:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=111, command_id=615, function=0, data=[4, 0, 0, 0, 160]))

2022-12-07 09:07:29.031 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0EF3), dst_ep=1, source_route=None, extended_timeout=False, tsn=111, profile_id=260, cluster_id=61184, data=Serialized[b'\x05A\x11o\x00\x00og\x02\x00\x04\x00\x00\x00\xa0'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2022-12-07 09:07:29.032 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (30, 112, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x0ef3 endpoint=1>, 260, 61184, 1, b'\x05A\x11o\x00\x00og\x02\x00\x04\x00\x00\x00\xa0', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)

2022-12-07 09:07:29.052 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 112]

2022-12-07 09:07:29.052 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 112]

2022-12-07 09:07:30.292 DEBUG (MainThread) [zigpy_deconz.api] Received command mac_poll[5, <DeconzAddress address_mode=AddressMode.NWK address=0x0ef3>, 255, -45]

2022-12-07 09:07:30.331 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]

2022-12-07 09:07:30.332 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]

2022-12-07 09:07:30.333 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)

2022-12-07 09:07:30.349 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x0ef3>, 1, 260, 61184, b'\x18o\x0b\x00\x83', 0, 175, 255, 152, 153, 0, 0, -46]

2022-12-07 09:07:30.350 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x0ef3>, 1, 260, 61184, b'\x18o\x0b\x00\x83', 0, 175, 255, 152, 153, 0, 0, -46]

2022-12-07 09:07:30.351 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0EF3), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=61184, data=Serialized[b'\x18o\x0b\x00\x83'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-46)

2022-12-07 09:07:30.352 DEBUG (MainThread) [zigpy.zcl] [0x0EF3:1:0xef00] Received ZCL frame: b'\x18o\x0b\x00\x83'

2022-12-07 09:07:30.353 DEBUG (MainThread) [zigpy.zcl] [0x0EF3:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=111, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)

2022-12-07 09:07:30.354 DEBUG (MainThread) [zigpy.zcl] [0x0EF3:1:0xef00] Decoded ZCL frame: SaswellManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)

2022-12-07 09:07:30.356 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0x0ef3>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'186f0b0083'

[... cut ...]
2022-12-07 09:07:30.441 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 112: 00

2022-12-07 09:07:30.444 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x70 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x0ef3 endpoint=1>, status: 0x00

2022-12-07 09:07:30.445 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0EF3:1:0x0201]: wrote {'occupied_heating_setpoint': 1600} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]

[image]

Am I missing something?

DaClemens, 2022-12-14:
Is there interest in adding support for _TZE200_h4cgnbzg (Hama, [this device](https://www.hama.com/00176593/hama-wlan-heating-control-2-x-smart-radiator-thermostats-+-central-control), recently bought)?

I have two around, willing to debug but got little experience with Zigbee yet.

When i try to add the device using the quirk from [here](https://github.com/zigpy/zha-device-handlers/issues/1189#issue-1065350787), plus two diffs mentioned above, it pairs successfully, but I see neither data nor am able to issue commands (all data is Unknown, commands get sent out, but I assume the device doesn't understand them.

My patches to the original post:
```diff
@@ -1,7 +1,7 @@
 """Map from manufacturer to standard clusters for thermostatic valves."""
 import logging
 from typing import Optional, Union
-from zhaquirks.tuya.ts0601_trv_sas import BATTERY_REPORTED
+from zhaquirks.tuya.ts0601_trv_sas import SASSWEL_BATTERY_LOW_ATTR
 
 from zigpy.profiles import zha
 import zigpy.types as t
@@ -156,7 +156,9 @@
         global SaswellManufClusterSelf
         SaswellManufClusterSelf[self.endpoint.device.ieee] = self
 
-    manufacturer_attributes = {
+    # This is deprecated now
+    #manufacturer_attributes = {
+    attributes = {
         SASWELL_ONOFF_ATTR: ("on_off", t.uint8_t),
         SASWELL_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t),
         SASWELL_ROOM_TEMP_ATTR: ("current_room_temp", t.uint32_t),
@@ -451,6 +453,7 @@
             ("_TZE200_9gvruqf5", "TS0601"),
             ("_TZE200_zuhszj9s", "TS0601"),
             ("_TZE200_zr9c0day", "TS0601"),
+            ("_TZE200_h4cgnbzg", "TS0601"),
         ],
         ENDPOINTS: {
             1: {
 ```

Attached here is also the debug log (filtered to this device) from adding the device to trying to change a couple settings: [debug.log](https://github.com/zigpy/zha-device-handlers/files/10230248/debug.log)

StrangeD0s, 2022-12-18:
> Is there interest in adding support for _TZE200_h4cgnbzg (Hama, [this device](https://www.hama.com/00176593/hama-wlan-heating-control-2-x-smart-radiator-thermostats-+-central-control), recently [...]

I also have two Hama _TZE200_h4cgnbzg devices which I am very much interested in getting to work like my _TZE200_yw7cahqs. So any help there would be appreciated!

lukasmd1989, 2022-12-23:
Recently received the "_TZE200_h4cgnbzg" version of this TRV. It is not correctly working in HomeAssistant using ZHA. No custom quirks solve all issues. 

This is a similar issue about the same model: #1901 

ZHA diagnose file: https://justpaste.it/4o5gv

What can I do to help you create a quirk for this model?

jacekk015, 2022-12-27:
@jzielke84 @niebergaeller @zillion42 
This line wasn't ever needed.
Below quirk with removed that import.

That import was a very old part of the code.
SASSWEL_BATTERY_LOW_ATTR is not needed also.
You have SASWELL_BATTERY_ALARM_ATTR which points to the 0x569 attribute for battery alarm.
You don't need to import that from anywhere.

[trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/10310990/trv_saswell.py.zip)

jacekk015, 2022-12-27:
@BigBadPhysicist @DaClemens @StrangeD0s @lukasmd1989 
Added your device to the quirk.
Check it and post what doesn't work.
Before that enable debug level in configuration.yaml and restart HA
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
Then do some action on the TRV/HA and post the HA full logs.
I need logs 2-3 minutes before action start and minimal 5 minutes after action end.
Zigbee is slow.

Best would be [after adding quirk and enabling debug level] to remove the TRV from HA.
Wait a bit. Pair it like a new.
Post the logs from pairing process and 10 minutes after.
TRV after pairing sends all data to HA.

[trv_saswell.py.zip](https://github.com/zigpy/zha-device-handlers/files/10311043/trv_saswell.py.zip)

RolandoMagico, 2022-12-28:
Hi @jacekk015 ,
thanks for the update. I used the latest file you provided wit _TZE200_h4cgnbzg.
What I did:
- Removed the device from HA
- Enalbed logging as described, restarted HA
- Paired the devlice again, that's how it looks like afterwards (sorry that is is in german):
[image]
- Switch to heating, but it switches back to "off" after a few seconds
- Toggled the available switches in the order they appear in the above screenshot (not sure which one is for which function), changed number temperature calibration
- Let HA run for further 5 minutes

Log:
[home-assistant_2022-12-28T07-14-17.158Z.log](https://github.com/zigpy/zha-device-handlers/files/10312749/home-assistant_2022-12-28T07-14-17.158Z.log)

BR
Roland

jacekk015, 2022-12-29:
I don't see any switching activities in the logs. That's a long used quirk and probably it works normally - like originally users wanted. Z2M uses also same code for those TRVs.
Firstly - I cannot change name of the switches from the quirk - that's the limitation - often bad when you have plenty of switches.

Switches are PROBABLY in order:
- SaswellWindowDectection
- SaswellChildLock
- SaswellAntiFreezeDectection
- SaswellLimescaleProtectionDectection
- SaswellScheduleModeDectection
- SaswellAwayModeDectection
You need to compare that with manual and TRV device settings. Best is to switch it on TRV and observe which switch in HA changed.

For more debugging please try to check which one is a Child Lock. Lets use one switch only.
Change it from TRV, then try from HA and post the logs and 10 minutes after.
Same for enabling Heating in HA. Logs of action and 10 minutes after.

BTW.
TRV sent to HA data that Heating is off:
`
2022-12-28 08:00:59.028 DEBUG (MainThread) [zigpy.zcl] [0xCE7F:1:0xef00] Received command 0x02 (TSN 51): set_data_response(param=Command(status=0, tsn=1, command_id=357, function=0, data=[1, 0]))
2022-12-28 08:00:59.030 DEBUG (MainThread) [zhaquirks.tuya] [0xce7f:1:0xef00] Received value [0] for attribute 0x0165 (command 0x0002)
`

0x0165 is an attribute for on/off state
`
SASWELL_ONOFF_ATTR = 0x0165  # [0/1] on/off 357
`

pticon91, 2022-12-30:
Hi @jacekk015 , probably this might help:

Filtered log: [debug_TZE200_h4cgnbzg.log](https://github.com/zigpy/zha-device-handlers/files/10322697/debug_TZE200_h4cgnbzg.log)
Full log: [debug_hass.log](https://github.com/zigpy/zha-device-handlers/files/10322699/debug_hass.log)

ACTION | TIME (approx) | RESULT
--- | --- | ---
[H] Pairing | 22:57 | OK
[P] Thermostat > Heat, 22deg | 23:08 | OK
[H] Target > 20deg | 23:19 | Fail
[H] Thermostat > Off | 23:29 | Fail
[P] Child Lock > On | 23:49 | Change not visible in HASS
[P] Child Lock  > Off | 00:00 | Change not visible in HASS
[H] Switch 1 > Off | 00:12 | Switch Off in HASS, unable to check the change on TRV
[H] Switch 1 > On | 00:22 | Switch On in HASS, unable to check the change on TRV
[H] All Switches > OFF | 00:33 | All Switches Off in HASS, unable to check the change on TRV
[H] Target > 20deg | 00:43 | Fail

[P] = Physical / TRV
[H] = Home Assistant

[24 more replies]

closed 2023-07-29 by github-actions[bot]