zigpy/zha-device-handlers #1193: [Device Support Request] [Enhancement request] Swapped V and A dividers for TS011F smart plug from _TZ3000_w0qqde0g

MayeulC, 2021-11-29:
**Is your feature request related to a problem? Please describe.**
Amperage and voltage dividers seem to be switched around on the plugs I just bought. Here is a screenshot of the device in homeassistant (dev docker image ba3d1d8a811c4a9aac6c2c1d340912fb775a947752933a4276e0291c8c43f9b1) with a relatively powerful LED light bulb connected. Wattage seems correct:

[image]

Device is a TS011F by _TZ3000_w0qqde0g 

**Describe the solution you'd like**
Quirks file: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py

* Make the voltage divider 1000 lower (new divider=1000*old)
* Make current divider 1000 times higher

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_w0qqde0g",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

* Pairing log from a normal pairing. The log kept scrolling, so I waited before copying it here: https://gist.github.com/MayeulC/8cc966783fd76789e62fe57c8150f6d4#file-pairing2-log
* Other pairing log. I think that one corresponds to the one where there seemed to have been an issue, with a lot more controls and endpoints appearing afterwards, some with unknown values. I could not reproduce this, but it might have been due to my activating the plug during the pairing sequence: https://gist.github.com/MayeulC/8cc966783fd76789e62fe57c8150f6d4#file-pairing1-weird-log

**Additional context**
[Item on aliexpress](https://www.aliexpress.com/item/1005003444957152.html).

sophof, 2022-02-18:
> Same for _TZ3000_gjnozsaz

I have just received three of these and all I can do is pair them, but I get no functionality at all. (no power measuring, no switching). So if you say same, do you mean you do get functionality but just have the issue as mentioned here? Device looks the same as in the aliexpres link btw.

dentra, 2022-02-20:
@sophof yes, I mean only issue mentioned here, all other functions are works. And I fix this issue by copying `ts011f_plug.py` to cunstom_zha_quirks folder and modifying `TuyaZBElectricalMeasurement` class to
```python
class TuyaZBElectricalMeasurement(CustomCluster, ElectricalMeasurement):
    """Divides the Current for tuya."""

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603
    AC_VOLTAGE_MULTIPLIER = 0x0600
    AC_VOLTAGE_DIVISOR = 0x0601
    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000, AC_VOLTAGE_MULTIPLIER: 1, AC_VOLTAGE_DIVISOR: 1}
```
And now all sensors shows correct values.

sophof, 2022-02-21:
Strange, I'm completely confused why mine don't work then. I bought 3 and all can be included to zha, but then do basically nothing.

closed 2022-08-27 by github-actions[bot]

MayeulC, 2022-08-28:
This is still an issue:

[image]

Dear bot, I hate you. Sincerely,