zigpy/zha-device-handlers #1194: Tuya curtain motors TM616EGZT Calibration feature missing

PremiumUsername, 2021-11-30:
There is no way to set the end limits or the direction of the motor with the physical buttons, so calibration needs to be done using zigbee. Would it be possible to add the calibration cluster that exist for some other Tuya  motors (e.g. TS130F) to the TM616EGZT?

skygrip, 2022-03-12:
Ive had a look at the Tuya IOT builder, and i believe the calibration is controlled using Tuya command "border" with DP ID 16. This command takes a "Enum" data type, is R/W, and has the properties "Enum Value: up, down, up_delete, down_delete, remove_top_bottom"

I haven't had any real experience with writing ZHA quirks to expose these settings, but ill have a look

MattWestb, 2022-03-12:
I think the defrnition of the commands shall being placed in this part in the INIT:
https://github.com/zigpy/zha-device-handlers/blob/08f99237f165560921845abbc6b810cb5a0cd4d8/zhaquirks/tuya/__init__.py#L96-L102 or some line above that have some that looks like attributes and commands.

The device class is here and more classes below but i dont knowing  how to implementing it then outs out of my skills.
https://github.com/zigpy/zha-device-handlers/blob/08f99237f165560921845abbc6b810cb5a0cd4d8/zhaquirks/tuya/__init__.py#L929-L931

skygrip, 2022-03-12:
I managed to get the following commands to work using the developer tools > Services to set a "border" in Home Assistant. As you dont usually have to run the calibration often, this might be a good workaround for the moment. Im making use of the custom command function in the ZHA quirk. It worked on my device with firmware TZE200_gubdgai2

Grab the MAC for the device from your devices page, substitute it in the YAML, and then try the below codes. Remember to click "Go To YAML Mode" to paste the code.

Set Upper Bounds

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: INSERT YOUR DEVICE MAC HERE
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0410
      - 0
      - [1,0x0000]
```

Set lower bound

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: INSERT YOUR DEVICE MAC HERE
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0410
      - 0
      - [1,0x0001]
```

Delete upper bounds

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: INSERT YOUR DEVICE MAC HERE
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0410
      - 0
      - [1,0x0002]
```

Delete lower bounds

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: INSERT YOUR DEVICE MAC HERE
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0410
      - 0
      - [1,0x0003]
```

0x04 is data type Enum and then 0x10 is DP ID "Border". The hex in the last argument most likely corresponds to "up, down, up_delete, down_delete, remove_top_bottom"

Im not entirely sure how to make this more user friendly or expose it in the UI just yet, but i thought i would post this code in the hope it might be of interest to others. No guarantee it would work on your device though, test it before you install it so you don't destroy your blinds.

PremiumUsername, 2022-03-13:
> I managed to get the following commands to work using the developer tools > Services to set a "border" in Home Assistant.

Confirmed solution! ....works brilliantly. Thank you! 

How did you manage to find that command?

PremiumUsername, 2022-03-13:
> Ive had a look at the Tuya IOT builder. 

I don't have a Tuya developer account and I never found a public API. What other interesting settings are there to play with?
I assume there is a way to reverse the motor direction, so the buttons work the other way around.

skygrip, 2022-03-30:
Set motor direction Forward

Valid for at least the _TZE200_gubdgai2 variant

```
service: zha.issue_zigbee_cluster_command
data:
ieee: b4:e3:f9:ff:fe:af:58:c1
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0405 # DIP command 5, motor direction
      - 0
      - [1,0x0000] # Forward direction
```

Set Motor Direction Back

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: b4:e3:f9:ff:fe:af:58:c1
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x6
  args:
      - 0
      - 0
      - 0x0405 # DIP command 5, motor direction
      - 0
      - [1,0x0001] # Back direction
```

smarthomelawyer, 2022-03-30:
Oh nice! Any idea if there is a setting for increasing/decreasing the speed of the motor?

closed 2022-10-03 by github-actions[bot]

mikeysteele, 2024-09-19:
I'm trying to run those set motor direction commands in the latest version of HASS and I keep getting this error: `Failed to perform the action zha.issue_zigbee_cluster_command. Unknown error`

This is the relevant error from the logs:
```
Logger: homeassistant.helpers.script.websocket_api_script
Source: helpers/script.py:525
First occurred: September 18, 2024 at 7:50:19 PM (30 occurrences)
Last logged: 4:48:13 PM

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: 6
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: 3
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 525, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 763, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 726, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1099, in _async_admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/websocket_api.py", line 1389, in issue_zigbee_cluster_command
    await zha_device.issue_cluster_command(
  File "/usr/local/lib/python3.12/site-packages/zha/zigbee/device.py", line 904, in issue_cluster_command
    [field.name for field in commands[command].schema.fields],
                             ~~~~~~~~^^^^^^^^^
KeyError: 6
```

Any ideas?