zigpy/zha-device-handlers #1199: [Device Support Request]

ferrdo4, 2021-12-04:
**Is your feature request related to a problem? Please describe.**
Got an Immax neo smart 07732l TRV (https://zigbee.blakadder.com/Tuya_HY367.html) paired, but got no control over it.

**Describe the solution you'd like**
Would it be possible to add support for this TRV

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
_TZE200_4eeyebrt TS0601

**Additional context**
I was trying around to add it but im noob in this

MattWestb, 2021-12-04:
I have making 2 files with added your device ID that shall making it  working.
[ts0601_trv.zip](https://github.com/zigpy/zha-device-handlers/files/7654014/ts0601_trv.zip)
First you need configuring local quirks in HA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and then copy the unzipped `ts0601_trv.py` the your new mande local quirk forder.

Then you also need copy the `climate.py` in to the HA docker container or the presets is not working. 
First copy the unzipped `climate.py` to your HA configuration directory.
Then copy it in to the container with this commands from the HA docker container CLI:
```
cp /config/climate.py  /usr/src/homeassistant/homeassistant/components/zha/climate.py 
```
and restarting HA and you shall getting one TRV in your device card.

Instruction how open one CLI in HA docker container https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install.

This is the normal quirk but one user is making one new with more setting that you can trying if you like if this is working well.

MattWestb, 2021-12-04:
By the was CN-Hysen (manufacturer of this device) have before using the same device ID for all devices but this is the 3 new in 2 weeks so its looks they have changing strategy.

Do you have one brand and model name of this device  and also one link if you can finding it.

ferrdo4, 2021-12-04:
It is working now, many thanks for your help

closed 2021-12-04 by ferrdo4

MattWestb, 2021-12-04:
## Great !!
I making one PR for adding it in ZHA so you and other user dont need doing the addin after updating HA.

MattWestb, 2021-12-04:
If you like testing one new quirk with more functions look at this post https://github.com/zigpy/zha-device-handlers/issues/1178#issuecomment-980294492.
Its need more testing but is trying getting all out if the TRV in ZHA GUI.

(its needs the `climate.py` that you have installing but you shall deleting the custom quirrk `ts0601_trv.py` if you is using the new one so you dont  getting collisions)

ferrdo4, 2021-12-04:
Sure ill look into it im quite new to HA and ZHA, but i would like to get into it. thx once more

MattWestb, 2021-12-04:
I like brave users !!!

If you is testing you can always going back to the first version if you like and shall not being any problems only need restart HA between :-)))

ferrdo4, 2021-12-05:
ive tried it quicky - changed your ts0601_trv.py to ts0601_trv_moes.py(also tried to rename it to ts0601_trv.py), but then no thermostat was detected and got similar logs as above ... i probably messed up something.

MattWestb, 2021-12-05:
Sorry i was missing you need adding your device ID after line 1253 like this:
```
            ("_TZE200_4eeyebrt", "TS0601"),
```
or its not being loaded ( jacekk015  was not knowing your device then doing the new quirk).

Add the line and restart AH and you shall getting your TRV on the device card :-)))

ferrdo4, 2021-12-05:
Yes, you are right, after adding that line quirk is working fine and all settings params are listed. so far its working fine. thx

ferrdo4, 2021-12-13:
@MattWestb hello, sorry to bother you but im having good time playing around with HA and so, i am trying to control this TRV but with external thermometer .... so far i figured out to be able to do it i need to set [generic_thermstat](https://www.home-assistant.io/integrations/generic_thermostat/) an it needs to have [switch](https://www.home-assistant.io/integrations/switch.template/) ... what i dont know is what would be a way to control this TRV as an on off switch or if it is even possible .... or i just go totally wrong way how to do this :)

MattWestb, 2021-12-13:
I have not trying but i have reading some user have using the "external thermometer" and comparing it with your temperature you like have and updating the setpoint temperature from the diference you is getting. So the external temp is 22° and you like have 24° and the setpoint temperature is 23° then you shall sending one new setpoint temperature that is 2° higher = new setpoint temperature shall being 25°.

If like switching it off sed one setpoint 5° (i think its the mini temp) and on 30° (if its the max the TRV can handling).

On the device card you have "Automations" on the right side and one ` + ` with all automations that ZHA is suggesting to use but  on my i missing the set setpoint temperature so must doing it from normal automation action and call clima service and setting the new temperature in the device.

Also possible changing the working mode of the TRV (presets in ZHA) to away if not like to heating and manual then like heating with one setpoint temperature you like.

I dont knowing your TRV then my Siter is having much lesser functions and im using them for keeping the temperature 24/7 and dont like overheating and returning the still warm  weather. My ideal is warm upper part of the heating and cold water returned so not need paying for heating i dont using.

ferrdo4, 2021-12-13:
i understand, so i shall set up the switch template to set like 30 as on, and 5 as off ... did not thought about it this way ;) ... i actually made(with a help of internet) a simple py script with HACS to control the thermostat ... but i was looking for some attribute how to handle this with generic thermostat. thx for pointing out this option.

MattWestb, 2021-12-13:
One way is:

Action type
Device
 
Device
TYST11 TRV
 
Action
Change preset on TYST11 TRV thermostat
preset_mode
- none
- away
- schedule

My TRV is only having 3 presets and none = manual mode.

Or changing the target temperature:

Action type
Call service
 
Service
Climate: Set temperature
Set target temperature of climate device.

Targets
What should this service use as targeted areas, devices or entities.

Temperature
New target temperature for HVAC.

ferrdo4, 2021-12-15:
i found a strange behavior with "analog_output_temperature_calibration" on moes quirk the set bar gives user possibility to set temperature calib in range <-9, 9>... but it gets somehow rounded <-9, 5> will be set to value -50 (-0.5C) <-4, 4> = 0 and <5, 9> is 50 (+0.5C) when i try to set it in developer tool to -100 (expecting -1C) it will be set back to value in device options ... i would like to set this calib based on difference of trv current temp and ext. sensor current temp :)

MattWestb, 2021-12-15:
Great finding and the quirk is not much tested so very sure its more to correcting.

Can you puting the findings in the "main issue" https://github.com/zigpy/zha-device-handlers/issues/1178 so it can being fixed if its possible (if its the device its not possible if its the quirk its easy if cn coding).

PR #1201 (2021-12-10): Add 2 more CN-Hysen TRVs
Add 2 more CN-Hysen TRVs "_TZE200_pvvbommb" and "_TZE200_4eeyebrt".
Both is needing updated presets in ZHA for working OK.

Fixes #1199

PR for adding presets in ZHA https://github.com/home-assistant/core/pull/61002 that  must being merged for getting the device working OK in ZHA.