zigpy/zha-device-handlers #1203: Support for LIDL Livarno home staande led lamp zigbee

12christiaan, 2021-12-05:
**Is your feature request related to a problem? Please describe.**

This LED Lamp is discovered in HA as a generic light (no Quirk). 
- ON/OFF works Okay
- DIM/Level works Okay
- RGB color capability is reported, but not supported by this lamp. it only CCT. 
- Color temperaturedoes not work correct.
When i change the color temperature, the LED does change. but not correct. 
when putting it to minimal it is cold white (with little bit warm) and putting it to MAX, it become full cold.
While i expect it to be warm (yellow, orange). 

LAMP website: https://www.lidl.nl/p/livarno-home-staande-led-lamp-zigbee-smart-home/p100335194

**Describe the solution you'd like**
1.  I like to have the RGB detection removed, so it does not show to change the color.
2.  I like to have the Color temperature working as it should be.

For solution (1. I have added in the file **/zhaquirks/lidl/cct.py** the following signature:
signature = {  MODELS_INFO: [  ("_TZ3000_8uaoilu9", "TS0502A")
That solves the RGB problem.

For solution (2, i have no idea where to even start.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_8uaoilu9",
  "model": "TS0502A",
  "class": "zhaquirks.lidl.cct.CCTLight"
}`

**Additional context**
If you need any other info / logging, just let me know.

12christiaan, 2021-12-05:
And the pairing log.
`New device 0x3876 (cc:86:ec:ff:fe:13:55:92) joined the network
[0x3876] Scheduling initialization
Device 0x3876 (cc:86:ec:ff:fe:13:55:92) joined the network
[0x3876] Scheduling initialization
[0x3876] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x84v8\x92U\x13\xfe\xff\xec\x86\xcc\x8e'
[0x3876:zdo] ZDO request ZDOCmd.Device_annce: [0x3876, cc:86:ec:ff:fe:13:55:92, 142]
Tries remaining: 3
[0x3876] Requesting 'Node Descriptor'
Tries remaining: 2
[0x3876] Extending timeout for 0x47 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'G\x00v8\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0x3876] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x3876] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'I\x00v8\x02\x01\xf2'
[0x3876] Discovered endpoints: [1, 242]
[0x3876] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x3876:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'K\x00v8\x1c\x01\x04\x01\x0c\x01\x01\x08\x00\x00\x03\x00\x04\x00\x05\x00\x06\x00\x08\x00\x00\x03\x00\x10\x02\x19\x00\n\x00'
[0x3876:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=268, device_version=1, input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096], output_clusters=[25, 10])
[0x3876:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'M\x00v8\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x3876:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x3876:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=79 command_id=Command.Read_Attributes_rsp>
[0x3876] Read model 'TS0502A' and manufacturer '_TZ3000_8uaoilu9' from <Endpoint id=1 in=[basic:0x0000, identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, level:0x0008, light_color:0x0300, lightlink:0x1000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x3876] Discovered basic device information for <Device model='TS0502A' manuf='_TZ3000_8uaoilu9' nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=True>
Device is initialized <Device model='TS0502A' manuf='_TZ3000_8uaoilu9' nwk=0x3876 ieee=cc:86:ec:ff:fe:13:55:92 is_initialized=True>
Checking quirks for _TZ3000_8uaoilu9 TS0502A (cc:86:ec:ff:fe:13:55:92)
Considering <class 'zhaquirks.lidl.cct.CCTLight'>
Found custom device replacement for cc:86:ec:ff:fe:13:55:92: <class 'zhaquirks.lidl.cct.CCTLight'>
device - 0x3876:cc:86:ec:ff:fe:13:55:92 entering async_device_initialized - is_new_join: True
device - 0x3876:cc:86:ec:ff:fe:13:55:92 has joined the ZHA zigbee network
[0x3876](TS0502A): started configuration
[0x3876:ZDO](TS0502A): 'async_configure' stage succeeded
Error handling '_save_attribute' event with (cc:86:ec:ff:fe:13:55:92, 1, 0, 4, '_TZ3000_8uaoilu9') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (cc:86:ec:ff:fe:13:55:92, 1, 0, 5, 'TS0502A') params: FOREIGN KEY constraint failed
[0x3876:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x3876:1:0x0300]: bound 'light_color' cluster: Status.SUCCESS
[0x3876:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0x3876:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Configure_Reporting_rsp>
[0x3876:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x3876:1:0x0006]: finished channel configuration
[0x3876:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=89 command_id=Command.Configure_Reporting_rsp>
[... cut ...]
[0x3876:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=110 command_id=Command.Read_Attributes>
[0x3876:1:0x000a] ZCL request 0x0000: [[7]]
Error handling '_save_attribute' event with (cc:86:ec:ff:fe:13:55:92, 1, 768, 61443, <Array type=AnonymousLVList, value=[]>) params: Error binding parameter 4 - probably unsupported type.
Error handling '_save_attribute' event with (cc:86:ec:ff:fe:13:55:92, 1, 768, 3328, [0, 0, 0, 0, 200, 0]) params: Error binding parameter 4 - probably unsupported type.`

MattWestb, 2021-12-05:
You is getting errors on some strange not existing cluster attributes that i think is being used with the LIDL remote for direct control of the light without ZBGW but is shall not being any problems.

The color temperature you can read the min and max on the color cluster in the `color_temp_physical_min (id: 0x400b)` and `color_temp_physical_max (id: 0x400c)`.
Then try reading and changing the CT on attribute `color_temperature (id: 0x0007)`.

I think its not helping but  it can being you is finding the value is out  of range and the device is making strange things.

12christiaan, 2021-12-05:
when I read these value (via cluser) i get.
color_temp_physical_min (id: 0x400b) > GET > Value None
color_temp_physical_max (id: 0x400c) > GET > Value None

MattWestb, 2021-12-05:
;-(((((((

I think the device is made for only working with the LIDK remote.

Try changing `color_temperature (id: 0x0007)` and see if the light is it OK and you can getting mon and may you can setting.
(i was see it reported in the log but with  tuya its not the same its working OK)

I have hot seen that  some one have trying getting the commands the LIDL remote is using but perhaps some is sniffing the device and we can using the commands in the future.

12christiaan, 2021-12-05:
It is becoming very strange now. I have 2 of the same LED Lamps. one is working good now. the other still not warm light. 
When i put them together. 1 is at color temperature 500 warm. the other at 300 about the same color. when i put it from 300 > 500, it become cold again.

12christiaan, 2021-12-05:
Okay. I got it now. both working the same.
The LED Lamp was connected to the LIDL gateways before. 
When i put the the LED OFF - ON - OFF - ON - OFF - ON, it can again in pairing mode. 
I looked for it via HA, and not the Lamp is working correct.
It remembered some setting from the LIDL gateway? 
So for any one haveing problems. while using the LIDL gatewway before. make sure you reset de device.

Still 1 item open is the CCT only and removeing the RGB.
Thanks

MattWestb, 2021-12-05:
If reading the Zigbee specification **shall** lights forgetting all things (only remember the last EPAN -ID but not the network or hardware settings) but tuya is tuya and many device can working OK after being in tuya ZBGW and moving to one "normal" Zigbee system but is going crazy if removing the battery.

If you like getting only CT try adding your device ID in one new line after the last in:
https://github.com/zigpy/zha-device-handlers/blob/e9b2c334382b3cfda3ab4cc2afe345d9de6eb918/zhaquirks/lidl/cct.py#L39-L47
and configure local quirks in HA and putting the new quirk in the local quirk directory and restarting HA but i think you have trying it.

If it working OK pleas making one PR so other user can using there device without problems.

12christiaan, 2021-12-05:
Yes, as i wrote in the item above. 

For solution (1. I have added in the file /zhaquirks/lidl/cct.py the following signature:
signature = { MODELS_INFO: [ ("_TZ3000_8uaoilu9", "TS0502A")
That solves the RGB problem.

PR #1205 (2021-12-06): Add _TZ3000_8uaoilu9 for CT
Adding "_TZ3000_8uaoilu9" for getting CT only.
Fix #1203
closed 2021-12-06 by dmulcahey