zigpy/zha-device-handlers #1204: [BUG] IKEA E1812 shortcut button does not use quirks

RenWal, 2021-12-05:
**Describe the bug**
I have two IKEA E1812 Tradfri shortcut buttons that started to behave differently. ZHA claims that they have the same firmware (I also bought them together), but they now have different clusters: The one that is not picked up correctly has an extra `in_cluster` with ID 0xfc7c in its device signature. This causes `zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn` to not pick this button up, producing a *Fail because input cluster mismatch on at least one endpoint* message in the log.

The `zha_event` still fires as it should, so by using the raw event one can work around this, but maybe there's a way to get the quirks to work.

**To Reproduce**
- pair a shortcut button that has the 0xfc7c input cluster
- go to the device page and see that no quirks are applied
- try to add an automation, see that triggers for button press are missing

(Unfortunately I have no clue what caused the two buttons to present different signatures)

**Expected behavior**
The quirks should still apply

**Additional context**
signature of the button with the extra cluster (this is the only difference between my buttons)
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI SHORTCUT Button",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2021-12-05:
The IKEA manufacture cluster is one real mystery !!

The E1812 dont need it then its **not** being used for pairing devices with touch link (Ikea pairing).

I can being that the factory have doing bad data settings in the device and its working different, It have happening with the Open/Close button and some lights before.

Im curious which version your devices is having and if updating then can helping (if not on the last version). 
(dont configure IKEA OTA if you have other controllers then they is loosing group binding if being updated to last firmware)

The "strange" one shall working OK but  you is getting 1/2 battery and no device automations then the quirk is not loading.

You can doing one more device class in the quirk with the same as the normal and adding the missing cluster and it shall working.

Also possible flashing one working "user data" = hardware config in the device with SWD probe but its little to advance for most user. 
(I have one working and have dumped all flash and also reflashing it with older FW and its working OK)

RenWal, 2021-12-06:
Both of the buttons run the 0x23080631 firmware (different FW was my first thought as well, but that doesn't seem to be the case). The `zigpy` logs say that this is the newest version when I try to force an update. Maybe a botched firmware update?

The quirk definitely didn't load. The signature shows `zigpy.device.Device` as the class, I'm seeing incorrect battery readings, and triggers don't work.

I have the necessary equipment to dig into this with SWD, but I'm afraid I'll not have the time for this for a while, especially if adding an extra quirk fixes things on the software side.

MattWestb, 2021-12-06:
SWD flashing  is only some minutes is having one working probe and some cables that is fitting the test points. Look on this https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1563280.
You only need on user data" that you can getting from my or you is dumping it for the working one.

But i think its easier installing one patches local quirk for your device that adding  one device class with  the extra cluster. I have making one that  you can testing by configuring local quirks in HA and putting the file in the quirk folder.
Instruction for local quirk: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Patched quirk:
[shortcutbtn2.zip](https://github.com/zigpy/zha-device-handlers/files/7659856/shortcutbtn2.zip)
Unzip the quirk and putting it in the local quirk derectory and restarting HA.

RenWal, 2021-12-06:
Oh this is awesome, thank you so much! With the instructions for the local quirk it works flawlessly!

I had to make one minor change, since I missed an extra difference yesterday -- the "weird" button also doesn't have the WindowCovering output cluster, so I had to remove that from the signature.

MattWestb, 2021-12-06:
My bad i was not looking thru you log as i was doing yesterday and was missing it :-((

Great that you was fixing it !!

If no more user is having the problem we can letting it there and you is having the local quirk for getting it working OK and its not being away then updating  HA.
If more user is having the problem i think its better making one PR for adding the "weird" version of the device in the main branch.

dkluivingh, 2021-12-18:
Hi,

I have the same problem, 1 button works and the other button does not.
The button that doesn't work has been updated with the Tradfri gateway. The other button that does work comes straight out of the package.
manufacturer_specific:0xFC7C and no WindowCovering.

Thanks for this solution.

waldheinz, 2021-12-23:
I have two buttons here, *both* report firmware `0x23080631`. One is showing the "weird" behavior and one works as expected. So whatever the difference is, the FW seems not to be the cause. However I can confirm that the patched quirk, with everything related to `WindowCovering` removed, fixes the issue for me.

dsimoes, 2021-12-24:
Hi everyone.

So glad I found this thread :)
Also been pulling my hair out with the shortcut button. I updated the firmware on it, and I can get events but no trigger actions. I guess it's the exact same issue here.

What's the solution? use the local quirk mentioned? @RenWal  can you provide the final working file?
Thanks.

RenWal, 2021-12-24:
[shortcutbtn_extracluster_fixed.zip](https://github.com/zigpy/zha-device-handlers/files/7774558/shortcutbtn_extracluster_fixed.zip)

jsbrich, 2022-01-14:
> [shortcutbtn_extracluster_fixed.zip](https://github.com/zigpy/zha-device-handlers/files/7774558/shortcutbtn_extracluster_fixed.zip)

I'm very new to ZigBee, but having issues with these shortcut buttons showing up in HA, but not creating zha_event events when you press the button, so wanted to try the local quirk fix.  Is the .py file in your zip the file that I should put in /config/custom_zha_quirks/ directory?

MattWestb, 2022-01-14:
If you is getting ` "class": "zigpy.device.Device"` on the device card / device signature  = no name of the device in the quirk then the system have not matching the quirk and mot loading it.

Both posted quirks is zipped and need being unzipped and putting in the local quirk folder.

If your device have one quirk loaded and not getting any `zha_events` try using "configure device" from the device card and its fixing the binding and reporting (dont forgetting waking up the device the sending commands to it).

kaechele, 2022-03-15:
The fixed quirk does fix my buttons not registering press events with HA after they were updated to 2.3.080.

MattWestb, 2022-03-24:
@RenWal  Its looks its more the only some device with different setting that is making problems for user and its looks the updated signature is working well for this user.
Do like making one PR for adding it as one alternative version (`IkeaTradfriShortcutBtn2` ?)so user dont need adding local quirks in the future ?
If you dont like i can doing it but i have not one device with the strange setting for testing and its better then have it verified for getting merged.

PR #1462 (2022-04-07): Add alternative quirk for IKEA E1812 button
Some IKEA Tradfri shortcut buttons (E1812) expose the 0xFC7C manufacturer-specific input cluster but lack the WindowCovering cluster. This leads to the IkeaTradfriShortcutBtn quirk not being applied. This PR adds an alternative quirk that can be used with these buttons.

Closes #1204
closed 2022-04-07 by dmulcahey