zigpy/zha-device-handlers #1206: [BUG] TS0601 TRV reports wrong power status

Immelstorn, 2021-12-06:
**Describe the bug**
This HVAC puts a valve closure state value in the power status field. So when the thermostat is closing the valve it sends 5% and then returns to 100%.

**To Reproduce**
Steps to reproduce the behavior:
2. Go to device overview
2. See that "power" is set to 100%
3. Lower the temperature, so the device will close the radiator
4. See that "power" is set to 5%

*Important*: it will only briefly set the power to 5% and then change back to 100% within a second or two, so you need to be attentive to spot it, or set some trigger this.

**Expected behavior**
It is expected to see the battery status of the device in this field.

**Screenshots**
[image]
[image]

MattWestb, 2021-12-06:
I think this is what the device is sending to the system but i cant saying for 110%  without logs then doing the operation.

If you like its one user testing one new updated quirk with more GUI function in ZHA but its  need more testing for knowing if all it working OK. https://github.com/zigpy/zha-device-handlers/issues/1178#issuecomment-980294492

jacekk015, 2021-12-09:
@Immelstorn Do same steps:
https://github.com/zigpy/zha-device-handlers/issues/1212#issuecomment-990086102

BTW In case of errors DEBUG logs are needed.

Immelstorn, 2021-12-10:
@jacekk015  Did those steps. Got a bunch of new entities in the valve page. But the problem is still there - the power is always 100%, but when valve closes/opens it briefly becomes 5%.
[image]

Here are my debug logs: 
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/7691238/home-assistant.log)

jacekk015, 2021-12-10:
@Immelstorn 
Two possibilities:
1) As you see - your TRV sends battery low signal - 0x016E attribute switches to [1] - then you see 5% of battery
```
MOES_BATTERY_LOW_ATTR = 0x016E  # battery low warning
2021-12-10 09:29:50 DEBUG (MainThread) [zigpy.zcl] [0x68f7:1:0xef00] ZCL request 0x0001: [Command(status=54, tsn=197, command_id=366, function=0, data=[1, 1])]
2021-12-10 09:29:50 DEBUG (MainThread) [zhaquirks.tuya] [0x68f7:1:0xef00] Received value [1] for attribute 0x016e (command 0x0001)
```

2) Some user then, did Moes quirk originally, wrongly connected attribute 0x016E to low battery status.

I'm for option 1), so better replace battery.

MattWestb, 2021-12-10:
My Siter TRVs is stop doing many things like valve detect then having very low battery likely for not stop working then the valve is open and cant closing it.

I think for testing its one must having at least 30% battery left for not getting problem with interanta protection mechanism in the device and also all TRVs with 2 AA batteries must using alkaline batteries (1.5V) and **not** rechargeable one (1.2V) for not getting problems.
Device with 3 AA batteries (Rti-Tek) is little different but all "Moes" (CN-Hysen) is using 2 AA batteries.

Immelstorn, 2021-12-10:
@jacekk015 replaced the batteries - indeed, this is not happening anymore. So it seems the real bug here is that the low battery status is not preserved, and changed to 100% after a second.

jacekk015, 2021-12-10:
@Immelstorn Battery code is indeed strange, but in reality that's TRV is sending on/off for battery low power.
When TRV sends battery low signal code makes it 5%, so the warning appears, else is always 100%.
```
        elif attrid == MOES_BATTERY_LOW_ATTR:
            self.endpoint.device.battery_bus.listener_event(
                "battery_change", 5 if value else 100
            )
```
Checked your LOGS again. Don't see any attribute that even could look like a battery level.
So unfortunately Moes did a bad job not exposing an attribute with battery level.

Immelstorn, 2021-12-10:
@jacekk015 ok. thanks anyway, at lease now I know that if it sends 5% than it is a time to replace batteries :)

closed 2021-12-10 by Immelstorn