zigpy/zha-device-handlers #1207: Quirks for Opple 2-button remote - lumi.remote.b286opcn01 doesn't get used in some cases

Drizzt321, 2021-12-07:
**Is your feature request related to a problem? Please describe.**

I purchased a multiple Opple 2 button (really 4 button, but 2 switch) remotes in a batch from AliExpress, after giving a single one a try and liking it. The single one, and 2 of the new batch, gets the `zhaquirks.xiaomi.aqara.opple_remote.RemoteB486OPCN01V2` quirks applied to it. However the other ones do not, and are class type `zigpy.device.Device`.

This is with current ZHA/Home Assistant OS 6.6, core-2021.11.5, supervisor-2021.10.8 (stable).

Issue https://github.com/zigpy/zha-device-handlers/issues/661 and Pull https://github.com/zigpy/zha-device-handlers/pull/787 seem related, but unfortunately the quirks still isn't being applied and my additional comment hasn't been replied to. 

**Describe the solution you'd like**
Get the quirks to match the signature, or figure out the difference in otherwise apparently identical switches so a new quirks (if needed) can be added. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

The signature of one of the devices that doesn't have the quirks applied:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0300"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003",
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0103",
[... 153 lines cut ...]
[0x1D2A:2:0x0006]: initializing channel: from_cache: False
[0x1D2A:2:0x0006]: finished channel initialization
[0x1D2A:3:0x0006]: initializing channel: from_cache: False
[0x1D2A:3:0x0006]: finished channel initialization
[0x1D2A:4:0x0006]: initializing channel: from_cache: False
[0x1D2A:4:0x0006]: finished channel initialization
[0x1D2A:5:0x0006]: initializing channel: from_cache: False
[0x1D2A:5:0x0006]: finished channel initialization
[0x1D2A:6:0x0006]: initializing channel: from_cache: False
[0x1D2A:6:0x0006]: finished channel initialization
[0x1D2A:1:0x0000]: 'async_initialize' stage succeeded
[0x1D2A:1:0x0008]: 'async_initialize' stage succeeded
[0x1D2A:1:0x0006]: 'async_initialize' stage succeeded
[0x1D2A:1:0x0300]: 'async_initialize' stage succeeded
[0x1D2A:2:0x0006]: 'async_initialize' stage succeeded
[0x1D2A:3:0x0006]: 'async_initialize' stage succeeded
[0x1D2A:4:0x0006]: 'async_initialize' stage succeeded
[0x1D2A:5:0x0006]: 'async_initialize' stage succeeded
[0x1D2A:6:0x0006]: 'async_initialize' stage succeeded
[0x1D2A](lumi.remote.b486opcn01): power source: Mains
[0x1D2A](lumi.remote.b486opcn01): completed initialization
```

**Additional context**
[Add any other context or screenshots about the feature request here.]

MattWestb, 2021-12-07:
I dont having and Opple switches (is burned of Aquara switches) but normally its only making one new device class in the opple quirk that is adjusted signature so its loading and hopefully is working.

You new device looks little other then its have 0x0012 cluster (= MultistateInput cluster) that  is not used by Opple before as one input cluster but the function is being used in the Opple quirk.

You can try cooking one new class and if you getting problem some devs can helping you getting it working if its not do that.

MattWestb, 2021-12-09:
That is very true for endpoint 1 but your device 5 more that all is not matching so the quirk is not loaded.

All cluster on all endpoints must matching for the quirk to being loaded also other parameters in the signature must matching.

Drizzt321, 2021-12-15:
@MattWestb Ah, so I'll need to look at how the quirks match, and try and make a copy/custom quirks to get it to match?

And I see what you mean,  with the mismatch now, from the pairing logging.

```
Considering <class 'zhaquirks.xiaomi.aqara.opple_remote.RemoteB486OPCN01V3'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.xiaomi.aqara.opple_remote.RemoteB486OPCN01V2'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6}
```

MattWestb, 2021-12-15:
Yes **all** clusters on **all** endpoints **and** the device name **and** model is all must matching or the system is not loading the quirk.

Drizzt321, 2021-12-15:
@MattWestb  apparently there's a v4 now, https://github.com/zigpy/zha-device-handlers/pull/1138. I just updated HA, and it's there in the latest version. After removing/re-adding I got it to pick up the quirks, and then issue a Reconfigure before the zha_events came in as expected.

So glad I delayed so someone else could dig into it and get it fixed up for me, lol.

closed 2021-12-15 by Drizzt321