zigpy/zha-device-handlers #1209: [BUG]

unaiur, 2021-12-07:
**Describe the bug**
All light entities associated to Tuya TS0001 wall switches stopped working and new entities in switch domain were created after update to HomeAssistant 2021.12b1

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from HA 2021.11.x to 2021.12b1 that includes latest revision of this package as of today.

**Expected behavior**
Old entities remained as is and no automations are broken.

**Screenshots**
N/A

**Additional context**

Previously, it was handled as an standard Zigbee device, without any quirk.

After upgrade, it was handled using quirk zhaquirks.tuya.ts000x.Switch_1G_GPP

It seems that this quirk is exposing the device as a smart plug but is an "inwall" light switch designed to operate lights.

See
https://a.aliexpress.com/_vsVqkq

It only happen to the 1 gang switches. 2 gang switches still work okay.

unaiur, 2021-12-07:
It seems related with this line: https://github.com/zigpy/zha-device-handlers/blob/a4c6cdb9e531b9f643567c32e8cd8bacc64840cd/zhaquirks/tuya/ts000x.py#L71

Previous versions were using zha.DeviceType.ON_OFF_LIGHT instead.

Can we change it?

unaiur, 2021-12-07:
2 gang switches are still working because there are only quirks for TS0001 (1 gang) and TS0003 (3 gang devices) but not for TS0002. 

So it is still handled as an standard zigbee device.

MattWestb, 2021-12-07:
I agree 100% with you and i was asking the maintainer then adding the new functions then the user was liking having it as **ON_OFF_PLUG_IN_UNIT** that i think its little strange.

I can doing one PR and putting the "real " device type back and letting the maintainer deciding if hi like have it.

Can you posting the signature for your TS0002 so we can adding it and getting the extra functions the 1 and 3 gang have getting ?

unaiur, 2021-12-08:
I copied the TS0003 definition, removed the endpoint 3 and changed the MODEL to TS0002 and works almost perfectly.

Cluster TuyaZBExternalSwitchTypeCluster is not working for the second endpoint but it seems a hardware issue: the cluster in the first endpoint affects both endpoints and you cannot setup a momentary push button in one endpoint and a toggle switch in another. Maybe we should remove it.

unaiur, 2021-12-08:
Or maybe the manufacturer fixes the issue at some point... who knows!

MattWestb, 2021-12-08:
We have with tuya devices here so its not 110% the same on all devices like I have one LIDL 3 plug power strip with 4 USB and one user have one newer one and cant steering the plugs individual but with tuya ZBGW its possible (its missing endpoint 2 and 3).

And some attribute is not implanted and other not or not working = tuya.

I think its better letting it as it is and  waiting for if tuya is releasing one new firmware !!

Do you like making on PR with the change `DEVICE_TYPE` back to its original and adding the the 2 gang ?

I can doing it  but is better if having the device in the hand and can testing it working so no small errors in in the code (copy and paste error).

If you doing it write its fixing one braking thing.

unaiur, 2021-12-08:
I will try. I have to travel for next two weeks, so no promise.

MattWestb, 2021-12-08:
OK i was thinking getting one fast fix before the release of HA 2021.12 but i think its too late.

I making one PR with your changes and hoping it going well and perhaps it can going in the bug fix fro HA.

MattWestb, 2021-12-08:
PR made and waiting merger :-))

And have one nice tripp !!!

unaiur, 2021-12-08:
Regarding the external_switch_type attribute (I upgraded HA to get access to that attribute), I got this error in the logs for the second endpoint:

2021-12-08 15:33:33 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [57345] cluster_type: [in] endpoint_id: [2] attribute: [53296] manufacturer: [None] response: [None] failure: [{53296: <Status.UNSUPPORTED_ATTRIBUTE: 134>}],

For the first endpoint:
2021-12-08 15:42:58 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [57345] cluster_type: [in] endpoint_id: [1] attribute: [53296] manufacturer: [None] response: [ExternalSwitchType.Toggle] failure: [{}],

MattWestb, 2021-12-08:
EP 2 looks OK then its responding its not supported.
For EP 1 i think its not repaying at all on the command that is little strange.
Do you knowing if the device is supporting / have the function implanted in the firmware ?

The cluster and attribute is being sniffed from one tuya ZBGW but is not guarantee its working for all devices then tuya is making things little different in many devices.

You can look in this issue https://github.com/zigpy/zha-device-handlers/issues/1118 and the its one diferent version of the module and i have one dual dimmer switch that is one 3rd version of the switches from tuya.

unaiur, 2021-12-08:
Yes, for first endpoint is working. And I can change it to momentary but it changes for both endpoint 1 and endpoint 2. I want to have a momentary endpoint and a toggle endpoint, but it seems that the hardware does not support it. With two TS0001 devices, everything works correctly but I am not able to close the wall (not enough space for two devices).

I will have to buy something else. Do you know any other device with two gangs and that supports connecting one momentary push button and a toogle switch together?

MattWestb, 2021-12-08:
I have only experience of the double dimmer switch (TS110F) that dont have so much options but i have getting it fitting in one normal EU round box with putting one patched double switch (so its not being so deep as the original) that is normal no contact and having toggle and dimming up and then down then long pressing it * 2 (= one button dimmering).

I is using it for power on and off and can also dimming both lights but mostly is made in ZHA but its working then the system is down and its the most important for my.

Teardown https://github.com/MattWestb/tuya-Switches-Dimmer

PR #1210 (2021-12-09): Fixing the braking TS001X quirk
Reversing the braking `zha.DeviceType.ON_OFF_PLUG_IN_UNIT` to original `zha.DeviceType.ON_OFF_LIGHT` that was braking automations in HA.
Also adding the TS0012 the 2 gang version (usr have trying the new quirk changes)
Fix #1209