zigpy/zha-device-handlers #1212: [Device Support Request] _TZE200_cwnjrr72 TS0601

mhombach, 2021-12-09:
**Is your feature request related to a problem? Please describe.**
My TuYa valve seems to have a lot of properties/entities missing when connected over ZHA instead of e.g. zigbee2mqtt.
I was following youtube video (with timestamp: https://www.youtube.com/watch?v=iD1U6Q47yLU&t=502s) and also bought exactly the same devices as he did (followed his link). As you can see in the attached screenshot, my HA only recognizes 2/3 entities while in the youtube video there are a lot more.
Also, in the product page (where I bought them) it's stated that those devices have switches for child-lock and open-window-detection. So at least there should be 2 switch-entities be available. Not talking about the rest like schedlue-switching etc.
And btw, I'm using a zigbee ethernet coordinator.

**Describe the solution you'd like**
Since the id `cwnjrr72` is already in the code, the problem must lie somewhere else. I would love to have this fixed somehow, so I can start using my newly bought 6 tuya valves :)

**Device signature**
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cwnjrr72",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.MoesHY368_Type1"
}
```

**Additional context**
[image]

MattWestb, 2021-12-09:
The "Moes" type have all working on cluster also setting and reading the weekly schedules but its not so easy using in automations. Only window detection is on one switch in the GUI.

One user have making one with most functions in ZHA GUI but its not well tested and need more brave user testing it.
Look in this issue and trying it out if you like and give feedback for getting all things in place.
https://github.com/zigpy/zha-device-handlers/issues/1178#issuecomment-980294492

jacekk015, 2021-12-09:
@mhombach 
Add to configuration.yaml
```
zha:
  custom_quirks_path: /config/zha_custom_quirks/
```
Create that directory "zha_custom_quirks" under config.
Download, unzip, and place that file below into that folder.
Restart HA
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/7687364/ts0601_trv_moes.py.zip)

mhombach, 2021-12-09:
> @mhombach Add to configuration.yaml
> 
> ```
> zha:
>   custom_quirks_path: /config/zha_custom_quirks/
> ```
> 
> Create that directory "zha_custom_quirks" under config. Download, unzip, and place that file below into that folder. Restart HA [ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/7687364/ts0601_trv_moes.py.zip)
> 
> [edit] @MattWestb That's a bit new, refactored version.

Just to be precise here: You mean, in the folder where my configuration.yaml is, i create the new folder "config"? Or is that "root"-ish folder already the "config"? I'm using docker to run HA on my Synology NAS.
And after the restart, do I need to re-discover the valves or should it directly work?

MattWestb, 2021-12-09:
ZHA shall loading the new quirk if all is OK but it can being good deleting it so the old data is deleted and then adding it after one minute or more or best restarting HA.

@jacekk015 The` ForceValveState` is one good name but i think many user is not understanding the meaning of `Normal = 0x00`, `Open = 0x01` and `Close = 0x02`
Perhaps:
Automatic_mode= 0x00 (or PID_mode)
Force_Open = 0x01
Force_Close = 0x02

? ? ?

MattWestb, 2021-12-09:
By the was in both "ZHA Native" and the new quirk is scheduled supported but on zigbee  cluster (from the device card and manage cluster and the `MoesThermostat` and attributes get and setting).

mhombach, 2021-12-09:
> ZHA shall loading the new quirk if all is OK but it can being good deleting it so the old data is deleted and then adding it after one minute or more or best restarting HA.
>
> @jacekk015 The` [...]

I did as you told me, but that didn't change anything, sadly. I put the config there, restarted the server, even removed and re-discovered the valve. Not 1 additional setting/entity on that device.

jacekk015, 2021-12-09:
@mhombach 
Did you add it to configuration.yaml??
```
zha:
  custom_quirks_path: /config/zha_custom_quirks/
```
Did you create a folder "zha_custom_quirks" in HA config folder??
Did you unzip and put a file iside "zha_custom_quirks" folder?
Did you then restart HA???

If all yes, then put again a card screenshot.
Best would be to enable DEBUG logging in configuration.yaml and restart HA again:
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
Then put ALL logs here as a file. A time of minimum 5 minutes of logs is a min, best is 10 minutes.

mhombach, 2021-12-09:
@jacekk015 I'm 1000% sorry, I missed the line in the config xaml.
After I restarted HA many entities were added !
See my valve's signature now:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "8": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "9": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "10": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    },
    "11": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x000d"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_cwnjrr72",
  "model": "TS0601",
  "class": "ts0601_trv_moes.MoesHY368_Type1"
}
```

A few followup questions:
- `analog_output_valve_state` is this correct? The valve itself does not have a physical turnable knob. Or I'm misinterpreting this
- `switch.tze200_cwnjrr72_ts0601_37054afe_on_off_2`, `switch.tze200_cwnjrr72_ts0601_37054afe_on_off` and `switch.thermostat_esszimmer_37054afe_on_off` (that one was renamed before from me) all are switches, but i have no idea what they mean. I guess that one is the child lock, one is the open-window-recognition and the 3rd is... ?
- see below screenshot, the value sliders all give me a "not a number" and so they can't be changed.
- Is `_TZE200_cwnjrr72 TS0601 37054afe analog_output Temperature Calibration` the entry that would let me set the offset for the valve's turning? Because my valves are all like 4 °C off the real temperature
[image]

MattWestb, 2021-12-09:
switch.thermostat_esszimmer_37054afe_on_off = WindowDetectionswitch.tze200_cwnjrr72_ts0601_37054afe_on_off = ChildLock
switch.tze200_cwnjrr72_ts0601_37054afe_on_off_2 = ValveDetect

MOES_VALVE_STATE_ATTR = 0x026D  # opening percentage 

and calibration is the offset of the device local temperature 👍

mhombach, 2021-12-09:
@MattWestb Hmm, is that duplicated entity caused by the device beeing in HA before that additional config? Then I would just remove and re-discover them. I have 6 valves but only installed 2 until now because i want to get them working 100% first.`
And what does `ValveDetect` mean?
And the problem with `NaN` still needs to be fixed then.
Is it possible, by the quirk config, to configure the entity names a little bit?

MattWestb, 2021-12-09:
The first temperature is window detection (how much temperature is falling for triggering)  and the time is how long it shall stop heating then window open is detected.
Both must being sett for the switch is working if the device is not having it configured.

And i was forgetting the valve state is one read only function but we have not getting it working so its possible changing but its not doing anything but shall reporting valve state OK.

mhombach, 2021-12-09:
Aaaah. About the `NaN` btw, I figured I just have to click inside the slider somewhere to set a value. I just jas no number as initial value. Weird UI though^^°
So, if I leave all toggle switched in the default OFF position, does that mean that window detection is active or inactive?

mhombach, 2021-12-09:
And another discovery:
I think it's recommended to remove devices before using the new config.
I just now removed my valves from HA and discovered one again and now all entities have initial values that make sense.

[image]

jacekk015, 2021-12-09:
When device is paired again it sends all data to HA. So that's why values were updated on pairing

mhombach, 2021-12-09:
Makes sense, goot to know :)
I'm 99% done now. I'm just trying to calibrate my valves, because the temperature is off.
But for some reason I'm having trouble understanding thr workings of those 2 sliders.
top slider is the calibration one, bottom one is the valve state.
From what I realized now I think, the valve state is basically showing how far the valve's pin is physically pushed in (or to be more precise, how far the valve's motor has pushed the pin out).
The calibration slider is mysterious to me. It ranges from -9 to 9.
I attached a sketch for my current situation.
My valve is measuring 20°C and when the vale is set to 18°C, it's closed (from listening with my ear on the valve/waterflow). When I set it to 19°C, the valve will open.
So my valve is off by 2°C.
But most of the time when I use the calibration slider and set it to e.g. 4, it will stay on 4 for a few secs, then it will revert to 0.
Then the valve state slider will change a little bit, as if the calibration is meant to increase/decrease the valve state without keeping a state itself. But that didn't fully work. At some point my calibration slider got "stuck" at position -9 or 9.
What is the proposed way how to calibrate in my current case?

[image]

jacekk015, 2021-12-09:
All sliders have representation on the TRV itself.
If you change calibration to +2 then you should see same when you click the TRV physically.
We just read/write data HA<->TRV - nothing more.
What TRV allow - it works. Like Valve state - if it's read-only on TRV then there's no real possibility to change it from HA UI.

mhombach, 2021-12-09:
Hmm... I might be missing some information here... I was now calibrating the temperature on the valve itself and it seems that I can change not the temperature that I set but instead the measured temperature?
I was thinking that
- the valve has a temperature sensor that measures to some extend reliable the temperature around the valve
- the valve is pushing in the pin on the radiator and that can be a little bit off. So I would think that I need to adjust _that_ number (the temp I set)?
Because when I set the temp to 21°C and the TRV pushed the pin in an arbitrary amount, how should it be sure that this position is 21°C?

MattWestb, 2021-12-09:
One thing is that the TRVs Zigbee module is one sleeping end device and it can taking some seconds for one command being received of it from you is doing changes in ZHA.
Then doing changes on the TRV is nearly immediately changes in ZHA and we cant do anything for foxing it without keeping the device alyas awake then then your batteries is dead in some weeks.

The local temperature calibrating -9 - +9 shall being sent to the TRV and its sending the "new" local temperature back with the corrected temperature to ZHA and is using it until you is sending one new value for calibration. 
The TRV have one very pore (not good calibrated) temperature sensor but then calibrated its working OK.
That is not problem then the TRV is only comparing the setpoint and if need heating more or less and the calibration is more or less "cosmetic" for presentation in the host system.

teklabania, 2021-12-10:
@jacekk015, I have _TZE200_cwnjrr72. My findings on the quirk you prepared:
- Calibration slider doesn't work correctly. When I set the value to +/-5, the temperature is adjusted of 0,5C and the value displayed next to slider shows +/-50C:
[image]
Setting other values makes slider return to the previous value.
- Valve state percentage slider makes no change (the valve's display indicates another value).
- There's no power button for disabling the thermostat:
[image]

There are 3 switches, but have no idea what they are. My guess:
- window detection
- children lock
- ???

Do you think there's a way to set up hysteresis for this valve?
Is the quirk you made going to be merged into master branch?

jacekk015, 2021-12-10:
@teklabania Was the device re-paired??
Enable debug logs and put here as a file whole logs. The logs need to include action you take.
Without it I can say nothing.

> Valve state percentage slider makes no change

Read-only

> There's no power button for disabling the thermostat:

Yes. How do you disable thermostat? There's no kind of function on the device.

> There are 3 switches, but have no idea what they are.

- MoesValveDetect
- MoesWindowDetection
- MoesChildLock

> Do you think there's a way to set up hysteresis for this valve?

You do have Temperature Calibration. If the valve manual says nothing about functions there's no possibility.

> Is the quirk you made going to be merged into master branch?

Don't know. PR needs working Python tests - I can't/won't do this.

teklabania, 2021-12-10:
@jacekk015, thank you for quick response.
> @teklabania Was the device re-paired?? Enable debug logs and put here as a file whole logs. The logs need to include action you take. Without it I can say nothing.

No, the device wasn't re-paired. I just put a quirk on place and restarted HA. New switches/sliders appeared after restarting. Should I delete and pair the device again?

> Yes. How do you disable thermostat? There's no kind of function on the device.

Hmm... Generic thermostat has the option:
[image]

> * MoesValveDetect

What might that be?

> You do have Temperature Calibration. If the valve manual says nothing about functions there's no possibility.

So it seems there's no option for setting up hysteresis. I didn't find it in manual.

> Don't know. PR needs working Python tests - I can't/won't do this.

I'm keeping fingers crossed for adding this device to official support.

teklabania, 2021-12-10:
Current temperature shows 24C. I set calibration to -5 with the slider. After that current temperature is set to 23,5C, but the slider shows -50. Logs:
```
2021-12-10 16:10:33 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=124 command_id=Command.Default_Response>
2021-12-10 16:10:33 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL request 0x000b: [0, <Status.SUCCESS: 0>]
2021-12-10 16:10:35 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=106 command_id=1>
2021-12-10 16:10:35 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=255, command_id=515, function=0, data=[4, 0, 0, 0, 235])]
2021-12-10 16:10:35 DEBUG (MainThread) [zhaquirks.tuya] [0xc6a9:1:0xef00] Received value [0, 0, 0, 235] for attribute 0x0203 (command 0x0001)
2021-12-10 16:10:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC6A9:1:0x0201]: Attribute report 'MoesThermostat'[local_temp] = 2350
2021-12-10 16:10:35 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.termostat_sypialnia: Attribute 'local_temp' = 2350 update
2021-12-10 16:10:36 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=1>
2021-12-10 16:10:36 DEBUG (MainThread) [zigpy.zcl] [0xc6a9:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=0, command_id=556, function=0, data=[4, 255, 255, 255, 251])]
2021-12-10 16:10:36 DEBUG (MainThread) [zhaquirks.tuya] [0xc6a9:1:0xef00] Received value [255, 255, 255, 251] for attribute 0x022c (command 0x0001)
2021-12-10 16:10:36 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC6A9:1:0x0201]: Attribute report 'MoesThermostat'[local_temperature_calibration] = -50
2021-12-10 16:10:36 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.termostat_sypialnia: Attribute 'local_temperature_calibration' = -50 update
2021-12-10 16:10:36 DEBUG (MainThread) [zigpy.zcl] [0x37a8:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=33 command_id=Command.Report_Attributes>
2021-12-10 16:10:36 DEBUG (MainThread) [zigpy.zcl] [0x37a8:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int8s, value=0>)]]
2021-12-10 16:10:36 DEBUG (MainThread) [zigpy.zcl] [0x37a8:1:0x0006] Attribute report received: on_off=0
2021-12-10 16:10:40 DEBUG (MainThread) [zigpy.zcl] [0xba87:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
2021-12-10 16:10:40 DEBUG (MainThread) [zigpy.zcl] [0xba87:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int8s, value=0>)]]
2021-12-10 16:10:40 DEBUG (MainThread) [zigpy.zcl] [0xba87:1:0x0006] Attribute report received: on_off=0
```

mhombach, 2021-12-10:
I can confirm that the slider for temperature calubration doesn't work as excepted. Maybe there some number-format wrong or so (int/float/whatever)?
I manually set the offset on my valve to +2°C (by the menu on the valve itself). Then I paired it with HA (over ZHA). And the slider for temperature calibration shows 100°C :/

jacekk015, 2021-12-10:
You didn't read whole topic:
https://github.com/zigpy/zha-device-handlers/issues/1212#issuecomment-990213781

> Should I delete and pair the device again?

Yes - then all default values will be set according to what TRV sends to HA.

> Hmm... Generic thermostat has the option:

Does the TRV has this function in manual? Do you really can OFF that device? Like totally OFF???

> What might that be?

Don't know. You have the device in your hand - switch and look what changed. Probably manual says something about it. We only implement that what users provide.
Quirk was originally written by other user - didn't analyse functions and I don't have this device.

> Current temperature shows 24C. I set calibration to -5 with the slider. After that current temperature is set to 23,5C, but the slider shows -50. Logs:

Re-pair device. Then check again.
Other users would need to confirm that also.

jacekk015, 2021-12-10:
@mhombach No logs, no talks.
Logs show everything what happens in communication.
Do changes on TRV physically and then put logs here.

teklabania, 2021-12-10:
> You didn't read whole topic: [#1212 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1212#issuecomment-990213781)

I did. The comment recommends re-pairing in case of issues with initial values. I had no problem with that.

> 
> > Should I delete and pair the device again?
> 
> Yes - then all default values will be set according to what TRV sends to HA.
> 

Didn't help.

> > Hmm... Generic thermostat has the option:
> 
> Does the TRV has this function in manual? Do you really can OFF that device? Like totally OFF???
> 

Well, thermostat in HA is virtual representation of a physical device. It could be turned off programmatically. 

> > What might that be?
> 
> Don't know. You have the device in your hand - switch and look what changed. Probably manual says something about it. We only implement that what users provide. Quirk was originally written by other user - didn't analyse functions and I don't have this device.
> 

I wouldn't ask if the option were described in manual. Toggling the switch changes nothing. Some people may have this knowledge, the question was public. Don't get too excited.

> > Current temperature shows 24C. I set calibration to -5 with the slider. After that current temperature is set to 23,5C, but the slider shows -50. Logs:
> 
> Re-pair device. Then check again. Other users would need to confirm that also.

Nothing changes after re-pairing.

MattWestb, 2021-12-10:
@teklabania 
>Hmm... Generic thermostat has the option:
>>Does the TRV has this function in manual? Do you really can OFF that device? Like totally OFF???

The HA clima generic card have many functions for all kind of clima component.
If you have one heating thermostat you can shutting it off completely (the device is DEAD) and shall using off mode in clima component.

TRV is normally always living and cant being shutted off so ZHA is not enabling this function then most TRVs is having antifreeze function that cant being suted off.

Also some TRV have off mode but its not recommending using one TRV for open and closing the valve then you is losing all algorithms in the TRV for adjusting the heat demand to the room is shall keeping heated at one setpoint. Its takes days for one TRV for adjusting it so it not doing over and undersutting the heating and if you is turning it off and on its starts from zero.

If your TRV can being sutted off with commands and you like to using it its up to you but if its not have it as mode we cant implanting it in ZHA and cant getting the code accepted of HA maintainer then its not one real working mode of the device.

If like gettin all function working with GUI elements in ZHA we need testing the devices for knowing how the funktions is working or perhaps is not working at all but we dont have all device so we need help of users that can do the testing and reporting back and also providing logs so we can getting all function that is possible on the device working OK in our system.

Also we cant doing great magic thins with devices then all is locked in the firmware but with good feedback we can making the device working great in ZHA and also putting in little extras then its possible.

teklabania, 2021-12-10:
That's great explanation. Thank you @MattWestb.

MattWestb, 2021-12-10:
One very good example:
>Calibration slider doesn't work correctly. When I set the value to +/-5, the temperature is adjusted of 0,5C and the value displayed next to slider shows +/-50C:

If providing logs then changing the calibrating in ZHA and reporting it is very likely getting one updated quirk in some hours but with no feedback its not possible doing anything and its better using the standard quirk in ZHA without GUI element and doing all on the Zigbee cluster.

teklabania, 2021-12-12:
> Edit: For logging pleas use this setting in HA [#1212 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1212#issuecomment-990123605)

This is what I did. I put logs here: https://github.com/zigpy/zha-device-handlers/issues/1212#issuecomment-991054106

vincentlee20, 2021-12-13:
> @teklabania
>
>
> The HA clima generic card have many functions for all kind of clima component. If you have one heating thermostat you can shutting it off completely (the device is DEAD) and shall [...]

The value (TS0601/HY368) I am using will heat the room by changing itself from "away" to "schedule" when I am not at home, even it was set to "away" before (I still have no idea why it changed itself). Since I can't change the TRV algorithm, am thinking if we can make use of the valve_force_state to shutdown when no body home if we can have a simple button. Can also make it automated with HA....

MattWestb, 2021-12-13:
You can using the automations from the TRVs device card like "Condition `TYST11 TRV thermostat is set to a specific preset mode`" and is you is using one switch you can selecting witch present it shall using if the switch is in on or off state.

One PM of the schedule mode if the TRV have some programing in its memory it start using it on the next time is passing the time. So best not using the schedule mode if you is not knowing what is programmed in the TRV.

I think the best is using one manual setpoint temperature and away mode and doing automations in HA by changing mode / resets and also setpoint if needed.

jacekk015, 2021-12-13:
You can create a Helper with list or Switch and use change event to make further changes on TRV in Automation.

vincentlee20, 2021-12-14:
Thanks, I think I need to set the valve to close in automation. The TRV was again changed itself to "schedule" and heat the room when all people left the house for hours....
>.<

jacekk015, 2021-12-14:
Not possible - in the logs you will have a hour and minute when the has been switched. You can check what made that change.

Lyr3x, 2022-01-03:
I have som trouble with those TRVs. It happens multiple times a day that the valve opens, but the target temp is always way higher than it should and the rooms heat up to the worst case 26°
[image]
I tried to set the calibration to +x° but it always switch back to 0. 

Any ideas whats going on?

jacekk015, 2022-01-04:
@Lyr3x HA doesn't manage TRV behavior. It just read what TRV sends, and also sends what you set in GUI.
If TRV works wrong, that I assume it's the TRV mistake. I have 10pcs of zigbee TRVs and not all of the work OK.
Some of them overheat room. I needed to calibrate the TRV on the radiator many times. Some of them have 3D printed spacer, and some others has a coin between valve and TRV valve.

Lyr3x, 2022-01-04:
Well I know what the integration does and what not. My question is if anyone here do also have trouble with these TRVs? That the valve opens at maximum for no reason has obviously nothing to do with the quirk nor ZHA.

It just don't make any sense to open the valve 100% if set to 19C and it measures already 25...

LoreFist, 2022-01-10:
Hello. how can i change the value of the current temperature through automation?
My idea is to change the temperature using another sensor https://www.aqara.com/us/temperature_humidity_sensor.html

Tell me please

teklabania, 2022-01-10:
> It just don't make any sense to open the valve 100% if set to 19C and it measures already 25...

It behaves exactly the same in my case. It always overheats the room both in PID mode and when not in PID mode. This device is a piece of trash.

mhombach, 2022-01-10:
> > It just don't make any sense to open the valve 100% if set to 19C and it measures already 25...
> 
> It behaves exactly the same in my case. It always overheats the room both in PID mode and when not in PID mode. This device is a piece of trash.

Same here, I was thinking it's only my specific valve. Most of them are somewhat working fine, but some and sometimes they are in heating-mode when they measure already 26°C anda re set to 19/20°C... I have another model now from "Hama" and will try that one. If that works flawlessly, I will give back all of the 6 ones mentioned here...

dcoder42, 2022-02-12:
>> There's no power button for disabling the thermostat:

> Yes. How do you disable thermostat? There's no kind of function on the device.

What about Cluster id 0x0201 Attribute 0x4006 (valve_force_state)
0 = ForceValveState.Normal
1 = ForceValveState.Open
2 = ForceValveState.Close

These are writable and working for me. Power Off could be translated to ForceValveState.Close and Power On to ForceValveState.Normal

closed 2022-08-18 by github-actions[bot]

doubledutch2, 2022-10-18:
I have one of the _TZE200_cwnjrr72 TRVs as well and as described on [other posts ](https://community.home-assistant.io/t/zha-fails-after-adding-quirk-for-ts0601-tze200-cwnjrr72-trv/442726) - Installing the ts0601_moes_trv quirk kills ZHA. After you install it, all your ZHA devices will report unavailable until you remove it again.

FYI: @jacekk015

jacekk015, 2022-10-18:
@doubledutch2 Moes quirk wasn't updated to HA April changes. Later evening will try to update it.

doubledutch2, 2022-10-21:
Thank you @jacekk015! 

I'll watch this page for updates. I only have one so it's not urgent but I know others have the same issue.

jacekk015, 2022-10-22:
Updated to match HA April 2022 changes.
None of the logic has changed, so if you think something works wrong please report it back.
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844336/ts0601_trv_moes.py.zip)

cemizm, 2022-10-22:
> Updated to match HA April 2022 changes. None of the logic has changed, so if you think something works wrong please report it back. [ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844336/ts0601_trv_moes.py.zip)

Thank you @jacekk015 for your custom quirk. Installing the quirk with `Home Assistant 2022.10.4` gives following error:

```
2022-10-22 15:56:32.592 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ZBridge for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 101, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_trv_moes.py", line 718, in <module>
    class MoesUserInterface(TuyaUserInterfaceCluster):
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 84, in __init_subclass__
    raise TypeError(
TypeError: `manufacturer_attributes` is deprecated. Copy the parent class's `attributes` dictionary and update it with your manufacturer-specific `attributes`. Make sure to specify that it is manufacturer-specific through the  appropriate constructor or tuple!
```

The Hama Thermostat has the following signature. I also attached a full report of the device.

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x0001",
        "0x000a",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "_TZE200_yw7cahqs",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}
```

Any suggestions?

[zha-bb791876a15e4fd9b338fdb8a62f7a7c-_TZE200_yw7cahqs TS0601-1e58c5fe46fc979c9248b134be979524.json.txt](https://github.com/zigpy/zha-device-handlers/files/9844544/zha-bb791876a15e4fd9b338fdb8a62f7a7c-_TZE200_yw7cahqs.TS0601-1e58c5fe46fc979c9248b134be979524.json.txt)

jacekk015, 2022-10-22:
BUG corrected - should be OK now.
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844583/ts0601_trv_moes.py.zip)

cemizm, 2022-10-22:
> BUG corrected - should be OK now. [ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844583/ts0601_trv_moes.py.zip)

wow, fast response! removing the othet deprecated `manufacturer_attributes` from `MoesWindowDetection` fixed the loading/initialization issue... but still no luck with the `_TZE200_yw7cahqs` thermostat... i tested both types (MoesHY368_Type1, MoesHY368_Type2) by extending the signature... any other ideas? 

found this thread regarding this particular thermostat:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841

jacekk015, 2022-10-22:
@cemizm Because that's not Moes. It's supported under Saswell quirk
https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1287512365

jacekk015, 2022-10-22:
Moes corrected
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844739/ts0601_trv_moes.py.zip)

cemizm, 2022-10-22:
> @cemizm Because that's not Moes. It's supported under Saswell quirk [zigpy/zigpy#653](https://github.com/zigpy/zigpy/discussions/653) [#1189 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1189#issuecomment-1287512365)

that thread too and the temperature correction/calibration is finally working... thank you very much! appreciate your work...

geheim27, 2022-11-18:
@jacekk015 could you please have a look at https://github.com/zigpy/zha-device-handlers/issues/1439#issuecomment-1075594517? Is it possible to add the _TZE200_cpmgn2cf", and _TZE200_9sfg7gm0 also to the "official" ts0601_trv_moes.py? Or are those not compatible?

jacekk015, 2022-12-25:
Quirk code updated
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/10300266/ts0601_trv_moes.py.zip)

marcelodavanzo, 2023-01-13:
If anyone's having trouble with this custom quirk from HA 2023.1.2 and onwards like I was, just change all references of `foundation.Command` to `foundation.GeneralCommand` in the quirks file.

jacekk015, 2023-01-13:
@marcelodavanzo Or just use already corrected quirk
https://github.com/zigpy/zha-device-handlers/issues/1178#issuecomment-1376253412

marcelodavanzo, 2023-01-13:
> @marcelodavanzo Or just use already corrected quirk [#1178 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1178#issuecomment-1376253412) https://github.com/zigpy/zha-device-handlers/files/10376551/ts0601_trv_moes.py.zip

Ah I hadn't seen that!

jacekk015, 2023-12-04:
@IlGiock https://github.com/jacekk015/zha_quirks#usage
Try to use Windows Explorer.
You can use HA Samba Share add-on for that.