zigpy/zha-device-handlers #1213: [Device Support Request] Third Reality Door sensor not reporting correctly

jds11111, 2021-12-11:
**Is your feature request related to a problem? Please describe.**
I have the [Third Reality Door Sensor 3RDS17BZ](https://smile.amazon.com/dp/B08R9PH4JT?psc=1&ref=ppx_yo2_dt_b_product_details). It pairs quite easily with ZHA and the Sonoff Zigbee 3.0 Dongle Plus.  However, it does not report when the door is closed. If I move the magnet to the sensor, it momentarily reports closed, but then in about 1sec it reports open again, though the magnet is still there. 

**Describe the solution you'd like**
Obviously, it should stay 'closed', but the behavior suggests that ZHA is misinterpreting the message sent by the sensor.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4659, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Third Reality, Inc",
  "model": "3RDS17BZ",
  "class": "zigpy.device.Device"
}
```

**Additional context**
I can't find where anyone reports using this sensor with ZHA, so am not sure if it is possibly a problem with my particular device. [Blakadder does report that it works with Zigbee2MQTT](https://zigbee.blakadder.com/Third_Reality_3RDS17BZ.html). It is the only relatively inexpensive door sensor I have found that uses (rechargeable) AAA batteries, so it would be great to get it working with ZHA, but I need help to make that happen.

jacekk015, 2021-12-11:
Post DEBUG logs so we could see if that comes from sensor or not.

jds11111, 2021-12-11:
OK, I have posted an edited log [here in pastebin](https://pastebin.com/vkMJAUtG).  If I did too much editing, please let me know. I didn't want to expose personal information.

dmulcahey, 2021-12-14:
The contact sensor part of this should just work. can you put this on a flat surface and test it to rule out tolerance issues with magnet alignment?

jds11111, 2021-12-14:
Hmm. You might be right. I took both off the door, and tried to do more careful alignment. On the desk I could get better reproducibility.  It seems like these sensors are less forgiving than the Sonoff ones I have tried.  Thanks for the pointer.

closed 2022-06-19 by github-actions[bot]