zigpy/zha-device-handlers #1214: [Device Support Request] Samotech / MoesHouse 2 Gang Zigbee Dimmer Module MS-105BZ (_TZE200_e3oitdyu)

nworbneb, 2021-12-12:
**Is your feature request related to a problem? Please describe.**
Device is not working at all "out-the-box".  It supports two lights with two switches.

**Describe the solution you'd like**
Develop quirk to support Samotech / MoesHouse 2 Gang Zigbee Dimmer Module MS-105BZ

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_e3oitdyu",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

**Additional context**

It looks like some work has been done in deconz to support this device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5320

Hardware available from here:

https://www.amazon.co.uk/Samotech-Zigbee-Switches-Dimmers-MS104BZ/dp/B08N14QBZ8?th=1

https://www.moeshouse.com/collections/dimmer-switch-module/products/mini-diy-tuya-zigbee-3-0-smart-dimmer-switch-relay-module-1-2-way-2-gang-tuya-zigbee3-0-hub-required-smart-life-tuya-app-works-with-alexa-google-home-voice-control

nworbneb, 2021-12-12:
I had (very) limited success in getting the first gang to work by modifying the zhaquirks/tuya/ts0601_dimmer.py.  But second gang does not work.

`"""Tuya based touch switch."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, LevelControl, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaDimmerSwitch,
    TuyaLevelControl,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerLevelControl,
    TuyaManufCluster,
    TuyaOnOff,
)

class TuyaDoubleDimmer(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_e3oitdyu", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LevelControl.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                    TuyaManufacturerLevelControl,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
`

MattWestb, 2021-12-12:
Try the quirk in https://github.com/zigpy/zha-device-handlers/issues/1033 they have gettingsome function working but  al parts in not in place.

Then possible configuring more logging and getting feedback its possible getting it working with multi gana dimmers.

nworbneb, 2021-12-12:
Thank you @MattWestb.  Not sure why I didn't find that in my Google searches.  Looks like support is nearly there, I will move my attention to that issue.

closed 2021-12-12 by nworbneb