zigpy/zha-device-handlers #1216: [BUG] Aqara lumi.switch.l1aeu1 seen as end device instead of router and power source is unknow instead of mains

nlhomme, 2021-12-13:
**Describe the bug**
The in-wall switch Aqara lumi.switch.l1aeu1 ([link to the device on blakkader.com](https://zigbee.blakadder.com/Aqara_WS-EUK01.html)) is seen as an end device instead of a router and the power source is unkown instead of mains:

But if we look at Zigbee signature for the device, we got:
```
(...)
  "node_descriptor": (...) *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, (...)",
(...)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Aqara lumi.switch.l1aeu1
2. Go to 'Configuration' -> 'Devices and Services' -> 'Zigbee Home Automation'
2. Click on a Aqara lumi.switch.l1aeu1 switch
4. See the power source and device type as in the screenshot
5. Click on 'Zigbee Device Signature' to check the node inspector

**Expected behavior**
Aqara lumi.switch.l1aeu1 switches are supposed to be seen and act as router in the zigbee network
Aqara lumi.switch.l1aeu1 switches' source power is supposed to be 'Mains

**Screenshots**
[image]

**Additional context**
Device complete signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l1aeu1",
  "class": "zigpy.device.Device"
}
```

dmulcahey, 2021-12-13:
What am I missing… the data you posted doesn’t match what you are reporting. The node descriptor the device reported looks bad and matches the device screen screenshot.

dmulcahey, 2021-12-13:
> But if we look at Zigbee signature for the device, we got:
> 
> ```
> (...)
>   "node_descriptor": (...) *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, (...)",
> (...)
> ```

Where did this come from? It doesn’t match what is below it

dmulcahey, 2021-12-13:
All of this said… they have done the reverse of this before so I wouldn’t be shocked if we need to override it to correct it.

nlhomme, 2021-12-13:
> Where did this come from? It doesn’t match what is below it

It comes from the device signature (se also at the end on my first message)

nlhomme, 2021-12-14:
> Where did this come from? It doesn’t match what is below it

Please excuse me, I made a mistake on my copy-paste :(

But this device (WS-EUK01) is an in-wall switch, so the power source is mains.
And since this device is powered by mains, it should work as a router, right?

dmulcahey, 2021-12-14:
Not necessarily. Does it have a neutral wire?

nlhomme, 2021-12-14:
No it doesn't

dmulcahey, 2021-12-14:
Then it’s probably an end device. I have a mix of the US Aqara switches. The ones w/ neutral wires are routers and the ones without are end devices.

Adminiuga, 2021-12-14:
Yes, it is very unlikely for a no-neutral switch to be a ZigBee router.

Works as designed.

nlhomme, 2021-12-14:
All right, thanks for your indications.

Take care

closed 2021-12-14 by nlhomme

MattWestb, 2021-12-14:
You can looking on the network map if its only having one router as parent (= end device) or its have connections to more routers (= router) then its working as one router and having more connections to neighbors.

nlhomme, 2021-12-14:
That's what I did, but since I believed that every mains powered devices where router capable, I thought something went wrong with my aqara switches