zigpy/zha-device-handlers #1217: Reconfig failes for Innr SP120 power plug

erkr, 2021-12-14:
**Describe the bug**
After installing the latest core-2021.12.1 there was a quirk added for the Innr SP120 plug.
I guess this is the quirk: Innr SP120 multiplier fix #1174
A fix I'm really looking forward too!!!
But I don't observe that the values where corrected, So I initiated a reconfigure in ZHA (using a Conbee 2 stick). 
The reconfigure process  fails (see below) and the values remain wrong!

**To Reproduce**
- press reconfigure for the SP120 device in ZHA

**Expected behavior**
I expect the reconfigure succeeds and that the the values from then on get corrected by the new multipliers

**Screenshots**
The result of reconfigure:
[image]

**Additional context**
The device info:
[image]

The device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4454, maximum_buffer_size=127, maximum_incoming_transfer_size=80, server_mask=0, maximum_outgoing_transfer_size=80, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0010",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 49246,
      "device_type": "0x1000",
      "in_clusters": [
        "0x1000"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "innr",
  "model": "SP 120",
  "class": "zhaquirks.innr.innr_sp120_plug.SP120"
}
```

dmulcahey, 2021-12-14:
Can you do this again and get debug logs? It’s not apparent what failed…

MattWestb, 2021-12-14:
Then looking in tuya Zigbee dev dock its looks like tuya ZBGW is only binding the basic cluster on "normal" Zigbee devices and there devices is then reporting to the coordinator that looks being one "Silabs standard" then IKEA is doing the same (but they is using ZLL GW) so i think its not critical failing binding more cluster then basic on tuya devices.

erkr, 2021-12-14:
Hi

> Can you do this again and get debug logs? It’s not apparent what failed…

Here is 30 seconds of log, starting at the initiation of the reconfigure command:
[log.txt](https://github.com/zigpy/zha-device-handlers/files/7712753/log.txt)
I hope this is providing what you need,

Thx
Eric

dmulcahey, 2021-12-15:
is the plug actually working after doing this?

dmulcahey, 2021-12-15:
I see some unsupported attributes responses... those should be no big deal. there is one interesting response that I have to look at: ConfigureReportingResponseRecord(status=141, direction=0, attrid=0) not sure how this response code makes sense tbh.   INVALID_DATA_TYPE = 0x8D 

@Adminiuga  have you ever seen this come back in this response before?

Adminiuga, 2021-12-15:
INNR does response with INVALID_DATA_TYPE and that probably explains why it won't send the reports.
Either bug in zigpy or in INNR. Need to contact them to confirm what type are they expecting. or try a few different types?

Adminiuga, 2021-12-15:
The result:
```
2021-12-14 16:19:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xD777:1:0x0702]: Failed to configure reporting for '['instantaneous_demand', 900, 'current_summ_delivered']' on 'smartenergy_metering' cluster: [ConfigureReportingResponseRecord(status=134, direction=0, attrid=1024), ConfigureReportingResponseRecord(status=134, direction=0, attrid=900), ConfigureReportingResponseRecord(status=141, direction=0, attrid=0)]
```

is in response to the folowing request:
```
2021-12-14 16:19:34 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (51, 70, 0, <DeconzAddressEndpoint address_mode=2 address=0xD777 endpoint=1>, 260, 1794, 1, b'\x00E\x06\x00\x00\x04*\x05\x00\x84\x03\x01\x00\x00\x00\x00\x00%\x1e\x00\x84\x03\x01\x00\x00\x00\x00\x00\x00\x00\x02\x18\x01\x00\x84\x03', 2, 0)
```

it deserializes to:
```
<ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=69 command_id=Command.Configure_Reporting> [
  AttributeReportingConfig(direction=0, attrid=1024, datatype=42, min_interval=5, max_interval=900, reportable_change=1),
  AttributeReportingConfig(direction=0, attrid=0, datatype=37, min_interval=30, max_interval=900, reportable_change=1),
  AttributeReportingConfig(direction=0, attrid=512, datatype=24, min_interval=1, max_interval=900)
]
```

What the heck is `attrid=900` ? is Innr confused with attribute report configuration request? cause we never ask for attr id 900 configuration, but the max interval is 900

Adminiuga, 2021-12-15:
The datatype=37 is 0x25 and is an unsigned 48 but Analog Integer. Per specs, CurrentSummationDelivered is 
[image]

What is the correct data type for this attribute?

Adminiuga, 2021-12-15:
however,  ` "model": "SP 234"` with `Firmware: 0x28016610` does configure reporting successfully:

```
2021-12-15 15:46:17 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1794, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=179), 140, -65, 0x2884, 255, 255, b'\x08\\\x07\x86\x00\x00\x04']
2021-12-15 15:46:17 DEBUG (MainThread) [zigpy.zcl] [0x2884:1:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=92 command_id=Command.Configure_Reporting_rsp>
2021-12-15 15:46:17 DEBUG (MainThread) [custom_components.zha.core.channels.base] [0x2884:1:0x0702]: Successfully configured reporting for '{'current_summ_delivered', 'status'}' on 'smartenergy_metering' cluster
2021-12-15 15:46:17 DEBUG (MainThread) [custom_components.zha.core.channels.base] [0x2884:1:0x0702]: Failed to configure reporting for '['instantaneous_demand']' on 'smartenergy_metering' cluster: [ConfigureReportingResponseRecord(status=134, direction=0, attrid=1024)]
```

dmulcahey, 2021-12-15:
It’s prob a firmware bug

Adminiuga, 2021-12-15:
the Electrical Measurement reads at 4 and does not provide any divisor or multipliers, so those are supposed to be assumed as 1. In this case, 4W at 2.5 hours should produce 0.01kWh. Are we displaying it incorrectly?

dmulcahey, 2021-12-15:
I meant the error code in my reply above… I didn’t look at that part… maybe we are?

Adminiuga, 2021-12-15:
RMS_Voltage reads 121, RMS_Current reads 48, with 1000 divisor and 1 multiplier, which should give about 5.8W consumption. So 4W is not that far off, but why are we displaying it as 0.4W

dmulcahey, 2021-12-15:
https://github.com/home-assistant/core/blob/5bf67cac6681d0b477bbc22da430db9c81546808/homeassistant/components/zha/sensor.py#L284

This looks right no??

dmulcahey, 2021-12-15:
Argh are both of those using the multiplier???

Adminiuga, 2021-12-15:
nah, it does look right. just the AC_POWER_DIVISOR reads as 10, that's why it's 0.4W

Adminiuga, 2021-12-15:
But for OP's SP120, guess should contact Innr and ask them to share the firmware update.

erkr, 2021-12-17:
> is the plug actually working after doing this? 
Yes the plug can be operated and it provides readings:
[image]
Here, actual power (first value in Watt) and the voltage (last value in volts) are correct.
The total usage (kWh) is a factor 100 to high (multiple of 10W not kW).
The current wrong as well, it is ca 1400 times less. It should ca 0.13A

dieneuser, 2021-12-18:
@erkr 
I have the same power plug and no problems since the update. All values are displayed correctly. You may have to remove and pair the plug again.
[image]

erkr, 2021-12-18:
@dieneuser 
Thanks for the suggestion! Indeed after removal and pairing the readouts became realistic:
[image]
The a reconfigure of the device still fails, now with two errors:

Adminiuga, 2021-12-19:
It's a bug with innr firmware.

erkr, 2021-12-20:
Thx for helping out. They now work and I don't think innr support OTA firmware updates. So I will close this issue.

closed 2021-12-20 by erkr