zigpy/zha-device-handlers #1219: Command.aps_data_indication errors

RefineryX, 2021-12-16:
**Describe the bug**
I keep getting regular errors on my logs and struggling to understand where to look in resolve the 'issue'.

**Error 1**
The seq id often changes but the error is very much the same
```
Logger: zigpy_deconz.api
Source: /usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py:315 
First occurred: 02:34:01 (1 occurrences) 
Last logged: 02:34:01

No response to 'Command.aps_data_indication' command with seq id '0xf3'
```

**Error 2**
I also get template variable warnings which refreshes cluster_id and endpoint_id etc.

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1779 
First occurred: 09:34:15 (4 occurrences) 
Last logged: 09:34:15

Template variable warning: 'dict object' has no attribute 'command' when rendering '{{ trigger.event.data.command }}'
Template variable warning: 'dict object' has no attribute 'cluster_id' when rendering '{{ trigger.event.data.cluster_id }}'
Template variable warning: 'dict object' has no attribute 'endpoint_id' when rendering '{{ trigger.event.data.endpoint_id }}'
Template variable warning: 'dict object' has no attribute 'args' when rendering '{{ trigger.event.data.args }}'
```

SimpleUserHA, 2022-01-18:
> **Describe the bug** I keep getting regular errors on my logs and struggling to understand where to look in resolve the 'issue'.
>
> **Error 1** The seq id often changes but the error is very much the [...]
Having the same issue here. Similar also has been reported [here](https://community.home-assistant.io/t/zigbee-stopped-working-a-few-weeks-ago/370019) and [here](https://community.home-assistant.io/t/zigpy-deconz-error-in-logs/344569)

No idea how to start narrowing down the issue, but I am running latest firmware [0x26720700 Stable 2021-08-20]:
https://deconz.dresden-elektronik.de/deconz-firmware/
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

htodep, 2022-02-17:
Having the same issue. Running latest firmware: deCONZ_ConBeeII_0x26720700.bin.GCF
`
Logger: zigpy_deconz.api
Source: /usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py:315
First occurred: 16:30:53 (4 occurrences)
Last logged: 16:34:00

No response to 'Command.aps_data_indication' command with seq id '0x1f'
No response to 'Command.aps_data_indication' command with seq id '0x28'
No response to 'Command.aps_data_indication' command with seq id '0x2a'
No response to 'Command.aps_data_indication' command with seq id '0x7a'
`

closed 2022-08-23 by github-actions[bot]