zigpy/zha-device-handlers #1220: [BUG] Wrong "current" for Tuya plug TS0121 (_TZ3000_g5xawfcq)

avbor, 2021-12-17:
**Describe the bug**
Wrong current (A) sensor for Tuya plug TS0121 (_TZ3000_g5xawfcq).
It's impossible to get 220 V 11 A and 1 W at the same time =)
It seems to me that the multiplier is lost somewhere...

**Screenshots**
[image]

smiba, 2021-12-23:
This should be fixed, but as a workaround you can manually refresh the ac_current_divisor cluster attribute to fix this:

https://github.com/home-assistant/core/issues/61564#issuecomment-993299627

avbor, 2021-12-23:
> This should be fixed, but as a workaround you can manually refresh the ac_current_divisor cluster attribute to fix this:
> 
> [home-assistant/core#61564 (comment)](https://github.com/home-assistant/core/issues/61564#issuecomment-993299627)

Thanks! a completely unobvious action, I tried to reconfigure many times, but it did not help, but the cluster polling helped.

closed 2021-12-23 by avbor