zigpy/zha-device-handlers #1224: Zigate v1 Router

ruimarinho, 2021-12-19:
[Zigate](https://zigate.fr/) is a NXP-based Zigbee radio gateway comparable to zzh and the likes. Recently, a router firmware was published at https://github.com/badzz/Zigate-Router/releases/tag/V0, but ZHA seems unable to properly detect it as a router.

```
2021-10-14 12:17:38 INFO (MainThread) [zigpy_znp.zigbee.application] Permitting joins for 60 seconds
2021-10-14 12:17:38 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.Broadcast: 15>, address=0xFFFC), dst_ep=0, src_ep=0, cluster=54, sequence=124, options=TransmitOptions.NONE, radius=0, data=b'|<\x00'
2021-10-14 12:17:38 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request 54({'PermitDuration': 60, 'TC_Significant': <Bool.false: 0>}) and replaced with ZDO.MgmtPermitJoinReq.Req(AddrMode=<AddrMode.Broadcast: 15>, Dst=0xFFFC, Duration=60, TCSignificance=0)
2021-10-14 12:17:38 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.MgmtPermitJoinReq.Req(AddrMode=<AddrMode.Broadcast: 15>, Dst=0xFFFC, Duration=60, TCSignificance=0)
2021-10-14 12:17:38 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinReq.Rsp(Status=<Status.SUCCESS: 0>)
2021-10-14 12:17:38 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinRsp.Callback(Src=0x0000, Status=<Status.SUCCESS: 0>)
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0xCBEF, SrcIEEE=00:15:8d:00:01:ed:46:b0, ParentNwk=0x0000)
2021-10-14 12:17:49 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xCBEF, SrcIEEE=00:15:8d:00:01:ed:46:b0, ParentNwk=0x0000)
2021-10-14 12:17:49 INFO (MainThread) [zigpy.application] Device 0xcbef (00:15:8d:00:01:ed:46:b0) joined the network
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.application] Device 00:15:8d:00:01:ed:46:b0 changed id (0xcbef => 0xcbef)
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.device] [0xcbef] Skipping initialization, device is fully initialized
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='Zigate-Router' manuf='NXP' nwk=0xCBEF ieee=00:15:8d:00:01:ed:46:b0 is_initialized=True>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for NXP Zigate-Router (00:15:8d:00:01:ed:46:b0)
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0xCBEF, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xCBEF:00:15:8d:00:01:ed:46:b0 entering async_device_initialized - is_new_join: False
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xCBEF:00:15:8d:00:01:ed:46:b0 has been reset and re-added or its nwk address changed
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] skipping discovery for previously discovered device - 0xCBEF:00:15:8d:00:01:ed:46:b0
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCBEF](Zigate-Router): started configuration
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:ZDO](Zigate-Router): 'async_configure' stage succeeded
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0000]: finished channel configuration
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0xCBEF, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0xCBEF, NWK=0xCBEF, IEEE=00:15:8d:00:01:ed:46:b0, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2021-10-14 12:17:49 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: ZDO.EndDeviceAnnceInd.Callback(Src=0xCBEF, NWK=0xCBEF, IEEE=00:15:8d:00:01:ed:46:b0, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2021-10-14 12:17:49 INFO (MainThread) [zigpy.application] Device 0xcbef (00:15:8d:00:01:ed:46:b0) joined the network
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.zigbee.application] Pretending we received a ZDO message: b'\xFF\xEF\xCB\xB0\x46\xED\x01\x00\x8D\x15\x00\x8E'
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy.zdo] [0xcbef:zdo] ZDO request ZDOCmd.Device_annce: [0xCBEF, 00:15:8d:00:01:ed:46:b0, 142]
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.SUCCESS: 0>)
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCBEF](Zigate-Router): started initialization
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:ZDO](Zigate-Router): 'async_initialize' stage succeeded
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xCBEF), dst_ep=0, src_ep=0, cluster=ZDOCmd.Bind_req, sequence=125, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'}\xb0F\xed\x01\x00\x8d\x15\x00\x01\x06\x00\x03Z\xa7{#\x00K\x12\x00\x01'
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.zigbee.application] Intercepted AP ZDO request ZDOCmd.Bind_req({'SrcAddress': 00:15:8d:00:01:ed:46:b0, 'SrcEndpoint': 1, 'ClusterID': 6, 'DstAddress': MultiAddress(addrmode=3, ieee=00:12:4b:00:23:7b:a7:5a, endpoint=1)}) and replaced with ZDO.BindReq.Req(Dst=0xCBEF, Src=00:15:8d:00:01:ed:46:b0, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:23:7b:a7:5a, endpoint=1))
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.BindReq.Req(Dst=0xCBEF, Src=00:15:8d:00:01:ed:46:b0, SrcEndpoint=1, ClusterId=6, Address=MultiAddress(addrmode=3, ieee=00:12:4b:00:23:7b:a7:5a, endpoint=1))
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0000]: initializing channel: from_cache: False
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0000]: finished channel configuration
2021-10-14 12:17:49 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0006]: initializing channel: from_cache: False
2021-10-14 12:17:49 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.BindReq.Rsp(Status=<Status.SUCCESS: 0>)
[... cut ...]
2021-10-14 12:17:55 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0000]: 'async_initialize' stage succeeded
2021-10-14 12:17:55 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCBEF:1:0x0006]: 'async_initialize' stage succeeded
2021-10-14 12:17:55 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCBEF](Zigate-Router): power source: Mains
2021-10-14 12:17:55 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCBEF](Zigate-Router): completed initialization
```

It would be great to have ZHA supporting it.

Hedda, 2022-01-03:
@badzz maybe you would be willing to look at creating a quirk for zha-device-handlers as indirectly looking at it for zigpy?

That is, fix quirks requirements in [zha-device-handlers](https://github.com/zigpy/zha-device-handlers) for using your Zigate Router firmware when using for example a Texas Instruments CC2652 based Zigbee Coordinator with zigpy-znp and zigpy.

ruimarinho, 2022-07-03:
Still an issue for me.

badzz, 2022-07-03:
Hi, sorry I had seen this issue. I will try to have a look
Thx

ruimarinho, 2022-10-18:
@badzz have you had any chance of looking at this one?

closed 2023-04-23 by github-actions[bot]