zigpy/zha-device-handlers #1230: [Device Support Request] lumi.vibration.aq1 orientation

schwickster, 2021-12-26:
**Is your feature request related to a problem? Please describe.**
Currently it´s not possible to read what orientation the lumi.vibration.aq1 (aqara djt11lm) is. It's possible to read how many degrees it has tilted.

**Describe the solution you'd like**
Adding orientation gives the option to use this sensor as a open/close for a garage door.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
No logging, but I did find the code within Deconz that it's possible.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/aq1_vibration_sensor.json
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/aq1_vibration_sensor_orientation.js

**Additional context**
Thanks!

schwickster, 2021-12-26:
I made a commit on a fork I made. It's not done yet... still looking into why the orientation is always returning 90 degrees
https://github.com/schwickster/zha-device-handlers/commit/ead8b52361d04b818fc5a760ec3a0d93fe57b949 Once done I'll create a pull request

schwickster, 2021-12-27:
After testing it works fine. Made a pull request. Hopefully it will be good because it's my first time doing something in Python and github pull request.

closed 2021-12-28 by schwickster

AndyVRD, 2022-01-12:
Hi,

Will this mean if it gets approved it will be available in one of the next HA updates?

schwickster, 2022-01-12:
Correct. It´s currently approved and next release of zha-device-handlers it can be put into a pull request for Home Assistant´s next release next month.

AndyVRD, 2022-01-12:
Thanks for the quick reaction and for adding this feature. I guess I have to remove and pair my sensor again to get it supported? Just switched from Deconz to ZHA and was really missing the orientation, using it for my garagedoor state.

schwickster, 2022-01-12:
I don't think you have to repair. I also use it for the same purpose.

AndyVRD, 2022-02-04:
> zha

Is it correct that it not made the 2feb release?

schwickster, 2022-02-04:
It should be there. Try a repair maybe?

AndyVRD, 2022-02-05:
Just removed and repaired the sensor but still no orientation attribute.

schwickster, 2022-02-05:
That's because It's not an attribute but an event.

triggerx, 2022-04-19:
This really needs to be a state!  (or attribute?)

SamJongenelen, 2022-06-24:
Yeah ive been struggling with that too. Having events is the new way