zigpy/zha-device-handlers #1232: [BUG] smatthings tagv4

gromgsxr, 2021-12-27:
my smart things tags are not working any more since moving to 2021:12 release they now show a binary input that never used to be there and they are almost always at away status very occasionally they will show as home for a min or so then show away again.
they used to work quite reliably for detecting the state at home or away but now they don't.

shackrat, 2022-01-01:
Happy New Year to the awesome dev team here.  Tossing in my two cents, I am seeing the same behavior, with one exception.  After about 2 days one of mine went unavailable.  I cannot get it to check in again.  I dug out my old SmartThings hub and it paired right up.  It will not even pair as a new device now in ZHA.

gromgsxr, 2022-01-29:
These devices are completely useless now they are either away when home and occasionally check in. Or go to unavailable if away for long enough

shackrat, 2022-01-30:
Right after I made my post, I dug into the cluster settings and was able to get two mine to work and report correctly.  checkin_interval in the Poll Control cluster does not appear to be properly set.  When paired to SmartThings it is being set to 60.  I assume the value is seconds, but couldn’t find a credible source to confirm.

1. Go to Manage Clusters
2. Select Poll Control
3. Select checkin_interval under Attributes
4. Enter 60 for Value (I went with 240 in an attempt to save battery life)
5. Click Set Zigbee Attribute

Its been almost a month and both fobs are correctly reporting arrivals and departures and they’re not falsely going offline when we are home.  So far so good..

MattWestb, 2022-01-30:
If i remember right is the value 1/4 seconds so standard is 60 seconds and you custom is 15 seconds.

gromgsxr, 2022-01-30:
so far so good. with setting to 60. on a side note using set zigbee attribute I was able to make the tag beep quite useful so I can make an automation to make the keys beep in an aid to finding them when thy are misplaced at home

closed 2022-01-30 by gromgsxr