zigpy/zha-device-handlers #1236: [Device Support Request] Integrating a motion, temperature & humidity sensor in ZHA model NAS-PD07 manufactured by NEO

sassiahmed, 2021-12-27:
Hi all,

I'm new in this community. so sorry about any stupid question :slight_smile: 

I want to integrate a motion sensor manufactured by NEO in ZHA.
Here is the description : https://zigbee.blakadder.com/Neo_NAS-PD07.html

When I try to integrate it with ZHA, there is no entity found.

I tried the quirks (Tuya/ts0601_motion.py : https://github.com/zigpy/zha-device-handlers/blob/5228c54d70d5eee0be09ff6ca2eeb81a538eefa1/zhaquirks/tuya/ts0601_motion.py ==> but nothing.

I modified the ts0601_motion.py file adding a new class NeoMotion with the right signature and import the clusters to use (inspired by ts0601_trv.py) found in the same repo.
I have the motion sensor entity but didn't work (always clear)

Here is the zigbee signature :
```

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_7hfcudw5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

and here is my ts0601_motion.py modified :

```

"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

from typing import Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Identify, Ota
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster,
    TuyaUserInterfaceCluster,
)
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
ZONE_TYPE = 0x0001

class MotionCluster(LocalDataCluster, MotionOnEvent):
    """Tuya Motion Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Motion_Sensor}
    reset_s = 15

class TuyaManufacturerClusterMotion(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Motion device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufClusterAttributes.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_cmd = args[0]
        if hdr.command_id == 0x0001 and tuya_cmd.command_id == 1027:
            self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)

class TuyaMotion(CustomDevice):
    """BW-IS3 occupancy sensor."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.motion_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        #  endpoint=1 profile=260 device_type=0 device_version=0 input_clusters=[0, 3]
        #  output_clusters=[3, 25]>
        MODELS_INFO: [
                    ("_TYST11_i5j6ifxj", "5j6ifxj"),
                    ("_TYST11_7hfcudw5", "hfcudw5"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [Basic.cluster_id, Identify.cluster_id],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    MotionCluster,
                    TuyaManufacturerClusterMotion,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }

class NeoMotion(CustomDevice):
    """NAS-PD07 occupancy sensor."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.motion_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=0 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [
                    ("_TZE200_7hfcudw5", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                    ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                    MotionCluster,
                    TuyaManufacturerClusterMotion,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

Thank you for your help :pray:

bsfaxi, 2021-12-30:
+1
I have the same.

javicalle, 2022-01-23:
Can you try with that quirk?
```python
class NeoMotion(CustomDevice):
    """NAS-PD07 occupancy sensor."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.motion_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=0 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_7hfcudw5", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterMotion.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterMotion,
                    MotionCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

The device signature will tell you if the device match the quirk.

Regards.

sassiahmed, 2022-01-24:
Hi @javicalle,

Thank you for your reply.
I tried the quirk but always same thing (status always "clear" 🤷‍♂️).

This is the device signature with the proposed quirk : 
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_7hfcudw5",
  "model": "TS0601",
  "class": "tuya.ts0601_motion.NeoMotion"
}
```

Regards,

javicalle, 2022-01-24:
Could you set the trace level of `zhaquirks` to debug and attache the logs?

It may also be that some event is being generated, but that part of the code I am not able to fully unsderstand...

sassiahmed, 2022-01-26:
Hi @javicalle,
zhaquirks set to debug.
I have this line in the logs about zhaquirks : 
```
2022-01-26 01:49:48 DEBUG (MainThread) [zhaquirks.xiaomi] xx:xx:xx:xx:xx:xx:xx:xx - Attribute report. attribute_id: [65281] value: [{'0xff01-100': <Bool.false: 0>, '0xff01-101': <Bool.false: 0>, '0xff01-110': 0, '0xff01-111': 255, 'battery_voltage_mV': 3300, 'temperature': 19, 'X-attrib-5': 13, '0xff01-8': 13850, 'path': 0, '0xff01-153': 0, '0xff01-155': 0}]
```
all the other lines are about loading native and custom quirks when starting the core :
```
....
2022-01-26 01:45:14 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing.motion
2022-01-26 01:45:14 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/zigbee/quirks
2022-01-26 01:45:14 DEBUG (MainThread) [zhaquirks] Loading custom quirks module tuya
2022-01-26 01:45:14 DEBUG (MainThread) [zhaquirks] Loading custom quirks module tuya.ts0601_motion 
```

Regards,

javicalle, 2022-01-26:
> I have this line in the logs about zhaquirks :
>
> ```
> 2022-01-26 01:49:48 DEBUG (MainThread) [zhaquirks.xiaomi] xx:xx:xx:xx:xx:xx:xx:xx - Attribute report. attribute_id: [65281] value: [{'0xff01-100': [...]

This trace must not be from the Neo.
Is it possible that you have a Xiaomi/Aqara sensor configured?

You can try to add traces to see what is happening.
At your quirk file, modify the `TuyaManufacturerClusterMotion` adding some logs:
```python
class TuyaManufacturerClusterMotion(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Motion device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufClusterAttributes.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_cmd = args[0]
        self.debug("handle_cluster_request--> hdr: %s, args: %s", hdr, args)
        if hdr.command_id == 0x0001 and tuya_cmd.command_id == 1027:
            self.endpoint.device.motion_bus.listener_event(MOTION_EVENT)
```

Delete cache, restart HA and check in your logs for the sensor events.

javicalle, 2022-01-26:
It seems that is already supported in zigbee-herdsman:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9e6a47f0a7331a5e0548e9df0f83ccd1d7ce130/converters/fromZigbee.js#L2430..L2456

javicalle, 2022-01-26:
Can you try with this quick and dirty implementation:
[ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/7941318/ts0601_motion.zip)

vietpv89, 2022-01-28:
> Can you try with this quick and dirty implementation: [ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/7941318/ts0601_motion.zip)

I replaced old file ts0601_motion vy your new and restart HA but my device still don't have any entities.
i'm newbie and please help me for solve this problem. Thanks !

javicalle, 2022-01-28:
I'm sorry for the trouble, but try with the version from @MattWestb comment:
* https://github.com/zigpy/zha-device-handlers/issues/1312#issuecomment-1023286613

I have not yet combined the 2 versions in a single file.

Thank you.

sassiahmed, 2022-01-28:
Hi @javicalle,

Thank you for your analysis.

I tried the ts0601_motion.py sent by @MattWestb.
I added my device : 
```
MODELS_INFO: [
            ("_TZE200_mrf6vtua", "TS0601"),
            ("_TZE200_7hfcudw5", "TS0601"),
        ],
```
The bad news, is nothing change on the status of the entity ==> always clear (even when dancing in front of the sensor 🕺).
the good news, I have these logs : 
```
2022-01-28 10:44:20 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:06:7c:a4:fa - Received motion event message
2022-01-28 10:45:30 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:06:7c:a4:fa - Resetting motion sensor

```
at your disposal to try other modifications,

Best regards

javicalle, 2022-01-28:
Sorry the trouble, but @sassiahmed you mut test with my version of file from
* https://github.com/zigpy/zha-device-handlers/issues/1236#issuecomment-1022104724

You are 2 users with 2 diferent devices. MattWestb and I will create an unic PR for the 2 devices, but first we need that every of you to test yours file version.

sassiahmed, 2022-01-28:
Ok sorry for the misunderstanding @javicalle,

I have the same status (unchanged) and same logs after loading your quirk.
```
2022-01-28 11:45:12 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:06:7c:a4:fa - Received motion event message
2022-01-28 11:46:22 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:06:7c:a4:fa - Resetting motion sensor
```

javicalle, 2022-01-28:
Could you enable the logs for the quirk and attach the trace?

You should have traces with the message `handle_cluster_request-->`

sassiahmed, 2022-01-28:
just to understand well : 
this is what you mean by "enable the logs" ? 
```
logger:
  default: info
  logs:
    ...
    zhaquirks: debug
```

javicalle, 2022-01-28:
If you have a custom quirk folder, also include:
```
logger:
  default: info
  logs:
    ...
    zhaquirks: debug
    custom_zha_quirks: debug
```

And probably `zigpy.zcl: debug` is needed, but this can be very verbose.

sassiahmed, 2022-01-28:
Thank you @javicalle,
It works only when I set `zigpy.zcl: debug`

Here is the trace : (begining with the first time I have `handle_cluster_request`)

```
2022-01-28 12:13:47 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=89 command_id=2>, args: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 39])]
2022-01-28 12:13:58 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=90 command_id=2>
2022-01-28 12:13:58 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 39])]
2022-01-28 12:13:58 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=90 command_id=2>, args: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 39])]
2022-01-28 12:14:08 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=91 command_id=2>
2022-01-28 12:14:08 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 40])]
2022-01-28 12:14:08 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=91 command_id=2>, args: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 40])]
2022-01-28 12:14:19 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=92 command_id=2>
2022-01-28 12:14:19 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 40])]
2022-01-28 12:14:19 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=92 command_id=2>, args: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 40])]
2022-01-28 12:14:22 DEBUG (MainThread) [zigpy.zcl] [0x312f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=138 command_id=Command.Read_Attributes>
2022-01-28 12:14:22 DEBUG (MainThread) [zigpy.zcl] [0x312f:1:0x000a] ZCL request 0x0000: [[0]]
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.bootstrap] Home Assistant initialized in 54.53s
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.core] Starting Home Assistant
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.setup] Setting up google_assistant
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.setup] Setup of domain google_assistant took 0.0 seconds
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.core] Timer:starting
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.components.zeroconf] Starting Zeroconf broadcast
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.components.automation.lights_set_time_helpers_values_based_on_sunrise_sunset] Initialized trigger Lights - Set time Helpers' values based on Sunrise/Sunset
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.components.automation.lights_time_helpers_edition_enabled] Initialized trigger Lights - Time Helpers - edition enabled
2022-01-28 12:14:29 INFO (MainThread) [homeassistant.components.automation.lights_time_helpers_edition_disabled] Initialized trigger Lights - Time Helpers - edition disabled
2022-01-28 12:14:34 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=93 command_id=2>
2022-01-28 12:14:34 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 42])]
2022-01-28 12:14:34 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=93 command_id=2>, args: [Command(status=0, tsn=0, command_id=617, function=0, data=[4, 0, 0, 0, 42])]
2022-01-28 12:14:40 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=94 command_id=2>
[... cut ...]
2022-01-28 12:16:14 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
2022-01-28 12:16:14 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=103 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
```

javicalle, 2022-01-28:
In your local `ts0601_motion.py` change this value:
```python
MOTION_EVENT_NEO = 617
```

sassiahmed, 2022-01-28:
same thing : status always clear
here is the trace after restarting : 
```
2022-01-28 12:37:51 DEBUG (MainThread) [zigpy.zcl] [0x312f:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=1 command_id=Command.Read_Attributes_rsp>
2022-01-28 12:37:52 DEBUG (MainThread) [zigpy.zcl] [0x312f:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Read_Attributes_rsp>
2022-01-28 12:37:53 DEBUG (MainThread) [zigpy.zcl] [0x312f:4:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Read_Attributes_rsp>
2022-01-28 12:37:56 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=2>
2022-01-28 12:37:56 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=357, function=0, data=[1, 0])]
2022-01-28 12:37:56 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=2>, args: [Command(status=0, tsn=0, command_id=357, function=0, data=[1, 0])]
2022-01-28 12:38:06 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=2>
2022-01-28 12:38:06 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
2022-01-28 12:38:06 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
2022-01-28 12:38:08 DEBUG (MainThread) [zigpy.zcl] [0x312f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=165 command_id=Command.Read_Attributes>
2022-01-28 12:38:08 DEBUG (MainThread) [zigpy.zcl] [0x312f:1:0x000a] ZCL request 0x0000: [[0]]
2022-01-28 12:38:17 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=24 command_id=2>
2022-01-28 12:38:17 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
2022-01-28 12:38:17 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=24 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
2022-01-28 12:38:27 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
2022-01-28 12:38:27 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
2022-01-28 12:38:27 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
2022-01-28 12:38:38 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=2>
2022-01-28 12:38:38 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
2022-01-28 12:38:38 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 229])]
2022-01-28 12:38:38 INFO (MainThread) [homeassistant.bootstrap] Home Assistant initialized in 55.24s
2022-01-28 12:38:38 INFO (MainThread) [homeassistant.core] Starting Home Assistant
[... cut ...]
2022-01-28 12:39:20 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
2022-01-28 12:39:20 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] handle_cluster_request--> hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=30 command_id=2>, args: [Command(status=0, tsn=0, command_id=616, function=0, data=[4, 0, 0, 0, 228])]
```

javicalle, 2022-01-28:
It is necessary to be able to identify the values of the `command_id`\DP with each of the functionalities.

It would be very helpful if you could add the traces of the device after powering it off and on again. At that moment, the device usually sends a report of all the device values, which helps us to have an image of all the DP of the device.

javicalle, 2022-01-30:
For future references:

* blakadder.com reports that this device supports:
  * battery
  * humidity
  * tamper
  * temperature
  * occupancy
* herdsman-converters implements endpoints for all entities
  * https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9e6a47f0a7331a5e0548e9df0f83ccd1d7ce130/converters/fromZigbee.js#L2430..L2456
* at https://github.com/zigpy/zha-device-handlers/issues/1312#issuecomment-1024192556 there is an example of implementation with the `TuyaNewManufCluster` class

javicalle, 2022-01-30:
@sassiahmed, @vietpv89 I've created a PR that should support this device and allow display the occupancy, temperature and humidity values.
You should also be able to query the values of `battery_level` and `tamper` (whatever that defines) as attributes of NeoMotionManufCluster.

If you would be so kind as to validate that the implementation shows the correct values, it would be of great help.

Thanks in advanced.

sassiahmed, 2022-01-31:
Hi @javicalle 
Thank you for all your investigation.
Where can I find out the PR to test it please ?

javicalle, 2022-01-31:
Ouuuch, here:
* https://github.com/zigpy/zha-device-handlers/pull/1315

sassiahmed, 2022-01-31:
I don't have a good news 😔
Here is a screenshot what the new quirk gives : new entities appear but not well valued.

I can tell you that I don't have 0,2°C in my room 😊. 
[image]

sassiahmed, 2022-01-31:
I wrote the comment very quickly !

The occupancy seems to work well 👍
[image]

sassiahmed, 2022-01-31:
Hi @javicalle ,

I made some modifications on the quirk and i think it's ok now. I have to verify how the values evolve along this day.

here is the modification : 
```
        104: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            lambda x: x * 10,
        ),
        105: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            lambda x: x * 100,
```
[image]

javicalle, 2022-01-31:
Hi @sassiahmed thanks for testing. I'll add your changes to the PR.

Could you also check the 2 cluster attributes `battery_level` and `tamper`?. 
From device configuration --> Manage Clusters --> look for attributes from `NeoMotionManufCluster`.

Not sure how `tamper` attribute can be checked. At least verify that it is present.
The `battery_level` attribute should change if connected via USB.

I'd also appreciate it if you could check the traces for any `No 'handle_get_data' tuya handler found for` messages.

sassiahmed, 2022-01-31:
Hi @javicalle 

Here is the attributes from `NeoMotionManufCluster`
[image]

Both of attributes exist but I can't view any value (same thing for both)
FYI, my devise is always connected via USB.

and below is the traces for `No 'handle_get_data' tuya handler found for` (30 min of logs)
```
2022-01-31 15:00:29 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:00:29 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:00:29 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:00:29 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:07:16 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:07:16 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:07:16 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:07:16 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:07:17 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:07:17 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:07:18 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:07:18 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:14:05 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:14:05 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:14:05 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:14:05 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:14:06 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:14:06 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
2022-01-31 15:14:06 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:14:06 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
[... cut ...]
2022-01-31 15:27:44 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))
2022-01-31 15:27:44 WARNING (MainThread) [zigpy.zcl] [0x6077:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=0, dp=113, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>))]
```

javicalle, 2022-01-31:
@sassiahmed thanks again for testing.

> Both of attributes exist but I can't view any value (same thing for both)

Just to be sure, once you select the attribute, have you press the "GET ZIGBEE ATTRIBUTE"?
[image]

> FYI, my devise is always connected via USB.

So the `battery_level` attribute should have a value of 4 (or something like NeoBatteryLevel.USB_POWER: 4). It should also change if unplugged from USB.

Your logs  show that there is also a DP with value 113 which is of type ENUM and has value 3. I'm not sure, but I think it is a response (`set_data_response`) to some request from the device.
Without knowing what it corresponds to, it is useless. I can add it as an 'extra' unidentified attribute in case someone gets to know its meaning in the future.

sassiahmed, 2022-02-01:
Hi @javicalle 
`Just to be sure, once you select the attribute, have you press the "GET ZIGBEE ATTRIBUTE"?`
Yes, I did and I did again after your message 😊.
No value appears

For example, in the `TuyaOccupancySensing`, I have this value for `occupancy` attribute : 
[image]

javicalle, 2022-02-01:
> Yes, I did and I did again after your message 😊.

Thanks again for testing 😅 

I just went again through all the logs in this issue and see no indication that those attributes never exist. If I added them, it must have been because they exist in the 'herdsman-converters' library and I must have accepted them as good. Now I see no reason to keep them.

What I am going to keep is the attribute related to `dp=113`. It may be related to one of those 2 attributes (`tamper`, `battery_level`). Or it may be a DP who doesn't use the device and is useless. 🤷🏻‍♂️ 
For now I'll keep it in case someone can deduce its value.

Later tonight I will upload the new version.

sassiahmed, 2022-02-01:
Thank you very much for your help @javicalle.
I will close this issue right now 👍

I'll try to see the values of the attribute related to dp=113 when the version is updated and keep you informed if it makes sens.

closed 2022-02-01 by sassiahmed

bsfaxi, 2022-02-02:
Hi there,
I have tested the quirk with the latest modifications yesterday, with and without the USB plugged, and I have the same results as @sassiahmed.
So thank you for your great support on this.
[image]

javicalle, 2022-02-02:
Thanks @bsfaxi for confirming. 🚀 
Can you check which value do you have for attribute `dp_113` in cluster `NeoMotionManufCluster`?

sassiahmed, 2022-02-02:
Hi all,
thanks @bsfaxi for your test.

On my side, the `dp_113` attribute gives `None` (same thing as Temper and Battery level).
[image]

javicalle, 2022-02-02:
> On my side, the `dp_113` attribute gives `None` (same thing as Temper and Battery level).

Ummmm, that is weird.
In your traces you had messages every few minutes, so the value should be updated...
Something suspicious in your traces?

sassiahmed, 2022-02-02:
searching for 113 in the logs, I found only this line (many times) : 
```
2022-02-02 19:46:15 DEBUG (MainThread) [zigpy.zcl] [0x6077:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=113 command_id=2>
```

javicalle, 2022-02-02:
No that's not it, it should be something like `dp=113` (`tsn` is the transaction number and is an incremental value between 0 - 255)
The 'problem' is that (in the current implementation) if all goes well no log is generated, only if something goes wrong.
No traces = the message is processed correctly, but if the attribute does not update, maybe I have missed something...

PR #1315 (2022-02-15): Add some ts0601_motion devices (with Tuya MCU Cluster approach)
This PR will add support for some `ts0601_motion` devices.

The new devices will follow the new 'Tuya MCU Cluster' approach to update device attributes.

Fixes: #1236
Reported motion sensor 'NAS-PD07' manufactured by NEO.
Most of the code came from 'herdsman-converters'
Those DPs that have been possible are reported as a new cluster.
Those that I have not found a match for are reported as attributes of the MCU cluster.

Not sure about possible `tamper` values. Probably there are diferent values >0 with other meanings.

Fixes: #131 [... cut]