zigpy/zha-device-handlers #1244: [Device Support Request] L&S Emotion Home Base Gateway

dieneuser, 2022-01-01:
**Is your feature request related to a problem? Please describe.**
I have this zigbee device/gateway to control/integrate my existing kitchen countertop lights/LED drivers by L&S to zigbee. The light color can be controlled with the ZHA integration. But the brightness can not be controlled. The light does not change the brightness.
https://zigbeealliance.org/zigbee_products/ls-emotion-home-base/
It is supposed to be fully compatible with Philips Hue.

The CW/WW LEDs are controlled antiparallel with only two connecting wires. Unfortunately, there are no LED drivers for this (anymore). Therefore, this gateway is the only way to use the LEDs with Zigbee.

**Describe the solution you'd like**
It would be great if the brightness can be controlled as well.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4686, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LS Deutschland GmbH",
  "model": "Frankfurt",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Link to the manual: https://cdn.highspeed-network.com/1001/1576109026/media/products/original/S13812_S13812_300.pdf
Link to the product sheet (only german): https://www.opo.de/medias/50.592.30-CAT-LS-Emotion-Home-Base-Werbebroschuere-2019-v3.pdf?context=bWFzdGVyfHJvb3R8NDk0NzQ4M3xhcHBsaWNhdGlvbi9wZGZ8aGZjL2hmZi84OTYyMDQyMTAxNzkwLnBkZnxlM2IyMGQ3MDQxYmFiNDgwZWJiMmJhNmJhODQxNjYzYjk5NDgxNTI0ZDZmODhmYTExOTJmNWM3MTMxM2E4ZWU3

LordDoggo, 2024-10-25:
I have the same issue... and also its not reporting the recent state to homeassistant, If i control it via the remote. Is there already a solution to this?

LordDoggo, 2024-10-25:
Could you please reopen the issue or give a valid solution? Thanks.