zigpy/zha-device-handlers #1249: [BUG] Livarno Lux E27 G95 Filament Bulb doesn't support color change, but that is available in ZHA

RasmusLauritsen, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
The Livarno Lux E27 G95 Filament bulb only supports dimming and on/off. In ZHA it shows up with support for color change as well.
It is this bulb: https://zigbee.blakadder.com/Lidl_HG06463B.html

**Describe the solution you'd like**
I would like to remove the color change option, as the bulb doesn't support it.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
New device 0x2958 (84:2e:14:ff:fe:9b:2b:1f) joined the network
[0x2958] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x83X)\x1f+\x9b\xfe\xff\x14.\x84\x8e'
[0x2958:zdo] ZDO request ZDOCmd.Device_annce: [0x2958, 84:2e:14:ff:fe:9b:2b:1f, 142]
Tries remaining: 3
[0x2958] Requesting 'Node Descriptor'
Tries remaining: 2
[0x2958] Extending timeout for 0xa8 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xa8\x00X)\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0x2958] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x2958] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\xaa\x00X)\x02\x01\xf2'
[0x2958] Discovered endpoints: [1, 242]
[0x2958] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x2958:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xac\x00X)\x1c\x01\x04\x01\x01\x01\x01\x08\x00\x00\x03\x00\x04\x00\x05\x00\x06\x00\x08\x00\x00\x03\x00\x10\x02\x19\x00\n\x00'
[0x2958:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=257, device_version=1, input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096], output_clusters=[25, 10])
[0x2958:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xae\x00X)\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x2958:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x2958:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=176 command_id=Command.Read_Attributes_rsp>
[0x2958] Read model 'TS0501A' and manufacturer '_TZ3000_nosnx7im' from <Endpoint id=1 in=[basic:0x0000, identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, level:0x0008, light_color:0x0300, lightlink:0x1000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x2958] Discovered basic device information for <Device model='TS0501A' manuf='_TZ3000_nosnx7im' nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=True>
Device is initialized <Device model='TS0501A' manuf='_TZ3000_nosnx7im' nwk=0x2958 ieee=84:2e:14:ff:fe:9b:2b:1f is_initialized=True>
Checking quirks for _TZ3000_nosnx7im TS0501A (84:2e:14:ff:fe:9b:2b:1f)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 242}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 242}
device - 0x2958:84:2e:14:ff:fe:9b:2b:1f entering async_device_initialized - is_new_join: True
device - 0x2958:84:2e:14:ff:fe:9b:2b:1f has joined the ZHA zigbee network
[0x2958](TS0501A): started configuration
[0x2958:ZDO](TS0501A): 'async_configure' stage succeeded
Error handling '_save_attribute' event with (84:2e:14:ff:fe:9b:2b:1f, 1, 0, 4, '_TZ3000_nosnx7im') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (84:2e:14:ff:fe:9b:2b:1f, 1, 0, 5, 'TS0501A') params: FOREIGN KEY constraint failed
[0x2958:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x2958:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0x2958:1:0x0300]: bound 'light_color' cluster: Status.SUCCESS
[0x2958:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=184 command_id=Command.Configure_Reporting_rsp>
[0x2958:1:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x2958:1:0x0006]: finished channel configuration
[0x2958:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=186 command_id=Command.Configure_Reporting_rsp>
[0x2958:1:0x0008]: Successfully configured reporting for '{'current_level': (1, 900, 1)}' on 'level' cluster: [ConfigureReportingResponseRecord(status=0)]
[0x2958:1:0x0008]: finished channel configuration
[0x2958:1:0x0000]: finished channel configuration
[0x2958:1:0x0019]: finished channel configuration
[0x2958:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=188 command_id=Command.Configure_Reporting_rsp>
[0x2958:1:0x0300]: Successfully configured reporting for '{'current_x': (30, 900, 1), 'current_y': (30, 900, 1), 'color_temperature': (30, 900, 1)}' on 'light_color' cluster: [ConfigureReportingResponseRecord(status=0)]
[... 26 lines cut ...]
[0x2958:1:0x0008]: received attribute: 0 update with value: 255
[0x2958:1:0x0008]: finished channel initialization
[0x2958:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=198 command_id=Command.Read_Attributes_rsp>
[0x2958:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=200 command_id=Command.Read_Attributes_rsp>
[0x2958:1:0x0300]: finished channel initialization
[0x2958:1:0x0006]: 'async_initialize' stage succeeded
[0x2958:1:0x0008]: 'async_initialize' stage succeeded
[0x2958:1:0x0300]: 'async_initialize' stage succeeded
[0x2958:1:0x1000]: 'async_initialize' stage succeeded
[0x2958:1:0x0000]: 'async_initialize' stage succeeded
[0x2958:1:0x0019]: 'async_initialize' stage succeeded
[0x2958](TS0501A): power source: Mains
[0x2958](TS0501A): completed initialization
[... cut ...] sp>
[0x1182:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=70 command_id=Command.Report_Attributes>
[0x1182:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x1182:1:0x0006] Attribute report received: on_off=0
```

**Additional context**
Add any other context or screenshots about the feature request here.

gromgsxr, 2022-01-05:
i have this issue to with this device and a tuya device they both show as colour capable but are only dimmable

gromgsxr, 2022-01-07:
i was able to get this working by editing one of the tuya quirks here:  https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 shows how to make a custom quirk directory.   

"""Tuya dimmable led controller."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
TuyaManufCluster,
)

class TuyaDimmableColorCluster(CustomCluster, Color):
    """Tuya dimmable led custom cluster."""

    _CONSTANT_ATTRIBUTES = {0x400A: 0}

class DimmableLedController(CustomDevice):
    """Tuya dimmable led controller."""

    signature = {
        MODELS_INFO: [("_TZ3000_nosnx7im", "TS0501A")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=257
            # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
				],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

RasmusLauritsen, 2022-01-07:
> 
It doesn't seem to work for me.
Do I need to add the bulb again?

Or could you perhaps share the code on a place like pastebin?
I think there might be an issue with indentation.

gromgsxr, 2022-01-07:
try this.   https://github.com/gromgsxr/zha-device-handlers/blob/dev/zhaquirks/lidl/TS0501A.py

i just followed the link on how to enable the local quirk folder put the quirk in the folder restarted homeassistant and it loaded the new quirk without having to re add the device

RasmusLauritsen, 2022-01-07:
That fixed it, so I guess it was a indentation problem. Thanks!

closed 2022-01-07 by RasmusLauritsen