zigpy/zha-device-handlers #1260: [Device Support Request] TS0601 GDC311ZBQ1 Tuya Loratap Garage Switcher

tiagoqpinto, 2022-01-06:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_wfxuhoea",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

Tuya device that has no entities/controls in home assistant and its a simple Zigbee Tuya garage switcher. It has a sensor if the garage is opened or not, but it should be very similar to many other devices like this.

Thanks in advance for the support.

EagleAdam, 2022-01-10:
I have one of these as well, happy to help anyone who is able to help integrate it in to zha, I’m not a developer but happy to help in any way I can!

cfnascimento, 2022-01-19:
Hi! We are together in pain...
Someone who have one of these working fine could help to integrate it in to ZHA?

EagleAdam, 2022-01-22:
> Device operation traces are needed to see if it is possible to implement a quirk.
> Each trace must be able to be identified with the action, eg: open, close, stop, etc.
>
> If any of you can include the information, I'll see what can be done.
>
> Regards.

As this is a garage door opener, it doesn't send any messages until a command to open the door is received - it primarily receives messages. How can we capture traces if we can't send it messages?

MattWestb, 2022-01-22:
Its have 3 function and i think its sending all of them the repower the device.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3714/files
You can trying "signal cable" and see if its sending that the door closed then closing the contact.

javicalle, 2022-01-22:
In the herdsman-converter seems that the door status is not reported.
The most similar device doc that I have found it's the 'curtain switch'. Maybe the status report will be the same:
* https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-12-Report%20the%20open%2C%20close%2C%20and%20stop%20status

MattWestb, 2022-01-22:
Its not so easy saying than the device is working like one switch (its parallel with the normal installed) and can only detecting the door is closed or not closed. 
Bets is getting all DPs its using by having it pared in ZHA and taking the power away and reconnecting it  and it shall sending al DPs is using (its still tuya).

javicalle, 2022-01-26:
In case someone wants to try it, it seems that it has already been included as an on/off switch:
* https://github.com/zigpy/zha-device-handlers/pull/1281/files

juanjoSanz, 2022-02-13:
There is a similar device: _TZE200_nklqjk62  "MOES  Garage opener" ( supported by zigbee2mqtt)
difference is that door sensor is wired instead of using RF.

Have anyone tried the changed pointed by @javicalle ?

javicalle, 2022-02-13:
Without logs from device operation, nothing can be done.

Enable logs and attach the relevant information:
```yaml
  zigpy: debug
  zigpy.zcl: debug
  custom_zha_quirks: debug
  zhaquirks.tuya: debug
  homeassistant.components.zha: debug
```

MattWestb, 2022-02-14:
@juanjoSanz Your device is having one GPP end point so you need making one new device class and adding the GPP endpoint and your device IDs.

juanjoSanz, 2022-02-14:
> @juanjoSanz Your device is having one GPP end point so you need making one new device class and adding the GPP endpoint and your device IDs.

Thanks for the advice, I have been trying to define my own class:
https://raw.githubusercontent.com/juanjoSanz/zha-device-handlers/dev/zhaquirks/tuya/ts0601_switch_door.py

With my class the switch control is exposed, I have to figure out how to get the door sensor too.

cfnascimento, 2022-02-15:
Hi!
What should I do to make it work in Zigbee2MQTT?

cfnascimento, 2022-02-15:
It was interviewed and recognized (unsupported), but doesn't exposes none of it's entities...

Smart Home ZigBee Garage Door Opener (GDC311ZBQ1)
por Loratap

Zigbee2MQTT:debug 2022-02-14 23:43:07: Received Zigbee message from 'Garage door', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-14 23:43:07: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_wfxuhoea'
Zigbee2MQTT:warn  2022-02-14 23:43:07: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Tks...

MattWestb, 2022-02-15:
@juanjoSanz can you taking the power off and putting it back so the device is restarting ?
If its being normal its sending the DP that is using and wee needing for fixing the missing parts.
Have some extra logging for ZHA is needed as @javicalle was writing.

javicalle, 2022-02-15:
I think that the appropriate thing would be to create a new file for cover type entities (maybe `ts0601_cover.py`) and implement this class inside (better than in a switch class).
Currently there are already some implementations of Tuya covers, but I can try to create an implementation with the new approach and create the attributes that can be extracted from the logs.

This device is supposed to have 3 DP (garage trigger, door contact and door status):
https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9e6a47f0a7331a5e0548e9df0f83ccd1d7ce130/converters/fromZigbee.js#L7773-L7781

'Door status' seems to be not reliable, but maybe can get some information from logs.

juanjoSanz, 2022-02-16:
If that helps, here are additional logs.

- Home Assistant debug logs for related components (zha, zigpy, quirks...):
https://raw.githubusercontent.com/juanjoSanz/zha-device-handlers/dev/zhaquirks/tuya/ts0601_switch_door.log

-  Logs extracted when initializing device directly from ZHA interface:

```console

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0x5a61:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x5a61:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
New device 0xff96 (a4:c1:38:3b:09:ba:bb:13) joined the network
[0xff96] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x96\xff\x13\xbb\xba\t;8\xc1\xa4\x8e'
[0xff96:zdo] ZDO request ZDOCmd.Device_annce: [0xFF96, a4:c1:38:3b:09:ba:bb:13, 142]
Tries remaining: 3
[0xff96] Requesting 'Node Descriptor'
Tries remaining: 2
[0xff96] Extending timeout for 0x1b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x1b\x00\x96\xff\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0xff96] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xff96] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1c\x00\x96\xff\x02\x01\xf2'
[0xff96] Discovered endpoints: [1, 242]
[0xff96] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0xff96:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1d\x00\x96\xff\x14\x01\x04\x01Q\x00\x01\x04\x04\x00\x05\x00\x00\xef\x00\x00\x02\x19\x00\n\x00'
[0xff96:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
Unknown cluster 61184
[0xff96:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1e\x00\x96\xff\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0xff96:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0xff96:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=31 command_id=Command.Read_Attributes_rsp>
[0xff96] Read model 'TS0601' and manufacturer '_TZE200_nklqjk62' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xff96] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_nklqjk62' nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_nklqjk62' nwk=0xFF96 ieee=a4:c1:38:3b:09:ba:bb:13 is_initialized=True>
Checking quirks for _TZE200_nklqjk62 TS0601 (a4:c1:38:3b:09:ba:bb:13)
Considering <class 'zhaquirks.tuya.ts0601_switch_door.TuyaGarageSwitchTO'>
Found custom device replacement for a4:c1:38:3b:09:ba:bb:13: <class 'zhaquirks.tuya.ts0601_switch_door.TuyaGarageSwitchTO'>
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0xFF96:a4:c1:38:3b:09:ba:bb:13 entering async_device_initialized - is_new_join: True
device - 0xFF96:a4:c1:38:3b:09:ba:bb:13 has joined the ZHA zigbee network
[0xFF96](TS0601): started configuration
[0xFF96:ZDO](TS0601): 'async_configure' stage succeeded
[0xFF96:1:0x0000]: finished channel configuration
[0xFF96:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (a4:c1:38:3b:09:ba:bb:13, 1, 0, 4, '_TZE200_nklqjk62') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (a4:c1:38:3b:09:ba:bb:13, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0xFF96:1:0x0006]: bound 'on_off' cluster: Status.INVALID_EP
[... cut ...]
[0xFF96:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
[0xFF96:1:0x0006]: async_initialize: retryable request #2 failed: . Retrying in 0.8s
[0xFF96:1:0x0006]: initializing channel: from_cache: False
[0xff96:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=72 command_id=2>
[0xff96:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=97, command_id=259, function=0, data=[1, 1])]

```

sigalou, 2022-03-26:
Good morning all,
I also have this GDC311ZBQ1 module? Were you finally able to integrate it? Do we have any leads or is it dead?
Thank you

EagleAdam, 2022-04-30:
So I saw that Zigbee2MQTT now seems to support this device. What's needed to get ZHA to support it?

https://github.com/Koenkk/zigbee2mqtt/issues/11570

BenjaminOe, 2022-05-25:
Same problem here. The device is recognized with ZHA but no entities are available. Would be very happy if there is any solution for it. 😄

jeison-souza, 2022-05-27:
> So I saw that Zigbee2MQTT now seems to support this device. What's needed to get ZHA to support it?
> 
> [Koenkk/zigbee2mqtt#11570](https://github.com/Koenkk/zigbee2mqtt/issues/11570)

Yeap, I saw this too, and works great, I tested in a dev network. But, my real network is under ZHA. I tried create a qirk, but it didn't work at all.

For example, in debug of ZHA I found that the door sensor sends some commands, some bytes changes and I presume that is because it means the timestamp of the command or something like that. For the closing action the last 4 bytes are always the same, for the opening action the last 5 are the same. And always the secondish byte changes.
For the closing action i can see 15 times one command then a single time other command. For the opening, it's just a 15 times the same command. Very weird indeed.

I don't know how to help, I tried to modify the @juanjoSanz file. But didn't worked very well, I had a on_off button, but it only toggles when I press the device button and when I try to switch the button on HA, I get an error saying something command error/unknown

`Closing

Test #1
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\x95\x03\x01\x00\x01\x00'
 1x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\x96\x0c\x04\x00\x01\x02'

Test #2
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\x99\x03\x01\x00\x01\x00'
 1x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\x9a\x0c\x04\x00\x01\x02'

Test #3
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x002\x03\x01\x00\x01\x00'
 1x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x003\x0c\x04\x00\x01\x02'

Opening

Test #1
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00(\x03\x01\x00\x01\x01'

Test #2
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\xda\x03\x01\x00\x01\x01'

Test #3
15x WARNING (MainThread) [zigpy.zcl] [0x9075:1:0xef00] Unknown cluster command 2 b'\x00\x1d\x03\x01\x00\x01\x01'

`

javicalle, 2022-05-28:
Hi, there is my proposal for `_TZE200_nklqjk62` device and maybe others:

<details>

<summary>ts0601_garage.py</summary>

```python
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

ZONE_TYPE = 0x0001

class ContactSwitchCluster(IasZone, TuyaLocalCluster):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
            dp_type=TuyaDPType.VALUE,
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
            dp_type=TuyaDPType.VALUE,
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
            dp_type=TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
            dp_type=TuyaDPType.BOOL,
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
            dp_type=TuyaDPType.ENUM,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_nklqjk62", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

</details>

Implementation must give a switch for open/close (already implemented and working)
Added the contact sensor (DP3). It may not work, but the traces can provide information on its use.
And all the other reported DPs as `TuyaGarageManufCluster` attributes:
* dp_2 (numeric value: 0) unknown function
* dp_4 (numeric value: 10) unknown function
* dp_5 (numeric value: ¿4110?) unknown function
* dp_11 (boolean: 0) unknown function
* dp_12 (enum: 2) garage door status (open, closed, ...)

For anyone who wants to test on another device, add your signature to the `MODELS_INFO`:
```python
        MODELS_INFO: [
            ("_TZE200_nklqjk62", "TS0601"),
        ],
```

javicalle, 2022-05-28:
Probably a `WindowCovering` device would be more appropriate, but not sure how status is updated. I suppose that device doesn't have any 'door position', and `WindowCovering` seems that only have `current_position` but not a `window_status`.

I have a question? This door can be stopped alfway? If yes, how is it? It seems that only accepts open and close buttons (no stop button).

jeison-souza, 2022-05-28:
> Probably a `WindowCovering` device would be more appropriate, but not sure how cluster status is updated. I suppose that device doesn't have any 'door position', and `WindowCovering` seems that only have `current_position` but not a 'window status'.
> 
> I have a question? This door can be stopped halfway? If yes, how is it? It seems that only accepts open and close buttons (no stop button).

In reality, physically the device it only has a "Action button" momentary switch, that activate a relay for 2 seconds then turn it off.  And also a wireless contact switch for door status detection. (Closed or not closed, but with no way to determine the position)

The garage door itself, start a movement if it isn't moving ou stops if it's moving. If you press it again, it inverts the direction, but I guess it will change from depending on the garage motor vendor.

Looking at the zb2mqtt converter code, we can see this:

`switch (dpValue.dp) {
                case tuya.dataPoints.garageDoorTrigger: // dp 1
                    result.action = 'trigger';
                    break;
                case tuya.dataPoints.garageDoorContact: // dp 3
                    result.garage_door_contact = Boolean(!value);
                    break;
                case tuya.dataPoints.garageDoorStatus: // dp 12
                    // This reports a garage door status (open, closed), but it is very naive and misleading
                    break;
                default:
                    meta.logger.debug(`zigbee-herdsman-converters:matsee_garage_door_opener: NOT RECOGNIZED ` +
                        `DP #${dpValue.dp} with data ${JSON.stringify(dpValue)}`);
                }`
                
They don't use the dp 12 for statut as you can see above, they use the 3 (wireless contact switch).

My device model is a little bit different than yours, but I'll give your code a shoot then I'll let you know the results.
"_TZE200_wfxuhoea", "TS0601"

Thanks a lot. !

jeison-souza, 2022-05-28:
Hi @javicalle , unfortunately it didn't work, on the on_off dp_1, I have a "toggle" when I press the button, but when I toggle in HA, nothing is sent to the device, no relay click sound.
The sensor doesn't work as a IAS_Zone, always shows off. But if I put it as TuyaOnOff cluster, it reports the state of the sensor, but in form of a switch. Not ideal, but it is something already.

I'll try to change some bits in your code to see if I can have the command sent to the device at least.

javicalle, 2022-05-29:
So, from your comments (thanks for that) it would be more appropiate represent in HA as a button, isn't it? I am going to see how a button can be defined in Zigpy.
The dp_1 was not already working? or am I confused? The logs from the HA interaction could give to us some clues about whats happening here.

Can you try with another iteration for the sensor? Just add the converter as:
```python

        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=2,
            converter=lambda x: 1 if x else 0,
        ),
```

Can you also check if (with the `IasZone` cluster) the `zone_status` is updating its value?

javicalle, 2022-05-29:
Another random test that can be done is to replace the `TuyaOnOff` cluster:

1. add the `TuyaOnOffNM` import:
```python
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)
```

2. replace the class (in the `replacement` part):
```python
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
```

jeison-souza, 2022-05-30:
Hummmm, I don't have the NonManufacturer versions in my HomeAssistant instalation, weird, My home assistant is up to date 2022.5.5, but the zhaquirks files are not.

[image]

It seems that I'm at v 0.0.73 of zha-quirks.  
I think I need to check how to update the quirks version before continuing debugging. 

I create a Repo to help share the files https://github.com/jeison-souza/Debug_Loratap
But my debugs are very polluted with all other devices, I'll move this to my dev clean instance with a controller that has nothing but this device. Il commit the logs latter.

Thanks again

javicalle, 2022-05-30:
The changes in `NoManufacturerCluster` are recent...

I believe that you can get it with a few new classes:

1. add the `` import from ts0601_dimmer.py:
```python
from zhaquirks.tuya.ts0601_dimmer import NoManufacturerCluster
```

2. create a new local `` implementation:
```python
class TuyaOnOffNM(NoManufacturerCluster, TuyaOnOff):
    """Tuya OnOff cluster with NoManufacturerID."""

```

The rest will be more or less the same.

The `ContactSwitchCluster` is working in other quirk. I can't see why is not working in yours...

AwiaKE, 2022-06-09:
Hi,
I have tried ts0601_garage.py proposal from javicalle and after replacing TuyaOnOff with TuyaOnOffNM currently I can see On/Off switch in HA. Switching on turn the garage relay but switch in HA stays On, while the unit turns relay off after short time. Switch in HA looks like light switch. But it is working.

I can see also Sensor Ias_zone but it is not working. It should work like typical Door/Window sensor.  - Open or closed state.

I have device with wired door sensor like this: https://www.aliexpress.com/item/1005003610757320.html [garage_door_z_desc.txt](https://github.com/zigpy/zha-device-handlers/files/8868653/garage_door_z_desc.txt)

Can you help haw to make sensor work?
Thank you

javicalle, 2022-06-10:
Hello,
  as reported by other users, the operation of the device should be represented by a button and not a switch. The problem is that I haven't found an 'easy' way from the quirk to render a button in HA.

There was an error in the Sensor code (a forgotten `endpoint_id=2,` in DP_3). Please correct it and try again.

In order to help you I need the traces of the device when you interact with it, identifying the events.
I do not have the device and I have no way of verifying its behavior or the operation of the quirk without the traces of each of the events or actions to be fixed.

cvjensen, 2022-06-22:
Hello @javicalle , 

I've just received this device: 
https://www.aliexpress.com/item/1005003610757320.html?spm=a2g0s.12269583.0.0.5fa33ac7Ybi80z

ZHA tells me its a: TS0601 by _TZE200_nklqjk62

If you instruct me what to do I would gladly test and upload any necessary traces etc.

cvjensen, 2022-06-24:
Hi @AwiaKE , 

How did you get HA to use the ts0601_garage.py proposal file from @javicalle .
Right now I have the following in my config:

```
zha:
  custom_quirks_path: config/zhaquirks/
  enable_quirks: true
```

And placed the file under config/zhaquirks/tuya/ts0601_garage.py

But still this is what I get:
[image]

Nothing changes

javicalle, 2022-06-24:
Follow the instructions to enable debug loggin:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Look at the logs while you interact with the device and attach the relevant info. Logs will be very verbose and we nned to identify logs with the events.

@cvjensen there is no need to install `zha-quirks` it would come with the HA instalation.

Try to put the file in `config/zhaquirks/ts0601_garage.py`

cvjensen, 2022-06-25:
@javicalle that helped .

[image]

what about the ts0601_dimmer.py ? should it also be in the same folder as ts0601_garage.py ?

Right now my folder look like this:

[image]

Do I need the init and const files?

cvjensen, 2022-06-26:
@javicalle 

I have added the following to configuration.yaml

```
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

Whenever I turn the "light switch" on nothing happens , but when I turn it off the relay clicks and then after 1 second clicks again . 

The magnet sensor does nothing when I take a cord from one connector to the other .

Which traces / debug informations do you need ?

Something like this?
[garageopener.log](https://github.com/zigpy/zha-device-handlers/files/8986415/garageopener.log)

cvjensen, 2022-06-27:
> There was an error in the Sensor code (a forgotten `endpoint_id=2,` in DP_3). Please correct it and try again.

I finaly understood what you meant here.

Now I have the following:

```
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
```

And it seems to be working fine . 
Now its only the toggle switch which could be great if it were a simple button :-)

javicalle, 2022-06-27:
Sorry for the late response. I have been busy these days.

> Now its only the toggle switch which could be great if it were a simple button :-)

This is an HA 'issue' and I'm not familiar with the code. It is possible that it can be done, but I don't think it will be in the short term.
After switching in HA, the entity returns to the `off` state or do you need to switch off to switch the entity again?

Could you post the actual version of the quirk?
If I'm not wrong, now you have a switch that can trigger the garage door and a contact sensor that report the door sensor state, is it like that?
I hope you'll excuse me, but I'm having a hard time keeping track of all the issues and remembering everything that has been mentioned.

If I remember the information correctly, this device also reports the status of the DPs: 2, 4, 5, 11 and 12
These have been defined as attributes of the `TuyaGarageManufCluster` cluster, but there is no information on the function of each of them.

> Which traces / debug informations do you need ?

It will be better if you try to grep the log with the device ID. In your case it seems to be `0x9acc`. Maybe this way you can get some info from others DPs.

AwiaKE, 2022-06-28:
@javicalle 

I have turned on logs and I try to simulate the door sensor function and extract logs from home-assistant.log

Simulating DOOR OPEN

2022-06-28 15:22:04 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xC904, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=141, SecurityUse=<Bool.false: 0>, TimeStamp=3333797, TSN=0, Data=b'\x09\x63\x02\x00\x99\x03\x01\x00\x01\x01', MacSrcAddr=0xC904, MsgResultRadius=29)
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Received ZCL frame: b'\x09\x63\x02\x00\x99\x03\x01\x00\x01\x01'
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=99, command_id=2, *is_reply=True)
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Decoded ZCL frame: TuyaGarageManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=153, dp=3, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Received command 0x02 (TSN 99): set_data_response(data=TuyaCommand(status=0, tsn=153, dp=3, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=99, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xC904), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=99, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x63\x0B\x02\x00')
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-28 15:22:04 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=99)
2022-06-28 15:22:05 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.APS_NO_ACK: 183>, Endpoint=1, TSN=20)
2022-06-28 15:22:05 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0xC904, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2022-06-28 15:22:05 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)

Simulating DOOR CLOSE

2022-06-28 15:25:28 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xC904, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=147, SecurityUse=<Bool.false: 0>, TimeStamp=322072, TSN=0, Data=b'\x09\x6E\x02\x00\xA8\x03\x01\x00\x01\x00', MacSrcAddr=0xC904, MsgResultRadius=29)
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Received ZCL frame: b'\x09\x6E\x02\x00\xA8\x03\x01\x00\x01\x00'
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=110, command_id=2, *is_reply=True)
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Decoded ZCL frame: TuyaGarageManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=168, dp=3, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Received command 0x02 (TSN 110): set_data_response(data=TuyaCommand(status=0, tsn=168, dp=3, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=110, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy.zcl] [0xC904:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xC904), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=110, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x6E\x0B\x02\x00')
2022-06-28 15:25:28 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
[... cut ...]
2022-06-28 15:25:28 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xED1C](TS0601): Device seen - marking the device available and resetting counter
2022-06-28 15:25:28 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xED1C](TS0601): Update device availability -  device available: True - new availability: True - changed: False

[Garage-door-sensor.log](https://github.com/zigpy/zha-device-handlers/files/9001512/Garage-door-sensor.log)

Hope it hels you identify what is going on.
Thank you

cvjensen, 2022-06-28:
> After switching in HA, the entity returns to the `off` state or do you need to switch off to switch the entity again?

Yes now I have made an automation that automatically switches it back whenever I use it . Not the most beautiful solution but it works .

> Could you post the actual version of the quirk?

```
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff
)

from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

ZONE_TYPE = 0x0001

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
            dp_type=TuyaDPType.VALUE,
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
            dp_type=TuyaDPType.VALUE,
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
            dp_type=TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
            dp_type=TuyaDPType.BOOL,
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
            dp_type=TuyaDPType.ENUM,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_nklqjk62", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

> If I'm not wrong, now you have a switch that can trigger the garage door and a contact sensor that report the door sensor state, is it like that?

Yes thats correct . It is actually working for me . I went from having a device that didnt work at all and now I can use it so thats good . Thanks alot.

> It will be better if you try to grep the log with the device ID. In your case it seems to be `0x9acc`. Maybe this way you can get some info from others DPs.

[garageport.log](https://github.com/zigpy/zha-device-handlers/files/9001894/garageport.log)

AwiaKE, 2022-06-29:
[cvjensen](https://github.com/cvjensen) Do you have door sensor - Open / Closed working? 
For my only switch is working in a way you described, with automation to to turn it off just short time after turn on.

javicalle, 2022-06-30:
I have looked at the logs and that is what I have seen:
* device reports DP_3 (contact sensor) most of the time
* there are some DP_1 (open switch) logs 
* there are some DP_12 (¿door status?) reports, probably related to the DP_1 operation

At some point in the logs, after pairing again the device, more reports are coming from device:
* DP_12: enum (0,1,2)
* DP_2: value=0
* DP_4: value=10
* DP_5: value=3600
* DP_11: boolean=False

⬆️ ⬆️ ⬆️ ⬆️ ⬆️ ⬆️ 
Up to here the facts.

Now my assumptions.
⬇️ ⬇️ ⬇️ ⬇️ ⬇️ ⬇️ 

I believe that DP_2, DP_4, DP_5 and DP_11 are some kind of device configuration.
We know that DP_1 is the switch to activate the garage door motor.
We also know that the DP_3 is the contact sensor.
And the DP_12, could be the 'door status' or maybe motor movement (open, stop, close). The device reports the DP_12 value after 'pressing' the switch. 

We also don't know what function the rest of the DPs do. Surely they are configuration parameters, but we also do not know their function and the values they accept.
⬆️ ⬆️ ⬆️ ⬆️ ⬆️ ⬆️ 

For my part, there is little more I can do. Analyzing the function of the rest of the DPs is a matter of trial and error (and I don't have the device). DPs are available as `TuyaGarageManufCluster` attributes if someone wants to try.

And it is pending, to be able to treat the DP_1 as a button (and not a switch). That requires a specific implementation in HA.

javicalle, 2022-06-30:
Ahhh, and feel free to create the PR to add the quirk. The version from https://github.com/zigpy/zha-device-handlers/issues/1260#issuecomment-1168792380 seems to be the good one. 😉

cchobanIL, 2022-09-30:
Ok all... so, it turns out that I submitted issue 1793 for this very same Device Support Request, and was referred to this request (1260).  That said, I didn't even know that quirks existed or what they were until the last couple of days, since I read up on them.  I'm an IT guy, and 100% understand the concept, but I am no coder.  I see that there are claims of success listed in this thread, getting this device working.

- I have read a primer (external site) on quirks, and "get it", but can't code.
- I have copied the custom quirks/enable quirks code into my configuration.yaml file.
- I see that there seems to be a somewhat complete quirk listed for this device, but I also see someone said they added "#3" code to it, in order to get it to work.  Does that #3 code go after the #242??  (Like I said, not a coder here.)
- Once I understand the full quirk text from what guidance you might give me, I have created a "config/quirks" folder in Home Assistant's config folders, but: 1) in what form/file type to I save the quirk code?  And 2), does it matter if I create subfolders in the quirks folder and/or what those subfolder names are?

Seriously... I know how to learn, but even the primer I read doesn't really tell me these things.  If there's a better source, please do let me know, but if you can be of quick help, that's amazing too.

Thanks!

kimme1024, 2022-10-13:
> Ok all... so, it turns out that I submitted issue 1793 for this very same Device Support Request, and was referred to this request (1260). That said, I didn't even know that quirks existed or what they were until the last couple of days, since I read up on them. I'm an IT guy, and 100% understand the concept, but I am no coder. I see that there are claims of success listed in this thread, getting this device working.
> 
> * I have read a primer (external site) on quirks, and "get it", but can't code.
> * I have copied the custom quirks/enable quirks code into my configuration.yaml file.
> * I have created the config/zhaquirks directory.
> * I see that there seems to be a somewhat complete quirk listed for this device, so I have saved it as config/zhaquirks/tuya/ts0601_garage.py
> * I have uninstalled the device, rebooted HA, and re-discovered it, only to have it be the exact same.
> 
> What am I doing wrong?
> 
> Seriously... I know how to learn, but even the primer I read doesn't really tell me these things. If there's a better source, please do let me know, but if you can be of quick help, that's amazing too.
> 
> Thanks!
> 
> **EDIT: Figured it out! Literally went through the .py file and found a bad indent. Everything came up working after that! Well, thanks for all your past work here!**

Hi, I'm stuck at the same point as you were before your edit. Could you share your solution for the bad indentation?

kimme1024, 2022-11-04:
Hi All, 

As there's I'm being sent back here from my separate issue (https://github.com/zigpy/zha-device-handlers/issues/1870) for another device I hope I'll find a solution here.

My garage door opener (_TZE200_wfxuhoea) works using the quirk made in this issue. However it automatically opens about every hour without any automation referring to it. Also, when there has been a power outage, the garage door opens when power is restored to the device. This is obviously not ok.
The only thing I've adapted in the quirk is the `MODELS_INFO`, the rest is identical. (See below)

```
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff
)

from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

ZONE_TYPE = 0x0001

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
            dp_type=TuyaDPType.VALUE,
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
            dp_type=TuyaDPType.VALUE,
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
            dp_type=TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
            dp_type=TuyaDPType.BOOL,
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
            dp_type=TuyaDPType.ENUM,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_wfxuhoea", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

Any idea's on how to fix this?

javicalle, 2022-11-04:
Please, enable the debug level and attach the logs from your device.
If you can reproduce it , attach the logs from that moment.
From your simptoms I will say that can be some kind the polling mecanism that triggers the command. Can you attach a screenshoot from the device view? Which controls does the device have in that view?

kimme1024, 2022-11-04:
> To enable the debug logging check the documentation:
> 
> * https://www.home-assistant.io/integrations/zha/#debug-logging
> 
> Be carefull because debug level can be very verbose.
> 
> Have you checked wich values take the `binary_sensor.garagepoort` during time? If it goes from unavailable to off it could be that trigger the automation?

Hi,

I've enabled debug for core, zha and zhaquirks, these are the outputs related to the garage door opener when connecting the power (this time the garage door didn't open on power) and on opening, stopping, and closing (signal is the same for all 3):

```
2022-11-04 16:05:48.595 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.garagepoort, old_state=<state binary_sensor.garagepoort=unavailable; device_class=garage_door, friendly_name=Garagepoort @ 2022-11-04T16:03:04.051224+01:00>, new_state=<state binary_sensor.garagepoort=off; device_class=garage_door, friendly_name=Garagepoort @ 2022-11-04T16:05:48.595108+01:00>>
2022-11-04 16:05:48.597 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.garage_door, old_state=<state cover.garage_door=open; device_class=garage, icon=mdi:garage-variant, friendly_name=Garagepoort, supported_features=11 @ 2022-11-04T16:02:53.793409+01:00>, new_state=<state cover.garage_door=closed; device_class=garage, icon=mdi:garage-variant, friendly_name=Garagepoort, supported_features=11 @ 2022-11-04T16:05:48.597051+01:00>>

2022-11-04 16:06:34.022 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=open_cover, service_data=entity_id=cover.garage_door>
2022-11-04 16:06:34.026 INFO (MainThread) [homeassistant.helpers.script.garagepoort] Garagepoort: Running template script
2022-11-04 16:06:34.027 INFO (MainThread) [homeassistant.helpers.script.garagepoort] Garagepoort: Test condition state: True
2022-11-04 16:06:34.027 INFO (MainThread) [homeassistant.helpers.script.garagepoort] Garagepoort: Executing step call service
2022-11-04 16:06:34.028 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=['light.garagepoort']>
2022-11-04 16:06:34.031 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4A34:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-11-04 16:06:34.032 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: turned on: {'on_off': Default_Response(command_id=1, status=<Status.SUCCESS: 0>)}
2022-11-04 16:06:34.032 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.garagepoort, old_state=<state light.garagepoort=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-04T16:05:48.588246+01:00>, new_state=<state light.garagepoort=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-04T16:06:34.032422+01:00>>
2022-11-04 16:06:34.039 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Garagepoort switch auto reset, entity_id=automation.garagepoort_button, source=state of light.garagepoort>
2022-11-04 16:06:34.039 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.garagepoort_button, old_state=<state automation.garagepoort_button=on; last_triggered=2022-10-17T12:05:14.751095+02:00, mode=single, current=0, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-04T16:02:55.567344+01:00>, new_state=<state automation.garagepoort_button=on; last_triggered=2022-11-04T16:06:34.039390+01:00, mode=single, current=1, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-04T16:02:55.567344+01:00>>
2022-11-04 16:06:34.041 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Running automation actions
2022-11-04 16:06:34.041 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step delay 0:00:00.050000

2022-11-04 16:06:34.094 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step device automation
2022-11-04 16:06:34.095 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=light.garagepoort>
2022-11-04 16:06:34.099 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: setting transitioning flag to True
2022-11-04 16:06:34.100 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4A34:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-11-04 16:06:34.100 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: starting transitioning timer for 1.25
2022-11-04 16:06:34.100 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: turned off: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
[... cut ...]
      milliseconds: 50
  - type: turn_off
    device_id: 3a69d3eb16206bc5693f6d6e92defa4e
    entity_id: light.garagepoort
    domain: light
mode: single

```

In the attached screenshot you can see, the device being turned on automatically and the automation reacting to it right after. The strange thing is that after the first time at 1:47 I've unplugged the device (as it was during the night) but even when the device was unplugged, an hour later, at 2:47, HA tried to open the garage door again.

[image]

cloudbr34k84, 2022-11-05:
I just got this device and reading through all the comments it appears the Switch part is still not 100%, is that correct?

kimme1024, 2022-11-09:
Little update:

I've been testing it for a couple of days and somehow it doesn't trigger the relay on power-on and the automatic openings also didn't happen anymore. However I didn't test at night as I don't trust it yet to be safe.
I hope I'll have some time next week to do a 24h test on it and to report back afterwards.

LarrxX, 2022-11-09:
Update for @kimme1024, I installed the quirk and changed the device type to `_TZE200_wfxuhoea`  to match the ID sent by my hardware. No problems so far.

But would definitely be better if the switch behaved like a button without the need of a separate automation.

kimme1024, 2022-11-09:
> Update for @kimme1024, I installed the quirk and changed the device type to `_TZE200_wfxuhoea` to match the ID sent by my hardware. No problems so far.
> 
> But would definitely be better if the switch behaved like a button without the need of a separate automation.

Hi, I've also changed the `MODELS_INFO` to `_TZE200_wfxuhoea` to make it work so it seems that we'd have the same config. It's strange that you're not having the same issues... As for the separate automation, that's not really an issue imo, I'd rather have the quirk working without any issues ;)

kimme1024, 2022-11-15:
Yesterday the garage door opened again without any input. Sadly I wasn't able to get the logs before an automatic restart.
"Luckily" today it happened again.
Here are the logs from around the period it happened:

```
2022-11-15 17:08:25.673 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x57eb](unk_model): Device seen - marking the device available and resetting counter
2022-11-15 17:08:25.673 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x57eb](unk_model): Update device availability -  device available: True - new availability: True - changed: False
2022-11-15 17:08:26.965 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4A34](TS0601): Device seen - marking the device available and resetting counter
2022-11-15 17:08:26.966 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4A34](TS0601): Update device availability -  device available: True - new availability: True - changed: False
2022-11-15 17:08:26.966 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC0D6](TS011F): Device seen - marking the device available and resetting counter
2022-11-15 17:08:26.966 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC0D6](TS011F): Update device availability -  device available: True - new availability: True - changed: False
2022-11-15 17:08:27.527 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCEB8](WB01): Device seen - marking the device available and resetting counter
2022-11-15 17:08:27.527 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCEB8](WB01): Update device availability -  device available: True - new availability: True - changed: False

2022-11-15 17:08:29.773 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.garagepoort, old_state=<state light.garagepoort=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T16:39:28.998846+01:00>, new_state=<state light.garagepoort=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.773432+01:00>>
2022-11-15 17:08:29.776 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Garagepoort switch auto reset, entity_id=automation.garagepoort_button, source=state of light.garagepoort>
2022-11-15 17:08:29.776 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.garagepoort_button, old_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T16:39:28.945493+01:00, mode=single, current=0, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>, new_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00, mode=single, current=1, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>>
2022-11-15 17:08:29.781 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Running automation actions
2022-11-15 17:08:29.782 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step delay 0:00:00.050000
2022-11-15 17:08:29.834 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step device automation
2022-11-15 17:08:29.835 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=light.garagepoort>
2022-11-15 17:08:29.837 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: setting transitioning flag to True
2022-11-15 17:08:29.838 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4A34:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-11-15 17:08:29.838 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: starting transitioning timer for 1.25
2022-11-15 17:08:29.839 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: turned off: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-11-15 17:08:29.839 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.garagepoort, old_state=<state light.garagepoort=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.773432+01:00>, new_state=<state light.garagepoort=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.839390+01:00>>
2022-11-15 17:08:29.846 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.garagepoort_button, old_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00, mode=single, current=1, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>, new_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00,  [... cut ...]
2022-11-15 17:08:39.015 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.atelier_humidity, old_state=<state sensor.atelier_humidity=67.7; unit_of_measurement=%, icon=mdi:water-percent, friendly_name=Atelier Humidity @ 2022-11-15T17:07:38.875223+01:00>, new_state=<state sensor.atelier_humidity=67.8; unit_of_measurement=%, icon=mdi:water-percent, friendly_name=Atelier Humidity @ 2022-11-15T17:08:39.015674+01:00>>
```

I've deleted some entries from my vacuum cleaner that contained some access tokens.
I hope this might be useful for finding the issue.
Thanks in advance!

javicalle, 2022-11-15:
I miss the zigpy traces, but the problem is that they can be very very verbose.

The traces provided seem normal, but nothing is visible before the button changes state, so there is no source information.
Your device (from the screenshoot)) seems to be the `0x4A34`.
```
2022-11-15 17:08:26.965 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4A34](TS0601): Device seen - marking the device available and resetting counter
2022-11-15 17:08:26.966 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4A34](TS0601): Update device availability -  device available: True - new availability: True - changed: False
2022-11-15 17:08:29.773 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.garagepoort, old_state=<state light.garagepoort=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T16:39:28.998846+01:00>, new_state=<state light.garagepoort=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.773432+01:00>>
2022-11-15 17:08:29.776 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Garagepoort switch auto reset, entity_id=automation.garagepoort_button, source=state of light.garagepoort>
2022-11-15 17:08:29.776 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.garagepoort_button, old_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T16:39:28.945493+01:00, mode=single, current=0, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>, new_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00, mode=single, current=1, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>>
2022-11-15 17:08:29.781 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Running automation actions
2022-11-15 17:08:29.782 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step delay 0:00:00.050000
2022-11-15 17:08:29.834 INFO (MainThread) [homeassistant.components.automation.garagepoort_button] Garagepoort switch auto reset: Executing step device automation
2022-11-15 17:08:29.835 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=light.garagepoort>
2022-11-15 17:08:29.837 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: setting transitioning flag to True
2022-11-15 17:08:29.838 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4A34:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-11-15 17:08:29.838 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: starting transitioning timer for 1.25
2022-11-15 17:08:29.839 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: turned off: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-11-15 17:08:29.839 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.garagepoort, old_state=<state light.garagepoort=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.773432+01:00>, new_state=<state light.garagepoort=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Garagepoort, supported_features=0 @ 2022-11-15T17:08:29.839390+01:00>>
2022-11-15 17:08:29.846 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.garagepoort_button, old_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00, mode=single, current=1, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>, new_state=<state automation.garagepoort_button=on; last_triggered=2022-11-15T17:08:29.776529+01:00, mode=single, current=0, id=1665696187259, friendly_name=Garagepoort switch auto reset @ 2022-11-15T08:00:10.234585+01:00>>
2022-11-15 17:08:30.034 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: received onoff Bool.false while transitioning - skipping update
2022-11-15 17:08:30.668 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.garagepoort, old_state=<state binary_sensor.garagepoort=off; device_class=garage_door, friendly_name=Garagepoort @ 2022-11-15T16:39:49.204434+01:00>, new_state=<state binary_sensor.garagepoort=on; device_class=garage_door, friendly_name=Garagepoort @ 2022-11-15T17:08:30.668233+01:00>>
2022-11-15 17:08:30.671 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.garage_door, old_state=<state cover.garage_door=closed; device_class=garage, icon=mdi:garage-variant, friendly_name=Garagepoort, supported_features=11 @ 2022-11-15T16:39:49.207333+01:00>, new_state=<state cover.garage_door=open; device_class=garage, icon=mdi:garage-open-variant, friendly_name=Garagepoort, supported_features=11 @ 2022-11-15T17:08:30.671251+01:00>>
2022-11-15 17:08:31.089 DEBUG (MainThread) [homeassistant.components.zha.entity] light.garagepoort: transition complete - future attribute reports will write HA state
```

In the logs I can't identify any reason why the button triggers.

javicalle, 2022-11-15:
For anyone who might be interested I want to propose a different approach.
### DISCLAIMER:
I do not have the device and that I cannot validate or test the implementation.
The proposed code may (_will_) have errors or simply not work.

The approach I propose has 2 parts:
* a new quirk for the device
* HA implementation for entities

In order to test it, in addition to setting the quirk locally, it is necessary to implement the ZHA integration as a `custom_component` in HA. This is not complicated to do, but requires some experience to be able to identify and analyze any problems. For this reason, I am not going to give instructions on this part, whoever wants to try it must be able to do this part and help me with the errors.

The main advantage that the new approach offers is that it configures the button as a button. It also defines the sensor that was already implemented, although it is possible that the format/look and feel will change from the previous one.

The proposed quirk:

<details open>
<summary>ts0601_garage.py</summary>

```python
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
)

TUYA_MANUFACTURER_GARAGE = "tuya_manufacturer_garage"

class TuyaGarageManufCluster(TuyaMCUCluster):
    """Tuya garage door opener."""

    ep_attribute = TUYA_MANUFACTURER_GARAGE

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF01: ("button", t.Bool, True),
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF03: ("contact_sensor", t.Bool, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TUYA_MANUFACTURER_GARAGE,
            "button",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
[... 107 lines cut ...]
@MULTI_MATCH(channel_names="tuya_manufacturer_garage")
class Garage(BinarySensor):
    """ZHA BinarySensor."""

    SENSOR_ATTR = "contact_sensor"
    _attr_device_class: BinarySensorDeviceClass = BinarySensorDeviceClass.GARAGE_DOOR
```

</details>

<details open>
<summary>zha/button.py</summary>

```python
@MULTI_MATCH(channel_names="tuya_manufacturer_garage")
class GarageButton(ZHAAttributeButton, id_suffix="button"):
    """Defines a ZHA garage button."""

    _attribute_name = "button"
    _attr_name = "Button"
    _attribute_value = 0
    _attr_device_class = ButtonDeviceClass.RESTART
```

</details>

toxic0berliner, 2022-11-24:
wow, took me quite a while because for a beginner even without fear of shell, having ZHA run as a custom component is quite some bar to step up to, downloading only a subdirectory from the core repo, making sure it gets laoded (not renaming it zha_custom...).. Well, all said and done, I think I have it running with the button.py as well as binary_sensor.py and the quirk.

Sadly, I am not getting any controls on it : 
[image]
I have 300kb of homeassistant.log in which there is a removal then pairing of the garage dorr in question.
Sadly I fear zha logs are at warning only since I just modified my config to say this : 
```
logger:
  default: warning
  logs:
    custom_components.zha_toolkit: debug
    custom_components.zha: debug #just added this was not enabled during my first attempt
```

As you see I have also just installed zha_toolkit hoping to get you some juicy logs is device specific communication is ongoing.

There's a lot of stuff including the key of my zigbee network in the logs, I'm going to msg you directly, I have little fear you come hack me at home but I'd rather not post it widely here ;)

I'll send homeassistant.log as it's the only thing I know of right now but please feel free to point me to anything I need to share to help ! (running hass_OS on qemu VM on PVE normally everything updated to the latest stable)

javicalle, 2022-11-24:
@toxic0berliner thanks for testing.

Probably this is also needed, modify this part:
<details open>
<summary>zha/core/channels/manufacturerspecific.py</summary>

```python
@registries.CHANNEL_ONLY_CLUSTERS.register(registries.TUYA_MANUFACTURER_CLUSTER)
@registries.ZIGBEE_CHANNEL_REGISTRY.register(registries.TUYA_MANUFACTURER_CLUSTER)
class TuyaChannel(ZigbeeChannel):
    """Channel for the Tuya manufacturer Zigbee cluster."""

    REPORT_CONFIG = ()

    def __init__(self, cluster: zigpy.zcl.Cluster, ch_pool: ChannelPool) -> None:
        """Initialize TuyaChannel."""
        super().__init__(cluster, ch_pool)

        if self.cluster.endpoint.manufacturer in (
            "_TZE200_7tdtqgwv",
            "_TZE200_amp6tsvy",
            "_TZE200_oisqyl4o",
            "_TZE200_vhy3iakz",
            "_TZ3000_uim07oem",
            "_TZE200_wfxuhoea",
            "_TZE200_tviaymwx",
            "_TZE200_g1ib5ldv",
            "_TZE200_wunufsil",
            "_TZE200_7deq70b8",
            "_TZE200_tz32mtza",
            "_TZE200_2hf7x9n3",
            "_TZE200_aqnazj70",
            "_TZE200_1ozguk6x",
            "_TZE200_k6jhsr0q",
            "_TZE200_9mahtqtg",
        ):
            self.ZCL_INIT_ATTRS = {  # pylint: disable=invalid-name
                "backlight_mode": True,
                "power_on_state": True,
            }
        if self.cluster.endpoint.manufacturer in (
            "_TZE200_nklqjk62",
            "_TZE200_wfxuhoea",
        ):
            self.ZCL_INIT_ATTRS = {  # pylint: disable=invalid-name
                "button": True,
                "contact_sensor": True,
            }
```

</details>

Edit the file adding the last if block, save changes, delete any `__pycache__` folder and restart HA.

toxic0berliner, 2022-11-24:
No more luck sorry.

Got this trying to reconfigure : 
[image]

Tried a fresh join again:

<details>
<summary>homeassistant.log</sumarry>

```
2022-11-25 00:16:07.533 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x08C1, SrcIEEE=a4:c1:38:15:74:26:c7:c8, ParentNwk=0xC177)
2022-11-25 00:16:07.533 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x08C1, SrcIEEE=a4:c1:38:15:74:26:c7:c8, ParentNwk=0xC177)
2022-11-25 00:16:07.534 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x08C1, SrcIEEE=a4:c1:38:15:74:26:c7:c8, ParentNwk=0xC177)
2022-11-25 00:16:07.534 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x08C1, SrcIEEE=a4:c1:38:15:74:26:c7:c8, ParentNwk=0xC177)
2022-11-25 00:16:07.534 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x08C1, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2022-11-25 00:16:07.549 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-25 00:16:07.687 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x08C1, NWK=0x08C1, IEEE=a4:c1:38:15:74:26:c7:c8, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded|RXWhenIdle|MainsPowered|Router: 142>)
2022-11-25 00:16:07.688 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
2022-11-25 00:16:07.688 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x08C1, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=0, MacDst=0xFFFF, Data=b'\xC1\x08\xC8\xC7\x26\x74\x15\x38\xC1\xA4\x8E')
2022-11-25 00:16:07.689 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x08C1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=0, profile_id=0, cluster_id=19, data=Serialized[b'\x00\xc1\x08\xc8\xc7&t\x158\xc1\xa4\x8e'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-25 00:16:07.689 INFO (MainThread) [zigpy.application] New device 0x08c1 (a4:c1:38:15:74:26:c7:c8) joined the network
2022-11-25 00:16:07.689 DEBUG (MainThread) [zigpy.device] [0x08c1] Scheduling initialization
2022-11-25 00:16:07.689 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x08C1 ieee=a4:c1:38:15:74:26:c7:c8 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xc1\x08\xc8\xc7&t\x158\xc1\xa4\x8e'
2022-11-25 00:16:07.690 DEBUG (MainThread) [zigpy.zdo] [0x08c1:zdo] ZDO request ZDOCmd.Device_annce: [0x08C1, a4:c1:38:15:74:26:c7:c8, 142]
2022-11-25 00:16:07.690 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2022-11-25 00:16:07.690 INFO (MainThread) [zigpy.device] [0x08c1] Requesting 'Node Descriptor'
2022-11-25 00:16:07.690 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2022-11-25 00:16:07.691 DEBUG (MainThread) [zigpy.device] [0x08c1] Extending timeout for 0x8b request
2022-11-25 00:16:07.691 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x08C1), dst_ep=0, source_route=None, extended_timeout=True, tsn=139, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'\x8b\xc1\x08'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-25 00:16:07.691 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0x08C1, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2022-11-25 00:16:07.699 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.FAIL: 1>)
2022-11-25 00:16:08.850 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x08C1), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=2, TSN=139, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x8B\xC1\x08')
2022-11-25 00:16:08.888 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-25 00:16:08.925 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.NodeDescRsp.Callback(Src=0x08C1, Status=<Status.SUCCESS: 0>, NWK=0x08C1, NodeDescriptor=NullableNodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False))
2022-11-25 00:16:08.926 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
[... cut ...] EST: 48>, Radius=0, Data=b'\x18\xFA\x01\x07\x00\x00\x23\xCF\xC5\x12\x2B')
2022-11-25 00:16:15.046 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-25 00:16:15.090 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=250)
```

</details>

Before doing that I did remove` __pycache__` from zhaquirks folder, seems there is none in zha custom component code itself:
[image]

javicalle, 2022-11-27:
In my tests the `custom_components` creates a `__pycache__` folder.
Can you check that do you have copied all the `components/zha` folder?
* https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

At the startup log you have to get a WARNING log like:
```
2022-11-25 12:10:52.348 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration zha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

toxic0berliner, 2022-11-28:
I used https://download-directory.github.io/ to download the zha subfolder, so I should be fine but just in case : 
[image]
<details>
<summary>/tmp/zha-files.txt</summary>

```
./zha
./zha/config_flow.py
./zha/strings.json
./zha/translations
./zha/translations/hu.json
./zha/translations/fy.json
./zha/translations/sk.json
./zha/translations/sl.json
./zha/translations/cs.json
./zha/translations/hr.json
./zha/translations/uk.json
./zha/translations/nl.json
./zha/translations/ar.json
./zha/translations/ja.json
./zha/translations/zh-Hans.json
./zha/translations/et.json
./zha/translations/th.json
./zha/translations/de.json
./zha/translations/bg.json
./zha/translations/bn.json
./zha/translations/el.json
./zha/translations/pt-BR.json
./zha/translations/ru.json
./zha/translations/fr.json
./zha/translations/nn.json
./zha/translations/da.json
./zha/translations/es.json
./zha/translations/sv.json
./zha/translations/en.json
./zha/translations/it.json
./zha/translations/id.json
./zha/translations/lb.json
./zha/translations/he.json
./zha/translations/zh-Hant.json
./zha/translations/pl.json
./zha/translations/pt.json
./zha/translations/tr.json
./zha/translations/es-419.json
./zha/translations/no.json
./zha/translations/fi.json
./zha/translations/ca.json
./zha/translations/ko.json
./zha/entity.py
./zha/light.py
./zha/select.py
./zha/fan.py
./zha/climate.py
./zha/button.py
./zha/services.yaml
./zha/alarm_control_panel.py
./zha/number.py
./zha/sensor.py
./zha/home-assistant core dev homeassistant-components_zha.zip
./zha/device_tracker.py
./zha/core
./zha/core/discovery.py
./zha/core/decorators.py
./zha/core/device.py
./zha/core/registries.py
./zha/core/channels
./zha/core/channels/manufacturerspecific.py
./zha/core/channels/base.py
./zha/core/channels/protocol.py
./zha/core/channels/smartenergy.py
[... 656 lines cut ...]
2022-11-28 08:44:59.317 DEBUG (MainThread) [zigpy.zcl] [0xC177:1:0x0b04] Received command 0x0A (TSN 217): Report_Attributes(attribute_reports=[Attribute(attrid=0x050F, value=TypeValue(type=uint16_t, value=490))])
2022-11-28 08:44:59.317 DEBUG (MainThread) [zigpy.zcl] [0xC177:1:0x0b04] Attribute report received: apparent_power=490
2022-11-28 08:45:00.843 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x92E2, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=236, SecurityUse=<Bool.false: 0>, TimeStamp=12187281, TSN=0, Data=b'\x09\xF4\x02\x00\x39\x01\x04\x00\x01\x01', MacSrcAddr=0x92E2, MsgResultRadius=29)
2022-11-28 08:45:00.843 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x92E2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\xf4\x02\x009\x01\x04\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=236, rssi=None)
2022-11-28 08:45:00.843 DEBUG (MainThread) [zigpy.device] Ignoring message (b'09f40200390104000101') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:af:64:30:62:59, 1)'
2022-11-28 08:45:01.090 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4CB7](TS130F): Device seen - marking the device available and resetting counter
2022-11-28 08:45:01.090 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4CB7](TS130F): Update device availability -  device available: True - new availability: True - changed: False
2022-11-28 08:45:01.091 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x453F](TS130F): Device seen - marking the device available and resetting counter
2022-11-28 08:45:01.091 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x453F](TS130F): Update device availability -  device available: True - new availability: True - changed: False

```
</details>

I believe you might be true in thinking somehow I'm not using the custom_component or it's not overwriting the zha fully or properly, but I'm still very new to this sorry. I just added a custom log in the async_setup_entry of the zha custom component to check : on line 104 of __init__.py I added `_LOGGER.debug("TOXIC: loading async_setup_entry of the custom component zha")`
I also have this in my `configuration.yml` : 
```
logger:
  default: warning
  logs:
    custom_components.zha_toolkit: debug
    custom_components.zha: debug
```
So it should show up but in fact there is nothing in homeassistant.log when I search for "TOXIC"... :(

Maybe I did need more help in "setting up zha as a custom component" sorry :'(

javicalle, 2022-11-28:
ZHA isn't loading:
```
2022-11-28 08:07:57.657 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration zha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-11-28 08:07:57.658 ERROR (SyncWorker_0) [homeassistant.loader] The custom integration 'zha' does not have a version key in the manifest file and was blocked from loading. See https://developers.home-assistant.io/blog/2021/01/29/custom-integration-changes#versions for more details
```

Edit the `./zha/manifest.json` file and add your current HA version like:
```json
  "version": "2022.11.4",
```

[139 more replies]

closed 2025-08-20 by github-actions[bot]