zigpy/zha-device-handlers #1261: [BUG] Ikea E1810 remote matching wrong quirk

bwickman97, 2022-01-07:
**Describe the bug**
I have an Ikea E1810 remote (the 5 button one) that is only partially working. The long press events for the "dim" and "brighten" (upper and lower) buttons don't trigger automations in HA. The quirk applied is zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1 but I believe that it should be zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote3

The reason I believe that it should be matching the IkeaTradriRemote3 quirk instead is because when I look at the events sent during a long press the "args" array contains an 84 instead of an 83 which appears to be the replacement for the v3 vs the v1 quirk.

When I check the device signature (copied below), it also seems like it should be matching quirk v3 as well, but it does not (log output below).

I've already done a firmware update and that did not make a difference. I've also done a Reconfigure Device which ran successfully but also did not fix the issue.

I apologize if this is not truly a bug but something that I've overlooked, I'm still new to HA and Zigbee.

**Expected behavior**
ZHA should match the v3 quirk instead of the v1 quirk to allow using the long press events on the dim/brighten buttons.

**Additional context**
FW: `0x23080631`

Device signature:
```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1"
}
```

Logging:
```2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden TRADFRI remote control (b4:e3:f9:ff:fe:7c:37:45)
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote3'>
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Fail because output cluster mismatch on at least one endpoint
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote2'>
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1'>
2022-01-06 00:04:19 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for b4:e3:f9:ff:fe:7c:37:45: <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1'>
```

Sample long press event:
```{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "b4:e3:f9:ff:fe:7c:37:45",
        "unique_id": "b4:e3:f9:ff:fe:7c:37:45:1:0x0008",
        "device_id": "6fdf4a9c901e73001d724bc2c2ccf1c0",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_with_on_off",
        "args": [
            0,
            84
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-01-07T01:31:47.819668+00:00",
    "context": {
        "id": "d4e09a199d396f4f437ab0c4f8b92d33",
        "parent_id": null,
        "user_id": null
    }
}
```

bwickman97, 2022-01-07:
Forgot to include: Home Assistant is version 2021.12.0, python is 3.9.7, and zha-quirks is version 0.0.65

MattWestb, 2022-01-07:
I think ZHA is using the "old" cluster that was red then pairing the device then it was pared in your system. I have seen device that must being repaired after firmware upgrade then its have changing its endpoint configuration and was not working OK.
I have 2 5 buttons and one was updated in IKEA GW then moved to ZHA and the other was  in my test system so its being repaired many times before and after so its working with the new settings. Also little strange in My production system is that the old ZLL E27 CWS 600 lm have being updated to Zigbee 3 and ZHA have reading the new endpoint settings and its working OK but it can being that is have doing some magic after the OTA.

Try open the network for joining and reset the remote with 4 * pressing the pairing button and see if its getting the new cluster setting (then you is not loosing your name and automations but binding is gone) and looking if the device signature is updating to the new one. If OK you need restart HA for the quirk being rematched.
If its not working you need deleting the device and waiting one minute and adding it (or better restart HA between) and look if the cluster is updated.
It can also being that the 5 button is having wrong data setting like the shortcut button is 2 user having devices without window covering cluster that is coming from data setting made in the factors and is for the open/close button and both have making one local quirk that is fixing it (the open/close button also having one batch with wrong name (KA) that is also coming from wrong data setting in the factory).

bwickman97, 2022-01-08:
Thanks so much for responding. Rejoining didn't work until I completely removed the device, restarted HA and then joined it again after which it finally matched to the v3 quirk. I'm now able to use the long press events in automations. I think when I had tried re-adding it before I never thought to restart HA first. Much appreciated and thanks to all who have contributed to this project.

closed 2022-01-08 by bwickman97