zigpy/zha-device-handlers #1265: [Device Support Request] Bosch Motion Detector RFDL-ZB-MS (battery percentage = unknown)

Pau1ey, 2022-01-08:
**Is your feature request related to a problem? Please describe.**
Does not display remaining battery percentage for [HA integration: Zigbee HubZ Smart Home Controller - Zigbee Home Automation](https://www.home-assistant.io/integrations/zha)

**Describe the solution you'd like**
Would like to see battery percentage remaining battery as I am using several of these in my HA setup.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4403, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0400",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Bosch",
  "model": "RFDL-ZB-MS",
  "class": "zigpy.device.Device"
}`

**Additional context**

- The other features: motion detection, temperature, and illuminance work properly. Only the battery is unknown.  Unpairing and repairing, fully rebooting did not fix the issue.  Also for what it's worth, this model has 2 new batteries inserted but shows as in developer tools below as "battery_quantity: 1".

- Developer tools shows this under State attributes (YAML, optional):

`state_class: measurement
battery_size: Other
battery_quantity: 1
battery_voltage: 2.8
unit_of_measurement: '%'
device_class: battery
friendly_name: Motion Sensor battery`

- This is some what similar to [issue #25361](https://github.com/home-assistant/core/issues/25361) but that was a different model.

- Please note this is my first request, so if more info is needed just tell me what to provide and possibly where to find it. Thank you in advance!

MattWestb, 2022-01-08:
If you like getting the battery reporting working you can using the old quirk but adding one new device class with updated signature for the new version.
https://github.com/zigpy/zha-device-handlers/blob/b0bbbfb6b1f267b7d87d833b1ba5a5b0b127a9a1/zhaquirks/bosch/motion.py#L27-L70
It shall being 2 empty lined between the first (original) device class and the new and no one after the new (end of file).
Changing the name and signature and if cluster is changed also you need getting then matching the new device.
If all is working ZHA shall loading it then you is restarting HA.

Easiest is using local quirk for testing https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Fixing battery quantity and adding batter types is possible but need more work.
You can looking in the IKEA quirk INIT that i have adding and fixing there remotes batteries how its don.

PS: IKEA remotes is sending false batter % (1/2) and the BOCH is not sending % only voltage and its prity command "defects" in devices.

Pau1ey, 2022-01-14:
@MattWestb After making some changes to that quirk I am now able to see the battery level as shown in the code below but have not pursued the battery quantity since they are in parallel and able to operate with either 1 or 2 installed.  I really appreciate your help and response as I was also able to create a working quirk to also show battery level for the Visonic MCT-350 SMA door/window contact sensors that are from Xfinity/Comcast if that is of interest to anyone I can upload as well.

`"""Device handler for Bosch motion sensors."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PollControl
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.measurement import TemperatureMeasurement, IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import PowerConfigurationCluster
from zhaquirks.bosch import BOSCH
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class BoschPowerConfiguration(PowerConfigurationCluster):
    """Bosch power configuration cluster for motion sensor."""

    MAX_VOLTS = 3.0
    MIN_VOLTS = 1.9

class RFDLZBMS(CustomDevice):
    """Custom device representing Bosch motion sensors."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=402
        #  device_version=0
        #  input_clusters=[0, 1, 3, 20, 400, 402, 500, b05]
        #  output_clusters=[19]>
        MODELS_INFO: [(BOSCH, "RFDL-ZB-MS")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Diagnostic.cluster_id,
                    IasZone.cluster_id,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    PollControl.cluster_id,
                    PowerConfigurationCluster.cluster_id,
                    TemperatureMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Diagnostic.cluster_id,
                    IasZone.cluster_id,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    PollControl.cluster_id,
                    BoschPowerConfiguration,
                    TemperatureMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }`

MattWestb, 2022-01-14:
## GREAT !!

IKEAs motion sensors is also having 2 batteries in parallel and the attribute is shown in ZHA and is for easter replacing batteries so having the right "in stock" and its not saying if they is in serie or parallel.

Can you making one PR for adding the updated quirk so other user can using it in the future ?

Also peas put ` ``` ` on one empty  line before and after the code so its not losing the formation and user can copy it easier.

Pau1ey, 2022-01-14:
Sure, will do.  I updated the code format above with `` ``` ``

cbarth3, 2022-03-26:
@Pau1ey I have been looking for an updated quirk for this Visonic MCT-350 SMA door/window contact sensors that are from Xfinity/Comcast. 

I would greatly appreciate if you could share it. As mine is connecting as URC4460BC0-X-R by Universal Electronics Inc and I have not yet figured out the location of the basic file to try and update it.

TheJulianJES, 2022-04-17:
Is the custom quirk above still working with Home Assistant 2022.4.5?

(Someone could then PR the changes to this repo, so the custom quirk would be merged into Home Assistant.)

closed 2022-10-21 by github-actions[bot]