zigpy/zha-device-handlers #1267: Blinds motor not recognized as such

iskobkarev, 2022-01-08:
It is one more of the big family of TS0601 devices.
Would it be possible to introduce an easy way to edit/update these signatures? 

At the moment I can't find a way to access these files with hassio and portainer is no longer available. 
I'll try custom folder.

Here is my device signature - "YIDI" branded inside-the-blind-tube + battery + rf433 + zigbee motor.

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_zuz7f94z",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

iskobkarev, 2022-01-08:
It shows no controls when added.

schwickster, 2022-01-10:
This needs a quirk I guess seeing as device_type 0x0051 is a SMART_PLUG. You could try to create a quirk yourself and in replacements use zha.DeviceType.WINDOW_COVERING_DEVICE

iskobkarev, 2022-01-10:
Thank you - that is interesting.

I have a hassio version. any guidance how to test the quirk without having
RW access to the host?
I tried this below (from here
https://github.com/zigpy/zha-device-handlers/discussions/693)  but it does
not seem to pick up the files from there.

zha:
  custom_quirks_path: /config/custom_zha_quirks/

iskobkarev, 2022-01-23:
so, after going around it a bit,
I've added a quirk based on tuya_moes window covering (TuyaMoesCover0601)
which had the same input/output clusters and device type, and injected  via
custom_quirks_path config directive.

It is being identified and applied:
2022-01-23 14:29:34 DEBUG (MainThread) [zigpy.quirks.registry] Considering
<class 'tuya_ts0601_cover.TuyaYIDICover0601'>
2022-01-23 14:29:34 DEBUG (MainThread) [zigpy.quirks.registry] Found custom
device replacement for cc:86:ec:ff:fe:b8:b2:41: <class
'tuya_ts0601_cover.TuyaYIDICover0601'>

However when I click the up/down in the Lovelace it goes:
2022-01-23 14:40:00 DEBUG (MainThread) [zhaquirks.tuya]
cc:86:ec:ff:fe:b8:b2:41 Sending Tuya Cluster Command.. Manufacturer is
_TZE200_zuz7f94z Cluster Command is 0x0002, Arguments are ()
2022-01-23 14:40:00 ERROR (MainThread)
[homeassistant.components.websocket_api.http.connection] [2967549456]
'_TZE200_zuz7f94z'
KeyError: '_TZE200_zuz7f94z'

The profile now looks like this:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>,
complex_descriptor_available=0, user_descriptor_available=0, reserved=0,
aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice:
142>, manufacturer_code=4098, maximum_buffer_size=82,
maximum_incoming_transfer_size=82, server_mask=11264,
maximum_outgoing_transfer_size=82,
descriptor_capability_field=<DescriptorCapability.NONE: 0>,
*allocate_address=True, *is_alternate_pan_coordinator=False,
*is_coordinator=False, *is_end_device=False, *is_full_function_device=True,
*is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True,
*is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_zuz7f94z",
  "model": "TS0601",
  "class": "tuya_ts0601_cover.TuyaYIDICover0601"
}

The quirk is attached and any suggestions are appreciated.

Thanks

Igor

-- 
Thanks,

Igor Skobkarev,
***@***.***

"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaYIDICover0601(TuyaWindowCover):
    """Tuya blind controller device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] }
        # },
        # "manufacturer": "_TZE200_zah67ekd",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [("_TZE200_zuz7f94z", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

iskobkarev, 2022-01-23:
The remova/join generates one odd message:

2022-01-23 14:54:55 DEBUG (MainThread) [bellows.zigbee.application]
Received incomingMessageHandler frame with
[<EmberIncomingMessageType.INCOMING_UNICAST: 0>,
EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1,
destinationEndpoint=1,
options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0,
sequence=18), 255, -69, 0xeb8a, 255, 255,
b'\x18\xda\x01\x04\x00\x00B\x10_TZE200_zuz7f94z\x05\x00\x00B\x06TS0601']
2022-01-23 14:54:55 INFO (MainThread) [zigpy.device] [0xeb8a] Read model
'TS0601' and manufacturer '_TZE200_zuz7f94z' from <Endpoint id=1
in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00]
out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2022-01-23 14:54:55 INFO (MainThread) [zigpy.device] [0xeb8a] Discovered
basic device information for <Device model='TS0601'
manuf='_TZE200_zuz7f94z' nwk=0xEB8A ieee=cc:86:ec:ff:fe:b8:b2:41
is_initialized=True>
2022-01-23 14:54:55 DEBUG (MainThread) [zigpy.application] Device is
initialized <Device model='TS0601' manuf='_TZE200_zuz7f94z' nwk=0xEB8A
ieee=cc:86:ec:ff:fe:b8:b2:41 is_initialized=True>
2022-01-23 14:54:55 DEBUG (MainThread) [zigpy.quirks.registry] Checking
quirks for _TZE200_zuz7f94z TS0601 (cc:86:ec:ff:fe:b8:b2:41)
*2022-01-23 14:54:55 DEBUG (MainThread) [zigpy.appdb] Error handling
'_save_attribute' event with (cc:86:ec:ff:fe:b8:b2:41, 1, 0, 4,
'_TZE200_zuz7f94z') params: FOREIGN KEY constraint failed*

TheJulianJES, 2022-04-17:
You need to add the command to this file: https://github.com/zigpy/zha-device-handlers/blob/16f298e9ce293d241592f86573e909f9c32809d0/zhaquirks/tuya/__init__.py#L104-L123
Possibly see some of my comments here for some instructions: https://github.com/zigpy/zha-device-handlers/issues/1294

If you get the custom quirk working, please PR the changes or send the quirk, so that it can be merged into Home Assistant.

closed 2022-10-21 by github-actions[bot]