zigpy/zha-device-handlers #1268: [Device Support Request] ERS-10TZBVK-AA TUYA KNOB

tayfunulu, 2022-01-08:
it's a new TuYa Smart knob -  **ERS-10TZBVK-AA** - like [https://expo.tuya.com/product/935005](url)

there is a support with zigbee2mqtt - [https://github.com/Koenkk/zigbee2mqtt/discussions/10473](url)

[image]

**Device signature** 

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_4fjiwweb",
  "model": "TS004F",
  "class": "zhaquirks.tuya.ts004f.TuyaSmartRemote004F"
}
```

MattWestb, 2022-01-08:
Its already supported in ZHA then it have the quirk loaded.
Is the device working as is suld or s it not working ?

By the way looks like one copy of IKEA Symphonisk :-)

tayfunulu, 2022-01-08:
thanks for fast response... 

it's pairing without any problem to home-assistant but there is no zha_event in events.

maybe, problem is model no because of model no, it is TS004F but it's different 4 button switch.

MattWestb, 2022-01-08:
Ops i have implanting the mode switch in tuya INIT but i have not putting the `TuyaZBOnOffAttributeCluster` in the ts004f quirk so you can't changing mode of it.

It shall having one light command mode Click mode but it can being that the default mode is clock mode that we have not getting working on the 4 button remote.

Can you putting little more debug in ZHA and looking what is sending to the system ??

MattWestb, 2022-01-08:
Updated old quirk (from oktober) with added attribute for switch mode `switch_mode (id: 0x8004)` on on/off cluster.
Configure local quirks in HA and putting the unzipped quirk in the folder and restart HA. 
[ts004f.zip](https://github.com/zigpy/zha-device-handlers/files/7833976/ts004f.zip)
Attribute and functions:
https://github.com/zigpy/zha-device-handlers/blob/b0bbbfb6b1f267b7d87d833b1ba5a5b0b127a9a1/zhaquirks/tuya/__init__.py#L835-L839
Dont forgetting waking the remote then sending attribute to it.

tayfunulu, 2022-01-08:
thanks, i tried this one. but device is different than 4 button switch 

it's rotate remote control 

debug

button press: 
`2022-01-08 23:25:42 DEBUG (MainThread) [zigpy.zcl] [0xd6f1:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=70 command_id=Command.Report_Attributes>
2022-01-08 23:25:42 DEBUG (MainThread) [zigpy.zcl] [0xd6f1:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=0>)]]
2022-01-08 23:25:42 DEBUG (MainThread) [zigpy.zcl] [0xd6f1:1:0x0b04] Attribute report received: active_power=0
2022-01-08 23:25:44 DEBUG (MainThread) [zigpy.zcl] [0xd6f1:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>

`

MattWestb, 2022-01-08:
If looking at Z2M is shall sending toggle and move up and down:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/629bbb0e1714e4ac83dabd27dc3b603bbbf7505b/devices/tuya.js#L1825-L1833
I think its also sending `move_hue` but they is not using it and i think (=not knowing) its then pressing the knob and rotating or double and triple poressing.

The step and move up and down is having automations in the quirk but toggle need being added for getting the automation triggers in ZHA.

Logs with all possible and impossible commands sent is more than welcome.

Its looks Z2M is not using events mode (aka scene mode / sending clicks and not light commands).

Sorry but your log is sending from one ElectricalMeasurement cluster (0x000a) so it must being one other device.

Can you verifying the `0xd6f1` is the remote and not one power plug ?

If not getting longs from the device deleting it wait one minute and paring it again.

MattWestb, 2022-01-08:
It can pairing lights and steering them with ZLL like the 4 button remote can without ZBGW https://expo.tuya.com/product/935005.

You also have all combinations in the picture for pressing and rotating.

Do you getting some manual with the device how to paring it with ZLL to lights and more interesting things and explanting the functions of it ?

I was tuch liking one LIDL LED strips by mistake then testing my 4 button version by pressing one button long.

tayfunulu, 2022-01-09:
> If looking at Z2M is shall sending toggle and move up and down: https://github.com/Koenkk/zigbee-herdsman-converters/blob/629bbb0e1714e4ac83dabd27dc3b603bbbf7505b/devices/tuya.js#L1825-L1833 I think [...]

yes you are right it's not my device ... i'll try again. device number [0xada3]

MattWestb, 2022-01-09:
Good !!
Some users was having problem getting the device sending commands to the system in the original TS004f testing and both devices (or more correct more devices then tuya have small remotes with 1 - 6 buttons with the same type) is having the same Zigbee module and the ground of functionality only that the new have one rotary encoder instead of buttons.

If not working deleting and adding the device try deleting it wait one minute and restarting HA and let it sattel and adding the remote and look if you is getting commands in the log   and / or zha_event.

I have playing with the original and some times zha is not loading the quirk then adding it then restart HA and its working.

PS: You dont need the updated quirk for getting all commands from the device then first need getting it sending "normal" commands and then testing if it can sending clock actions but its coms later.

tayfunulu, 2022-01-09:
i tried a lot to debug mode... 

i deleted zigbee.db file in HA and setup from beginning.... there are debug code... i am not expert and i do my best.... i hope it works 

TSN is probably counter... 

**SINGLE PRESS**

```
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=26 command_id=253>
2022-01-09 22:27:01 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 253
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fd: b'\x00'
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 253
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=26 command_id=253>
2022-01-09 22:27:01 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 253
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fd: b'\x00'
2022-01-09 22:27:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 253
2022-01-09 22:27:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=26 command_id=253>
2022-01-09 22:27:02 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 253
2022-01-09 22:27:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fd: b'\x00'
2022-01-09 22:27:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 253
2022-01-09 22:27:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=252>
2022-01-09 22:27:03 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 252
2022-01-09 22:27:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fc: b'\x00'
2022-01-09 22:27:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 252
```

**DOUBLE PRESS**

```
2022-01-09 22:28:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=32 command_id=252>
2022-01-09 22:28:01 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 252
2022-01-09 22:28:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fc: b'\x01'
2022-01-09 22:28:01 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 252
2022-01-09 22:28:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=32 command_id=252>
2022-01-09 22:28:02 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 252
2022-01-09 22:28:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fc: b'\x01'
2022-01-09 22:28:02 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 252
2022-01-09 22:28:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=33 command_id=253>
2022-01-09 22:28:03 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 253
2022-01-09 22:28:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fd: b'\x00'
2022-01-09 22:28:03 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 253
2022-01-09 22:28:05 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=34 command_id=253>
2022-01-09 22:28:05 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 253
2022-01-09 22:28:05 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fd: b'\x00'
2022-01-09 22:28:05 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 253
```

**HOLD - LONG PRESS**

```
2022-01-09 22:21:45 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=7 command_id=252>
2022-01-09 22:21:45 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 252
2022-01-09 22:21:45 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fc: b'\x00'
2022-01-09 22:21:45 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 252
[... cut ...]
2022-01-09 22:08:30 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=113 command_id=252>
2022-01-09 22:08:30 WARNING (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] Unknown cluster-specific command 252
2022-01-09 22:08:30 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] ZCL request 0x00fc: b'\x01'
2022-01-09 22:08:30 DEBUG (MainThread) [zigpy.zcl] [0x5fb0:1:0x0006] No handler for cluster command 252

```

MattWestb, 2022-01-09:
Great debug logging !!

If command is Zigbee standard the stack is sending one default response and the device is only sending one command but its not recognized so the device is repeating the same command with the same TSN (Transaction number) = normal until the system is knowing the command (the quirk) and doing it right.

The strange thing is the device `0x5fb0` is sending on endpoint `1`  and cluster `0x0006` (OnOff cluster) only 2 command 252 and 253 with x00 (0) or 0x1 (1).

First i was expecting it sending command 0 1 and 2 (on/Off and toggle) on cluster 0x0006 and also sending command on light level cluster 0x0008 and perhaps scene cluster 0x0005.

I think it can being in some other mode but i dont knowing how its shall working but Z2M is only getting light commands.

Can you deleting the device and resetting it and taking out the battery for some minutes and putting it back and paring it again and look wath is sending in the log.

Do you have some manuale or description with the device ? I can only imagine how it shall working from the Z2M code and the tuya homepage.

MattWestb, 2022-01-09:
I have patching the "normal" quirk so its using the same commands as the TS0044 and for command 253 is sending default response and logging the press types.
It need doing the same for command 252 but i cant doit so fast and need testing if its working.

My ts004f is "detecting" shor, double and long press with this quirk.
```
2022-01-09 22:11:16 DEBUG (MainThread) [zigpy.zcl] [0x6443:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=118 command_id=253>
2022-01-09 22:11:16 DEBUG (MainThread) [zigpy.zcl] [0x6443:1:0x0006] ZCL request 0x00fd: [2]
2022-01-09 22:11:16 DEBUG (MainThread) [zigpy.zcl] [0x6443:1:0x0006] TS004X: send default response
```
If you is having the firs  quirk installed delete it and putting this in your custom quirk folder and restart HA and look what you is getting in the log

MattWestb, 2022-01-09:
zha_event:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "60:a4:23:ff:fe:fd:dc:14",
        "unique_id": "60:a4:23:ff:fe:fd:dc:14:1:0x0006",
        "device_id": "fd5671bb8e2a9f4f6fc35fdaaf2c7a0a",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "press_type",
        "args": [
            1
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-01-09T21:11:29.487480+00:00",
    "context": {
        "id": "ee6ae47761346f959d864ebdfd1cc905",
        "parent_id": null,
        "user_id": null
    }
}
Event 21 fired 22:11:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "60:a4:23:ff:fe:fd:dc:14",
        "unique_id": "60:a4:23:ff:fe:fd:dc:14:1:0x0006",
        "device_id": "fd5671bb8e2a9f4f6fc35fdaaf2c7a0a",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_double_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-01-09T21:11:29.487349+00:00",
    "context": {
        "id": "d23ac7714fab66031ac3c50c73875a38",
        "parent_id": null,
        "user_id": null
    }
}
```

tayfunulu, 2022-01-10:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

tayfunulu, 2022-01-10:
quick question: did you update ts004f file. do i have to use quirk file or not ?

MattWestb, 2022-01-10:
For light remote mode no quirk is needed or the first one i was sending.
For scene mode shall the last sendt quirk its partly working (then the remote is sending the strange commands).

MattWestb, 2022-01-12:
Z2M have implantation the "scene mode" that is also possible making in ZHA and using it but the problem is how combinating it with the normal function light function nad getting both working without changing the quirk.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3660/files

With good logs i think it possible implanting both working mode.

MattWestb, 2022-01-12:
I have adding the second cmmandset (0xFC = 252) and is getting the rotate right and left in event mode with my TS004F if faking the commend number = very good.

First the remote have 2 working modes: 
Commands / remote = sending commands to (groups of) lights.
Events / scene mode = sending click and rotate.
Changing mode with pressing the button 3 times.
(my TS004F its long pressing right upper and lower button)

In command mode this is expected:
```
Single press: Toggle *
Double press:  (?) (not documented but likely implanted)
Rotate R/L: Step level up/down *
Press and rotate R/L: Step CT up/down *
Long press: Set color (not knowing the commands) *
Long press release: (?) (not documented but likely implanted)
```
and in event mode:
```
Single press: OK
Double press: OK
Long press: OK
Long press release: (?) (not documented but likely implanted)
Rotate right: OK (simulated) *
Rotate left: OK (simulated) *
Press rotate right: (can being click + rotate) *
Press rotate left: (can being click + rotate) *
```
I need debug logs and `zha_evet` from all commands so i can implanting the device automations for the device.

xatr0z, 2022-02-03:
@MattWestb I might be able to help as well. I suppose you mean the same commands as @tayfunulu posted, but with debug logs added?
And how should i configure debug logging (which zha/zigpy and what logging level do you need)

MattWestb, 2022-02-03:
We is trying getting the command mode working and getting the device automations corect in the linked issues before your post.
Then is made we shall trying the same with events / scene mode but its coming after the first step.

xatr0z, 2022-02-03:
OK great so let me know if you need anything tested!

sitthorn-ch, 2022-02-05:
Joined in the waiting list...

marcog20, 2022-02-06:
Hi, me too now a part of this thread; I bought two Smart Knobs and the pairing has been successfully done with a Xiaomi Gateway unlocked under ZHA integration, but there are no events on debug log...

MattWestb, 2022-03-05:
Nearly all in for the knob !!
https://github.com/zigpy/zha-device-handlers/issues/1322#issuecomment-1059771390

MattWestb, 2022-03-19:
PR is made for adding all working  futures we have found and one broken that we cant fixing https://github.com/zigpy/zha-device-handlers/pull/1437.

closed 2022-09-22 by github-actions[bot]

SolusCado, 2024-12-14:
Are we supposed to be able to bind this device's events to a Zigbee light group? This is my first time trying to bind a switch to a Zigbee group with ZHA, and cannot seem to get it to work.